zigpy/bellows #1: Cannot form network

rcloran, 2017-04-23:
@dale3h reports at https://github.com/home-assistant/home-assistant/pull/6263#issuecomment-296404854

rcloran, 2017-04-23:
@dale3h, could you run the form CLI with debug logging (ie, `bellows -v debug form`).

rcloran, 2017-04-23:
Actually, I think I see where this comes from.

rcloran, 2017-04-23:
@dale3h could you try with ed8278035f6dd25bbc2c08a2b85ccc3a006820c7 (https://github.com/rcloran/bellows/tree/form-fix), and let me know if it works?

dale3h, 2017-04-23:
Still a no-go:

```
$ bellows -v debug form
debug: Using selector: EpollSelector
debug: Loading application state from /home/homeassistant/.config/bellows/app.db
debug: Sending: b'\x1a\xc08\xbc~'
debug: RSTACK frame: b'\xc1\x02\x0b\nR~'
debug: Send command version
debug: Sending: b'\x00B!\xa8P\xed,~'
debug: Data frame: b'\x01B\xa1\xa8P(\x05\xe6\x7fb~'
debug: Sending: b'\x81`Y~'
debug: Application frame 0 (version) received
debug: Send command setConfigurationValue
debug: Sending: b"}1C!\xfbX(\x15\xc3'~"
debug: Data frame: b'\x12C\xa1\xfbT\xfbs~'
debug: Sending: b'\x82P:~'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'"@!\xfbY/\x15"o~'
debug: Data frame: b'#@\xa1\xfbT\xc6\x10~'
debug: Sending: b'\x83@\x1b~'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'3A!\xfby+\x15\xa2\xd7~'
debug: Data frame: b'4A\xa1\xfbT\xd3*~'
debug: Sending: b'\x840\xfc~'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'DF!\xfby+\x15\xf7\xb3~'
debug: Data frame: b'EF\xa1\xfbT5\xd0~'
debug: Sending: b'\x85 \xdd~'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'UG!\xfb}^)\x15\xde\xfb~'
debug: Data frame: b'VG\xa1\xfbT\xa9\xec~'
debug: Sending: b'\x86\x10\xbe~'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'fD!\xfbM(\x15>\xd0~'
debug: Data frame: b'gD\xa1\xfbT\x94\x8f~'
debug: Sending: b'\x87\x00\x9f~'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'wE!\xfbU\xd5\x15{5~'
debug: Data frame: b'pE\xa1\xfbT\x81\xb5~'
debug: Sending: b'\x80px~'
debug: Application frame 83 (setConfigurationValue) received
Traceback (most recent call last):
  File "/srv/homeassistant/bin/bellows", line 11, in <module>
    sys.exit(main())
  File "/srv/homeassistant/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/srv/homeassistant/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/srv/homeassistant/lib/python3.5/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/srv/homeassistant/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/srv/homeassistant/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/srv/homeassistant/lib/python3.5/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/srv/homeassistant/lib/python3.5/site-packages/bellows/cli/application.py", line 30, in form
    return util.app(inner, app_startup=False)(ctx)
  File "/srv/homeassistant/lib/python3.5/site-packages/bellows/cli/util.py", line 63, in inner
    loop.run_until_complete(async_inner(*args, **kwargs))
  File "/usr/lib/python3.5/asyncio/base_events.py", line 387, in run_until_complete
    return future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/srv/homeassistant/lib/python3.5/site-packages/bellows/cli/util.py", line 50, in async_inner
    yield from f(ctx, *args, **kwargs)
  File "/srv/homeassistant/lib/python3.5/site-packages/bellows/cli/application.py", line 28, in inner
    yield from app.form_network(channel, pan_id, extended_pan_id)
  File "/srv/homeassistant/lib/python3.5/site-packages/bellows/zigbee/application.py", line 86, in form_network
    channel = t.uint8_t(channel)
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
```

rcloran, 2017-04-23:
Ah, OK, different problem now. I've just pushed an update to that branch (e82d2a5616b75d0566a3e2d1f0f9d2a970a23f76) which adds a default value for the channel. Alternatively, you can specify a channel with `-c`. I recommend channel 15.

dale3h, 2017-04-24:
We have formation!

```
$ bellows form
$ bellows info
[00:0d:6f:00:0a:ff:73:4d]
[0]
[<EmberNetworkStatus.JOINED_NETWORK: 2>]
[<EmberStatus.SUCCESS: 0>, <EmberNodeType.COORDINATOR: 1>, <EmberNetworkParameters extendedPanId=[50, 179, 52, 40, 5, 170, 14, 178] panId=18288 radioTxPower=8 radioChannel=15 joinMethod=EmberJoinMethod.USE_MAC_ASSOCIATION nwkManagerId=0 nwkUpdateId=0 channels=134215680>]
[<EmberStatus.SUCCESS: 0>, <EmberCurrentSecurityState bitmask=116 trustCenterLongAddress=00:0d:6f:00:0a:ff:73:4d>]
```

rcloran, 2017-04-24:
Thanks for confirming.

closed 2017-04-24 by rcloran

dale3h, 2017-04-24:
Still having permit issues with my Zigbee devices.

dale3h, 2017-04-24:
Trying with an IKEA Tradfri motion detector, I get this error:

```
17-04-24 01:50:30 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback SerialTransport._read_ready()
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/events.py", line 125, in _run
    self._callback(*self._args)
  File "/srv/homeassistant/lib/python3.5/site-packages/serial_asyncio/__init__.py", line 106, in _read_ready
    self._protocol.data_received(data)
  File "/srv/homeassistant/lib/python3.5/site-packages/bellows/uart.py", line 54, in data_received
    self.frame_received(frame)
  File "/srv/homeassistant/lib/python3.5/site-packages/bellows/uart.py", line 64, in frame_received
    self.data_frame_received(data)
  File "/srv/homeassistant/lib/python3.5/site-packages/bellows/uart.py", line 84, in data_frame_received
    self._application.frame_received(self._randomize(data[1:-3]))
  File "/srv/homeassistant/lib/python3.5/site-packages/bellows/ezsp.py", line 159, in frame_received
    result, data = t.deserialize(data, schema)
  File "/srv/homeassistant/lib/python3.5/site-packages/bellows/types/__init__.py", line 9, in deserialize
    value, data = type_.deserialize(data)
  File "/srv/homeassistant/lib/python3.5/site-packages/bellows/types/basic.py", line 12, in deserialize
    r = cls.from_bytes(data[:cls._size], 'little', signed=cls._signed)
  File "/usr/lib/python3.5/enum.py", line 241, in __call__
    return cls.__new__(cls, value)
  File "/usr/lib/python3.5/enum.py", line 476, in __new__
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 128 is not a valid EmberMacPassthroughType
```

commit ed827803 (2017-04-23): Fix CLI form

Fixes #1
commit a24c0b72 (2017-04-23): Fix CLI form

Fixes #1
commit d56ed7c2 (2017-04-23): Fix CLI form

Fixes #1
commit e82d2a56 (2017-04-23): Fix CLI form

Fixes #1