zigpy/bellows #4: AttributeError: 'EzspStatus' object has no attribute '_name_'

patchedsoul, 2017-04-28:
(homeassistant_venv) homeassistant@hac1:~/.homeassistant$ bellows -d /dev/ttyUSB1 devices
warning: Unknown cluster 64580
warning: Unknown cluster 65280
Device:
  NWK: 0x56e0
  IEEE: 00:0d:6f:00:05:98:02:f9
  Endpoints:
    1: profile=0x104, device_type=DeviceType.REMOTE_CONTROL
      Clusters:
        Basic (0)
        Power Configuration (1)
        Identify (3)
        On/Off Switch Configuration (7)
        Poll Control (32)
        Diagnostic (2821)
    2: profile=0x104, device_type=DeviceType.REMOTE_CONTROL
      Clusters:
        On/Off Switch Configuration (7)
    3: profile=0x104, device_type=DeviceType.REMOTE_CONTROL
      Clusters:
        On/Off Switch Configuration (7)
    4: profile=0x104, device_type=DeviceType.REMOTE_CONTROL
      Clusters:
        On/Off Switch Configuration (7)
Device:
  NWK: 0xda82
  IEEE: 7c:e5:24:00:00:0c:21:d1
  Endpoints:
Device:
  NWK: 0xfbfe
  IEEE: ff:ff:00:0f:e7:ff:4d:d6
  Endpoints:
    1: profile=0xc05e, device_type=DeviceType.CONTROLLER
      Clusters:
        Basic (0)
        LightLink (4096)
        Cluster (64580)
        Cluster (65280)
(homeassistant_venv) homeassistant@hac1:~/.homeassistant$ bellows -d /dev/ttyUSB1 config --all
NOTE: Configuration changes do not persist across resets
Traceback (most recent call last):
  File "/srv/homeassistant/homeassistant_venv/bin/bellows", line 11, in <module>
    sys.exit(main())
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/bellows/cli/util.py", line 32, in inner
    loop.run_until_complete(f(*args, **kwargs))
  File "/usr/lib/python3.4/asyncio/base_events.py", line 276, in run_until_complete
    return future.result()
  File "/usr/lib/python3.4/asyncio/futures.py", line 277, in result
    raise self._exception
  File "/usr/lib/python3.4/asyncio/tasks.py", line 235, in _step
    result = coro.send(value)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/bellows/cli/ncp.py", line 25, in config
    s = yield from util.setup(ctx.obj['device'], util.print_cb)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/bellows/cli/util.py", line 107, in setup
    yield from cfg(c.CONFIG_STACK_PROFILE, 2)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/bellows/cli/util.py", line 102, in cfg
    check(v[0], 'Setting config %s to %s: %s' % (config_id, value, v[0]))
  File "/usr/lib/python3.4/enum.py", line 464, in __str__
    return "%s.%s" % (self.__class__.__name__, self._name_)
AttributeError: 'EzspStatus' object has no attribute '_name_'
error: Task was destroyed but it is pending!
error: task: <Task pending coro=<_send_task() running at /srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/bellows/uart.py:127> wait_for=<Future pending cb=[Task._wakeup()]>>
(homeassistant_venv) homeassistant@hac1:~/.homeassistant$ bellows -d /dev/ttyUSB1 -v debug scan
debug: Using selector: EpollSelector
debug: Connected. Resetting.
debug: Sending: b'\x1a\xc08\xbc~'
debug: RSTACK frame: b'\xc1\x02\x0b\nR~'
debug: Send command version
debug: Sending: b'\x00B!\xa8P\xed,~'
debug: Data frame: b'\x01B\xa1\xa8P(\x05\xe6\x7fb~'
debug: Sending: b'\x81`Y~'
debug: Application frame 0 (version) received
debug: Configuring...
debug: Send command setConfigurationValue
debug: Sending: b"}1C!\xfbX(\x15\xc3'~"
debug: Data frame: b'\x12C\xa1\xfbT\xfbs~'
debug: Sending: b'\x82P:~'
debug: Application frame 83 (setConfigurationValue) received
Traceback (most recent call last):
  File "/srv/homeassistant/homeassistant_venv/bin/bellows", line 11, in <module>
    sys.exit(main())
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/bellows/cli/util.py", line 32, in inner
    loop.run_until_complete(f(*args, **kwargs))
  File "/usr/lib/python3.4/asyncio/base_events.py", line 276, in run_until_complete
    return future.result()
  File "/usr/lib/python3.4/asyncio/futures.py", line 277, in result
    raise self._exception
  File "/usr/lib/python3.4/asyncio/tasks.py", line 235, in _step
    result = coro.send(value)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/bellows/cli/network.py", line 136, in scan
    s = yield from util.setup(ctx.obj['device'])
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/bellows/cli/util.py", line 107, in setup
    yield from cfg(c.CONFIG_STACK_PROFILE, 2)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/bellows/cli/util.py", line 102, in cfg
    check(v[0], 'Setting config %s to %s: %s' % (config_id, value, v[0]))
  File "/usr/lib/python3.4/enum.py", line 464, in __str__
    return "%s.%s" % (self.__class__.__name__, self._name_)
AttributeError: 'EzspStatus' object has no attribute '_name_'
error: Task was destroyed but it is pending!
error: task: <Task pending coro=<_send_task() running at /srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/bellows/uart.py:127> wait_for=<Future pending cb=[Task._wakeup()]>>
(homeassistant_venv) homeassistant@hac1:~/.homeassistant$

rcloran, 2017-04-28:
You can work around this by commenting out that check() line in `cli/util.py`. The return code is 0 in this case.

What version of Python are you using? I see 3.4 in the path names, but which minor version? I'm unable to repro this on 3.5.

patchedsoul, 2017-04-29:
python version is 3.4.2. Commenting out that check throws another error 
```
(homeassistant_venv) homeassistant@hac1:~/.homeassistant$ bellows -d /dev/ttyUSB1 -v debug scan
debug: Using selector: EpollSelector
debug: Connected. Resetting.
debug: Sending: b'\x1a\xc08\xbc~'
debug: RSTACK frame: b'\xc1\x02\x0b\nR~'
debug: Send command version
debug: Sending: b'\x00B!\xa8P\xed,~'
debug: Data frame: b'\x01B\xa1\xa8P(\x05\xe6\x7fb~'
debug: Sending: b'\x81`Y~'
debug: Application frame 0 (version) received
debug: Configuring...
debug: Send command setConfigurationValue
debug: Sending: b"}1C!\xfbX(\x15\xc3'~"
debug: Data frame: b'\x12C\xa1\xfbT\xfbs~'
debug: Sending: b'\x82P:~'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'"@!\xfbY/\x15"o~'
debug: Data frame: b'#@\xa1\xfbT\xc6\x10~'
debug: Sending: b'\x83@\x1b~'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'3A!\xfby+\x15\xa2\xd7~'
debug: Data frame: b'4A\xa1\xfbT\xd3*~'
debug: Sending: b'\x840\xfc~'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'DF!\xfbU\xd5\x154\xda~'
debug: Data frame: b'EF\xa1\xfbT5\xd0~'
debug: Sending: b'\x85 \xdd~'
debug: Application frame 83 (setConfigurationValue) received
Scanning channels 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26
debug: Send command startScan
debug: Sending: b'UG!\xb2U*\xedM^\x96\xd4K~'
debug: Data frame: b'VG\xa1\xb2T\x1e\xb8~'
debug: Sending: b'\x86\x10\xbe~'
debug: Application frame 26 (startScan) received
debug: Data frame: b'fG\xb1\xb3_\x87\x7f\xfa\xb6\x18J%\x8cM\x16I\x9eN\xd8\x12l\xf4~'
debug: Sending: b'\x87\x00\x9f~'
debug: Application frame 27 (networkFoundHandler) received
debug: Data frame: b'vG\xb1\xb3_\x87\x7f\xfa\xb6\x18J%\x8cM\x16I\x9eN\xd8\x13\x17\x86~'
debug: Sending: b'\x80px~'
debug: Application frame 27 (networkFoundHandler) received
debug: Data frame: b'\x06G\xb1\xb3[w\x04\x82\xd1\xd6K%+R\x92I\x9eN\xd8\x01\x10*~'
debug: Sending: b'\x81`Y~'
debug: Application frame 27 (networkFoundHandler) received
debug: Data frame: b"\x16G\xb1\xb3[\xad$\xd6\xd3\xd6K%+R\x92I\x9eN%\n'n~"
debug: Sending: b'\x82P:~'
debug: Application frame 27 (networkFoundHandler) received
debug: Data frame: b'&G\xb1\xb3[\xcd\x10LW\xd0K%+R\x92I\x9eN\xd8\x07K"~'
debug: Sending: b'\x83@\x1b~'
debug: Application frame 27 (networkFoundHandler) received
debug: Data frame: b'6G\xb1\xb3[]\x0c\xf2\xd3\xd6K%+R\x92I\x9eN\xd8\x02\xe6\xb9~'
debug: Sending: b'\x840\xfc~'
debug: Application frame 27 (networkFoundHandler) received
debug: Data frame: b'FG\xb1\xb3[\xb1\xad:?\xb4\xf4\x08(rWI\x9eN\xd8\x15r\xb7~'
debug: Sending: b'\x85 \xdd~'
debug: Application frame 27 (networkFoundHandler) received
debug: Data frame: b'VG\xb1\xb3[\xb1\xad:?\xb4\xf4\x08(rWI\x9eN\xd8\x13y"~'
debug: Sending: b'\x86\x10\xbe~'
debug: Application frame 27 (networkFoundHandler) received
debug: Data frame: b'fG\xb1\xb3[\xb1\xad:?\xb4\xf4\x08(rWI\x9eN\xdc\x01:`~'
debug: Sending: b'\x87\x00\x9f~'
debug: Application frame 27 (networkFoundHandler) received
debug: Data frame: b'vG\xb1\xb3[\xb1\xad:?\xb4\xf4\x08(rWI\x9eN\xd8\x1fn\x08~'
debug: Sending: b'\x80px~'
debug: Application frame 27 (networkFoundHandler) received
debug: Data frame: b'\x06G\xb1\xb3M\xd4\x1c\x86\x7f\xcfK%+R\x92I\x9eNZ\n\xc6u~'
[... 18 lines cut ...]
    return ctx.invoke(self.callback, **ctx.params)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/bellows/cli/util.py", line 32, in inner
    loop.run_until_complete(f(*args, **kwargs))
  File "/usr/lib/python3.4/asyncio/base_events.py", line 276, in run_until_complete
    return future.result()
  File "/usr/lib/python3.4/asyncio/futures.py", line 277, in result
    raise self._exception
  File "/usr/lib/python3.4/asyncio/tasks.py", line 235, in _step
    result = coro.send(value)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/bellows/cli/network.py", line 157, in scan
    click.echo(network)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/click/utils.py", line 221, in echo
    message = text_type(message)
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/bellows/types/struct.py", line 29, in __repr__
    ['%s=%s' % (f[0], getattr(self, f[0], None)) for f in self._fields]
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/bellows/types/struct.py", line 29, in <listcomp>
    ['%s=%s' % (f[0], getattr(self, f[0], None)) for f in self._fields]
  File "/usr/lib/python3.4/enum.py", line 464, in __str__
    return "%s.%s" % (self.__class__.__name__, self._name_)
AttributeError: 'Bool' object has no attribute '_name_'
```

rcloran, 2017-04-29:
Cause: https://bugs.python.org/issue23640

rcloran, 2017-04-29:
Thanks for the report. Since it only affects the CLI, I don't think I'm going to bump bellows in hass unless something else comes up.

commit 72bc167a (2017-04-29): Construct uint_t objects from an int

https://bugs.python.org/issue23640 means that we can't use from_bytes directly
on Python 3.4.

Fixes #4
closed 2017-04-29 by rcloran