zigpy/bellows #14: Can't initialize NCP

presslab-us, 2017-05-16:
Bellows can't initialize my NCP properly.  The stack version is 5.7.3-40.  It seems my NCP sends a CANCEL but bellows doesn't handle it properly.  Here is the debug output:

root@assistant:/home/pi/bellows# bellows -v DEBUG -d /dev/ttyAMA0 info
debug: Using selector: EpollSelector
debug: Connected. Resetting.
debug: Sending: b'1ac038bc7e'
debug: Data frame: b'00c1020b0a527e'
debug: Sending: b'8160597e'
debug: Application frame 163 (dsaVerify) received
^C
Aborted!

What is actually received is b'00' then b'1ac1020b0a527e'.  The x00 is still in _buffer and is not cleared on CANCEL.  When the CANCEL is received it only clears the previous contents of data and not _buffer.  That's why b'1a' is gone but b'00' is still there.

If I add <self._buffer = b''> here [https://github.com/rcloran/bellows/blob/master/bellows/uart.py#L50](url) it works and everything starts up correctly.  Is that the right way to fix it?

rcloran, 2017-05-16:
I believe that would be the correct fix. It seems that the SUBSTITUTE character suffers the same incorrect handling.

Do you plan to submit a PR? If so, a test case to cover this would be great. If not, I'll try get to this within the next few days.

presslab-us, 2017-05-17:
Hey thanks!  I have tested the cancel-partial branch and it solved my initialization issue.

commit 134070a7 (2017-05-17): Better handling of partial data and CANCEL/SUBSTITUTE

Fixes #14
closed 2017-05-17 by rcloran