zigpy/bellows #18: Invalid EmberOutgoingMessageType on permit (EZSP 5)

rcloran, 2017-06-09:
https://community.home-assistant.io/t/one-zigbee-gateway-to-rule-them-all/16803/30

Exception:

```
ERROR:homeassistant.core:Error doing job: Exception in callback SerialTransport.readready()
Traceback (most recent call last):
File "/usr/lib/python3.4/asyncio/events.py", line 120, in _run
self._callback(*self.args)
File "/opt/hass/.homeassistant/deps/serial_asyncio/__init__.py", line 106, in readready
self.protocol.data_received(data)
File "/opt/hass/.homeassistant/deps/bellows/uart.py", line 61, in data_received
self.frame_received(frame)
File "/opt/hass/.homeassistant/deps/bellows/uart.py", line 73, in frame_received
self.data_frame_received(data)
File "/opt/hass/.homeassistant/deps/bellows/uart.py", line 94, in data_frame_received
self._application.frame_received(self._randomize(data[1:-3]))
File "/opt/hass/.homeassistant/deps/bellows/ezsp.py", line 179, in frame_received
result, data = t.deserialize(data, schema)
File "/opt/hass/.homeassistant/deps/bellows/types/__init__.py", line 9, in deserialize
value, data = type_.deserialize(data)
File "/opt/hass/.homeassistant/deps/bellows/types/basic.py", line 13, in deserialize
r = cls(int.from_bytes(data[:cls._size], 'little', signed=cls._signed))
File "/usr/lib/python3.4/enum.py", line 222, in __call__
return cls.__new__(cls, value)
File "/usr/lib/python3.4/enum.py", line 457, in __new__
raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 6 is not a valid EmberOutgoingMessageType
```

MindrustUK, 2017-06-09:
Hi,

As requested output below:

```
debug: Using selector: EpollSelector
debug: Loading application state from zigbee.db
debug: Sending: b'1ac038bc7e'
debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
debug: Send command version
debug: Sending: b'004221a850ed2c7e'
debug: Data frame: b'0142a1a8512805ebd87b7e'
debug: Sending: b'8160597e'
debug: Application frame 0 (version) received
debug: Switching to eszp version 5
debug: Send command version
debug: Sending: b'7d31432157542a1025c57e'
debug: Data frame: b'1243a157542a10b049cd89937e'
debug: Sending: b'82503a7e'
debug: Application frame 0 (version) received
debug: Send command setConfigurationValue
debug: Sending: b'22402157547919b059c46e7e'
debug: Data frame: b'2340a157547915fbd27e'
debug: Sending: b'83401b7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'3341215754797d38b75995c77e'
debug: Data frame: b'3441a157547915904d7e'
debug: Sending: b'8430fc7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'44462157547938b35902ad7e'
debug: Data frame: b'4546a157547915640c7e'
debug: Sending: b'8520dd7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'55472157547938b359fda37e'
debug: Data frame: b'5647a157547915ce557e'
debug: Sending: b'8610be7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'6644215754793fb1590f627e'
debug: Data frame: b'6744a157547915e1597e'
debug: Sending: b'87009f7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'7745215754790cb0595fa87e'
debug: Data frame: b'7045a1575479158ac67e'
debug: Sending: b'8070787e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'004a21575479144d597d33457e'
debug: Data frame: b'014aa15754791573c57e'
debug: Sending: b'8160597e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command networkInit
debug: Sending: b'7d314b2157543d7d33827e'
debug: Data frame: b'124ba557543d1591927e'
debug: Sending: b'82503a7e'
debug: Application frame 23 (networkInit) received
debug: Data frame: b'224bb157543385e5f57e'
debug: Sending: b'83401b7e'
debug: Application frame 25 (stackStatusHandler) received
debug: Send command getNetworkParameters
debug: Sending: b'2348215754029e207e'
debug: Data frame: b'3348a157540215b35e64e824545a9fdc873dd9a5edce678bfd3e9c8e338b7e'
debug: Sending: b'8430fc7e'
debug: Application frame 40 (getNetworkParameters) received
debug: Send command setPolicy
debug: Sending: b'34492157547f10e2190d7e'
[... 44 lines cut ...]
Traceback (most recent call last):
  File "/usr/lib/python3.4/asyncio/events.py", line 120, in _run
    self._callback(*self._args)
  File "/usr/local/lib/python3.4/dist-packages/pyserial_asyncio-0.4-py3.4.egg/serial_asyncio/__init__.py", line 106, in _read_ready
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.4/dist-packages/bellows-0.2.7-py3.4.egg/bellows/uart.py", line 61, in data_received
    self.frame_received(frame)
  File "/usr/local/lib/python3.4/dist-packages/bellows-0.2.7-py3.4.egg/bellows/uart.py", line 73, in frame_received
    self.data_frame_received(data)
  File "/usr/local/lib/python3.4/dist-packages/bellows-0.2.7-py3.4.egg/bellows/uart.py", line 94, in data_frame_received
    self._application.frame_received(self._randomize(data[1:-3]))
  File "/usr/local/lib/python3.4/dist-packages/bellows-0.2.7-py3.4.egg/bellows/ezsp.py", line 179, in frame_received
    result, data = t.deserialize(data, schema)
  File "/usr/local/lib/python3.4/dist-packages/bellows-0.2.7-py3.4.egg/bellows/types/__init__.py", line 9, in deserialize
    value, data = type_.deserialize(data)
  File "/usr/local/lib/python3.4/dist-packages/bellows-0.2.7-py3.4.egg/bellows/types/basic.py", line 13, in deserialize
    r = cls(int.from_bytes(data[:cls._size], 'little', signed=cls._signed))
  File "/usr/lib/python3.4/enum.py", line 222, in __call__
    return cls.__new__(cls, value)
  File "/usr/lib/python3.4/enum.py", line 457, in __new__
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 6 is not a valid EmberOutgoingMessageType

Done
```

rcloran, 2017-07-03:
Should be fixed by #24. Thanks @AndreasBomholtz!

closed 2017-07-03 by rcloran

MindrustUK, 2017-07-03:
Looks like it's still failing to me:

```
bellows -v debug -d /dev/ttyUSB1 permit -D zigbee.db
debug: Using selector: EpollSelector
debug: Loading application state from zigbee.db
debug: Sending: b'1ac038bc7e'
debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
debug: Send command version
debug: Sending: b'004221a850ed2c7e'
debug: Data frame: b'0142a1a8512805ebd87b7e'
debug: Sending: b'8160597e'
debug: Application frame 0 (version) received
debug: Switching to eszp version 5
debug: Send command version
debug: Sending: b'7d31432157542a1025c57e'
debug: Data frame: b'1243a157542a10b049cd89937e'
debug: Sending: b'82503a7e'
debug: Application frame 0 (version) received
debug: Send command setConfigurationValue
debug: Sending: b'22402157547919b059c46e7e'
debug: Data frame: b'2340a157547915fbd27e'
debug: Sending: b'83401b7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'3341215754797d38b75995c77e'
debug: Data frame: b'3441a157547915904d7e'
debug: Sending: b'8430fc7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'44462157547938b35902ad7e'
debug: Data frame: b'4546a157547915640c7e'
debug: Sending: b'8520dd7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'5547215754793fb1591e517e'
debug: Data frame: b'5647a157547915ce557e'
debug: Sending: b'8610be7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'6644215754790cb059a0a67e'
debug: Data frame: b'6744a157547915e1597e'
debug: Sending: b'87009f7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'774521575479144d59d0f77e'
debug: Data frame: b'7045a1575479158ac67e'
debug: Sending: b'8070787e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command networkInit
debug: Sending: b'004a2157543de6f77e'
debug: Data frame: b'014aa557543d153bcb7e'
debug: Sending: b'8160597e'
debug: Application frame 23 (networkInit) received
debug: Data frame: b'114ab157543385215a7e'
debug: Sending: b'82503a7e'
debug: Application frame 25 (stackStatusHandler) received
debug: Send command getNetworkParameters
debug: Sending: b'124b215754027d3ade7e'
debug: Data frame: b'224ba157540215b35e64e824545a9fdc873dd9a5edce678bfd3e9c8e41b77e'
debug: Sending: b'83401b7e'
debug: Application frame 40 (getNetworkParameters) received
debug: Send command setPolicy
debug: Sending: b'23482157547f10e25bc07e'
debug: Data frame: b'3348a157547f156b4d7e'
debug: Sending: b'8430fc7e'
debug: Application frame 85 (setPolicy) received
debug: Send command setPolicy
debug: Sending: b'34492157547f7d33d36a2c7e'
debug: Data frame: b'4449a157547f15b3c87e'
[... 192 lines cut ...]
    self._protocol.data_received(data)
  File "/opt/hass/.homeassistant/deps/bellows/uart.py", line 61, in data_received
    self.frame_received(frame)
  File "/opt/hass/.homeassistant/deps/bellows/uart.py", line 73, in frame_received
    self.data_frame_received(data)
  File "/opt/hass/.homeassistant/deps/bellows/uart.py", line 94, in data_frame_received
    self._application.frame_received(self._randomize(data[1:-3]))
  File "/opt/hass/.homeassistant/deps/bellows/ezsp.py", line 179, in frame_received
    result, data = t.deserialize(data, schema)
  File "/opt/hass/.homeassistant/deps/bellows/types/__init__.py", line 9, in deserialize
    value, data = type_.deserialize(data)
  File "/opt/hass/.homeassistant/deps/bellows/types/basic.py", line 13, in deserialize
    r = cls(int.from_bytes(data[:cls._size], 'little', signed=cls._signed))
  File "/usr/lib/python3.4/enum.py", line 222, in __call__
    return cls.__new__(cls, value)
  File "/usr/lib/python3.4/enum.py", line 457, in __new__
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 6 is not a valid EmberOutgoingMessageType
DEBUG:bellows.uart:Data frame: b'6253b157546f11b259874a25aa5593499c36d94e0af29874f1db84b5058df1a7eb40cb6f0143a27e'
DEBUG:bellows.uart:Sending: b'87009f7e'
DEBUG:bellows.ezsp:Application frame 69 (incomingMessageHandler) received
DEBUG:bellows.zigbee.zdo:[0x3ce7:zdo] ZDO request 0x0013: [15591, 00:15:8d:00:00:ce:f3:f9, 142]
^CINFO:homeassistant.core:Bus:Handling <Event homeassistant_stop[L]>
INFO:homeassistant.core:Bus:Handling <Event homeassistant_close[L]>
```

rcloran, 2017-07-03:
653c74b4598090397fe642f944660160f3b44cf6 fixes a typo in the EmberOutgoingMessageType enum. But that shouldn't affect it.

This definitely works for me on Python 3.4,

```
In [1]: import bellows.types as t

In [2]: t.EmberOutgoingMessageType.deserialize(b'\x06')
Out[2]: (<EmberOutgoingMessageType.OUTGOING_BROADCAST: 6>, b'')
```

All your line numbers suggest your bellows in `/opt/hass/.homeassistant/deps/bellows` is up to date, but can you verify this for me, please?

```
grep OUTGOING_ /opt/hass/.homeassistant/deps/bellows/types/named.py
```

I can't think what could be causing this other than that file not being updated :-/

MindrustUK, 2017-07-03:
Ok, just blatted everything and brought it up to date to be sure as I suspect there maybe some version problems, currently getting this error on startup:

```
INFO:homeassistant.setup:Setting up zha
DEBUG:bellows.zigbee.appdb:Loading application state from /opt/hass/zigbee.db
DEBUG:bellows.uart:Sending: b'\x1a\xc08\xbc~'
DEBUG:bellows.uart:RSTACK frame: b'\xc1\x02\x0b\nR~'
DEBUG:bellows.ezsp:Send command version
DEBUG:bellows.uart:Sending: b'\x00B!\xa8P\xed,~'
DEBUG:bellows.uart:Data frame: b'\x01B\xa1\xa8Q(\x05\xeb\xd8{~'
DEBUG:bellows.uart:Sending: b'\x81`Y~'
DEBUG:bellows.ezsp:Application frame 0 (version) received
DEBUG:bellows.ezsp:Send command setConfigurationValue
DEBUG:bellows.uart:Sending: b"}1C!\xfbX(\x15\xc3'~"
DEBUG:bellows.uart:Data frame: b'\x12C\xa1WT\xa3`~'
DEBUG:bellows.uart:Sending: b'\x82P:~'
ERROR:homeassistant.core:Error doing job: Exception in callback SerialTransport._read_ready()
Traceback (most recent call last):
  File "/usr/lib/python3.4/asyncio/events.py", line 120, in _run
    self._callback(*self._args)
  File "/opt/hass/.homeassistant/deps/serial_asyncio/__init__.py", line 106, in _read_ready
    self._protocol.data_received(data)
  File "/opt/hass/.homeassistant/deps/bellows/uart.py", line 54, in data_received
    self.frame_received(frame)
  File "/opt/hass/.homeassistant/deps/bellows/uart.py", line 64, in frame_received
    self.data_frame_received(data)
  File "/opt/hass/.homeassistant/deps/bellows/uart.py", line 84, in data_frame_received
    self._application.frame_received(self._randomize(data[1:-3]))
  File "/opt/hass/.homeassistant/deps/bellows/ezsp.py", line 144, in frame_received
    frame_name = self.COMMANDS_BY_ID[frame_id][0]
KeyError: 255
WARNING:homeassistant.setup:Setup of zha is taking over 10 seconds.
```

Also output as requested:
```
    VALUE_MAXIMUM_OUTGOING_TRANSFER_SIZE = 0x06
    OUTGOING_DIRECT = 0x00
    OUTGOING_VIA_ADDRESS_TABLE = 0x01
    OUTGOING_VIA_BINDING = 0x02
    OUTGOING_MULTICAST = 0x03
    OUTGOING_BROADCAST = 0x04
    KEY_HAS_OUTGOING_FRAME_COUNTER = 0x0002
```

rcloran, 2017-07-03:
You now have very old bellows, without any of the fixes. If you want to test whether the fixes work, you have to apply them.

rcloran, 2017-07-03:
I will update hass to require 0.3.1 soon - the code for that is at https://github.com/rcloran/home-assistant/tree/bellows-split-cluster - I just need to wait for home-assistant/home-assistant#8308 and home-assistant/home-assistant#8310 to be merged.

MindrustUK, 2017-07-03:
Yup I screwed that up, fixed it now. Think the code issue is resolved, looks like a HASS issue now:

```
DEBUG:bellows.uart:Data frame: b'3348a157547f156b4d7e'
DEBUG:bellows.uart:Sending: b'8430fc7e'
DEBUG:bellows.ezsp:Application frame 85 (setPolicy) received
DEBUG:bellows.ezsp:Send command setPolicy
DEBUG:bellows.uart:Sending: b'34492157547f7d33d36a2c7e'
DEBUG:bellows.uart:Data frame: b'4449a157547f15b3c87e'
DEBUG:bellows.uart:Sending: b'8520dd7e'
DEBUG:bellows.ezsp:Application frame 85 (setPolicy) received
DEBUG:bellows.ezsp:Send command setPolicy
DEBUG:bellows.uart:Sending: b'454e2157547f15b342b77e'
DEBUG:bellows.uart:Data frame: b'554ea157547f15f4937e'
DEBUG:bellows.uart:Sending: b'8610be7e'
DEBUG:bellows.ezsp:Application frame 85 (setPolicy) received
DEBUG:bellows.ezsp:Send command getNodeId
DEBUG:bellows.uart:Sending: b'564f2157540d0e867e'
DEBUG:bellows.uart:Data frame: b'664fa157540d15b23ba27e'
DEBUG:bellows.uart:Sending: b'87009f7e'
DEBUG:bellows.ezsp:Application frame 39 (getNodeId) received
DEBUG:bellows.ezsp:Send command getEui64
DEBUG:bellows.uart:Sending: b'674c2157540c9a597e'
DEBUG:bellows.uart:Data frame: b'774ca157540ccf0450944a86885527a57e'
DEBUG:bellows.uart:Sending: b'8070787e'
DEBUG:bellows.ezsp:Application frame 38 (getEui64) received
INFO:homeassistant.core:Bus:Handling <Event service_registered[L]: domain=zha, service=permit>
INFO:homeassistant.core:Bus:Handling <Event component_loaded[L]: component=zha>
ERROR:homeassistant.core:Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/lib/python3.4/asyncio/tasks.py", line 237, in _step
    result = next(coro)
  File "/usr/local/lib/python3.4/dist-packages/homeassistant/components/zha/__init__.py", line 136, in async_device_initialized
    zha_const.populate_data()
  File "/usr/local/lib/python3.4/dist-packages/homeassistant/components/zha/const.py", line 51, in populate_data
    COMPONENT_CLUSTERS[component].update(clusters)
TypeError: unhashable type: 'list'
INFO:homeassistant.bootstrap:Home Assistant initialized in 7.35s
INFO:homeassistant.core:Starting Home Assistant core loop
INFO:homeassistant.core:Starting Home Assistant
INFO:homeassistant.core:Bus:Handling <Event homeassistant_start[L]>
```

rcloran, 2017-07-03:
Right - 8a999039a5583d4600a1407b7fa5beaab42f7b46 changes the bellows API, and is unfortunately before the EmberOutgoingMessageType fix.

The easiest is going to be to wait for https://github.com/rcloran/home-assistant/tree/bellows-split-cluster in hass, but you should be able to apply it to your hass cleanly.

You could also try to revert 8a999039a5583d4600a1407b7fa5beaab42f7b46 from your bellows.