zigpy/bellows #22: a

ausil, 2017-06-26:
I have the reciever from http://www.homedepot.com/p/Home-Decorators-Collection-Gardinier-52-in-LED-Brushed-Nickel-Ceiling-Fan-43260/206648825 and was wanting to add support for it.  it paired okay. and shows up as a switch, which seems to be the switch to turn everything on/off.  Turning it on works, however turning it back off does not seem to, the light is a dimmer and the fan speed control ihas 4 speeds, there is a comfort breeze mode that randomly changes the speed of the fan.  There is support in smartthings for the receiver,  I was wondering if you had any docs on how to best add support for new devices.

rcloran, 2017-06-26:
I don't have any docs.

Are you trying to add the device to Home Assistant? If you think this thing is supposed to be ZHA compliant, we can work through figuring out why it's not working properly. If it's not ZHA compliant, I'd prefer to avoid adding special code in hass until I have a better framework in place for handling non-compliant devices.

Can you send me the output of "bellows devices", and if you're doing this in Home Assistant, debug logs of the startup, and when you attempt to switch on/off?

ausil, 2017-06-27:
AFAIK it is, seems that there is support for 2 clusters missing 0x0008 and 0x0202 

0x0008 is the light dimmer and 0x0202 is the fan controller, at least its what I get when taking the list of clusters in https://github.com/dcoffing/KOF-CeilingFan/blob/master/devicetypes/dcoffing/kof-zigbee-fan-controller.src/kof-zigbee-fan-controller.groovy#L81 and looking at the list in bellows/zigbee/zdo/types.py 

I am using Home Assistant. Will get some logs tonight.

ausil, 2017-06-28:
logs on startup are 

2017-06-28 03:38:26 INFO (MainThread) [homeassistant.setup] Setting up zha
2017-06-28 03:38:27 DEBUG (MainThread) [bellows.zigbee.appdb] Loading application state from /home/dennis/zigbee.db
2017-06-28 03:38:27 DEBUG (MainThread) [bellows.uart] Sending: b'\x1a\xc08\xbc~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] RSTACK frame: b'\xc1\x02\x0b\nR~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Send command version
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'\x00B!\xa8P\xed,~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Data frame: b'\x01B\xa1\xa8P(\x05\xe6\x7fb~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'\x81`Y~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Application frame 0 (version) received
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b"}1C!\xfbX(\x15\xc3'~"
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Data frame: b'\x12C\xa1\xfbT\xfbs~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'\x82P:~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'"@!\xfbY/\x15"o~'
2017-06-28 03:38:28 INFO (MainThread) [homeassistant.setup] Setting up updater
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Data frame: b'#@\xa1\xfbT\xc6\x10~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'\x83@\x1b~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'3A!\xfby+\x15\xa2\xd7~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Data frame: b'4A\xa1\xfbT\xd3*~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'\x840\xfc~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'DF!\xfby+\x15\xf7\xb3~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Data frame: b'EF\xa1\xfbT5\xd0~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'\x85 \xdd~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'UG!\xfb}^)\x15\xde\xfb~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Data frame: b'VG\xa1\xfbT\xa9\xec~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'\x86\x10\xbe~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'fD!\xfbM(\x15>\xd0~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Data frame: b'gD\xa1\xfbT\x94\x8f~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'\x87\x00\x9f~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'wE!\xfbU\xd5\x15{5~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Data frame: b'pE\xa1\xfbT\x81\xb5~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'\x80px~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Send command networkInit
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'\x00J!\xbf=O~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Data frame: b'\x01J\xa5\xbfT\xffD~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'\x81`Y~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Application frame 23 (networkInit) received
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Data frame: b'\x11J\xb1\xb1\xc4\xc4\x0b~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'\x82P:~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Application frame 25 (stackStatusHandler) received
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Send command getNetworkParameters
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'\x12K!\x80;\x0c~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Data frame: b'"K\xa1\x80T+\xc8\'\x97\x00S\x8eP*\xcf\x9d\x94A\'\xab\xed\xcegs\x02\xc1\xde\xc9~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'\x83@\x1b~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Application frame 40 (getNetworkParameters) received
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.ezsp] Send command setPolicy
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Sending: b'#H!\xfdQz\xe4\x17~'
2017-06-28 03:38:28 DEBUG (MainThread) [bellows.uart] Data frame: b'3H\xa1\xfdT\xed/~'
[... cut ...]
2017-06-28 03:43:07 DEBUG (MainThread) [bellows.uart] Sending: b'\x86\x10\xbe~'
2017-06-28 03:43:07 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received
2017-06-28 03:43:07 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-06-28 03:43:07 WARNING (MainThread) [bellows.zigbee.endpoint] [0x07cb:1] Message on unknown cluster 0x0019

At this point the light turns on, but not off. The dimmer and fan do not show up, I suspect it is because the zdo bits for their clusters is missing.

rcloran, 2017-07-02:
Can you send me the output of "bellows devices"?

ausil, 2017-07-02:
(homeassistant) [dennis@wandboard02 bellows (master)]$ bellows devices
Usage: bellows [OPTIONS] COMMAND [ARGS]...

Error: Missing option "-d" / "--device".
(homeassistant) [dennis@wandboard02 bellows (master)]$ bellows --device /dev/ttyUSB1 devices
Traceback (most recent call last):
  File "/home/dennis/homeassistant/bin/bellows", line 11, in <module>
    load_entry_point('bellows==0.2.7', 'console_scripts', 'bellows')()
  File "/home/dennis/homeassistant/lib/python3.5/site-packages/click-6.7-py3.5.egg/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/dennis/homeassistant/lib/python3.5/site-packages/click-6.7-py3.5.egg/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/dennis/homeassistant/lib/python3.5/site-packages/click-6.7-py3.5.egg/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/dennis/homeassistant/lib/python3.5/site-packages/click-6.7-py3.5.egg/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/dennis/homeassistant/lib/python3.5/site-packages/click-6.7-py3.5.egg/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/dennis/homeassistant/lib/python3.5/site-packages/click-6.7-py3.5.egg/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/dennis/homeassistant/lib/python3.5/site-packages/bellows-0.2.7-py3.5.egg/bellows/cli/application.py", line 59, in devices
  File "/home/dennis/homeassistant/lib/python3.5/site-packages/bellows-0.2.7-py3.5.egg/bellows/zigbee/application.py", line 28, in __init__
  File "/home/dennis/homeassistant/lib/python3.5/site-packages/bellows-0.2.7-py3.5.egg/bellows/zigbee/appdb.py", line 29, in __init__
sqlite3.OperationalError: unable to open database file

there does not seem to be an option to specify the database needed

ausil, 2017-07-02:
Device:
  NWK: 0x07cb
  IEEE: 00:22:a3:00:00:16:b4:27
  Endpoints:
    1: profile=0x104, device_type=14
      Clusters:
        Basic (0)
        Identify (3)
        Groups (4)
        Scenes (5)
        On/Off (6)
        Level control (8)
        Fan Control (514)
changed the code and hardcoded in the location of my zigbe db from home assistant 

Device:
  NWK: 0x07cb
  IEEE: 00:22:a3:00:00:16:b4:27
  Endpoints:
    1: profile=0x104, device_type=14
      Clusters:
        Basic (0)
        Identify (3)
        Groups (4)
        Scenes (5)
        On/Off (6)
        Level control (8)
        Fan Control (514)

rcloran, 2017-07-03:
I don't see any calls to the "turn_off" service in hass, so either the logs don't include it, or the UI toggle isn't actually generating the service call. So, basically, I have no information about why it's failing.

What I can see is that getting the response from the device is timing out during the turn on (I think, your log timestamps seems weirdly out of order).

What I wanted to check with the devices output was how the fan control was exposed. Since it's a different cluster, I don't think there'll be any weirdness about how the fan and light control interact.

Also, adding support in hass for the fan shouldn't be too hard. However, what I do have in bellows isn't the full spec: https://github.com/rcloran/bellows/blob/master/bellows/zigbee/zcl/clusters/hvac.py#L111-L123 , so it'll need a change there first.

rcloran, 2017-07-03:
You should be able to get the dimmer working by adding something like this to your config:

```
zha:
  ...
  device_config:
      "00:22:a3:00:00:16:b4:27-1":
          type: light
```

rcloran, 2017-07-03:
Oh, it looks like the Fan cluster in bellows is complete. So only work in hass will be needed to make it work.

ausil, 2017-07-04:
okay, setting the device_config I can turn the light on and control the brightness, something about the on off swicth does not work. it just stays as off.  I will look at what needs doing in hass. 

I do wonder if in https://github.com/rcloran/bellows/blob/master/bellows/zigbee/profiles/zll.py and https://github.com/rcloran/bellows/blob/master/bellows/zigbee/profiles/zha.py we would have to add some profile and device info for the fans and the device to be recognised correctly.

rcloran, 2017-07-04:
Are you saying that you can't turn it off because the switch just flips back to off? That would make sense if the turn_on is failing (timeout from your logs).

You won't need to add to the profiles. Your device reports a device type of 14 (0x0e), which I cannot find in the spec. You would need to add to https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/zha/const.py#L38-L42

ausil, 2017-07-04:
yeah the switch just flips back to off

rcloran, 2017-07-04:
Aha. Good, now I understand the problem better.

If you call the light.turn_off service "directly" without the convenience UI (eg, from /dev-service), does it switch off?

Can you send me debug logs from the turn on again, but make sure to include triple back-ticks before and after so that github doesn't treat it as markdown? I'd like to decode the hex from packets being sent and received, but it's mangled.

cytech, 2017-07-04:
Also dove in and installed the same king of fans receiver as ausil.
same results as noted.
I will watch this thread and test when available.
thanx

cytech, 2017-07-06:
calling "the light.turn_off service "directly" does switch off the light

cytech, 2017-07-06:
debug turn on from UI
```
2017-07-06 06:13:35 DEBUG (MainThread) [homeassistant.components.websocket_api] WS 139860690201232: Received {'type': 'call_service', 'domain': 'homeassistant', 'service_data': {'entity_id': 'light.king_of_fans__inc_hdc52eastwindfan_0016b404_1'}, 'service': 'turn_on', 'id': 12}
2017-07-06 06:13:35 INFO (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: service=turn_on, domain=homeassistant, service_data=entity_id=light.king_of_fans__inc_hdc52eastwindfan_0016b404_1, service_call_id=139861610363480-20>
2017-07-06 06:13:35 INFO (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: service=turn_on, domain=light, service_data=entity_id=['light.king_of_fans__inc_hdc52eastwindfan_0016b404_1'], service_call_id=139861610363480-21>
2017-07-06 06:13:35 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast
2017-07-06 06:13:35 DEBUG (MainThread) [bellows.uart] Sending: b'TW!\x9cT\xe9\xaf\xb6X\x9cJ$\xab\x15\x93I\x9cI \xad\xec\xc9ct\xf8\xc6\xd05~'
2017-07-06 06:13:35 DEBUG (MainThread) [bellows.uart] Data frame: b'FW\xa1\x9cT\xf7W\x1e~'
2017-07-06 06:13:35 DEBUG (MainThread) [bellows.uart] Sending: b'\x85 \xdd~'
2017-07-06 06:13:35 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received
2017-07-06 06:13:35 DEBUG (MainThread) [bellows.uart] Data frame: b'VW\xb1\x97T\xe9\xaf\xb6X\x9cJ$\xab\x15\x93I\x9c\x93 \xab\xed\x9c\xee~'
2017-07-06 06:13:35 DEBUG (MainThread) [bellows.uart] Sending: b'\x86\x10\xbe~'
2017-07-06 06:13:35 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received
```

ausil, 2017-07-11:
I have started working on adding a fan.zha device https://github.com/ausil/home-assistant/tree/zha-fan right now the fan device shows up. but none of the controls work yet. though it will turn the light off. so i suspect I have not got it right

cytech, 2017-07-11:
just a note, i am under the impression that this KOF controller has 4 fan speeds
HIGH, MED-HI, MED, LOW
as well as a "breeze" setting and light on/off/dim
also (i don't know if this means much) there is a fan component added to z-wave for a fan switch (on,off,high,med,low) in zwave/workaround.py

ausil, 2017-07-13:
It is a 4 speed fan with "breeze" mode that randomly changes the speed, as well as on/off/dimmer light. I beleiev there is also a control to turn everything on or off, though that may be implemented in software

closed 2017-08-01 by rcloran

cytech, 2017-08-01:
Hi Russell
I noticed you closed this.
Does that mean there is no interest in this king of fans component or is this the wrong place to ask for it?
There are a few people looking for this but I have not seen anyway to implement it.
https://community.home-assistant.io/t/zha-zigbee-tested-devices-please-add-your-device-results/17718/49.
If this is the wrong place to ask, please point me in the right direction.
I am happy to help where I can, and although I can follow the logic for creating a new fan component, I get lost the deeper I get with interacting with bellows.
my alternative at this point  would be to find a cheap Wink hub, and use the Wink Fan component (built specifically for this fan controller) to drive these. I would prefer not to go that route and keep it in home assistant.
Thank you,
Dave

rcloran, 2017-08-01:
I don't believe there's any work needed in bellows to make this work, which is why I closed this issue. I left some comments on @ausil's home-assistant branch linked above, but it looks like it's just a copy of the ZWave fan component so far, so there's not much to discuss. I'd be happy to provide pointers, but I'm not in a position to write this code myself. As far as I can tell, the zigbee part of this simply involves writing to an attribute, and bellows already provides an API for that.