zigpy/bellows #31: Not getting any events from Zigbee remote

mjg59, 2017-07-05:
I'm trying to add support for a Philips Hue dimmer remote to HASS. I've done the following so far:

* Updated zha/const.py and added zha.DeviceType.SIMPLE_SENSOR: 'binary_sensor', to the ZHA device profiles
* Added DeviceType.SIMPLE_SENSOR: [0x0001, 0x0003, 0x000f], to bellows/zigbee/profiles/zha.py
* Hacked binary_sensor/zha.py to look for BinaryInput rather than IAS and removed the IAS-specific code for now

I then get a binary sensor device in HASS. However, hitting buttons never calls the cluster_command() method and I see no debug output from bellows. zha and bellows debug output:

```
Jul 04 16:59:32 alloneword hass[2557]: INFO:homeassistant.loader:Loaded zha from homeassistant.components.zha
Jul 04 16:59:34 alloneword hass[2557]: INFO:homeassistant.setup:Setting up zha
Jul 04 16:59:34 alloneword hass[2557]: DEBUG:bellows.zigbee.appdb:Loading application state from /root/.homeassistant/zigbee.db
Jul 04 16:59:34 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x1a\xc08\xbc~'
Jul 04 16:59:35 alloneword hass[2557]: DEBUG:bellows.uart:RSTACK frame: b'\xc1\x02\x0b\nR~'
Jul 04 16:59:35 alloneword hass[2557]: DEBUG:bellows.ezsp:Send command version
Jul 04 16:59:35 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x00B!\xa8P\xed,~'
Jul 04 16:59:35 alloneword hass[2557]: DEBUG:bellows.uart:Data frame: b'\x01B\xa1\xa8P(\x05\xe6\x7fb~'
Jul 04 16:59:35 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x81`Y~'
Jul 04 16:59:35 alloneword hass[2557]: DEBUG:bellows.ezsp:Application frame 0 (version) received
Jul 04 16:59:35 alloneword hass[2557]: DEBUG:bellows.ezsp:Send command setConfigurationValue
Jul 04 16:59:35 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b"}1C!\xfbX(\x15\xc3'~"
Jul 04 16:59:35 alloneword hass[2557]: DEBUG:bellows.uart:Data frame: b'\x12C\xa1\xfbT\xfbs~'
Jul 04 16:59:35 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x82P:~'
Jul 04 16:59:35 alloneword hass[2557]: DEBUG:bellows.ezsp:Application frame 83 (setConfigurationValue) received
Jul 04 16:59:35 alloneword hass[2557]: DEBUG:bellows.ezsp:Send command setConfigurationValue
Jul 04 16:59:35 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'"@!\xfbY/\x15"o~'
Jul 04 16:59:35 alloneword hass[2557]: DEBUG:bellows.uart:Data frame: b'#@\xa1\xfbT\xc6\x10~'
Jul 04 16:59:35 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x83@\x1b~'
Jul 04 16:59:35 alloneword hass[2557]: DEBUG:bellows.ezsp:Application frame 83 (setConfigurationValue) received
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.ezsp:Send command setConfigurationValue
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'3A!\xfby+\x15\xa2\xd7~'
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.uart:Data frame: b'4A\xa1\xfbT\xd3*~'
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x840\xfc~'
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.ezsp:Application frame 83 (setConfigurationValue) received
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.ezsp:Send command setConfigurationValue
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'DF!\xfby+\x15\xf7\xb3~'
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.uart:Data frame: b'EF\xa1\xfbT5\xd0~'
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x85 \xdd~'
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.ezsp:Application frame 83 (setConfigurationValue) received
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.ezsp:Send command setConfigurationValue
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'UG!\xfb}^)\x15\xde\xfb~'
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.uart:Data frame: b'VG\xa1\xfbT\xa9\xec~'
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x86\x10\xbe~'
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.ezsp:Application frame 83 (setConfigurationValue) received
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.ezsp:Send command setConfigurationValue
Jul 04 16:59:43 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'fD!\xfbM(\x15>\xd0~'
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.uart:Data frame: b'gD\xa1\xfbT\x94\x8f~'
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x87\x00\x9f~'
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.ezsp:Application frame 83 (setConfigurationValue) received
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.ezsp:Send command setConfigurationValue
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'wE!\xfbU\xd5\x15{5~'
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.uart:Data frame: b'pE\xa1\xfbT\x81\xb5~'
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x80px~'
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.ezsp:Application frame 83 (setConfigurationValue) received
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.ezsp:Send command networkInit
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x00J!\xbf=O~'
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.uart:Data frame: b'\x01J\xa5\xbfT\xffD~'
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x81`Y~'
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.ezsp:Application frame 23 (networkInit) received
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.uart:Data frame: b'\x11J\xb1\xb1\xc4\xc4\x0b~'
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x82P:~'
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.ezsp:Application frame 25 (stackStatusHandler) received
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.ezsp:Send command getNetworkParameters
Jul 04 16:59:44 alloneword hass[2557]: WARNING:homeassistant.setup:Setup of zha is taking over 10 seconds.
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x12K!\x80;\x0c~'
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.uart:Data frame: b'"K\xa1\x80T+R\x91\xc8\xb8`\xfcH\x91\xc3M\x94A\'\xab\xed\xcegs\x02\xc1\xa2m~'
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x83@\x1b~'
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.ezsp:Application frame 40 (getNetworkParameters) received
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.ezsp:Send command setPolicy
Jul 04 16:59:44 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'#H!\xfdQz\xe4\x17~'
[... 135 lines cut ...]
Jul 04 17:00:28 alloneword hass[2557]: INFO:homeassistant.core:Bus:Handling <Event platform_discovered[L]: service=load_platform.binary_sensor, discovered=discovery_key=00:17:88:01:02:19:57:81-2, platform=zha>
Jul 04 17:00:28 alloneword hass[2557]: DEBUG:homeassistant.components.zha:7
Jul 04 17:00:28 alloneword hass[2557]: DEBUG:homeassistant.components.zha:7
Jul 04 17:00:28 alloneword hass[2557]: DEBUG:homeassistant.components.zha:7
Jul 04 17:00:28 alloneword hass[2557]: DEBUG:homeassistant.components.zha:7
Jul 04 17:00:28 alloneword hass[2557]: DEBUG:homeassistant.components.zha:7
Jul 04 17:00:28 alloneword hass[2557]: INFO:homeassistant.loader:Loaded binary_sensor.zha from homeassistant.components.binary_sensor.zha
Jul 04 17:00:28 alloneword hass[2557]: INFO:homeassistant.components.binary_sensor:Setting up binary_sensor.zha
Jul 04 17:00:28 alloneword hass[2557]: DEBUG:homeassistant.components.binary_sensor.zha:BINARY SENSOR ZHA FOUND
Jul 04 17:00:28 alloneword hass[2557]: DEBUG:homeassistant.components.binary_sensor.zha:[<bellows.zigbee.zcl.clusters.general.PowerConfiguration object at 0xa56d33ec>, <bellows.zigbee.zcl.clusters.general.Identify object at 0xa8636f0c>, <bellows.zigbee.zcl.clusters.general.BinaryInput object at 0xa8299e4c>]
Jul 04 17:00:28 alloneword hass[2557]: DEBUG:homeassistant.components.binary_sensor.zha:FOUND BINARY INPUT
Jul 04 17:00:28 alloneword hass[2557]: DEBUG:homeassistant.components.binary_sensor.zha:<bellows.zigbee.zcl.clusters.general.BinaryInput object at 0xa8299e4c>
Jul 04 17:00:28 alloneword hass[2557]: DEBUG:homeassistant.components.binary_sensor.zha:{'endpoint': <bellows.zigbee.endpoint.Endpoint object at 0xa56ddd0c>, 'clusters': [<bellows.zigbee.zcl.clusters.general.PowerConfiguration object at 0xa56d33ec>, <bellows.zigbee.zcl.clusters.general.Identify object at 0xa8636f0c>, <bellows.zigbee.zcl.clusters.general.BinaryInput object at 0xa8299e4c>], 'new_join': True, 'model': 'RWL020', 'manufacturer': 'Philips'}
Jul 04 17:00:28 alloneword hass[2557]: DEBUG:bellows.ezsp:Send command sendUnicast
[... cut ...]
Jul 04 17:00:28 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x85 \xdd~'
Jul 04 17:00:28 alloneword hass[2557]: DEBUG:bellows.ezsp:Application frame 63 (messageSentHandler) received
Jul 04 17:00:29 alloneword hass[2557]: DEBUG:bellows.uart:Data frame: b'VW\xb1\xedT*\x15\x93\xd9\x94J%\xabU\x92\xd5c\xf0\xe1\xb0\x121e\x8d\xfd\xd2\\~'
Jul 04 17:00:29 alloneword hass[2557]: DEBUG:bellows.uart:Sending: b'\x86\x10\xbe~'
Jul 04 17:00:29 alloneword hass[2557]: DEBUG:bellows.ezsp:Application frame 69 (incomingMessageHandler) received
```

mjg59, 2017-07-05:
bellows devices gives:
```
Device:
  NWK: 0x1bc6
  IEEE: 00:17:88:01:02:19:57:81
  Endpoints:
    1: profile=0xc05e, device_type=DeviceType.SCENE_CONTROLLER
      Input Clusters:
        Basic (0)
    2: profile=0x104, device_type=DeviceType.SIMPLE_SENSOR
      Input Clusters:
        Basic (0)
        Power Configuration (1)
        Identify (3)
        Binary Input (Basic) (15)
        Manufacturer Specific (64512)
```

rcloran, 2017-07-07:
You may be interested in:

 - https://github.com/rcloran/home-assistant/commits/remotes
 - https://github.com/rcloran/bellows/tree/dup-init

The main thing you missed is the difference between input and output clusters. Those branches include the changes for that (I'm waiting on some other merges to home-assistant before I can request a merge of the input/output cluster split).

The remote works fine when it is first connected. I've presented it as a binary sensor with a level. If you leave the remote for a while without using it, it fails to properly reconnect to the network. I haven't figured out what needs to be done for that to work.

mjg59, 2017-07-07:
Thanks! I have root on a Hue hub, so I'll see if I can dump the Zigbee traffic there.

rcloran, 2017-07-07:
That may be useful, but it'll almost certainly use the zll profile, and the differences between zll and zha may be important. 

I did some research yesterday, and it looks like it needs some changes in the way that bellows configures the radio to handle rejoins, but I haven't figured out exactly what from the docs yet.

scottcmace, 2018-01-04:
I'm fairly new to zigbee, but I'm not understanding exactly what's missing here... after today's "Smart"things outage, I'm renewing my push to get everything on Home Assistant... and  I rely on these Hue buttons a little too much.

So, as above, bellows finds the device, sees that its a "SCENE_CONTROLLER", adds it to the database, and then this  is where with my Cree bulbs I saw HASS setting up the device... I see nothing past the Hue buttons joining the network.  The device types match the zll.py file, and the output clusters line up as well, so where is the next step taking place beyond it joining the network?

untergeek, 2018-01-04:
@scottcmace 

> after today's "Smart"things outage, I'm renewing my push to get everything on Home Assistant...

I don't have Hue buttons, but I'm with you on that push.  I just unplugged my SmartThings hub last night.  Best feeling ever!

scottcmace, 2018-01-21:
So, is the problem that the device type isn't in the table, or is it that HA doesn't know how to handle a device of this type?

This is my single biggest hurdle at this point... these infernal buttons.

egorian, 2018-01-25:
I'm on the same boat with a Lutron Bulb Remote, it get registered on Bellows but after that, buttons events doesn't appear to be registered.

Device:
  NWK: 0x1e96
  IEEE: ff:ff:00:0f:e7:ff:96:dd
  Endpoints:
    1: profile=0xc05e, device_type=DeviceType.CONTROLLER
      Input Clusters:
        Basic (0)
        LightLink (4096)
        Manufacturer Specific (64580)
        Manufacturer Specific (65280)
      Output Clusters:
        Basic (0)
        Identify (3)
        Groups (4)
        Scenes (5)
        On/Off (6)
        Level control (8)
        LightLink (4096)
        Manufacturer Specific (65280)

rcloran, 2018-02-05:
Please re-open in the new zigpy project (https://github.com/zigpy/zigpy) if this is still an issue. There appears to be no way to move issues between projects.

Since these remotes are all implemented differently, I think the only sensible approach is to add proper support for custom device type handlers outside of zigpy, and implement each of them separately.

closed 2018-02-05 by rcloran