zigpy/bellows #32: Iris Motion sensor breaks zha in 0.3.2

zombu2, 2017-07-17:
It seems iris sensors break the zha database when added to homeassistant

When DB is deleted and HA is started it creates a new db and everything is fine .
As soon as a gen 2 iris motion sensor is added it works fine until restart of HA.

after restart 
```
2017-07-17 17:24:12 INFO (MainThread) [homeassistant.loader] Loaded zha from homeassistant.components.zha
2017-07-17 17:24:17 INFO (MainThread) [homeassistant.setup] Setting up zha
2017-07-17 17:24:18 ERROR (MainThread) [homeassistant.setup] Error during setup of component zha
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/zha/__init__.py", line 87, in async_setup
* [zha](https://home-assistant.io/components/zha/)
* [zha](https://home-assistant.io/components/zha/)

```
adding other devices like hue motion sensor or remote does not break anything just IRIS motion sensor 
any ideas?

rcloran, 2017-07-18:
Do you have any more stack trace information in the logs? This looks incomplete.

zombu2, 2017-07-18:
well i can reproduce the problem if you let me know what you need i can get it if you tell me where to look

rcloran, 2017-07-18:
Is there no stack trace after what you put in the initial report? If not, I suspect it's around reading the database, can you add a dump of your bellows database here? (`sqlite3 zigbee.db .dump`, where zigbee.db is the path to your bellows database).

zombu2, 2017-07-18:
k gimme a minute i try to get you all the stuff

zombu2, 2017-07-18:
here is what i found 
this is when the iris motion sensor is beeing added 
for some reason it added iris motion sensors that are not even in pairing mode 
next post i post everything that comes when i restart hass

```
2017-07-17 23:53:15 INFO (MainThread) [homeassistant.components.zha] Permitting joins for 60s
2017-07-17 23:53:20 INFO (MainThread) [homeassistant.core] Bus:Handling <Event platform_discovered[L]: platform=zha, discovered=discovery_key=00:0d:6f:00:0e:25:fe:1a-1-1026, service=load_platform.sensor>
2017-07-17 23:53:20 INFO (MainThread) [homeassistant.core] Bus:Handling <Event platform_discovered[L]: platform=zha, discovered=discovery_key=00:0d:6f:00:0e:25:fe:1a-1-1280, service=load_platform.binary_sensor>
2017-07-17 23:53:20 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.zha
2017-07-17 23:53:20 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.zha
```

```
2017-07-17 23:48:57 INFO (MainThread) [bellows.zigbee.application] Device 0x6096 (00:0d:6f:00:0e:35:8d:eb) joined the network
2017-07-17 23:48:57 INFO (MainThread) [bellows.zigbee.device] [0x6096] Discovering endpoints
2017-07-17 23:48:58 INFO (MainThread) [bellows.zigbee.device] [0x6096] Discovered endpoints: [1, 2]
2017-07-17 23:48:58 INFO (MainThread) [bellows.zigbee.endpoint] [0x6096:1] Discovering endpoint information
2017-07-17 23:48:59 INFO (MainThread) [bellows.zigbee.application] Device 0xd3c4 (00:0d:6f:00:0e:25:fe:bf) joined the network
2017-07-17 23:48:59 INFO (MainThread) [bellows.zigbee.device] [0xd3c4] Discovering endpoints
2017-07-17 23:49:00 INFO (MainThread) [bellows.zigbee.device] [0xd3c4] Discovered endpoints: [1, 2]
2017-07-17 23:49:00 INFO (MainThread) [bellows.zigbee.endpoint] [0xd3c4:1] Discovering endpoint information
2017-07-17 23:49:01 INFO (MainThread) [bellows.zigbee.endpoint] [0xd3c4:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 32, 1026, 1280, 2821] output_clusters=[25]>
2017-07-17 23:49:01 INFO (MainThread) [bellows.zigbee.endpoint] [0xd3c4:2] Discovering endpoint information
2017-07-17 23:49:01 INFO (MainThread) [bellows.zigbee.application] Device 0xe067 (00:0d:6f:00:0e:26:06:9b) joined the network
2017-07-17 23:49:01 INFO (MainThread) [bellows.zigbee.device] [0xe067] Discovering endpoints
2017-07-17 23:49:02 INFO (MainThread) [bellows.zigbee.endpoint] [0xd3c4:2] Discovered endpoint information: <SimpleDescriptor endpoint=2 profile=49887 device_type=263 device_version=0 input_clusters=[0, 1, 3, 2821, 64582] output_clusters=[3]>
2017-07-17 23:49:02 WARNING (MainThread) [bellows.zigbee.util] Error calling listener.device_initialized: No such command name: cluster_id
2017-07-17 23:49:02 INFO (MainThread) [bellows.zigbee.device] [0xe067] Discovered endpoints: [1, 2]
2017-07-17 23:49:02 INFO (MainThread) [bellows.zigbee.endpoint] [0xe067:1] Discovering endpoint information
2017-07-17 23:49:03 INFO (MainThread) [bellows.zigbee.endpoint] [0xe067:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 32, 1026, 1280, 2821] output_clusters=[25]>
2017-07-17 23:49:03 INFO (MainThread) [bellows.zigbee.endpoint] [0xe067:2] Discovering endpoint information
2017-07-17 23:49:04 INFO (MainThread) [bellows.zigbee.application] Device 0xd470 (00:0d:6f:00:0e:35:5c:d4) joined the network
2017-07-17 23:49:04 INFO (MainThread) [bellows.zigbee.device] [0xd470] Discovering endpoints
2017-07-17 23:49:04 INFO (MainThread) [bellows.zigbee.endpoint] [0xe067:2] Discovered endpoint information: <SimpleDescriptor endpoint=2 profile=49887 device_type=263 device_version=0 input_clusters=[0, 1, 3, 2821, 64582] output_clusters=[3]>
2017-07-17 23:49:04 WARNING (MainThread) [bellows.zigbee.util] Error calling listener.device_initialized: No such command name: cluster_id
2017-07-17 23:49:05 INFO (MainThread) [bellows.zigbee.device] [0xd470] Discovered endpoints: [1, 2]
2017-07-17 23:49:05 INFO (MainThread) [bellows.zigbee.endpoint] [0xd470:1] Discovering endpoint information
2017-07-17 23:49:05 INFO (MainThread) [bellows.zigbee.endpoint] [0x6096:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 32, 1026, 1280, 2821] output_clusters=[25]>
2017-07-17 23:49:05 INFO (MainThread) [bellows.zigbee.endpoint] [0x6096:2] Discovering endpoint information
2017-07-17 23:49:06 INFO (MainThread) [bellows.zigbee.endpoint] [0xd470:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 32, 1026, 1280, 2821] output_clusters=[25]>
2017-07-17 23:49:06 INFO (MainThread) [bellows.zigbee.endpoint] [0xd470:2] Discovering endpoint information
2017-07-17 23:49:06 INFO (MainThread) [bellows.zigbee.endpoint] [0x6096:2] Discovered endpoint information: <SimpleDescriptor endpoint=2 profile=49887 device_type=263 device_version=0 input_clusters=[0, 1, 3, 2821, 64582] output_clusters=[3]>
2017-07-17 23:49:06 WARNING (MainThread) [bellows.zigbee.util] Error calling listener.device_initialized: No such command name: cluster_id
2017-07-17 23:49:07 INFO (MainThread) [bellows.zigbee.endpoint] [0xd470:2] Discovered endpoint information: <SimpleDescriptor endpoint=2 profile=49887 device_type=263 device_version=0 input_clusters=[0, 1, 3, 2821, 64582] output_clusters=[3]>
[... cut ...]
2017-07-17 23:54:36 WARNING (MainThread) [bellows.zigbee.zcl] [0x1e2e:1:0x0019] No handler for cluster command 1
2017-07-17 23:54:39 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-17 23:54:39 WARNING (MainThread) [bellows.zigbee.zcl] [0x1e2e:1:0x0019] No handler for cluster command 1
2017-07-17 23:54:42 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-17 23:54:42 WARNING (MainThread) [bellows.zigbee.zcl] [0x1e2e:1:0x0019] No handler for cluster command 1
```

zombu2, 2017-07-18:
2017-07-18 00:03:11 ERROR (MainThread) [homeassistant.setup] Error during setup of component zha
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/setup.py", line 187, in _async_setup_component
    result = yield from component.async_setup(hass, processed_config)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/zha/__init__.py", line 87, in async_setup
    APPLICATION_CONTROLLER = ControllerApplication(ezsp_, database)
  File "/etc/homeassistant/deps/bellows/zigbee/application.py", line 30, in __init__
    self._dblistener.load()
  File "/etc/homeassistant/deps/bellows/zigbee/appdb.py", line 204, in load
    clus = ep.in_clusters[cluster]
KeyError: 0

rcloran, 2017-07-18:
OK, that KeyError suggests database badness. I suspect your device is leaving and rejoining the network (but I don't see that in the logs). I'm about to push a branch which will stop Home Assistant from exploding on load. https://github.com/rcloran/bellows/commits/dup-init may also help, and then the actual fix I'll get ready soon.

zombu2, 2017-07-18:
cool cool 
these sensors where working perfect for month now just with 0.49 it exploded haha
if you need to i can let you have a look at my HA for debugging purposes 

Its also very hard to get HA to issue the permit command the call button needs to be pushed multiple times 

what i really don't get why would all the sensors that are not in pairing mode also pair ?
it did that earlyer with one of my ZLL bulbs ...it paird without the bulb in pair mode right after HA restartand fresh DB

rcloran, 2017-07-18:
Oh ... I think this should fix, actually: https://github.com/rcloran/bellows/tree/manuf-specific-fix

zombu2, 2017-07-18:
404

rcloran, 2017-07-18:
Oh, I just deleted it because it's merged with master now.

zombu2, 2017-07-18:
well i replaced the file and still hard time to even get zha permit called and if it does it adds sensors that are not even in pairing mode but not the one it's supposed to lol

rcloran, 2017-07-18:
Does it fix the problem that this bug is about?

closed 2017-07-22 by rcloran