zigpy/bellows #33: Upgrade to 0.3.2 (ha .49)

cytech, 2017-07-18:
CLI - attempt to use -D option returns
Error: no such option: -D
----------
after upgrade, clusters 6 &  8 were added to attributes for 3 of 5 identical lights.
manually added the other 2. These were Sengled bulbs.
A note to anyone editing the database with DB browser for Sqlite;
when a record is added, it stores the integer as a text (1  -> '1') which causes a key error upon startup.
I am not sure how to get around this in DB browser for Sqlite, but i had to export database to sql, correct the offending fields in text editor, and reimport.
---------------
Upgrade created empty output_clusters table.
manually added ieee, endpoint_id, cluster  (xx:xx:xx:xx:xx:xx:xx:xx    1    25)
upon restart still receiving different errors addressed in closed issue # 23
now i get a "No handler for cluster command"
---
2017-07-18 08:16:44 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-18 08:16:44 WARNING (MainThread) [bellows.zigbee.zcl] [0xb559:1:0x0019] No handler for cluster command 1
2017-07-18 08:16:47 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-18 08:16:47 WARNING (MainThread) [bellows.zigbee.zcl] [0xb559:1:0x0019] No handler for cluster command 1
2017-07-18 08:16:50 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-18 08:16:50 WARNING (MainThread) [bellows.zigbee.zcl] [0xb559:1:0x0019] No handler for cluster command 1
2017-07-18 08:16:53 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-18 08:16:53 WARNING (MainThread) [bellows.zigbee.zcl] [0xb559:1:0x0019] No handler for cluster command 1
2017-07-18 08:16:56 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-18 08:16:56 WARNING (MainThread) [bellows.zigbee.zcl] [0xb559:1:0x0019] No handler for cluster command 1
2017-07-18 08:17:23 WARNING (MainThread) [bellows.zigbee.application] Unexpected message send notification
2017-07-18 08:17:25 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-18 08:17:25 WARNING (MainThread) [bellows.zigbee.zcl] [0xada6:1:0x0019] No handler for cluster command 1
2017-07-18 08:17:28 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-18 08:17:28 WARNING (MainThread) [bellows.zigbee.zcl] [0xada6:1:0x0019] No handler for cluster command 1
2017-07-18 08:17:31 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-18 08:17:31 WARNING (MainThread) [bellows.zigbee.zcl] [0xada6:1:0x0019] No handler for cluster command 1
2017-07-18 08:17:34 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-18 08:17:34 WARNING (MainThread) [bellows.zigbee.zcl] [0xada6:1:0x0019] No handler for cluster command 1
2017-07-18 08:17:37 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-18 08:17:37 WARNING (MainThread) [bellows.zigbee.zcl] [0xada6:1:0x0019] No handler for cluster command 1
2017-07-18 08:17:46 WARNING (MainThread) [bellows.zigbee.application] Unexpected message send notification
2017-07-18 08:17:48 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-18 08:17:48 WARNING (MainThread) [bellows.zigbee.zcl] [0x5f0d:1:0x0019] No handler for cluster command 1
2017-07-18 08:17:51 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-18 08:17:51 WARNING (MainThread) [bellows.zigbee.zcl] [0x5f0d:1:0x0019] No handler for cluster command 1
2017-07-18 08:17:54 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-18 08:17:54 WARNING (MainThread) [bellows.zigbee.zcl] [0x5f0d:1:0x0019] No handler for cluster command 1
2017-07-18 08:17:57 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-18 08:17:57 WARNING (MainThread) [bellows.zigbee.zcl] [0x5f0d:1:0x0019] No handler for cluster command 1
2017-07-18 08:18:00 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-07-18 08:18:00 WARNING (MainThread) [bellows.zigbee.zcl] [0x5f0d:1:0x0019] No handler for cluster command 1

rcloran, 2017-07-18:
The warning log messages are innocuous. -D exists (some output would help me point you in the right direction, but a bug report is not the best place to ask how to use a program). The rest is hard to understand, or not a bug report.

closed 2017-07-18 by rcloran

cytech, 2017-07-18:
Hi Russell,
I apologize for not following proper protocol. I am pretty new to this.

I was not really asking how to use the program. In the prior version I hardcoded the database in opts.py.
I saw the new -D option in the new version and tried
 bellows  -d/dev/ttyUSB1 -D/home/homeassistant/.homeassistant/zigbee-0.3.2.db devices
which returned the Error: no such option: -D

second comment was just trying to explain that the new version upgraded existing attributes but not on all devices. (and an issue I ran into using DB Browser to edit)

last was self explanatory. It corrected my original report of issue #23 but introduced a new set of errors to the log. i will continue to block the bellows warning with 
logger:
  default: warning
  logs:
    bellows.zigbee.application: critical
    bellows.zigbee.zcl: critical
    bellows.zigbee.endpoint: critical
in HA

thank you
dave

rcloran, 2017-07-18:
-D goes after "devices".

cytech, 2017-07-18:
thanx