zigpy/bellows #35: ZHA Permit not working in homeassistant and other oddities

zombu2, 2017-07-26:
Since 0.49 i'm having trouble with zha ...now the exploding database has been fixed but zha permit does not work when called and zha just randomly adds everything it can get a hold of without running permit it even adds a19 bulbs that are paired with hue including the hue motion sensors and remotes .
Even iris sensors that are not in pair mode randomly get added throughout the day without permission but devices you want to pair will not .

Any help would be appreciated 
thnx 

if needed i can give access to HA to get to the bottom of this

rolf-tx, 2017-08-01:
I have trouble adding zigbee devices unless I disable zwave before doing the zigbee permit

rolf-tx, 2017-10-08:
@montagp in configuration.yaml comment out the section starting with “zwave:” and restart HA

closed 2018-02-05 by rcloran