zigpy/bellows #40: First steps - documentation update

andyka, 2017-08-11:
I got my hands on Telegesis ETRX3USB8M stick (ETRX357 via USB) and I am trying to make it work with HA. Since HA has trouble starting `zha` (`Setup of zha is taking over 10 seconds.`), I figured I would try using bellows directly first. 

This issue is about the documentation (`README.md`) deficiencies. I think it should be updated, however, I don't feel like I know enough to do a PR. If you wish I can try though.

*Versions:*
```
# python3 --version
Python 3.6.2
# pip3 freeze | grep bellows
bellows==0.3.4
```

Following the instructions in README:

```
# bellows devices
Usage: bellows [OPTIONS] COMMAND [ARGS]...

Error: Missing option "-d" / "--device".
```

Trying again:
```
# bellows -d /dev/ttyUSB0 devices
Traceback (most recent call last):
  File "/usr/local/bin/bellows", line 11, in <module>
    load_entry_point('bellows==0.3.4', 'console_scripts', 'bellows')()
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/bellows/cli/application.py", line 69, in devices
    app = bellows.zigbee.application.ControllerApplication(ezsp, database)
  File "/usr/local/lib/python3.6/site-packages/bellows/zigbee/application.py", line 29, in __init__
    self._dblistener = bellows.zigbee.appdb.PersistingListener(database_file, self)
  File "/usr/local/lib/python3.6/site-packages/bellows/zigbee/appdb.py", line 29, in __init__
    detect_types=sqlite3.PARSE_DECLTYPES)
sqlite3.OperationalError: unable to open database file
```

But it looks like database file is not a parameter I can specify?

```
# bellows --help
Usage: bellows [OPTIONS] COMMAND [ARGS]...

Options:
  -v, --verbosity LVL  Either CRITICAL, ERROR, WARNING, INFO or DEBUG
  -d, --device TEXT    [required]
  --help               Show this message and exit.

Commands:
  config   Get/set configuration on the NCP
  devices  Show device database
  dump     Capture frames on CHANNEL and write to FILE...
  form     Form a new ZigBee network
  info     Get NCP information
  join     Join an existing ZigBee network as an end...
  leave    Leave the ZigBee network
  permit   Allow devices to join this ZigBee network
  scan     Scan for networks or radio interference
  zcl      Peform ZCL operations against a device
  zdo      Perform ZDO operations against a device
```

Correct me if I'm wrong, but I think the more relevant command for starting would be:

```
# bellows -v debug -d /dev/ttyUSB0 info
debug: Using selector: EpollSelector
debug: Connected. Resetting.
debug: Sending: b'1ac038bc7e'
```

(this command hangs on me, but I am guessing this is because I didn't upgrade the firmware, so I'll start looking how to do that now - any hints appreciated ;) )

rcloran, 2017-08-11:
The README is not supposed to be complete documentation. I'd be happy to review PRs for wiki pages which document the CLI better. I've updated the README to include showing `export EZSP_DEVICE=/dev/ttyUSB1`, which is usually easier than specifying the -d on every invocation.

Options are also positional -- that is, database parameter is a parameter of the "devices" subcommand (and many other subcommands).

```
$ export EZSP_DEVICE=/dev/ttyUSB1
$ bellows devices --help
Usage: bellows devices [OPTIONS]

  Show device database

Options:
  -D, --database PATH
  --help               Show this message and exit.
```

andyka, 2017-08-17:
Thank you for your fast response - and sorry for answering this late, I didn't get the e-mail. :-/

In the mean time I gave up trying to upgrade firmware (I couldn't find a way to do it) and focused my efforts on XBee XStick2 ZB instead (using python-xbee library)... I ordered different ZB sticks because I need to make at least one of them to work. 

As far as I am concerned, adding environment variable to README mostly solves this issue, so I guess this issue can be closed? Feel free to reopen.

closed 2017-08-17 by andyka

nk3750, 2019-06-11:
Hi, can i get some instructions/ steps to get started with zigpy and bellows? I'm new to the filed and has been trying to set it up for past 1.5 days now.

Adminiuga, 2019-06-11:
1. Get Home Assistant running 1st. https://www.home-assistant.io/getting-started/
2. Once running, configure ZHA integration from HA UI -> Configuration -> Integration -> ZHA
you'll need to know the Serial Port you radio is connected to and type of the radio: EZSP - HusBZB-1, Telegis, Elelabs shield, XBee based or Deconz.

nk3750, 2019-06-11:
Thank you for the quick response, i just bought a GoControl CECOMINOD016164 HUSBZB-1 USB Hub and i'm trying to get it work with bellows. I don't have use for the HA as yet, i want to communicate directly with the zigbee device through the USB hub, basic command attributes communication. How do i go forward with that?

nk3750, 2019-06-11:
What exactly is a db file in bellows? Do i need to create it seperately?

Adminiuga, 2019-06-11:
create `~/.config/bellows` folder in your home directory, bellows will create the db automatically. Otherwise you have to specify `-D /path/to/dbfile.db` option for bellow CLI.

Set environment variable `EZSP_DEVICE=/dev/ttyUSB1` pointing to the serial port of the Zigbee radio. Keep in mind, GoControl has two serial ports. Usually port #0 is ZWave and port #1 is Zigbee. 
I recommend to setup that variable to `/dev/serial/by-id/xxx` as those won't change. Alternatively you could use `-d /dev/ttyUSB1` CLI parameter.

bellows CLI was really designed for testing purposes, so for any further questions check the source code 1st.

nk3750, 2019-06-11:
Thank you so much for the prompt response, i'll surely go ahead and check the source code first before posting any more questions.

nk3750, 2019-06-11:
I do have one more question though, what are the zigbee devices i can buy which i'll be able to communicate with through GoControl on ZigBee? I'm sorry if the question doesn't belong here.

Adminiuga, 2019-06-11:
Most Zigbee HA Profile 1.2 devices should be compatible. Xiaomi devices are not quite following the specs, so require some workarounds not present in `bellows`. Just keep in mind, `bellows` is really intended to be used as a module and not as a standalone app. 

Check https://community.home-assistant.io/t/zha-zigbee-tested-devices-please-add-your-device-results/17718/978 for more details. 

And still, unless you are working on a research project, you probably want to use home assistant with ZHA components. It does exactly what you are asking: communicates with devices directly using HUSBZB-1 stick. `Bellows` is not a hub, it is just a library for communication with EZSP based radios. Home Assistant is your smart hub.

nk3750, 2019-06-11:
Thank you for the detailed clarification, i'm actually working on a research project and won't be needing the  smart hub as for now. I'll go through the list and order any one of the working devices for now. Again, thank you so much for all your help.

nk3750, 2019-06-26:
Hi,
I'm getting this error in another setup i did for bellows 
Invalid value for "-D" / "--database": File "/root/.config/bellows/app.db" does not exist.
i have to specify -D option every time. Is there any way i can escape this? In my other environment bellows CLI is working without passing the DB file every time. Is there a reason for this?

nk3750, 2019-07-09:
Hi, i'm trying to list all the attributes supported by a particular cluster for a device using bellows. Is there a way to go forward? I don't see the support for that currently in the bellows.

Adminiuga, 2019-07-09:
`bellows` is just the library for the radio interface. It extends `https://github.com/zigpy/zigpy` library which implements the actual Zigbee protocol + ZCL (Zigbee Cluster Library) + ZDO (Zigbee Device Object)

Google for `Zigbee Cluster Library` you should be able to find `revision 6` of the document, which defines Clusters and associated attributes + cluster commands specific for the cluster. Keep in mind, most devices do not implement all attributes or commands of a cluster. Silabs had some good videos on youtube about Zigbee fundamentals.

nk3750, 2019-07-09:
Thank you for the prompt response. 
I'm aware that most devices don't implement all the attributes and the commands of the cluster. I'm just trying to extract the attributes supported by a given cluster on a particular device.

Adminiuga, 2019-07-10:
`bellows devices -D yourzigbee.db` would print all devices and for each device: all endpoint + input and output clusters. 
To see what actually supported on a particular cluster of a device, you need to send ZCL command: `discover_attributes_extended` and ZCL commands:  `discover_commands_recevied`, `discover_commands_generated`

See https://github.com/Adminiuga/zha_custom/blob/80e82ac910c1508147652e94e02d79135c4f6173/scan_device.py#L24 for an example

nk3750, 2019-07-10:
Hi, does the current version of bellows module on github support this feature? As per my understanding, the currently supported features are:
  command
  commands
  configure-reporting
  read-attribute
  write-attribute
Is there any way to integrate this code somewhere in the zigpy and bellows module to support this?

nk3750, 2019-07-10:
Thank you for the guidance, i was able to play with the code a bit and find a way to do it.

nk3750, 2019-07-11:
You have provided CLI for doing all the stuff related to zigbee, i am trying to convert those CLI in to API, meaning trying to return values and call the methods instead of using CLI. The deep integration of click in the whole project is making it really difficult. Do you have any advice regarding the same?

Adminiuga, 2019-07-11:
Not quite so. `bellows` and `zigpy` is the API. CLI for bellows is built on top of that API.

Adminiuga, 2019-07-12:
Essentially you need to 
- start an asyncio loop
- instantiate `bellows.ezsp.EZSP`
- connect EZSP instance to the serial port
- Instantiate `bellows.zigbee.application.ContollerApplication` with EZSP instance
- call `startup()` method of the ControllerApplication instance

see https://gist.github.com/Adminiuga/494a2574d96b1be98ca03d52b374fc52 for an example.

But maybe it easier just to use HA

nk3750, 2019-07-12:
Hi, i tried the following and i'm getting this error, 
zigpy.exceptions.DeliveryError: [0x1511:1:0x0000]: message send failure: EmberStatus.DELIVERY_FAILED
Any clue what might be causing this?

dmulcahey, 2019-07-12:
Are you trying to do this with a battery powered device? If so, they sleep frequently and you will have to interact with the device while issuing the commands in order to ensure that it stays awake.

nk3750, 2019-07-12:
I think there is some problem with the device network address as well, bellows CLI is giving different network address, the network address i'm getting using this script by printing 
print(app.devices)
    for dev in app.devices.values():
        print(dev.nwk)
        print(dev.ieee)
is different.

nk3750, 2019-07-12:
When i'm entering the correct network address, it's throwing a key error, so basically the real network address and the network address i'm getting through this script  https://gist.github.com/Adminiuga/494a2574d96b1be98ca03d52b374fc52 are different, trying to figure out the reason.

nk3750, 2019-07-12:
One more issue i have is, **bellows devices** keeps showing the devices even when the device isn't available right now. What i mean is, once it has an entry for that device in the database, no matter if the devices is actually connected or not, it will keep showing the device.

Adminiuga, 2019-07-12:
Yes, that's by design. Look at zigpy code for more info

nk3750, 2019-08-07:
Hi, I just shifted my whole prject from a ubuntu VM to raspberry pi and i'm getting this warnng message all the time. 
warning: <class 'bellows.zigbee.application.EZSPCoordinator'> signature update is required. Support for `signature = {1: { replacement }}` will be removed in the future releases. Use `signature = {'endpoints': {1: { replacement }}}
What exactly might be causing this?

smdjeff, 2019-11-17:
The way responsibilities are divided between the separate projects is a bit confusing. Clearly documenting the layered responsibilities of bellows, zigpy, and home assistant would go a long way to helping folks get started. Right up top on the readmes.

> `bellows` is just the library for the radio interface. It extends `https://github.com/zigpy/zigpy` library which implements the actual Zigbee protocol + ZCL (Zigbee Cluster Library) + ZDO (Zigbee Device Object)

junsaw, 2020-02-09:
I am trying to use the bellows command, I am getting the following errors:

bellows devices
Usage: bellows devices [OPTIONS]
Try "bellows devices --help" for help.

Error: Invalid value for "-D" / "--database": File "/root/.config/bellows/app.db" does not exist.

Per the instructions above it should create the database automatically, am I missing something or doing something incorrect. 

Any help will be much appreciated.

Adminiuga, 2020-02-09:
it creates the database if you form/start the network, **but** it is not going to create database path. in other words, the folder where database would be located should exist before running the command.

smdjeff, 2020-02-09:
Why not?

smdjeff, 2020-02-09:
The error message is a little misleading as it suggests the problem is the database not the folder. In any case, why doesn't it create the folder?

bbrummer, 2020-02-17:
Stupid bug is stupid; It makes the example in the project's readme file more like a giant middle finger than a welcome letter.  Here's a workaround:

touch $HOME/.config/bellows/app.db

bellows form

bellows devices