zigpy/bellows #42: Regular freeze during dump of `form`

lesteenman, 2017-08-22:
While trying to figure out why my new Xiaomi devices don't work with Bellows, I'm attempting to make a dump to check what is happening in WireShark. I'm starting the dump as follows:

```
bellows -d /dev/ttyUSB0 -v INFO dump -c 15 -w zigbee_connect_dump
```

After that, I form a new network with a fresh database:
```
bellows -d /dev/ttyUSB0 -v DEBUG form -D zigbee.db
```

After a bit of output, however, both the dump and the `form` seem to freeze. This is the output where the `form` halts:
```
debug: Using selector: EpollSelector
debug: Loading application state from zigbee2.db
debug: Sending: b'1ac038bc7e'
```

and the following output is printed in the window where I'm running `dump`:

```
Capture started
Exception in callback SerialTransport._read_ready()
handle: <Handle SerialTransport._read_ready()>
Traceback (most recent call last):
  File "/usr/lib/python3.4/asyncio/events.py", line 120, in _run
    self._callback(*self._args)
  File "/home/pi/python3venv/lib/python3.4/site-packages/serial_asyncio/__init__.py", line 106, in _read_ready
    self._protocol.data_received(data)
  File "/home/pi/python3venv/lib/python3.4/site-packages/bellows-0.3.5-py3.4.egg/bellows/uart.py", line 61, in data_received
    self.frame_received(frame)
  File "/home/pi/python3venv/lib/python3.4/site-packages/bellows-0.3.5-py3.4.egg/bellows/uart.py", line 81, in frame_received
    self.rstack_frame_received(data)
  File "/home/pi/python3venv/lib/python3.4/site-packages/bellows-0.3.5-py3.4.egg/bellows/uart.py", line 128, in rstack_frame_received
    self._reset_future.set_result(True)
  File "/usr/lib/python3.4/asyncio/futures.py", line 338, in set_result
    raise InvalidStateError('{}: {!r}'.format(self._state, self))
asyncio.futures.InvalidStateError: FINISHED: <Future finished result=True>
```

After this, neither of the terminals give any output anymore until I kill the program.

This does not always happen, but it does most of the times. I'm using build 0.3.5.

lesteenman, 2017-08-22:
This actually does always seem to happen during a `form`. I confused it with `permit`, where this does not (always) happen.

Still a bug, I assume.

rcloran, 2017-08-22:
Are you saying you're running a dump in one terminal, and a form in another? That won't work (and there's no sensible way of supporting it unless you have a second radio).

lesteenman, 2017-08-22:
Well, that would explain it. That is what I'm doing. Is there another way to make a dump of what happens during a form?

rcloran, 2017-08-22:
"Form" does not transmit anything on the air (well, not anything interesting, and not anything bellows can control -- it's handled by the NCP). The debug logs include all communications to/from the NCP.

lesteenman, 2017-08-22:
Ok, thank you. I preferred WireShark to visualize the logs, but this can be closed.

closed 2017-08-22 by lesteenman

rcloran, 2017-08-22:
There's enough information in the logs to transform them to pcap files. I'd be happy to review a pull request to do that.