zigpy/bellows #43: Issue connecting Xiaomi devices

lesteenman, 2017-08-22:
I'm attempting to couple the following devices of Xiaomi using Bellows, to use them with Home Assistant: [Xiaomi Smart Door and Window Sensor, Human Body sensor and Wireless switch](https://www.gearbest.com/smart-light-bulb/pp_257677.html). However, after connecting them using `bellows permit`, I get `Message on unknown endpoint 1` errors.

As connecting seems a bit finnicky, it took me a while to get a somewhat reliable log. In this particular case, another exception was thrown by Bellows right after it started to look for endpoints. I think this might have been the case because the button was clicked right after (it is almost impossible to click the linking button without actually triggering the button).

It seems that the Active Endpoint Request never receives a response, due to which it never registers the endpoints.

I'm not well known with the whole Zigbee protocol stack, although I have been trying to read up on it. However, I have no device that does work to compare logs with, so I'm really not sure what I'm looking for in this case. I'm hoping for some input with this issue.

When the button is clicked without the endpoints being found, the following is logged:
```
2017-08-23 00:11:05.462 debug: Data frame: b'304db1ed542e14b459954b25ab5592ca7bf5bbf912316093fecc6389ec7ed1357e'
2017-08-23 00:11:05.471 debug: Sending: b'8430fc7e'
2017-08-23 00:11:05.480 debug: Application frame 69 (incomingMessageHandler) received
2017-08-23 00:11:05.491 warning: [0x529c] Message on unknown endpoint 1
2017-08-23 00:11:05.668 debug: Data frame: b'404db1ed542e14b459954b25ab5592cd72f5bbf912316093f9cc6389ec7f247b7e'
2017-08-23 00:11:05.678 debug: Sending: b'8520dd7e'
2017-08-23 00:11:05.688 debug: Application frame 69 (incomingMessageHandler) received
2017-08-23 00:11:05.698 warning: [0x529c] Message on unknown endpoint 1
```

Full log of `bellows -d /dev/ttyUSB0 -v DEBUG permit -t 150 -D zigbee.db`:

```
2017-08-23 00:10:23.100 debug: Using selector: EpollSelector
2017-08-23 00:10:23.308 debug: Loading application state from zigbee3.db
2017-08-23 00:10:23.320 debug: Sending: b'1ac038bc7e'
2017-08-23 00:10:23.449 debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
2017-08-23 00:10:23.461 debug: Send command version
2017-08-23 00:10:23.477 debug: Sending: b'004221a850ed2c7e'
2017-08-23 00:10:23.488 debug: Data frame: b'0142a1a8502815e23c957e'
2017-08-23 00:10:23.499 debug: Sending: b'8160597e'
2017-08-23 00:10:23.511 debug: Application frame 0 (version) received
2017-08-23 00:10:23.520 debug: Send command setConfigurationValue
2017-08-23 00:10:23.538 debug: Sending: b'7d314321fb582815c3277e'
2017-08-23 00:10:23.555 debug: Data frame: b'1243a1fb54fb737e'
2017-08-23 00:10:23.572 debug: Sending: b'82503a7e'
2017-08-23 00:10:23.585 debug: Application frame 83 (setConfigurationValue) received
2017-08-23 00:10:23.596 debug: Send command setConfigurationValue
2017-08-23 00:10:23.610 debug: Sending: b'224021fb592f15226f7e'
2017-08-23 00:10:23.631 debug: Data frame: b'2340a1fb54c6107e'
2017-08-23 00:10:23.641 debug: Sending: b'83401b7e'
2017-08-23 00:10:23.656 debug: Application frame 83 (setConfigurationValue) received
2017-08-23 00:10:23.665 debug: Send command setConfigurationValue
2017-08-23 00:10:23.675 debug: Sending: b'334121fb792b15a2d77e'
2017-08-23 00:10:23.685 debug: Data frame: b'3441a1fb54d32a7e'
2017-08-23 00:10:23.695 debug: Sending: b'8430fc7e'
2017-08-23 00:10:23.707 debug: Application frame 83 (setConfigurationValue) received
2017-08-23 00:10:23.716 debug: Send command setConfigurationValue
2017-08-23 00:10:23.726 debug: Sending: b'444621fb7d5e291514417e'
2017-08-23 00:10:23.736 debug: Data frame: b'4546a1fb5435d07e'
2017-08-23 00:10:23.748 debug: Sending: b'8520dd7e'
2017-08-23 00:10:23.758 debug: Application frame 83 (setConfigurationValue) received
2017-08-23 00:10:23.769 debug: Send command setConfigurationValue
2017-08-23 00:10:23.778 debug: Sending: b'554721fb4d2815713f7e'
2017-08-23 00:10:23.787 debug: Data frame: b'5647a1fb54a9ec7e'
2017-08-23 00:10:23.797 debug: Sending: b'8610be7e'
2017-08-23 00:10:23.806 debug: Application frame 83 (setConfigurationValue) received
2017-08-23 00:10:23.816 debug: Send command setConfigurationValue
2017-08-23 00:10:23.825 debug: Sending: b'664421fb55d515b18f7e'
2017-08-23 00:10:23.835 debug: Data frame: b'6744a1fb54948f7e'
2017-08-23 00:10:23.846 debug: Sending: b'87009f7e'
2017-08-23 00:10:23.856 debug: Application frame 83 (setConfigurationValue) received
2017-08-23 00:10:23.866 debug: Send command networkInit
2017-08-23 00:10:23.876 debug: Sending: b'774521bf02267e'
2017-08-23 00:10:23.886 debug: Data frame: b'7045a5bf549c7d7e'
2017-08-23 00:10:23.895 debug: Sending: b'8070787e'
2017-08-23 00:10:23.906 debug: Application frame 23 (networkInit) received
2017-08-23 00:10:23.916 debug: Data frame: b'0045b1b1c4beee7e'
2017-08-23 00:10:23.925 debug: Sending: b'8160597e'
2017-08-23 00:10:23.935 debug: Application frame 25 (stackStatusHandler) received
[... 120 lines cut ...]
2017-08-23 00:11:23.292 debug: Data frame: b'2152b12816b64797247e'
2017-08-23 00:11:23.302 debug: Sending: b'83401b7e'
2017-08-23 00:11:23.312 debug: Application frame 128 (incomingRouteErrorHandler) received
2017-08-23 00:11:24.858 debug: Data frame: b'3152b19754b647b259914a25aa1593499cf825cded56b67e'
2017-08-23 00:11:24.868 debug: Sending: b'8430fc7e'
2017-08-23 00:11:24.877 debug: Application frame 63 (messageSentHandler) received
2017-08-23 00:11:26.866 debug: Send command sendUnicast
2017-08-23 00:11:26.876 debug: Sending: b'1453219c54b647b259914a25aa1593499c4d24a8ee5235746b7e'
2017-08-23 00:11:26.886 debug: Data frame: b'4253a19c549d15557e'
2017-08-23 00:11:26.896 debug: Sending: b'8520dd7e'
2017-08-23 00:11:26.906 debug: Application frame 52 (sendUnicast) received
2017-08-23 00:11:29.812 debug: Data frame: b'5253b12816b6478e677e'
2017-08-23 00:11:29.821 debug: Sending: b'8610be7e'
2017-08-23 00:11:29.831 debug: Application frame 128 (incomingRouteErrorHandler) received
2017-08-23 00:11:34.305 debug: Data frame: b'6253b12816b647d7ea7e'
2017-08-23 00:11:34.315 debug: Sending: b'87009f7e'
2017-08-23 00:11:34.325 debug: Application frame 128 (incomingRouteErrorHandler) received
2017-08-23 00:11:38.800 debug: Data frame: b'7253b12816b647e0917e'
2017-08-23 00:11:38.809 debug: Sending: b'8070787e'
2017-08-23 00:11:38.819 debug: Application frame 128 (incomingRouteErrorHandler) received
2017-08-23 00:11:40.365 debug: Data frame: b'0253b19754b647b259914a25aa1593499cf924cdedcfcc7e'
2017-08-23 00:11:40.374 debug: Sending: b'8160597e'
2017-08-23 00:11:40.384 debug: Application frame 63 (messageSentHandler) received
2017-08-23 00:11:40.393 warning: [0x529c] Failed ZDO request during device initialization: Message send failure: EmberStatus.DELIVERY_FAILED
```

lesteenman, 2017-08-23:
I currently have a Philips Hue bulb that succesfully connects. Comparing the logging for both the Xiaomi device and the Hue:

- The Xiamoi device usually causes a crash in Bellows, as above
- With the Hue device, a message on the ZDO channel (0) arrives before the _discovering endpoints_ log. The Xiamoi device only receives this ZDO message after this log.
- A much longer data frame received after the `sendUniCast` that's sent in the try/catch to retrieve the channel information. Data frame for both:
  - Hue:  `b'304db1975464cdb259914a25aa1593499c2f26abed05f57e'`
  - Xiaomi: 
`b'304db1ed502a15a159944a2dab5592c162a4eab612316b6630db25617a7f3f2afecd5e4cdc7e'`
- The Xiaomi device does not receive a `messageSentHandler` after receiving the `sendUnicast` applicaiton frame. 
- The Xiaomi device never prints `Discovered endpoints`, while the Hue does. I assume this is the symptom, not related to a cause.

After this dataframe, both the Hue and the Xiaomi send/receive (not sure how to read the log, I think it's send) a ZDO request 0x0013:
 - Hue: `[0xd84e:zdo] ZDO request 0x0013: [55374, 00:17:88:01:10:50:ca:d2, 142]`
- Xiaomi: `[0x1dcd:zdo] ZDO request 0x0013: [7629, 00:15:8d:00:01:86:e8:46, 128]`

I'm assuming the last number is something of a callback identifier.

Updating this post as I figure out more differences.

lesteenman, 2017-08-24:
The error that occurs during the join:
```
2017-08-23 00:10:56.239 debug: Data frame: b'004db1ed542e14b259954b25ab5592cb2cf7bbf912317a97c9d46183fe8173a1ebdddf4e63f4e673d4f6698c4623a9cd123a85ba92057e'
2017-08-23 00:10:56.249 debug: Sending: b'8160597e'
2017-08-23 00:10:56.261 debug: Application frame 69 (incomingMessageHandler) received
2017-08-23 00:10:56.271 Exception running handler
2017-08-23 00:10:56.281 Traceback (most recent call last):
2017-08-23 00:10:56.292   File "/home/pi/python3venv/lib/python3.4/site-packages/bellows-0.3.5-py3.4.egg/bellows/ezsp.py", line 198, in handle_callback
2017-08-23 00:10:56.302     handler(*args)
2017-08-23 00:10:56.312   File "/home/pi/python3venv/lib/python3.4/site-packages/bellows-0.3.5-py3.4.egg/bellows/zigbee/application.py", line 146, in ezsp_callback_handler
2017-08-23 00:10:56.322     self._handle_frame(*args)
2017-08-23 00:10:56.332   File "/home/pi/python3venv/lib/python3.4/site-packages/bellows-0.3.5-py3.4.egg/bellows/zigbee/application.py", line 169, in _handle_frame
2017-08-23 00:10:56.342     tsn, command_id, is_reply, args = deserialize(aps_frame, message)
2017-08-23 00:10:56.352   File "/home/pi/python3venv/lib/python3.4/site-packages/bellows-0.3.5-py3.4.egg/bellows/zigbee/zcl/__init__.py", line 56, in deserialize
2017-08-23 00:10:56.362     value, data = t.deserialize(data, schema)
2017-08-23 00:10:56.372   File "/home/pi/python3venv/lib/python3.4/site-packages/bellows-0.3.5-py3.4.egg/bellows/types/__init__.py", line 9, in deserialize
2017-08-23 00:10:56.381     value, data = type_.deserialize(data)
2017-08-23 00:10:56.392   File "/home/pi/python3venv/lib/python3.4/site-packages/bellows-0.3.5-py3.4.egg/bellows/types/basic.py", line 128, in deserialize
2017-08-23 00:10:56.401     item, data = r._itemtype.deserialize(data)
2017-08-23 00:10:56.412   File "/home/pi/python3venv/lib/python3.4/site-packages/bellows-0.3.5-py3.4.egg/bellows/types/struct.py", line 22, in deserialize
2017-08-23 00:10:56.422     v, data = field_type.deserialize(data)
2017-08-23 00:10:56.432   File "/home/pi/python3venv/lib/python3.4/site-packages/bellows-0.3.5-py3.4.egg/bellows/zigbee/zcl/foundation.py", line 129, in deserialize
2017-08-23 00:10:56.442     actual_type = DATA_TYPES[self.type][1]
2017-08-23 00:10:56.452 KeyError: 76
```
Seems to be because the application receives a ZCL frame with a Structure type in it(?). This type has been uncommented in `zigbee/zcl/foundation.py` (`0x4c`).

Skorfulose, 2017-10-23:
Any update on this one? Seeing the same issues in debug output as @lesteenman.
I would love to see Xiaomi Zigbee device support in bellows and HASS respectively. Unfortunately my SW dev skills are non-existant. Willing to debug or donate some sensors to @rcloran if it helps.

lesteenman, 2017-10-23:
It seems there are currently two issues with supporting Xiaomi devices in Bellows. The first is the unrecognized Structure type, which a colleague of mine has fixed already (but I don't think he's made a pull request; I'll ask him about that later). 

The second is that the endpoint discovery does not work for Xiaomi, as they don't explicitly send the endpoints initially. Instead, they just start sending data over the endpoints, and it seems one should just use them when they arrive.

twendt, 2017-11-10:
I did get a little further with this. I simply created the endpoint and output_clusters in the salute db. Now I am away to see the devices in HASS and I also receive updates.
Unfortunately it stops working after an hour or so. Then I do not get any messages from the sensors anymore. The same effect happens with the button and the door sensor.
Maybe the the devices require some response and they stop sending messages then?
I am only on the phone right now. I will post the entries that I had to do in the db later.
The devices are then shown as switches in HASS for now. That requires the attribute updated function in the zha switch component to be implemented.

lesteenman, 2017-11-12:
What devices have you tried it with?

twendt, 2017-11-13:
I only tested with the door sensor and the button. The SmartThings community had the same issue that they stop working after an hour of inactivity.
I created a dump with bellows and there I have seen that there is am ieee message (Data Request) sent by both devices exactly 1 hour after the last usage. There is no response message. So I expect that this might be the issue. I haven't read the complete thread in the SmartThings forum yet, but it seems that there it was related to the Zigbee firmware.

Here is what I did to get it running for 1 hour:

The sqlite DB looks as follows after the device has been joined. There is only an entry in the devices table since it does not respond to the endpoint discovery:

Device table
ieee|nwk|status
xx:xx:xx:xx:xx:xx:xx:xx|12345|0

Now simply create the endpoint entry and entries for the output clusters:
```
sqlite3 zigbee.db
INSERT INTO "endpoints" VALUES('xx:xx:xx:xx:xx:xx:xx:xx',1,260,0,1);
INSERT INTO "output_clusters" VALUES('xx:xx:xx:xx:xx:xx:xx:xx',1,4);
INSERT INTO "output_clusters" VALUES('xx:xx:xx:xx:xx:xx:xx:xx',1,5);
INSERT INTO "output_clusters" VALUES('xx:xx:xx:xx:xx:xx:xx:xx',1,6);
```

This should be enough to get the device visible as switch in Home Assistant. To also retrive updates I added the following function to the switch/zha.py component in the Switch class at the end of the file.

```
    def attribute_updated(self, attribute, value):
        """Handle an attribute updated on this cluster."""
        _LOGGER.info("Value: %s", value)
        self._state = value
        self.schedule_update_ha_state()
```
I use the Bitron Zigbee Stick in Germany which has an EM 3587 chip. I did not have to flash any firmware since EZSP was working out of the box.

gazoscalvertos, 2017-11-21:
same here however I managed to capture one of the devices drop form the network if its any use:

2017-11-21 16:16:54 DEBUG (MainThread) [bellows.zigbee.zcl] [0x95e5:1:0x0000] Attribute report received: 65281=b'\x04!\xa8\x13\n!\x00\x00'
2017-11-21 16:16:55 DEBUG (MainThread) [bellows.uart] Data frame: b'4253b18cb1bfb5b07d954aa8bf55904a63b187607e'
2017-11-21 16:16:55 DEBUG (MainThread) [bellows.uart] Sending: b'8520dd7e'
2017-11-21 16:16:55 DEBUG (MainThread) [bellows.ezsp] Application frame 36 (trustCenterJoinHandler) received
2017-11-21 16:16:55 INFO (MainThread) [bellows.zigbee.application] Device 0x95e5 (00:15:8d:00:01:24:02:a0) left the network

jonatanolofsson, 2017-11-22:
@lesteenman Did you find a pull-request for the unrecongnized structure?

jonatanolofsson, 2017-11-22:
I would be happy to contribute to resolving this issue and get started as a contributor to bellows. While I am a programmer, I am new to bellows/home assistant however, so any pointers are appreciated. 

@twendt I tried your sql method and got something showing in the home assistant overview. However, I receive no updates. Could you please walk me trough the effect of the commands (and the "magic numbers" in the insert query)?

I am currently working on getting the xiaomi door sensor to work, and I have hardware to work with. I have some IKEA trådfri as well so I hope to get that working as well to have something to compare with. RIght now, I am looking into why I am not receiving any updates. Without the sql "injections", I got the warnings previously mentioned in this thread, so I am looking for an entry point where the event actually gets registered.. There doesn't seem to be a lot of documentation on bellows yet though, so I am reading code and testing manually at the moment.

puddly, 2017-12-08:
I've been trying to get these newer Xiaomi door sensors to work for the past few days. It looks like all of these sensors were properly integrated into [Smart Things](https://github.com/a4refillpad/Xiaomi/blob/master/devicetypes/a4refillpad/xiaomi-door-window-sensor.src/xiaomi-door-window-sensor.groovy#L167-L196), including the battery status. I wasn't able to get any of that code to work, however, because my door sensor doesn't actually implement all of the same clusters:

```
Device:
  NWK: 0x9adc
  IEEE: 00:15:8d:00:01:ab:40:52
  Endpoints:
    1: profile=0x104, device_type=24321
      Input Clusters:
        Basic (0)
        Identify (3)
        On/Off (6)
        Manufacturer Specific (65535)
      Output Clusters:
        Basic (0)
        Groups (4)
        Manufacturer Specific (65535)
```

When a new device joins or is loaded from bellows' database, Home Assistant's `zha` component tries to enumerate its endpoints with `_discover_endpoint_info`, which is essentially the following:

    await device.endpoints[1].in_clusters[0].read_attributes(['manufacturer'])

Here's a sample log of what happens:

```
DEBUG:bellows.zigbee.appdb:Loading application state from .homeassistant/zigbee.db
DEBUG:bellows.uart:Sending: b'1ac038bc7e'
DEBUG:bellows.uart:RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
DEBUG:bellows.ezsp:Send command version
DEBUG:bellows.uart:Sending: b'004221a850ed2c7e'
DEBUG:bellows.uart:Data frame: b'0142a1a8502805e67f627e'
DEBUG:bellows.uart:Sending: b'8160597e'
DEBUG:bellows.ezsp:Application frame 0 (version) received
DEBUG:bellows.ezsp:Send command setConfigurationValue
DEBUG:bellows.uart:Sending: b'7d314321fb582815c3277e'
DEBUG:bellows.uart:Data frame: b'1243a1fb54fb737e'
DEBUG:bellows.uart:Sending: b'82503a7e'
DEBUG:bellows.ezsp:Application frame 83 (setConfigurationValue) received
DEBUG:bellows.ezsp:Send command setConfigurationValue
DEBUG:bellows.uart:Sending: b'224021fb592f15226f7e'
DEBUG:bellows.uart:Data frame: b'2340a1fb54c6107e'
DEBUG:bellows.uart:Sending: b'83401b7e'
DEBUG:bellows.ezsp:Application frame 83 (setConfigurationValue) received
DEBUG:bellows.ezsp:Send command setConfigurationValue
DEBUG:bellows.uart:Sending: b'334121fb792b15a2d77e'
DEBUG:bellows.uart:Data frame: b'3441a1fb54d32a7e'
DEBUG:bellows.uart:Sending: b'8430fc7e'
DEBUG:bellows.ezsp:Application frame 83 (setConfigurationValue) received
DEBUG:bellows.ezsp:Send command setConfigurationValue
DEBUG:bellows.uart:Sending: b'444621fb7d5e291514417e'
DEBUG:bellows.uart:Data frame: b'4546a1fb5435d07e'
DEBUG:bellows.uart:Sending: b'8520dd7e'
DEBUG:bellows.ezsp:Application frame 83 (setConfigurationValue) received
DEBUG:bellows.ezsp:Send command setConfigurationValue
DEBUG:bellows.uart:Sending: b'554721fb4d2815713f7e'
DEBUG:bellows.uart:Data frame: b'5647a1fb54a9ec7e'
DEBUG:bellows.uart:Sending: b'8610be7e'
DEBUG:bellows.ezsp:Application frame 83 (setConfigurationValue) received
DEBUG:bellows.ezsp:Send command setConfigurationValue
DEBUG:bellows.uart:Sending: b'664421fb55d515b18f7e'
DEBUG:bellows.uart:Data frame: b'6744a1fb54948f7e'
DEBUG:bellows.uart:Sending: b'87009f7e'
DEBUG:bellows.ezsp:Application frame 83 (setConfigurationValue) received
DEBUG:bellows.ezsp:Send command setConfigurationValue
DEBUG:bellows.uart:Sending: b'774521fb4a2b1524a97e'
DEBUG:bellows.uart:Data frame: b'7045a1fb5481b57e'
DEBUG:bellows.uart:Sending: b'8070787e'
DEBUG:bellows.ezsp:Application frame 83 (setConfigurationValue) received
DEBUG:bellows.ezsp:Send command setConfigurationValue
[... 189 lines cut ...]
debug: [0xc282:1:0x0000] ZCL request 0x000a: [[<Attribute attrid=5 value=<bellows.zigbee.zcl.foundation.TypeValue object at 0x7fc9873bbe10>>, <Attribute attrid=1 value=<bellows.zigbee.zcl.foundation.TypeValue object at 0x7fc9873bbd30>>]]
debug: [0xc282:1:0x0000] Attribute report received: 5=b'lumi.sensor_magnet.aq2', 1=3
debug: Data frame: b'7152b1ed542e14b259954b25ab559260638ea56912314197a2d76283fd817dbaeaec31648cd7dfdff47a688967f1a9ea523aa5ea75824bb94c2677a1e1200f7e'
debug: Sending: b'8070787e'
debug: Application frame 69 (incomingMessageHandler) received
debug: [0xc282:1:0x0000] ZCL request 0x000a: [[<Attribute attrid=65281 value=<bellows.zigbee.zcl.foundation.TypeValue object at 0x7fc9873bbc18>>]]
debug: [0xc282:1:0x0000] Attribute report received: 65281=b'\x01!\xef\x0b\x03(\x18\x04!\xa8\x01\x05!\xd2\x00\x06$\x01\x00\x00\x00\x00\n!\x00\x00d\x10\x01'
debug: Data frame: b'0152b1ed542e14b459954b25ab55926363f3a56912316093ffcc6389ec7edc0a7e'
debug: Sending: b'8160597e'
debug: Application frame 69 (incomingMessageHandler) received
debug: [0xc282:1:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<bellows.zigbee.zcl.foundation.TypeValue object at 0x7fc9873bbe10>>]]
debug: [0xc282:1:0x0006] Attribute report received: 0=Bool.false
debug: Data frame: b'1152b1ed542e14b459954b25ab55926262f5a56912316093fecc6389ec7f428d7e'
debug: Sending: b'82503a7e'
debug: Application frame 69 (incomingMessageHandler) received
debug: [0xc282:1:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<bellows.zigbee.zcl.foundation.TypeValue object at 0x7fc9873bbc18>>]]
debug: [0xc282:1:0x0006] Attribute report received: 0=Bool.true
Done
```

The last few lines show bellows successfully receiving updates when I tap the two pieces together.

Home Assistant assumes that all binary sensors have the IAS Zone cluster (`0x0500`), but this one apparently does not. If I modify Home Assistant to work around that issue, it integrates perfectly for about an hour but then the sensor drops off the network until I pair it again.

Is this some profile issue with my ZigBee radio (Linear HUSBZB-1)? Or does this sensor just need some sort of special treatment?

Yoda-x, 2017-12-11:
It need some special treatment. I invested some time the last week and got the door/windows sensor working.
1. paring: you need to keep the sensor alive after joining. I push the button until it blinks 3x, and  then I click it every second for 20-30 sec. So it have enough time to finish the discovery process and initialize  correctly. Paring is not so easy and I have a look on the logfile with debugging enabled to be sure I don't missed a click and it went to sleep in the middle. 

2. device leaving after 1h. This is caused by the endpoint timeout of 320 sec. The controller expects to hear something from the sensor. , Otherwise it gets removed from the peer table. I extended the policy that bellows applies during the network initialization to increase this timeout (`CONFIG_END_DEVICE_POLL_TIMEOUT * 2^CONFIG_END_DEVICE_POLL_TIMEOUT_SHIFT) `to more than 3600 sec
There maybe also the option to query the device for it's update-timer and define a timeout per device. But this needs to be investigated. 

3. I also modified the zha code, to configure  the sensor to send attribute reports  for cluster/attribute 6/0 every 30min, as keep-alive during the binary_sensor initialization: `configure_reporting(self, attribute, min_interval, max_interval, reportable_change)`
Also the sensors reports itself as device_type:` 0x104, (DIMMER_SWITCH) `which provides in/out-cluster profiles, that are wrong for the sensor. So I modified the cluster  profile to include the xiaomi clusters. 

I also  modified code at many other places and added a lot of debug- logging statements during my  try and error testing . Now as it's working, I want to start over from the newest HA code and do only the minimal changes to get it working and provide this patches to github or the maintainer of the code. 
I started phython programming  a week ago and I neeed a lot of time to understand the source code. so  my programming style and capability is very basic. 
Currently I see only the On/Off information in HA. I need to check where to find the other information, like battery,  and forward them to the HA state machine. 

We need for the ZHA code more configurable option for  the configuration file. Like the option to set customized cluster profile  to override the default settings for such non-standard devices. 

I just wanted to give a heads up and provide  some hope:-)
I also have a temp/humidity sensor and PIR from Xiaomi, I will now also try to get these working. 

Many Thanks to the smartthings forum, the DH from a4refillpad and the discussions about the device were very helpful.

jonatanolofsson, 2017-12-11:
I actually just pushed support for this (as well as TRÅDFRI lights to my own branch: [jonatanolofsson/bellows](https://github.com/jonatanolofsson/bellows) . Have a look if you like.

It does also contain some database updates though, so you may want to start a new database if you run my code.

There's also a standalone command line option that you can use if you don't want to restart homeassistant all the time.

@Yoda-x Instead of extending the HA zha module and further cementing bellows as a HA-only zigbee driver, I think a better option is to separate them and see bellows as its own project which HA uses for zigbee communication. The [bellows abstractions should not leak](https://en.wikipedia.org/wiki/Leaky_abstraction) into HA more than necessary.

Gamester17, 2017-12-11:
@jonatanolofsson Great stuff! Could you maybe push those upstream as seperate pull requests here?

jonatanolofsson, 2017-12-11:
I'm intending to, I just finished it up this morning. However, I have been unable to get in contact with @rcloran so far, I hope he's not MIA..

Yoda-x, 2017-12-11:
Hello @jonatanolofsson 
the only change from bellows would be the timeout increase during the network init. 
And the network policy should be something that shoud be configurable  from the calling HA compoment and not be hardcoded only. So it's not realy a code change, more are policy change. 

The rest of the changes should be inside  the  ZHA HASS component  and should be configurable inside the configuration yaml, or an include file. This is the goal for me. The ZHA HA component already supports  configuration parameters, like the endpoint-id or device type. Thus, it should be doable to handover the clusters (or other paramters)  based  on the endpoint-id, that I already use define the door sensor as binary_sensor.
I like the fingerprinting inside smartthings and how they handle new devices. The final goal for me would be a configuration yaml with a device-model-fingerprint inside the zha section, where I can override non standard devices with the parameters that are needed and keep the code base unchanged without coding new vendors in it.  

At least for the xiaomi door sensor, bellows worked for me out of  the box,  for endpoint, cluster, manufacturer and model  discovery, (after I did the paring the right way)   The problem is more with the upstream to HA  inside the ZHA component, making  it a binary_sensor creating the entity  and sending the event. 
Not sure if this cements bellows inside HASS.

It would appricate a new bellows version. seeing that many requests pending.

BTW, good to see you were working on it. I was thinking this was dead and started my own way ;-)

puddly, 2017-12-11:
@Yoda-x excellent news! I implemented your changes and both of my sensors have stayed paired for the past two hours. 

@jonatanolofsson I'm not sure how much manufacturer-specific code you're suggesting bellows to handle, but here's a basic decoder for the heartbeat attribute (`0xFF01`) for Xiaomi devices:

```
from bellows.zigbee.zcl import foundation

def decode_xiaomi_heartbeat(data):
    heartbeat = {}
    unknown = {}

    while data:
        xiaomi_type = data[0]
        value, data = foundation.TypeValue.deserialize(data[1:])
        value = value.value

        if xiaomi_type == 0x01:
            # I'm sure there's a more accurate way to do this
            heartbeat['battery_percentage'] = (value - 2600) / (3200 - 2600)
            heartbeat['battery_voltage'] = value / 1000
        elif xiaomi_type == 0x03:
            heartbeat['soc_temperature'] = value
        elif xiaomi_type == 0x64:
            heartbeat['state'] = value
        else:
            unknown[xiaomi_type] = value

    if unknown:
        heartbeat['unknown'] = unknown

    return heartbeat
```

For the latest door sensor, a heartbeat is sent when you click the button. The temperature sensor's heartbeat that is triggered by a button press doesn't contain anything useful, but I'll check my logs later and see if it might be periodically sending a large one like the door sensor.

jonatanolofsson, 2017-12-12:
@puddly Awesome!

I'm looking at getting automatic battery/heartbeat reporting working now for my device, but so far I can't get it to respond to the bind command at all. @Yoda-x did you get reporting working? Do you have an example code?

From what I understand about your point 2 above, this only delays the problem, right? So instead of leaving after one hour, it leaves after a longer time. Point 3 is what really should fix the issue I suppose, if there's no way of disabling timeouts completely (?)

puddly, 2017-12-12:
@jonatanolofsson [Here's the hacky code](https://gist.github.com/puddly/7db5ecc9b4f618e37583922dce2f4b83) that I've been testing with. Attribute reporting has been working fine all day for me in Home Assistant with the increased timeout:

```
2017-12-11 17:46:20 DEBUG (MainThread) [bellows.zigbee.zcl] [0x8ddf:1:0x0000] Attribute report received: 65281=b'\x01!\xa9\x0b\x03(\x19\x04!\xa8\x13\x05!\xbb\x01\x06$!\x00\x00\x00\x00\n!\x00\x00d\x10\x01'
2017-12-11 18:36:28 DEBUG (MainThread) [bellows.zigbee.zcl] [0x8ddf:1:0x0000] Attribute report received: 65281=b'\x01!\xa9\x0b\x03(\x19\x04!\xa8\x13\x05!\xbb\x01\x06$\x01\x00\x00\x00\x00\n!\x00\x00d\x10\x01'
2017-12-11 19:26:44 DEBUG (MainThread) [bellows.zigbee.zcl] [0x8ddf:1:0x0000] Attribute report received: 65281=b'\x01!\xb3\x0b\x03(\x18\x04!\xa8\x13\x05!\xbb\x01\x06$\x01\x00\x00\x00\x00\n!\x00\x00d\x10\x01'

2017-12-11 17:12:07 DEBUG (MainThread) [bellows.zigbee.zcl] [0x0f73:1:0x0000] Attribute report received: 65281=b'\x01!\x9f\x0b\x04!\xa8\x13\x05!%\x00\x06$\x01\x00\x00\x00\x00d)\xb3\x07e!\xae\x0ef+\xbd{\x01\x00\n!\x00\x00'
2017-12-11 18:04:51 DEBUG (MainThread) [bellows.zigbee.zcl] [0x0f73:1:0x0000] Attribute report received: 65281=b'\x01!\x9f\x0b\x04!\xa8\x13\x05!%\x00\x06$\x01\x00\x00\x00\x00d)a\x07e!%\x12f+k{\x01\x00\n!\x00\x00'
2017-12-11 18:57:54 DEBUG (MainThread) [bellows.zigbee.zcl] [0x0f73:1:0x0000] Attribute report received: 65281=b'\x01!\x9f\x0b\x04!\xa8\x13\x05!%\x00\x06$\x01\x00\x00\x00\x00d)*\x07e!\xe7\x12f+W{\x01\x00\n!\x00\x00'
2017-12-11 19:50:48 DEBUG (MainThread) [bellows.zigbee.zcl] [0x0f73:1:0x0000] Attribute report received: 65281=b'\x01!\xa9\x0b\x04!\xa8\x13\x05!%\x00\x06$\x01\x00\x00\x00\x00d)(\x07e!\x8e\x13f+\\{\x01\x00\n!\x00\x00'
```

I increased the timeout to about an hour and a half and both of my sensors have been working continuously for the past four hours. I think that as long as the device sends an attribute report before the timeout occurs, everything will work fine. Pressing the button or triggering the sensor during the timeout period doesn't seem to prevent it from being disconnected due to a timeout.

Hedda, 2017-12-13:
FYI; I have posted a suggestion for Home Assistant to take owership of this library while @rcloran is M.I.A.

https://github.com/home-assistant/home-assistant/issues/11119

@Yoda-x @jonatanolofsson @puddly  and others, perhaps you care to comment regarding that there?

Yoda-x, 2017-12-13:
Here is some sample code I used during my try and errors:, it's from the binary_sensor/zha and the  async_setup_platform routine: It's very hacky.
The problem is that bellows implement configure reporting on the cluster and not on the endpoint. This seems not correct. Others implement it as ZCL global command. So you have to initialize a cluster first, if it's not there already. And the power cluster (1) is  not  reported during discovery.  This code below is for setting reports on the on/off-cluster.
 >        _LOGGER.debug("CLUSTER BIND")
>         yield from cluster.bind()
>         _LOGGER.debug("CLUSTER BIND DONE")
>         ieee = cluster.endpoint.device.application.ieee
>         _LOGGER.debug("CLUSTER WRITE")
>         yield from cluster.write_attributes({'cie_addr': ieee})
>         _LOGGER.debug("CLUSTER WRITE DONE")
>         if 6 in endpoint.in_clusters:
>             yield from endpoint.in_clusters[6].configure_reporting( 0, 1, 3600, '01')
>         else:
>             yield from endpoint.out_clusters[6].configure_reporting( 0, 1, 3600, '01')
>         _LOGGER.debug("Config report")
For the xiaomi keep-alive it should be ok with the timeout increase, but other reports like battery every day and and an on/off status every 30min would be helpful. 
I made some code cleanup and created a ZHA customs_component based on existing zha component.  I think it's easier to have only a git with the importandt files and for others it' easy to test it, just changing the config and go back to the the HA build-in component, if they don't like it.
[ha-zha-new](https://github.com/Yoda-x/ha-zha-new.git)
Please have a look, I added cluster definition override from the config.yaml and working on configure_reports from config.yaml currently.
The only change from bellows I need is the time out increase so far.

good to see we get on the track again for this sensors.

jonatanolofsson, 2017-12-14:
For some reason, my binding requests just times out. I have tried with both the old type of door sensor and the aqara one.. Did you experience this as well? I get the updates from the sensor, but it's like it doesn't listen to anything you send to it..

jonatanolofsson, 2017-12-14:
For some reason I only get a timeout on the bind request.. Did you experience this as well at some point?

I managed to get a response once, but mostly I get
>Traceback (most recent call last):
>  File "/home/jonatan/git/bellows/bellows/standalone.py", line 50, in wrapper
>    result = yield from handler_func(request)
>  File "/home/jonatan/git/bellows/bellows/standalone.py", line 109, in _get_sensor
>    res = yield from dev[1].on_off.bind()
>  File "/home/jonatan/git/bellows/bellows/zigbee/application.py", line 296, in request
>    v = yield from send_fut
>bellows.zigbee.exceptions.DeliveryError: Message send failure: EmberStatus.DELIVERY_FAILED

It's like it's not listening to me.. :(
I managed to get a successful bind once, now I'm trying to figure out what the difference was.. :)

jonatanolofsson, 2017-12-14:
@Yoda-x cie_addr is an attribute of the IasZone cluster, right? Did you add this to the OnOff cluster as well?

puddly, 2017-12-16:
@jonatanolofsson my temp/humidity/pressure sensor and the door sensor have been working for the past 5 days with just the binding to the OnOff input cluster and configuring reporting on attribute 0. When my Xiaomi gateway arrives I'll try and see what their actual binding process is. I have a feeling the sensors don't expect to be used outside of the gateway's network, so they just send attribute reports without you asking.

Yoda-x, 2017-12-16:
I agree, I don't think reporting is needed as long as the global timeout is higher than the build in reporting from the xiaomi devices. 
I did a bind on the ias cluster and set the cie from the ias cluster.  For the report configuration you need a added cluster. because the command is part of the cluster methods. 
I saw the  DeliveryFailed error in most cases where the device went to sleep, because I stopped clicking.

Yoda-x, 2017-12-18:
Got the xiaomi HT sensor working, I modifed some of the zha and sensors code and added the option to configure reports from the configuration file
[image]

dmulcahey, 2017-12-18:
@Yoda-x what do I have to add to HA to make humidity show up? (Ex temp and humidity = temp only, motion and lux = motion only, etc.) I have several multi function sensors and only the primary function gets a sensor in HA. I’d also like to get battery reports and sensors working as well.

Yoda-x, 2017-12-19:
The Sensors/zha.py only has support for the temperature cluster. I extented it with  the humitidy cluster

puddly, 2017-12-19:
It seems like the only thing left to do is make the `zha` component in Home Assistant support these multi-function ZigBee sensors and integrate the Xiaomi-specific code for battery reporting.

My sensors have been working perfectly in home assistant for over a week straight with battery reporting and everything else. Interestingly, the undocumented built-in thermometer in the window and door sensor is accurate to within 1˚C, so it would be nice to expose that as a low-resolution temperature sensor usable from within Home Assistant.

Is there any reason to keep this issue open now that we know that the only required change is just increasing the timeout? Or are there concrete plans to integrate manufacturer-specific extensions into bellows instead of the `zha` component of Home Assistant?

dmulcahey, 2017-12-19:
@puddly do you have a fork of HA that I can look at?

puddly, 2017-12-19:
@dmulcahey [it's not pretty](https://github.com/puddly/home-assistant/commits/xiaomi-zigbee) and probably won't work for you without tweaking, but you can try. I never tested pairing with `zha.permit`.

Will there be any future problems if the longer timeout configuration is moved into bellows? Or should this be kept in `zha`?

Yoda-x, 2017-12-20:
Imho, the configuration timeout should be set on the application level and not in bellows. which for me is the underlaying api. We can discuss about the duration, but I think 6min is far to low, even if it's the default. 

@puddly : did you tried to create a xiaomi cluster inside the bellows/zigbee/zcl/clusters/manufacturer_specific.py file?  This should identify and handle the heartbeats inside the bellows code and create attribute_reports. Which means less vendor specific code. 

BTW, I think we can close this issue. It's not really a bellows problem. But we need to open a homeassistant  issue to get all the zha changes addressed for xiaomi

jonatanolofsson, 2017-12-21:
@puddly I tried to mimick your code, but the reponse I get is: 
`Configure reporting response is: [6, <Status.UNSUP_GENERAL_COMMAND: 130>]`

Just to confirm - all you did was 
1) bind to cluster 0
2) Configure reporting on cluster 0, attribute 0

and that should be it to get periodical reports?

jonatanolofsson, 2017-12-21:
To reply to myself: It seems that even though the response 130 is returned, it actually start reporting anyway.. Yay =)

puddly, 2017-12-21:
@jonatanolofsson It does seem like the attribute reporting command is extraneous. Is binding to the OnOff cluster even necessary then?

@Yoda-x I have looked at the manufacturer-specific clusters but the Xiaomi sensors send a manufacturer-specific attribute `0xFF01` over the standard `General` cluster. The Aqara sensors report a device type of `0x5F01`, which is "reserved" in the Intruder Alarm System range.

ZigBee already defines battery reporting and device temperature capabilities, so I think it would be easiest to create some system that would proxy these manufacturer-specific attributes and clusters and transform them into attribute reports over standard ZigBee clusters (even if the end device doesn't actually support them). Home Assistant would then contain no specialized code for Xiaomi devices and they'd show up as normal sensors with battery reporting capabilities.

For example, the door/window sensor currently reports supporting the following:

```
  Endpoints:
    1: profile=0x104, device_type=24321
      Input Clusters:
        Basic (0)
        Identify (3)
        On/Off (6)
        Manufacturer Specific (65535)
```

The heartbeat provides data about its internal temperature and battery voltage, so two virtual clusters could be created that transform the heartbeat information into reports on the appropriate clusters:

```
  Endpoints:
    1: profile=0x104, device_type=24321
      Input Clusters:
        Basic (0)
        Power Configuration (1) (virtual)
        Device Temperature (2) (virtual)
        Identify (3)
        On/Off (6)
        Manufacturer Specific (65535)
```

Any thoughts?

ChristianHackAMS, 2017-12-31:
Hi @Yoda-x 

I'm trying to give your version a go. I'm a bit confused with configuration. What's the difference between the zha.yaml and configuration.yaml in your git repo? Do I need both? I'm trying with an Aqara temp sensor which initially recorded as this (when using the 0.60 HA "release" version of bellows):

```Device:
  NWK: 0x84cd
  IEEE: 00:15:8d:00:01:9f:df:de
  Endpoints:
    1: profile=0x104, device_type=24321
      Input Clusters:
        Basic (0)
        Identify (3)
        Temperature Measurement (1026)
        Pressure Measurement (1027)
        Relative Humidity Measurement (1029)
        Manufacturer Specific (65535)
      Output Clusters:
        Basic (0)
        Groups (4)
        Manufacturer Specific (65535)
```

This sort of lines up with your example but not quite.

I have this in configuration.yaml

```zha_new:
    usb_path: /dev/ttyUSB1
    database_path: /config/zigbee.db
    device_config:

# HT sensor
      "00:15:8d:00:01:9f:df:de-1":
        type: sensor
        in_cluster: [0x0000, 0x0001, 0x0003, 0x0009, 0xff01 , 0xffff]
#        in_cluster: [0x0000, 0x0003, 0xffff]
        config_report:
          - [ 0x0405, 0, 1, 120, 5]
          - [ 0x0402, 0, 1, 120, 5]
```

Is zha.yaml needed in addition to this? I didn't see all the in_clusters you have listed and there's a few different ones.

Yoda-x, 2018-01-04:
@ChristianHackAMS 
your config should do it if you start the section with  "zha_new:" and do the indent right.
 I moved out the zha part out of the configuration yaml to focus on the zha part. You will not see the pressure part. I have only temp and humidity implemented right now. Adding pressure should not be a problem.  If you run in problems please open an issue in my spin-off. 
Let me know if it works for you.
Thanks

Hedda, 2018-01-15:
FYI; @jonatanolofsson created a Bellows dicussion group on Discord @ https://discord.gg/D5dRgXQ

rcloran, 2018-02-05:
Please re-open in the new zigpy project (https://github.com/zigpy/zigpy). There appears to be no way to move issues between projects.

FWIW, I've spent a bit of time playing with the Xiaomi devices. AFAICT, they implement some bits of the spec in some way. IMHO, the right approach here is to add support for custom device handlers to zigpy and let HA install those. A library of separate handlers could be inside of outside of HA.

closed 2018-02-05 by rcloran

ruippeixotog, 2018-02-07:
I have just created https://github.com/zigpy/zigpy/issues/14 to track this issue.

Gamester17, 2018-02-15:
@twendt sorry if off-topic but did you manage to get your Bitron Video ZigBee USB Funkstick to work?

http://www.bitronvideo.eu/index.php/produkte/smart-home-produkte/zb-funkstick/

Looks to be cheapest ZigBee stick in Europe based on Silicon Labs Telegesis ETRX357USB with CP2104:

https://www.amazon.de/Telekom-Smart-ZigBee-Funkstick-USB-Hub/dp/B071YVVQD1/

also sold as "Deutsche Telekom Smart Home Kit Zigbee V.2 with USB Hub" in Germany:

https://www.smarthome.de/geraete/bitron-video-smarthome-funkstick-fuer-zigbee-v-2

Read that a few people haven't got it to work with bellows, some tried to flash other firmware and failed.

dmulcahey, 2018-12-08:
> Please re-open in the new zigpy project (https://github.com/zigpy/zigpy). There appears to be no way to move issues between projects.
>
> FWIW, I've spent a bit of time playing with the Xiaomi devices. [...]

@rcloran I have done just this: https://github.com/dmulcahey/zha-device-handlers let me know your thoughts...

Hedda, 2018-12-11:
@rcloran @AndreasBomholtz @jonatanolofsson @puddly @therve @lesteenman @Yoda-x @twendt I have just started a new Zigpy feature request to specifically discuss the ZHA Device Handler that @dmulcahey have written (and the quirks handlers for it) and whether or not it or something like it could be integrated into Zigpy :

https://github.com/zigpy/zigpy/issues/102

Hopefully, this type of modular ZHA Device Handler approach or something similar is something that the Zigpy project would like to adopt for dealing with the custom / unique exception handling of some devices.