zigpy/bellows #44: Failed ZDO request during device initialization for Ikea bulb

rklomp, 2017-08-23:
I am unable to join my ikea bulb to my home assistant setup.

`2017-08-23 20:33:55 DEBUG (MainThread) [homeassistant.components.websocket_api] WS 1948625904: Received {'type': 'call_service', 'domain': 'zha', 'service': 'permit', 'service_data': {}, 'id': 11}
2017-08-23 20:33:55 INFO (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=zha, service=permit, service_data=, service_call_id=1961586576-2>
2017-08-23 20:33:55 INFO (MainThread) [homeassistant.components.zha] Permitting joins for 60s
2017-08-23 20:33:55 DEBUG (MainThread) [bellows.ezsp] Send command permitJoining
2017-08-23 20:33:55 DEBUG (MainThread) [bellows.uart] Sending: b'634c218a686fa77e'
2017-08-23 20:33:55 DEBUG (MainThread) [bellows.uart] Data frame: b'374ca18a543f167e'
2017-08-23 20:33:55 DEBUG (MainThread) [bellows.uart] Sending: b'8430fc7e'
2017-08-23 20:33:55 DEBUG (MainThread) [bellows.ezsp] Application frame 34 (permitJoining) received
2017-08-23 20:33:55 INFO (MainThread) [homeassistant.core] Bus:Handling <Event service_executed[L]: service_call_id=1961586576-2>
2017-08-23 20:33:55 DEBUG (MainThread) [homeassistant.components.websocket_api] WS 1948625904: Sending {'id': 11, 'type': 'result', 'success': True, 'result': None}
2017-08-23 20:34:00 DEBUG (MainThread) [bellows.uart] Data frame: b'474cb5cae1a78f4ca6c34125fa717e'
2017-08-23 20:34:00 DEBUG (MainThread) [bellows.uart] Sending: b'8520dd7e'
2017-08-23 20:34:00 DEBUG (MainThread) [bellows.ezsp] Application frame 98 (incomingSenderEui64Handler) received
2017-08-23 20:34:00 DEBUG (MainThread) [bellows.uart] Data frame: b'574cb1ed502a15a159944a2dae5592d36385cd4512316b0a1728d6046680c0f0e0cd50a34d7e'
2017-08-23 20:34:00 DEBUG (MainThread) [bellows.uart] Sending: b'8610be7e'
2017-08-23 20:34:00 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received
2017-08-23 20:34:00 DEBUG (MainThread) [bellows.zigbee.application] No such device 61162
2017-08-23 20:34:00 WARNING (MainThread) [bellows.zigbee.application] Message on unknown device 0xeeea
2017-08-23 20:34:03 DEBUG (MainThread) [bellows.uart] Data frame: b'674cb18c084ba03fc36ab572a15593499c4e701e7e'
2017-08-23 20:34:03 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
2017-08-23 20:34:03 DEBUG (MainThread) [bellows.ezsp] Application frame 36 (trustCenterJoinHandler) received
2017-08-23 20:34:03 INFO (MainThread) [bellows.zigbee.application] Device 0x615c (00:0b:57:ff:fe:9a:8d:b5) joined the network
2017-08-23 20:34:03 INFO (MainThread) [bellows.zigbee.device] [0x615c] Discovering endpoints
2017-08-23 20:34:03 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast
2017-08-23 20:34:03 DEBUG (MainThread) [bellows.uart] Sending: b'774d219c547674b259914a25aa1593499c4f26a8ec92065b347e'
2017-08-23 20:34:03 DEBUG (MainThread) [bellows.uart] Data frame: b'704da19c54e1c5707e'
2017-08-23 20:34:03 DEBUG (MainThread) [bellows.uart] Sending: b'8070787e'
2017-08-23 20:34:03 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received
2017-08-23 20:34:08 DEBUG (MainThread) [bellows.uart] Data frame: b'004db197547674b259914a25aa1593499c8526cdedc8847e'
2017-08-23 20:34:08 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2017-08-23 20:34:08 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received
2017-08-23 20:34:09 DEBUG (MainThread) [bellows.uart] Data frame: b'104db18ccb38a03fc36ab572a15593499c4e335d7e'
2017-08-23 20:34:09 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2017-08-23 20:34:09 DEBUG (MainThread) [bellows.ezsp] Application frame 36 (trustCenterJoinHandler) received
2017-08-23 20:34:09 INFO (MainThread) [bellows.zigbee.application] Device 0x129f (00:0b:57:ff:fe:9a:8d:b5) joined the network
2017-08-23 20:34:09 DEBUG (MainThread) [bellows.zigbee.application] Skip initialization for existing device 00:0b:57:ff:fe:9a:8d:b5
2017-08-23 20:34:09 DEBUG (MainThread) [bellows.uart] Data frame: b'204db5cae1a78f4ca6c3412599937e'
2017-08-23 20:34:09 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
2017-08-23 20:34:09 DEBUG (MainThread) [bellows.ezsp] Application frame 98 (incomingSenderEui64Handler) received
2017-08-23 20:34:09 DEBUG (MainThread) [bellows.uart] Data frame: b'304db1ed502a15a159944a2dae55923d6382b8b912316b0a62d4d6046680c0f0e0cd50156a7e'
2017-08-23 20:34:09 DEBUG (MainThread) [bellows.uart] Sending: b'8430fc7e'
2017-08-23 20:34:09 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received
2017-08-23 20:34:09 DEBUG (MainThread) [bellows.zigbee.zdo] [0x129f:zdo] ZDO request 0x0013: [4767, 00:0b:57:ff:fe:9a:8d:b5, 142]
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.uart] Sending: b'0452219c54b507b259914a25aa1593499c4c25a8ef920699167e'
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.uart] Data frame: b'4152a19c54e7ae397e'
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.uart] Sending: b'8520dd7e'
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.uart] Data frame: b'5152b1f1cb38a03fc36ab572a1556d849c0b797e'
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.uart] Sending: b'8610be7e'
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.ezsp] Application frame 89 (incomingRouteRecordHandler) received
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.uart] Data frame: b'6152b1f1cb38a03fc36ab572a1556d849c297a7e'
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.ezsp] Application frame 89 (incomingRouteRecordHandler) received
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.uart] Data frame: b'7152b1ed542a15b7d9944a65aa55923c6383b8b9123162897c9a0289eaf77e'
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.uart] Sending: b'8070787e'
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.uart] Data frame: b'0152b19754b507b259914a25aa1592499c8325abed52037e'
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2017-08-23 20:34:10 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received
2017-08-23 20:34:10 WARNING (MainThread) [bellows.zigbee.device] [0x129f] Failed ZDO request during device initialization: ('Endpoint request failed: %s', [129, 24924, []])
2017-08-23 20:34:18 DEBUG (MainThread) [bellows.uart] Data frame: b'1152b1ed502a15b459944a2dab55923f6382b8b912316e8a00396788fd673fa798b27e'
2017-08-23 20:34:18 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2017-08-23 20:34:18 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received
2017-08-23 20:34:18 DEBUG (MainThread) [bellows.zigbee.zdo] [0x129f:zdo] ZDO request 0x0006: [65533, 260, [25], []]
2017-08-23 20:34:18 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast
2017-08-23 20:34:18 DEBUG (MainThread) [bellows.uart] Sending: b'1253219c54b507b25992ca25aa1593499c4d24adeece678bfcc798c57e'
2017-08-23 20:34:18 DEBUG (MainThread) [bellows.uart] Data frame: b'2253a19c54e4a5f37e'
[... 25 lines cut ...]
2017-08-23 20:34:26 DEBUG (MainThread) [bellows.uart] Data frame: b'1253b1f1cb38a03fc36ab572a1556d859cd94e7e'
2017-08-23 20:34:26 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2017-08-23 20:34:26 DEBUG (MainThread) [bellows.ezsp] Application frame 89 (incomingRouteRecordHandler) received
2017-08-23 20:34:26 DEBUG (MainThread) [bellows.uart] Data frame: b'2253b1ed542e14ab59954b65aa5592306382b8b91231698affc762f5ed7c1d87bcdccf6e8f533e7e'
2017-08-23 20:34:26 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
2017-08-23 20:34:26 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received
2017-08-23 20:34:26 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-08-23 20:34:26 WARNING (MainThread) [bellows.zigbee.device] [0x129f] Message on unknown endpoint 1
2017-08-23 20:34:29 DEBUG (MainThread) [bellows.uart] Data frame: b'3253b1f1cb38a03fc36ab572a1556d859ce54c7e'
2017-08-23 20:34:29 DEBUG (MainThread) [bellows.uart] Sending: b'8430fc7e'
2017-08-23 20:34:29 DEBUG (MainThread) [bellows.ezsp] Application frame 89 (incomingRouteRecordHandler) received
2017-08-23 20:34:29 INFO (SyncWorker_0) [homeassistant.components.device_tracker.ubus] Checking ARP
[... cut ...]
2017-08-23 20:34:32 DEBUG (MainThread) [bellows.uart] Data frame: b'6253b1ed542e14ab59954b65aa5592326382b8b91231698af9c762f5ed7c1d87bcdccf6e8ff2db7e'
2017-08-23 20:34:32 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
2017-08-23 20:34:32 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received
2017-08-23 20:34:32 WARNING (MainThread) [bellows.zigbee.zcl] Data remains after deserializing ZCL frame
2017-08-23 20:34:32 WARNING (MainThread) [bellows.zigbee.device] [0x129f] Message on unknown endpoint 1`

AndreasBomholtz, 2017-09-15:
I just included my IKEA bump and it worked. But i have seen this kind of error before.
@rklomp What version of bellows are you using?
Because I fixed this problem and it was integrated into release 0.3.0.

rklomp, 2017-09-15:
I am using the bellows version included in latest home assistant running on hassio. 
According to this file it should be 0.3.4:
https://github.com/home-assistant/home-assistant/blob/release-0-52-1/homeassistant/components/zha/__init__.py

AndreasBomholtz, 2017-09-15:
I can see that the device tries to join two times, but with different network id's.

> Device 0x615c (00:0b:57:ff:fe:9a:8d:b5) joined the network
and
> Device 0x129f (00:0b:57:ff:fe:9a:8d:b5) joined the network
I do not know if this is the issue, because my bulb also tries two times, but with the same network id.

Can you try to include it again and see if it behaves the same?

rklomp, 2017-09-15:
This is the logging of another try:
```
2017-08-17 18:26:12 DEBUG (MainThread) [bellows.uart] Data frame: b'504db5cae1a78f4ca6c341251d237e'
2017-08-17 18:26:12 DEBUG (MainThread) [bellows.uart] Sending: b'8610be7e'
2017-08-17 18:26:12 DEBUG (MainThread) [bellows.ezsp] Application frame 98 (incomingSenderEui64Handler) received
2017-08-17 18:26:12 DEBUG (MainThread) [bellows.uart] Data frame: b'604db1ed502a15a159944a2dae5592636381a48c12316b0a7ee1d6046680c0f0e0cd50c0957e'
2017-08-17 18:26:12 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
2017-08-17 18:26:12 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received
2017-08-17 18:26:12 DEBUG (MainThread) [bellows.zigbee.application] No such device 10115
2017-08-17 18:26:12 WARNING (MainThread) [bellows.zigbee.application] Message on unknown device 0x2783
2017-08-17 18:26:15 DEBUG (MainThread) [bellows.uart] Data frame: b'704db18c6cc9a03fc36ab572a15593499c4e16657e'
2017-08-17 18:26:15 DEBUG (MainThread) [bellows.uart] Sending: b'8070787e'
2017-08-17 18:26:15 DEBUG (MainThread) [bellows.ezsp] Application frame 36 (trustCenterJoinHandler) received
2017-08-17 18:26:15 INFO (MainThread) [bellows.zigbee.application] Device 0xe338 (00:0b:57:ff:fe:9a:8d:b5) joined the network
2017-08-17 18:26:15 INFO (MainThread) [bellows.zigbee.device] [0xe338] Discovering endpoints
2017-08-17 18:26:15 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast
2017-08-17 18:26:15 DEBUG (MainThread) [bellows.uart] Sending: b'0052219c5412f6b259914a25aa1593499c4f26a8ecf6844aa57e'
2017-08-17 18:26:15 DEBUG (MainThread) [bellows.uart] Data frame: b'0152a19c54fc67737e'
2017-08-17 18:26:15 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2017-08-17 18:26:15 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received
2017-08-17 18:26:16 INFO (SyncWorker_10) [homeassistant.components.device_tracker.ubus] Checking ARP
2017-08-17 18:26:20 DEBUG (MainThread) [bellows.uart] Data frame: b'1152b1975412f6b259914a25aa1593499c9826cdedbc487e'
2017-08-17 18:26:20 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2017-08-17 18:26:20 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received
2017-08-17 18:26:21 DEBUG (MainThread) [bellows.uart] Data frame: b'2152b18cc756a03fc36ab572a15593499c4ecab47e'
2017-08-17 18:26:21 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
2017-08-17 18:26:21 DEBUG (MainThread) [bellows.ezsp] Application frame 36 (trustCenterJoinHandler) received
2017-08-17 18:26:21 INFO (MainThread) [bellows.zigbee.application] Device 0x7c93 (00:0b:57:ff:fe:9a:8d:b5) joined the network
2017-08-17 18:26:21 DEBUG (MainThread) [bellows.zigbee.application] Skip initialization for existing device 00:0b:57:ff:fe:9a:8d:b5
2017-08-17 18:26:21 DEBUG (MainThread) [bellows.uart] Data frame: b'3152b5cae1a78f4ca6c3412573b87e'
2017-08-17 18:26:21 DEBUG (MainThread) [bellows.uart] Sending: b'8430fc7e'
2017-08-17 18:26:21 DEBUG (MainThread) [bellows.ezsp] Application frame 98 (incomingSenderEui64Handler) received
2017-08-17 18:26:21 DEBUG (MainThread) [bellows.uart] Data frame: b'4152b1ed502a15a159944a2dae5592c86381b4d712316b0a6ebad6046680c0f0e0cd50c92e7e'
2017-08-17 18:26:21 DEBUG (MainThread) [bellows.uart] Sending: b'8520dd7e'
2017-08-17 18:26:21 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received
2017-08-17 18:26:21 DEBUG (MainThread) [bellows.zigbee.zdo] [0x7c93:zdo] ZDO request 0x0013: [31891, 00:0b:57:ff:fe:9a:8d:b5, 142]
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.uart] Sending: b'1553219c54b969b259914a25aa1593499c4c25a8eff68477057e'
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.uart] Data frame: b'5253a19c54f292987e'
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.uart] Sending: b'8610be7e'
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.uart] Data frame: b'6253b1f1c756a03fc36ab572a1556d869c7b9b7e'
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.ezsp] Application frame 89 (incomingRouteRecordHandler) received
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.uart] Data frame: b'7253b1f1c756a03fc36ab572a1556d869c659a7e'
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.uart] Sending: b'8070787e'
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.ezsp] Application frame 89 (incomingRouteRecordHandler) received
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.uart] Data frame: b'0253b1ed542a15b7d9944a65aa5592cb6381b4d7123162897cfe808975687e'
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.uart] Data frame: b'1253b19754b969b259914a25aa1592499c9625abedf7c07e'
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2017-08-17 18:26:22 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received
2017-08-17 18:26:22 WARNING (MainThread) [bellows.zigbee.device] [0x7c93] Failed ZDO request during device initialization: ('Endpoint request failed: %s', [129, 58168, []])
```
Looks like there are more people with the same kind of issue:
https://community.home-assistant.io/t/zha-zigbee-tested-devices-please-add-your-device-results/17718/113

andyzickler, 2017-09-30:
I'm also having the same problem. I tried with the latest version published on pip ( 0.3.5 ) on a raspberry pi with a [HUSBZB-1](https://www.amazon.com/GoControl-CECOMINOD016164-Linear-HUSBZB-1/dp/B01GJ826F8) usb stick. The light is the [TRÅDFRI 903.182.83](http://www.ikea.com/us/en/catalog/products/90318283/) on version 1.2.217 of the firmware (latest as of Sep 29th)

To your question above https://github.com/rcloran/bellows/issues/44#issuecomment-329727240 it looks like mine tried to join twice both times with the same id.

```
(venv) ~/python-zigbee $ bellows -v DEBUG permit -D ./app.db
debug: Using selector: EpollSelector
debug: Loading application state from ./app.db
debug: Sending: b'1ac038bc7e'
debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
debug: Send command version
debug: Sending: b'004221a850ed2c7e'
debug: Data frame: b'0142a1a8502805e67f627e'
debug: Sending: b'8160597e'
debug: Application frame 0 (version) received
debug: Send command setConfigurationValue
debug: Sending: b'7d314321fb582815c3277e'
debug: Data frame: b'1243a1fb54fb737e'
debug: Sending: b'82503a7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'224021fb592f15226f7e'
debug: Data frame: b'2340a1fb54c6107e'
debug: Sending: b'83401b7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'334121fb792b15a2d77e'
debug: Data frame: b'3441a1fb54d32a7e'
debug: Sending: b'8430fc7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'444621fb7d5e291514417e'
debug: Data frame: b'4546a1fb5435d07e'
debug: Sending: b'8520dd7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'554721fb4d2815713f7e'
debug: Data frame: b'5647a1fb54a9ec7e'
debug: Sending: b'8610be7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'664421fb55d515b18f7e'
debug: Data frame: b'6744a1fb54948f7e'
debug: Sending: b'87009f7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command networkInit
debug: Sending: b'774521bf02267e'
debug: Data frame: b'7045a5bf549c7d7e'
debug: Sending: b'8070787e'
debug: Application frame 23 (networkInit) received
debug: Data frame: b'0045b1b1c4beee7e'
debug: Sending: b'8160597e'
debug: Application frame 25 (stackStatusHandler) received
debug: Send command getNetworkParameters
debug: Sending: b'014a21808c477e'
debug: Data frame: b'114aa180542ba49b4fd23f94a204afce944127abedce677302c1b9947e'
debug: Sending: b'82503a7e'
debug: Application frame 40 (getNetworkParameters) received
debug: Send command setPolicy
debug: Sending: b'124b21fd517a60e97e'
debug: Data frame: b'224ba1fd54d8f87e'
debug: Sending: b'83401b7e'
debug: Application frame 85 (setPolicy) received
debug: Send command setPolicy
debug: Sending: b'234821fd524b97367e'
debug: Data frame: b'3348a1fd54ed2f7e'
debug: Sending: b'8430fc7e'
debug: Application frame 85 (setPolicy) received
debug: Send command setPolicy
debug: Sending: b'344921fd542b29a27e'
[... 52 lines cut ...]
debug: Send command sendUnicast
debug: Sending: b'1053219c54c4f7b259914a25aa1593499c4d24a8eeaee683237e'
debug: Data frame: b'0253a19c540cecdd7e'
debug: Sending: b'8160597e'
debug: Application frame 52 (sendUnicast) received
debug: Data frame: b'1253b1f1bac8bc2d7b6ab572a1556df49c66ff7e'
debug: Sending: b'82503a7e'
debug: Application frame 89 (incomingRouteRecordHandler) received
debug: Data frame: b'2253b1f1bac8bc2d7b6ab572a1556df59c77cd7e'
debug: Sending: b'83401b7e'
debug: Application frame 89 (incomingRouteRecordHandler) received
debug: Data frame: b'3253b1ed542a15b7d9944a65aa5592ab63f2c949123162887ca6e28982497e'
debug: Sending: b'8430fc7e'
debug: Application frame 69 (incomingMessageHandler) received
debug: Data frame: b'4253b19754c4f7b259914a25aa1592499c6824abed53af7e'
debug: Sending: b'8520dd7e'
debug: Application frame 63 (messageSentHandler) received
warning: [0xe2ee] Failed ZDO request during device initialization: ('Endpoint request failed: %s', [129, 33120, []])
Done
(venv) ~/python-zigbee $ bellows devices -D ./app.db
Device:
  NWK: 0x8160
  IEEE: 00:0b:57:ff:fe:22:9f:a9
  Endpoints:
```

AndreasBomholtz, 2017-10-05:
@andyzickler Your light also uses 2 different ID's and it is the first ID that is in the database.
ID's: 0x8160 and 0xe2ee.

I will try to make a version where I try to use the new ID. Can @andyzickler or @rklomp you try with a patch if I create it?

AndreasBomholtz, 2017-10-05:
In function `_handle_join` in file `bellows/zigbee/application.py`, try to out comment the `return`.

andyzickler, 2017-10-05:
@AndreasBomholtz Thanks for the reply. I commented out the `return` in `_handle_join`, unpaired my light from my TRÅDFRI hub, deleted the old device database and ran permit.

```
(venv) pi@raspberrypi:~/python-zigbee $ bellows -v DEBUG permit -D ./app.db
debug: Using selector: EpollSelector
debug: Loading application state from ./app.db
debug: Sending: b'1ac038bc7e'
debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
debug: Send command version
debug: Sending: b'004221a850ed2c7e'
debug: Data frame: b'0142a1a8502805e67f627e'
debug: Sending: b'8160597e'
debug: Application frame 0 (version) received
debug: Send command setConfigurationValue
debug: Sending: b'7d314321fb582815c3277e'
debug: Data frame: b'1243a1fb54fb737e'
debug: Sending: b'82503a7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'224021fb592f15226f7e'
debug: Data frame: b'2340a1fb54c6107e'
debug: Sending: b'83401b7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'334121fb792b15a2d77e'
debug: Data frame: b'3441a1fb54d32a7e'
debug: Sending: b'8430fc7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'444621fb7d5e291514417e'
debug: Data frame: b'4546a1fb5435d07e'
debug: Sending: b'8520dd7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'554721fb4d2815713f7e'
debug: Data frame: b'5647a1fb54a9ec7e'
debug: Sending: b'8610be7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'664421fb55d515b18f7e'
debug: Data frame: b'6744a1fb54948f7e'
debug: Sending: b'87009f7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command networkInit
debug: Sending: b'774521bf02267e'
debug: Data frame: b'7045a5bf549c7d7e'
debug: Sending: b'8070787e'
debug: Application frame 23 (networkInit) received
debug: Data frame: b'0045b1b1c4beee7e'
debug: Sending: b'8160597e'
debug: Application frame 25 (stackStatusHandler) received
debug: Send command getNetworkParameters
debug: Sending: b'014a21808c477e'
debug: Data frame: b'114aa180542ba49b4fd23f94a204afce944127abedce677302c1b9947e'
debug: Sending: b'82503a7e'
debug: Application frame 40 (getNetworkParameters) received
debug: Send command setPolicy
debug: Sending: b'124b21fd517a60e97e'
debug: Data frame: b'224ba1fd54d8f87e'
debug: Sending: b'83401b7e'
debug: Application frame 85 (setPolicy) received
debug: Send command setPolicy
debug: Sending: b'234821fd524b97367e'
debug: Data frame: b'3348a1fd54ed2f7e'
debug: Sending: b'8430fc7e'
debug: Application frame 85 (setPolicy) received
debug: Send command setPolicy
debug: Sending: b'344921fd542b29a27e'
debug: Data frame: b'4449a1fd54e1c97e'
debug: Sending: b'8520dd7e'
[... 102 lines cut ...]
(venv) pi@raspberrypi:~/python-zigbee $ bellows devices -D ./app.db
Device:
  NWK: 0x25bd
  IEEE: 00:0b:57:ff:fe:22:9f:a9
  Endpoints:
    1: profile=0xc05e, device_type=DeviceType.COLOR_TEMPERATURE_LIGHT
      Input Clusters:
        Basic (0)
        Identify (3)
        Groups (4)
        Scenes (5)
        On/Off (6)
        Level control (8)
        Color Control (768)
        Diagnostic (2821)
        LightLink (4096)
      Output Clusters:
        Scenes (5)
        Ota (25)
        Poll Control (32)
        LightLink (4096)
```

I'm happy to help test out any patches, and I have a different Ikea light that I can test on too if it will be helpful. It's a [TRÅDFRI 303.569.61
](http://www.ikea.com/us/en/catalog/products/30356961/) fixed color temperature light.

AndreasBomholtz, 2017-10-05:
This time it joined only once and the inclusion worked, with out hitting
the "patched" code.

Can you try again to see if you can hit the issue with double join?

patchedsoul, 2017-10-05:
I encountered the same issue today with respect to adding GE Link bulbs. I can confirm that your patch also resolved my issue too. my errors below for completeness...

2017-10-05 16:43:21 WARNING (MainThread) [bellows.zigbee.application] Message on unknown device 0xeea2
2017-10-05 16:44:56 WARNING (MainThread) [bellows.zigbee.device] [0x7d34] Failed ZDO request during device initialization: Message send failure: EmberStatus.DELIVERY_FAILED

gazoscalvertos, 2017-10-05:
Mine ikea bulb joins however timesout when trying to retrieve manufacturer data:

2017-10-05 22:55:16 DEBUG (MainThread) [bellows.ezsp] Application frame 36 (trustCenterJoinHandler) received
2017-10-05 22:55:16 INFO (MainThread) [bellows.zigbee.application] Device 0x8991 (00:0b:57:ff:fe:90:3a:f2) joined the network
2017-10-05 22:55:16 INFO (MainThread) [bellows.zigbee.device] [0x8991] Discovering endpoints
2017-10-05 22:55:16 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast
2017-10-05 22:55:16 DEBUG (MainThread) [bellows.uart] Sending: b'0652219c54bb9cb259914a25aa1593499c4f26a8ec5fee9ff87e'
2017-10-05 22:55:16 DEBUG (MainThread) [bellows.uart] Data frame: b'6152a19c54450e997e'
2017-10-05 22:55:16 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
2017-10-05 22:55:16 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received
2017-10-05 22:55:20 DEBUG (MainThread) [bellows.uart] Data frame: b'7152b19754bb9cb259914a25aa1593499c2126cded82667e'
2017-10-05 22:55:20 DEBUG (MainThread) [bellows.uart] Sending: b'8070787e'
2017-10-05 22:55:20 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Data frame: b'0152b18c517fe788c96ab572a15593499c4ed84e7e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.ezsp] Application frame 36 (trustCenterJoinHandler) received
2017-10-05 22:55:22 INFO (MainThread) [bellows.zigbee.application] Device 0x5505 (00:0b:57:ff:fe:90:3a:f2) joined the network
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.zigbee.application] Skip initialization for existing device 00:0b:57:ff:fe:90:3a:f2
2017-10-05 22:55:22 INFO (MainThread) [bellows.zigbee.device] [0x5505] Discovering endpoints
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Sending: b'7d3153219c542f40b259914a25aa1593499c4c25a8efcb32f9287e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Data frame: b'1253a19c545bdc4b7e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Data frame: b'2253b5caa610854ca6c34125c5c47e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.ezsp] Application frame 98 (incomingSenderEui64Handler) received
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Data frame: b'3253b1ed502a15a159944a2dae55920e638122fe12316b0af89391b36c80c0f0e0cd50cf637e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Sending: b'8430fc7e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.zigbee.zdo] [0x5505:zdo] ZDO request 0x0013: [21765, 00:0b:57:ff:fe:90:3a:f2, 142]
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Data frame: b'4253b1f1517fe788c96ab572a1556d869cca3e7e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Sending: b'8520dd7e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.ezsp] Application frame 89 (incomingRouteRecordHandler) received
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Data frame: b'5253b1f1517fe788c96ab572a1556d879ce70e7e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Sending: b'8610be7e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.ezsp] Application frame 89 (incomingRouteRecordHandler) received
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Data frame: b'6253b1ed542a15b7d9944a65ab559201638122fe12316189fdc33688fd23087e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Data frame: b'7253b197542f40b259914a25aa1593499c3f25abedbf007e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Sending: b'8070787e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received
2017-10-05 22:55:22 INFO (MainThread) [bellows.zigbee.device] [0x5505] Discovered endpoints: [1]
2017-10-05 22:55:22 INFO (MainThread) [bellows.zigbee.endpoint] [0x5505:1] Discovering endpoint information
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Sending: b'2050219c542f40b259904a25aa1593499c4d24afeecb328a7c427e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Data frame: b'0350a19c54585dde7e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Data frame: b'1350b1f1517fe788c96ab572a1556d879c67a57e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.ezsp] Application frame 89 (incomingRouteRecordHandler) received
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Data frame: b'2350b1f1517fe788c96ab572a1556d879c45a67e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
2017-10-05 22:55:22 DEBUG (MainThread) [bellows.ezsp] Application frame 89 (incomingRouteRecordHandler) received
[... cut ...]
    result = yield from self.read_attributes_raw(to_read)
  File "/srv/homeassistant/lib/python3.5/site-packages/bellows/zigbee/zcl/__init__.py", line 160, in read_attributes_raw
    v = yield from self.request(True, 0x00, schema, attributes)
  File "/srv/homeassistant/lib/python3.5/site-packages/bellows/zigbee/application.py", line 266, in request
    v = yield from asyncio.wait_for(reply_fut, timeout)
  File "/usr/lib/python3.5/asyncio/tasks.py", line 396, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError

gazoscalvertos, 2017-10-05:
Output direct from bellows:

Device 0x997e (00:0b:57:ff:fe:90:3a:f2) joined the network
debug: Skip initialization for existing device 00:0b:57:ff:fe:90:3a:f2
[0x997e] Discovering endpoints
debug: Send command sendUnicast
debug: Sending: b'2050219c54548cb259914a25aa1593499c4c25a8efb0fef7847e'
debug: Data frame: b'0350a19c54dc8cd27e'
debug: Sending: b'8160597e'
debug: Application frame 52 (sendUnicast) received
debug: Data frame: b'1350b5caa610854ca6c34125e7bb7e'
debug: Sending: b'82503a7e'
debug: Application frame 98 (incomingSenderEui64Handler) received
debug: Data frame: b'2350b1ed502a15a159944a2dae55921c6387593212316b0a835f91b36c80c0f0e0cd50e6d67e'
debug: Sending: b'83401b7e'
debug: Application frame 69 (incomingMessageHandler) received
debug: [0x997e:zdo] ZDO request 0x0013: [39294, 00:0b:57:ff:fe:90:3a:f2, 142]
debug: Data frame: b'3350b1f12ab3e788c96ab572a1556d809c065e7e'
debug: Sending: b'8430fc7e'
debug: Application frame 89 (incomingRouteRecordHandler) received
debug: Data frame: b'4350b1f12ab3e788c96ab572a1556d819c6f687e'
debug: Sending: b'8520dd7e'
debug: Application frame 89 (incomingRouteRecordHandler) received
debug: Data frame: b'5350b1ed542a15b7d9944a65ab55921f6387593212316189fdb8fa88fd9c6d7e'
debug: Sending: b'8610be7e'
debug: Application frame 69 (incomingMessageHandler) received
debug: Data frame: b'6350b19754548cb259914a25aa1593499cb825abede4e97e'
debug: Sending: b'87009f7e'
debug: Application frame 63 (messageSentHandler) received
[0x997e] Discovered endpoints: [1]
[0x997e:1] Discovering endpoint information
debug: Send command sendUnicast
debug: Sending: b'3751219c54548cb259904a25aa1593499c4d24afeeb0fe8ac9367e'
debug: Data frame: b'7451a19c54ddbb7f7e'
debug: Sending: b'8070787e'
debug: Application frame 52 (sendUnicast) received
debug: Data frame: b'0451b1f12ab3e788c96ab572a1556d819ccde67e'
debug: Sending: b'8160597e'
debug: Application frame 89 (incomingRouteRecordHandler) received
debug: Data frame: b'1451b1f12ab3e788c96ab572a1556d809ce0d67e'
debug: Sending: b'82503a7e'
debug: Application frame 89 (incomingRouteRecordHandler) received
debug: Data frame: b'2451b1ed542a15b6d9944a65ab55921e6386593212314288fdb8faa9fd20ffa7eacfd66f8ffcc7dfd5d7698a462ba9e97d3bb5ee708258986c2613a1aeec7e'
debug: Sending: b'83401b7e'
debug: Application frame 69 (incomingMessageHandler) received
debug: Data frame: b'3451b19754548cb259904a25aa1593499cb924abed0f907e'
debug: Sending: b'8430fc7e'
debug: Application frame 63 (messageSentHandler) received
[0x997e:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=49246 device_type=256 device_version=2 input_clusters=[0, 3, 4, 5, 6, 8, 2821, 4096] output_clusters=[5, 25, 32, 4096]>
debug: Send command sendUnicast
debug: Sending: b'4456219c54548cb259914a25aa1593499c4a23a8e9fe7662cb7e'
debug: Data frame: b'4556a19c54d247687e'
debug: Sending: b'8520dd7e'
debug: Application frame 52 (sendUnicast) received
debug: Data frame: b'5556b1f12ab3e788c96ab572a1556d809ce0af7e'
debug: Sending: b'8610be7e'
debug: Application frame 89 (incomingRouteRecordHandler) received
debug: Data frame: b'6556b1f12ab3e788c96ab572a1556d819cf19d7e'
debug: Sending: b'87009f7e'
debug: Application frame 89 (incomingRouteRecordHandler) received
debug: Data frame: b'7556b1ed542a15b7d9944a65ab559211638759321231628f7cf67289e2197e'
debug: Sending: b'8070787e'
debug: Application frame 69 (incomingMessageHandler) received
debug: Data frame: b'0556b19754548cb259914a25aa1593499cb623abed83027e'
debug: Sending: b'8160597e'
debug: Application frame 63 (messageSentHandler) received
warning: [0x997e] Failed ZDO request during device initialization: ('Endpoint request failed: %s', [129, 4400, []])
debug: Data frame: b'1556b1f12ab3e788c96ab572a1556d839ccdf87e'
debug: Sending: b'82503a7e'
debug: Application frame 89 (incomingRouteRecordHandler) received
debug: Data frame: b'2556b1ed542a15b7d9944a65ab55921f6387593212316189fdb8fa88fd9bec7e'
debug: Sending: b'83401b7e'
debug: Application frame 69 (incomingMessageHandler) received
warning: Unexpected response TSN=2 command=32773 args=[0, 39294, [1]]
debug: [0x997e:zdo] Unexpected ZDO reply 0x8005: [0, 39294, [1]]

gazoscalvertos, 2017-10-06:
I manually added manufacturer data etc to the attributes table yet the bulbs still don't work.

Could the Ikea ones be running proprietary commands?

I have two of these bulbs: http://www.ikea.com/gb/en/products/lighting/smart-lighting/tr%C3%A5dfri-led-bulb-e14-400-lumen-wireless-dimmable-warm-white-chandelier-opal-white-art-60365271/

I also have a couple of Philips Hue which work a treat with Bellows.

Has anyone had any joy with these?

sweco-secfha, 2017-10-23:
I also had the problem with 2 ids and managed to get the bulb to join by commenting out `return`. Although I get a warning:
```
Joins are permitted for the next 30s...
warning: Message on unknown device 0xacfe
warning: Message on unknown device 0xacfe
warning: Message on unknown device 0xacfe
warning: Message on unknown device 0xacfe
warning: Message on unknown device 0xacfe
warning: Message on unknown device 0xacfe
Device 0x2164 (00:0b:57:ff:fe:27:30:f5) joined the network
[0x2164] Discovering endpoints
Device 0x4477 (00:0b:57:ff:fe:27:30:f5) joined the network
[0x4477] Discovering endpoints
[0x4477] Discovered endpoints: [1]
[0x4477:1] Discovering endpoint information
[0x4477:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=49246 device_type=544 device_version=2 input_clusters=[0, 3, 4, 5, 6, 8, 768, 2821, 4096] output_clusters=[5, 25, 32, 4096]>
warning: [0x4477] Failed ZDO request during device initialization: ('Endpoint request failed: %s', [129, 8548, []])
warning: Unexpected response TSN=2 command=32773 args=[0, 17527, [1]]
Done
``` 

If you need any help testing I'm up for the task.

perosb, 2017-11-07:
Having the same issues.
I can get by the "double join" issue by using https://github.com/rcloran/bellows/pull/65

```
Joins are permitted for the next 30s...
debug: Data frame: b'2152b18c558aded3696ab572a15593499c4e13de7e'
debug: Sending: b'83401b7e'
debug: Application frame 36 (trustCenterJoinHandler) received
Device 0xa001 (00:0b:57:ff:fe:30:61:cb) joined the network
debug: Device 00:0b:57:ff:fe:30:61:cb changed id (0xfd8b => 0xa001)
debug: Data frame: b'3152b18c566a1cfd7d6ab572a15593499c4e1f327e'
debug: Sending: b'8430fc7e'
debug: Application frame 36 (trustCenterJoinHandler) received
Device 0x4002 (00:0b:57:ff:fe:24:4f:09) joined the network
debug: Device 00:0b:57:ff:fe:24:4f:09 changed id (0xeb1d => 0x4002)
```

Querying for an attribute fails though.
Timeout issue can also be seen in home-assistant

```Traceback (most recent call last):
  File "/home/perosb/build/home-assistant/venv/bin/bellows", line 11, in <module>
    load_entry_point('bellows==0.4.0', 'console_scripts', 'bellows')()
  File "/home/perosb/build/home-assistant/venv/lib/python3.6/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/perosb/build/home-assistant/venv/lib/python3.6/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/perosb/build/home-assistant/venv/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/perosb/build/home-assistant/venv/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/perosb/build/home-assistant/venv/lib/python3.6/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/perosb/build/home-assistant/venv/lib/python3.6/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/perosb/build/home-assistant/venv/lib/python3.6/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/perosb/build/home-assistant/venv/lib/python3.6/site-packages/bellows/cli/util.py", line 72, in inner
    loop.run_until_complete(async_inner(*args, **kwargs))
  File "/usr/lib/python3.6/asyncio/base_events.py", line 467, in run_until_complete
    return future.result()
  File "/home/perosb/build/home-assistant/venv/lib/python3.6/site-packages/bellows/cli/util.py", line 59, in async_inner
    yield from f(ctx, *args, **kwargs)
  File "/home/perosb/build/home-assistant/venv/lib/python3.6/site-packages/bellows/cli/application.py", line 241, in read_attribute
    v = yield from cluster.read_attributes([attribute], allow_cache=False)
  File "/home/perosb/build/home-assistant/venv/lib/python3.6/site-packages/bellows/zigbee/zcl/__init__.py", line 192, in read_attributes
    result = yield from self.read_attributes_raw(to_read)
  File "/home/perosb/build/home-assistant/venv/lib/python3.6/site-packages/bellows/zigbee/zcl/__init__.py", line 159, in read_attributes_raw
    v = yield from self.request(True, 0x00, schema, attributes)
  File "/home/perosb/build/home-assistant/venv/lib/python3.6/site-packages/bellows/zigbee/application.py", line 286, in request
    v = yield from send_fut
bellows.zigbee.exceptions.DeliveryError: Message send failure: EmberStatus.DELIVERY_FAILED```

perosb, 2017-11-07:
Complete debug output when `permit` new device:

```
Joins are permitted for the next 30s...
debug: Data frame: b'2152b18c3cdd05c96e6ab572a15593499c4ea6b47e'
debug: Sending: b'83401b7e'
debug: Application frame 36 (trustCenterJoinHandler) received
Device 0xf768 (00:0b:57:ff:fe:37:7b:10) joined the network
[0xf768] Discovering endpoints
debug: Send command sendUnicast
debug: Sending: b'7d3353219c5442e2b259914a25aa1593499c4f26a8eca690b0f57e'
debug: Data frame: b'3253a19c54d7b9477e'
debug: Sending: b'8430fc7e'
debug: Application frame 52 (sendUnicast) received
debug: Data frame: b'4253b1975442e2b259914a25aa1593499cb326cded21307e'
debug: Sending: b'8520dd7e'
debug: Application frame 63 (messageSentHandler) received
debug: Data frame: b'5253b18cc33905c96e6ab572a15593499c4e31ba7e'
debug: Sending: b'8610be7e'
debug: Application frame 36 (trustCenterJoinHandler) received
Device 0x1397 (00:0b:57:ff:fe:37:7b:10) joined the network
debug: Device 00:0b:57:ff:fe:37:7b:10 changed id (0xf768 => 0x1397)
debug: Canceling old initialize call
[0x1397] Discovering endpoints
debug: Send command sendUnicast
debug: Sending: b'2650219c54bd06b259914a25aa1593499c4c25a8ef59749daa7e'
debug: Data frame: b'6350a19c54d542e37e'
debug: Sending: b'87009f7e'
debug: Application frame 52 (sendUnicast) received
debug: Data frame: b'7350b5ca4451224ca6c34125e7207e'
debug: Sending: b'8070787e'
debug: Application frame 98 (incomingSenderEui64Handler) received
debug: Data frame: b'0350b1ed502a15a159944a2dae5592616384b0b812316b0a6ad573f2cb80c0f0e0cd5064387e'
debug: Sending: b'8160597e'
debug: Application frame 69 (incomingMessageHandler) received
debug: [0x1397:zdo] ZDO request 0x0013: [5015, 00:0b:57:ff:fe:37:7b:10, 142]
debug: Data frame: b'1350b1f1c33905c96e6ab572a1556d839c27c17e'
debug: Sending: b'82503a7e'
debug: Application frame 89 (incomingRouteRecordHandler) received
debug: Data frame: b'2350b1f1c33905c96e6ab572a1556d839c05c27e'
debug: Sending: b'83401b7e'
debug: Application frame 89 (incomingRouteRecordHandler) received
debug: Data frame: b'3350b1ed542a15b7d9944a65ab5592606384b0b812316189fd517088fdb5497e'
debug: Sending: b'8430fc7e'
debug: Application frame 69 (incomingMessageHandler) received
debug: Data frame: b'4350b19754bd06b259914a25aa1593499cb125abed5f477e'
debug: Sending: b'8520dd7e'
debug: Application frame 63 (messageSentHandler) received
[0x1397] Discovered endpoints: [1]
[0x1397:1] Discovering endpoint information
debug: Send command sendUnicast
debug: Sending: b'3551219c54bd06b259904a25aa1593499c4d24afee59748af6d67e'
debug: Data frame: b'5451a19c542a118f7e'
debug: Sending: b'8610be7e'
debug: Application frame 52 (sendUnicast) received
debug: Data frame: b'6451b1f1c33905c96e6ab572a1556d809cf21f7e'
debug: Sending: b'87009f7e'
debug: Application frame 89 (incomingRouteRecordHandler) received
debug: Data frame: b'7451b1f1c33905c96e6ab572a1556d809cec1e7e'
debug: Sending: b'8070787e'
debug: Application frame 89 (incomingRouteRecordHandler) received
debug: Data frame: b'0451b1ed542a15b6d9944a65ab5592636387b0b812314088fd5170abfd20ff87e9cfd76f8ffcc7dfd5d7698a462ba9ec753eaeea65864498552633b1e06057047e'
debug: Sending: b'8160597e'
debug: Application frame 69 (incomingMessageHandler) received
debug: Data frame: b'1451b19754bd06b259904a25aa1593499c4e24abed89297e'
debug: Sending: b'82503a7e'
debug: Application frame 63 (messageSentHandler) received
[0x1397:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=49246 device_type=544 device_version=2 input_clusters=[0, 3, 4, 5, 6, 8, 768, 2821, 4096] output_clusters=[5, 25, 32, 4096]>
debug: Send command sendUnicast
debug: Sending: b'4256219c54bd06b259914a25aa1593499c4a23a8e9a69088f37e'
debug: Data frame: b'2556a19c542b66467e'
debug: Sending: b'83401b7e'
debug: Application frame 52 (sendUnicast) received
debug: Data frame: b'3556b1f1c33905c96e6ab572a1556d809cec677e'
debug: Sending: b'8430fc7e'
debug: Application frame 89 (incomingRouteRecordHandler) received
debug: Data frame: b'4556b1f1c33905c96e6ab572a1556d809cb6607e'
debug: Sending: b'8520dd7e'
debug: Application frame 89 (incomingRouteRecordHandler) received
debug: Data frame: b'5556b1ed542a15b7d9944a65ab5592626384b0b81231628f7cae948971327e'
debug: Sending: b'8610be7e'
debug: Application frame 69 (incomingMessageHandler) received
debug: Data frame: b'6556b19754bd06b259914a25aa1593499c4f23abedbc977e'
debug: Sending: b'87009f7e'
debug: Application frame 63 (messageSentHandler) received
warning: [0x1397] Failed ZDO request during device initialization: ('Endpoint request failed: %s', [129, 63336, []])
debug: Data frame: b'7556b1f1c33905c96e6ab572a1556d839cc1307e'
debug: Sending: b'8070787e'
debug: Application frame 89 (incomingRouteRecordHandler) received
debug: Data frame: b'0556b1ed542a15b7d9944a65ab5592606387b0b812316189fd517088fd79087e'
debug: Sending: b'8160597e'
debug: Application frame 69 (incomingMessageHandler) received
warning: Unexpected response TSN=2 command=32773 args=[0, 5015, [1]]
debug: [0x1397:zdo] Unexpected ZDO reply 0x8005: [0, 5015, [1]]
Done
```

perosb, 2017-11-10:
@AndreasBomholtz are you still able to join your IKEA bulbs? Even after their upgrades etc.

AndreasBomholtz, 2017-11-10:
@perosb I'm not using the IKEA gateway, so my bulbs are not updated. But it might be an idea to try to update them and see what is going on :)

perosb, 2017-11-12:
@AndreasBomholtz @rcloran anything I can do to get more debug data needed? I guess all new trådfri bulbs sold are using the latest fw version. Unless this is an issue with my zigbee stick (ETRX357USB-LRS+8M).

Gamester17, 2017-11-13:
@perosb What exact firmware have flashed on the ETRX357USB-LRS+8M stick?

I have not got one myself (yet) but read in Home Assistant forum that need other firmware:

https://community.home-assistant.io/t/eu-usb-sticks-for-the-new-zigbee-component/16718

Frank there wrote:

> _I can confirm that the ETRX357USB-LRS+8M can be flashed with an appropriate EZSP firmware without the use of any programming hardware.

So, without any guarantees (you might -but probably won’t- brick your device), here’s how:

Download the firmware. I used some random blob someone put on github that at least seemed to have the right name. What could possibly go wrong, right? :smile: https://github.com/yqyunjie/Zigbee-Project/blob/master/firmware/EmberZNet/EM35x-EZSP/build/em35x-ezsp-images/EM357/em357-ncp-uart-xon-xoff-use-with-serial-uart-bl-500.ebl?raw=true34

(Install USB-to-serial drivers for the device. Linux will load the driver automatically when you plug in; not sure about other OSes.)

Install a serial port communication app that supports X-MODEM. (debian/ubuntu: sudo apt-get install minicom) Run it and configure it (sudo minicom -s) to use the fake serial port (/dev/ttyUSB0) at 19200 baud 8N1. Disable both hardware (RTS/CTS) and software (XON/XOFF) flow control.

Type ‘AT’ , you should get OK in response. Now type ‘AT+BLOAD’ . The device will reboot into the bootloader.

Change the baud rate in the serial port communication app setting to 115200 baud. (exit minicom using CTRL-A Q, run ‘sudo minicom -s’ again)

Pressing enter in the terminal should now show you a three-option boot loader menu. Choose option 1.

‘C’ characters will start showing. Don’t wait for this to finish, but start an X-MODEM upload of the firmware you downloaded earlier (use CTRL-A S in minicom). You have 60 seconds to start the upload.

After the upload finished, you should return to the menu. Now select option 2, to reboot into the new firmware. You’re done._

perosb, 2017-11-13:
@Gamester17 yes, that's the firmware I'm using.

guardmedia, 2017-11-13:
Same as @perosb , using #65 I'm able to get past the join issue, but any commands result in a timeout. Using a Linear/GoControl HUSBZB-1 here.

h3ndrik, 2017-11-14:
Similar errors with me. Is this a problem with newer versions of the tradfri firmwares or is tradfri just not supported in bellows (yet)?

Can I do something to see if the light bulb successfully joined the network? Or send a command to the bulb? Because all my previous attempts to read or write attributes failed.

ryanwinter, 2018-01-14:
I modified the _handle_join and was able to connect all my Ikea TRADFRI bulb E26 opal 1000lm:

```
    def _handle_join(self, nwk, ieee, device_update, join_dec, parent_nwk):
        LOGGER.info("Device 0x%04x (%s) joined the network", nwk, ieee)
        #if ieee in self.devices:
            #dev = self.get_device(ieee)
            #dev.nwk = nwk
            #if dev.initializing or dev.status == bellows.zigbee.device.Status.ENDPOINTS_INIT:
                #LOGGER.debug("Skip initialization for existing device %s", ieee)
                #return
        #else:
        dev = self.add_device(ieee, nwk)
        self.listener_event('device_joined', dev)

        dev.schedule_initialize()

```

perosb, 2018-01-14:
> I modified the _handle_join and was able to connect all my Ikea TRADFRI bulb E26 opal 1000lm:

Hmm..and you can control them how? HA?
I can get the devices added (It looks like). I'm using @AndreasBomholtz developer branch https://github.com/Seluxit/bellows/tree/develop which seem to have more potential fixes.

But I'm not able to get them added to HA.
```python
Traceback (most recent call last):
  File "/usr/lib/python3.6/asyncio/tasks.py", line 180, in _step
    result = coro.send(None)
  File "/home/perosb/build/home-assistant/homeassistant/components/zha/__init__.py", line 141, in async_device_initialized
    discovered_info = yield from _discover_endpoint_info(endpoint)
  File "/home/perosb/build/home-assistant/homeassistant/components/zha/__init__.py", line 285, in _discover_endpoint_info
    yield from read(['manufacturer', 'model'])
  File "/home/perosb/build/home-assistant/homeassistant/components/zha/__init__.py", line 281, in read
    allow_cache=True,
  File "/home/perosb/build/bellows/bellows/zigbee/zcl/__init__.py", line 227, in read_attributes
    result = yield from self.read_attributes_raw(to_read, manufacturer=manufacturer)
  File "/home/perosb/build/bellows/bellows/zigbee/zcl/__init__.py", line 194, in read_attributes_raw
    v = yield from self.request(True, 0x00, schema, attributes, manufacturer=manufacturer)
  File "/home/perosb/build/bellows/bellows/zigbee/application.py", line 299, in request
    v = yield from asyncio.wait_for(reply_fut, timeout)
  File "/usr/lib/python3.6/asyncio/tasks.py", line 362, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
```

Whats the bellowscli way of tuning a light on/off?

ryanwinter, 2018-01-15:
I'm using the standard version with the modification I outlined above. Looks like you are using the colour temp changing bulbs, I'm just using the dimmer variety (TRADFRI bulb E26 opal 1000lm).

```
Device:
  NWK: 0x157f
  IEEE: 00:0b:57:ff:fe:48:1c:a1
  Endpoints:
    1: profile=0x104, device_type=DeviceType.ON_OFF_LIGHT
      Input Clusters:
        Basic (0)
        Identify (3)
        Groups (4)
        Scenes (5)
        On/Off (6)
        Level control (8)
        Diagnostic (2821)
        LightLink (4096)
      Output Clusters:
        Scenes (5)
        Ota (25)
        Poll Control (32)
        LightLink (4096)
```

ryanwinter, 2018-01-16:
I actually found that _handle_join patch makes pairing my Smartthings multi-sensors more reliable. I kept trying to pair a new one and it said it was successful however it didn't actual produce events. After patching and re-pairing, everythings is good.

Yoda-x, 2018-01-26:
One problem I found was that bulbs on new firmware reports as ZLL, but you need to talk to them with the ZHA profile.  
the last trace from @perosb shows the timeout of the query for  model and manufacturer. If you do the same query with profile-Id=0x104 you get an answer. 
I created a template in my spin-off to override the profile_id with zha instead of zll and can manage my tradfri bulbs (dimmable and color-temp) without problems. 
Not sure, if this zll/zha mismatch relates to the firmware level. At least all of my bulbs report as zll. @ryanwinter has a zha bulb. maybe with older software?

perosb, 2018-01-27:
@Yoda-x cool, updated my endpoints to 0x104 and they show up as on/off switches in hass. at least I can toggle them on off.

Also testing out your repo of bellows.

perosb, 2018-01-27:
Ah,

I updated the "device_type" to 0x0102 and everything seem jolly good!
`update endpoints set profile_id = 0x0104, device_type = 0x0102;`

values from:
```
0x0100 | ON/OFF Light
0x0101 | Dimmable Light
0x0102 | Color Dimmable Light
0x0103 | ON/OFF Light Switch
0x0104 | Dimmer Switch
0x0105 | Color Dimmer Switch
0x0106 | Light Sensor
0x0107 | Occupancy Sensor
```

rcloran, 2018-02-05:
4cf12375111f3333d04c00159b07b0b004cc0538 should fix some of the issues reported here. 

Please re-open in the new zigpy project (https://github.com/zigpy/zigpy) if this is still an issue. There appears to be no way to move issues between projects.

closed 2018-02-05 by rcloran