zigpy/bellows #50: Support for RaspBee and ConBee hardware by Dresden Elektronik

Gamester17, 2017-09-01:
@rcloran would you please consider adding RaspBee ZigBee Shield for Raspberry Pi and ConBee USB dongle hardware, both made by Dresden Elektronik

RaspBee and ConBee (v1/v2) by Dresden Elektronik are the two most common/popular ZigBee radio adapters available in Europe, and they are of course also available in North America too. Supporting both ZigBee Home Automation (ZHA) and ZigBee Light Link (ZLL) 

https://www.dresden-elektronik.de/conbee/?L=1&ref=gh
https://www.dresden-elektronik.de/raspbee/?L=1&ref=gh

https://www.dresden-elektronik.de/funktechnik/service/download/driver/?L=1

https://github.com/dresden-elektronik
https://github.com/dresden-elektronik/basic-aps-plugin
https://github.com/dresden-elektronik/deconz-rest-plugin
https://github.com/dresden-elektronik/deconz-rest-doc

Dresden Elektronik today their own open source ZigBee gateway software called "deCONZ" which has APIs for RaspBee and ConBee hardware.

https://www.dresden-elektronik.de/funktechnik/products/software/pc/deconz/?L=1

The goal would be to use RaspBee and ConBee hardware devices in Home Assistant and Hass.io

rcloran, 2017-09-02:
I'm planning on splitting out the zigbee parts of bellows (soon -- work is in progress), which will allow using that with libraries that interface with other . Adding support for new radios is not a goal for bellows at the moment.

closed 2017-09-02 by rcloran

Gamester17, 2017-09-04:
@rcloran that is cool, however if you change your mind Dresden Elektronik have offered to donate RaspBee/ConBee hardware to developers Home Assistant if they would like to implement support for them, see @manup comment here:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/158

abmantis, 2017-09-16:
Hey @rcloran, any progress on the split?

Gamester17, 2017-10-03:
@rcloran FYI; here is a link to the serial UART protocol documentation used by ConBee and RaspBee hardware:

http://www.dresden-elektronik.de/rpi/deconz/deCONZ-Serial-Protocol-en.pdf

PS: If send a mail to support@dresden-elektronik.de and reference Manu they can send a RaspBee and a ConBee. 

This above information was provided my manup (Manuel Pietschmann) here => https://github.com/dresden-elektronik/deconz-rest-plugin/issues/158

Gamester17, 2017-10-30:
@rcloran any news on splitting out the Zigbee parts of bellows?

That is, to allow using it with libraries that interface with other.

Gamester17, 2018-01-09:
@rcloran any chance that you could reconsider adding deCONZ Serial Protocol support to bellows?

http://www.dresden-elektronik.de/rpi/deconz/deCONZ-Serial-Protocol-en.pdf

manup from Dresden-Elektronik said that they will send you a free ConBee and Raspbee if you post here:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/158

Gamester17, 2018-01-15:
@AndreasBomholtz is this maybe something that you would be interested in?

Equidamoid, 2018-03-06:
Hi everyone (@rcloran?). 
Is there any update or plans? I'm considering writing a python wrapper for their serial api and don't really want to reinvent the wheel if someone already works on it =)