zigpy/bellows #52: Proper remove/forget support

roothorick, 2017-09-02:
This is a pretty critical feature, seeing as bellows manages its own database, and it's a regular occurrence that a device is unavailable or has been reset and is no longer *actually* in the network to receive a leave command (if that even removes the device from the database -- which is unclear). This makes troubleshooting connection problems difficult, potentially impossible if direct (i.e. outside of bellows) access to the database isn't available.

rcloran, 2017-09-02:
Does #34 add the support you need?

roothorick, 2017-09-02:
Possibly. From a cursory glance with a lack of knowledge about the code, it's unclear to me whether that will remove the device from the database if bellows does not receive a response, which is the most important part here.

It also doesn't appear to add the functionality to the CLI; corresponding CLI commands and/or an implementation in e.g. Home Assistant is necessary.

rcloran, 2017-09-02:
New line 160 in application.py in that PR is where it removes the device from the database.

I'll update that PR to make the application leave CLI command use this instead of just sending the ZDO.

closed 2017-09-24 by rcloran