zigpy/bellows #55: State update after power turned on

andrewpc, 2017-09-11:
Hi @rcloran - Just wanted to say that Im now using your library (and HASS) with the Cortet MeshConnect EM357 stick. I did have to update the firmware, bootloader and baud rate setting using an ISA3 debug device, but now its working successfully with HASS and a Wemo bulb. Thanks for your hard work.

Quick question, not sure if its a bug or not? If I cut power to a zigbee bulb and then turn the switch back on, should I expect that bulb to re-establish communication with the USB stick/bellows and update its status or does it need to be polled? My scenario is that I have a zigbee light bulb in a location where there is a physical wall switch. If the physical switch is turned on or off, the bulb state does not seem to be reflected. I would expect either 'on', 'off' or 'unknown', similar to how a lifx state works in HASS if switched off at the wall.

If it needs to be polled, is there a way to enable that in HASS?

Cheers

rcloran, 2017-10-15:
This should be a hass improvement, not a bellows improvement. Maybe hass does need more information from bellows -- I'm not sure we propagate the ZDO device announce message in any way that hass can hook into, which would probably be the right place to re-query the light state.

jsb5151, 2017-10-31:
Same issue here.  Trying to report keypresses on a ZHA wall switch (GE Tapt) but am not getting anything back.   Any way we can get a manual status update (zcl command reportable_change maybe)?

mhofman, 2018-01-20:
I didn't know about this open issue and reported a similar one on hass: https://github.com/home-assistant/home-assistant/issues/11782

rcloran, 2018-02-05:
Please re-open in the new zigpy project (https://github.com/zigpy/zigpy) if this is still an issue. There appears to be no way to move issues between projects.

@jsb5151 : @andrewpc is talking about something entirely different.

closed 2018-02-05 by rcloran

mhofman, 2018-02-05:
@rcloran, I think there's 2 issues really in what @andrewpc describes:
- on the bulb power on, the actual state is not reported to the consumer (Hass in this case). That's probably an API (usage) issue ?
- on bulb power off, bellows is not receiving the off state update, even when the bulb is capable of transmitting it (as verified when used with other controllers). That's most likely a protocol implementation issue, so I believe it's an issue that pertains to bellows.

rcloran, 2018-02-05:
Sure. I don’t think I suggested otherwise.

rcloran, 2018-02-05:
Oh, by “protocol level” you mean EZSP? It doesn’t seem likely. But it could be. Which ezsp message type do you think needs to be handled better?

mhofman, 2018-02-05:
I don't really know what's what, I'm not familiar at all with zigbee. All I know is that the logs showed no message being received on power off. I don't know if it because some kind of subscription needs to happen at the zigbee level, or if it's something znet needs to take care of.

mhofman, 2018-02-05:
I guess no matter what I could reopen a specific issue on zigpy and reference this discussion.