zigpy/bellows #70: Zigbee home thermostat

titilambert, 2017-11-04:
Hello !
I have this heater http://www.zigbee.org/zigbee-products-2/#zigbeecertifiedproducts/productdetails3/576aefccf87038aa313e32f1/
I want to plug it to home assistant.
I already made several HA components, so I don't have any issue with HA.
My issue is more about the Zigbee protocol.
I'm trying some bellows commands without success, I need a database
I don't know how to "form" a new network :/

In fact, I know where to start :D 

Thanks for your help !

titilambert, 2017-11-06:
I got some new stuff !
```
# deps/bin/bellows  -d /dev/ttyUSB0 devices -D zigbee.db 
Device:
  NWK: 0xd99f
  IEEE: f8:f0:05:ff:ff:f8:93:e0
  Endpoints:
    25: profile=0x104, device_type=DeviceType.THERMOSTAT
      Input Clusters:
        Basic (0)
        Identify (3)
        Groups (4)
        Thermostat (513)
        Thermostat User Interface Configuration (516)
      Output Clusters:
        Temperature Measurement (1026)
Device:
  NWK: 0x12af
  IEEE: f8:f0:05:ff:ff:f9:6a:b2
  Endpoints:
    25: profile=0x104, device_type=DeviceType.THERMOSTAT
      Input Clusters:
        Basic (0)
        Identify (3)
        Groups (4)
        Thermostat (513)
        Thermostat User Interface Configuration (516)
      Output Clusters:
        Temperature Measurement (1026)
```

rcloran, 2018-02-05:
Please re-open in the new zigpy project (https://github.com/zigpy/zigpy). There appears to be no way to move issues between projects.

closed 2018-02-05 by rcloran