zigpy/bellows #80: ZHA not able to control light until HA is restart

sumitrajancse, 2017-12-15:
Following error occurs and Home assistant needs to restart to control the lights again.
Can you please guide what code can be modify to avoid restarting of the HA. 

`2017-12-15 06:43:04 WARNING (MainThread) [bellows.zigbee.application] Unexpected message send notification
2017-12-15 06:43:04 WARNING (MainThread) [bellows.zigbee.application] Unexpected response TSN=8 command=1 args=[[<ReadAttributeRecord attrid=0 status=0 value=254>]]
2017-12-15 06:48:08 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/lib/python3.6/asyncio/tasks.py", line 181, in _step
    result = coro.throw(exc)
  File "/usr/lib/python3.6/site-packages/homeassistant/core.py", line 1031, in _event_to_service_call
    yield from service_handler.func(service_call)
  File "/usr/lib/python3.6/site-packages/homeassistant/components/light/__init__.py", line 271, in async_handle_light_service
    yield from light.async_turn_on(**params)
  File "/usr/lib/python3.6/site-packages/homeassistant/components/light/zha.py", line 105, in async_turn_on
    duration
  File "/usr/lib/python3.6/site-packages/bellows/zigbee/application.py", line 284, in request
    v = yield from send_fut
  File "/usr/lib/python3.6/asyncio/futures.py", line 331, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.6/asyncio/tasks.py", line 244, in _wakeup
    future.result()
  File "/usr/lib/python3.6/asyncio/futures.py", line 244, in result
    raise self._exception
bellows.zigbee.exceptions.DeliveryError: Message send failure: EmberStatus.DELIVERY_FAILED
2017-12-15 06:48:13 WARNING (MainThread) [bellows.zigbee.application] Unexpected message send notification`

sumitrajancse, 2017-12-15:
I have following zigbee devices:

Nortek USB stick with Zwave and zigbee protocol.
1 Osram lightify RBGW buld
2 sengled tunable white light bulbs.
3 cree connected bulbs

List of errors in the logs and after sometimes not able to control zigbee devices until home assistant is restarted. The light stop to respond to the home assistant and when i verify the logs found following error.
It is full of error.

rcloran, 2018-02-05:
Is this a duplicate of #37?

Nayar, 2018-07-09:
I'm having the same issue when restarting HA. Any solution?

Adminiuga, 2018-12-19:
If this is still relevant, then all the fun is in https://github.com/zigpy/bellows/issues/124
otherwise we should close this one.

closed 2019-05-23 by Adminiuga