zigpy/bellows #85: Osram Lightify Dimming Switch

kalliste, 2017-12-31:
This device pairs ok but doesn't show up in the web interface and nothing shows up in the logs when I push the button. I have enabled debug logging for the bellows component.

Device:
  NWK: 0xc51a
  IEEE: 84:18:26:00:00:e8:f8:d2
  Endpoints:
    1: profile=0x104, device_type=DeviceType.LEVEL_CONTROL_SWITCH
      Input Clusters:
        Basic (0)
        Power Configuration (1)
        Identify (3)
        Poll Control (32)
        Temperature Measurement (1026)
        Diagnostic (2821)
      Output Clusters:
        Identify (3)
        On/Off (6)
        Level control (8)
        Ota (25)

Yoda-x, 2018-01-11:
currently LEVEL_CONTROL_SWITCH has no component assigned in ZHA code. You can try to add this to your configuration yaml in the zha section. 
Not sure for "Level control", but on/off should work then. 
Level control may need some custom code depending how it's reported.  
Be carefull about the indent. 

 ```
...
     database_path: /config/zigbee.db
#add below device settings
     device_config: 
       "84:18:26:00:00:e8:f8:d2-1":
          type: switch

```

rcloran, 2018-02-05:
Please re-open in the new zigpy project (https://github.com/zigpy/zigpy) if this is still an issue. There appears to be no way to move issues between projects.

closed 2018-02-05 by rcloran