zigpy/bellows #86: Electrical Measurement cluster support

untergeek, 2018-01-02:
```
Device:
  NWK: 0x2744
  IEEE: 00:0d:6f:00:0b:2c:f3:13
  Endpoints:
    1: profile=0x104, device_type=DeviceType.SMART_PLUG
      Input Clusters:
        Basic (0)
        Identify (3)
        Groups (4)
        Scenes (5)
        On/Off (6)
        Metering (1794)
        Electrical Measurement (2820)
        Diagnostic (2821)
        Manufacturer Specific (64515)
      Output Clusters:
        Ota (25)
```
**Full Disclosure**: I haven't looked at the code at all

I would love to be able to track power usage by my plugs and lights.  With some pointers and feedback, I'd be happy to help lend a hand to add support for this.  Any pointers on where to get started?

dmulcahey, 2018-01-02:
I started this in HA: https://github.com/home-assistant/home-assistant/pull/11305

But I closed it for now so I can work out a better way to dynamically determine power unit of measure and adjust the calculation appropriately.

dmulcahey, 2018-01-15:
@untergeek if you want to try the support for cluster 2820 (ElectricalMeasurement) the changes can be found [here](https://github.com/dmulcahey/home-assistant/commit/964f71797ad3c4a0f9d78baab01bd30a8ba76bc4) for the moment...

@rcloran do you know if this cluster isn't reportable? or is that more of a manufacturer specific thing? I tried setting this up w/ configure reporting like the rest of the sensors and it didn't seem to work. I set it up to poll from hass instead and it seems pretty stable like this. My fear is that this adds extra overhead (it is just inefficient) to the zigbee network because it will poll indefinitely.

rcloran, 2018-02-05:
Please re-open in the new zigpy project (https://github.com/zigpy/zigpy) if this is still an issue. There appears to be no way to move issues between projects.

closed 2018-02-05 by rcloran