zigpy/bellows #92: don't expect a reply?

igorbernstein2, 2018-01-30:
Hi,
This is the second issue I ran into trying to get bellows to communicate with
https://www.homedepot.com/p/Hampton-Bay-Universal-Wink-Enabled-White-Ceiling-Fan-Premier-Remote-Control-99432/206591100.

It appears that the device does not send a `incomingMessageHandler` in reply to the on/off commands in the On/Off cluster. I can see that the reply expectation can be disabled in [bellows.zigbee.application.ControllerApplication#request](https://github.com/rcloran/bellows/blob/4cf12375111f3333d04c00159b07b0b004cc0538/bellows/zigbee/application.py#L278). However it doesn't appear that the layers above it can pass that arg.

I'm still fairly new to zigbee, but skimming through the spec, it doesn't appear that On/Off commands require an explicit reply. It seems like the reply can be omitted if the device sends other commands in response using the same Transaction sequence number as the received command. See `2.5.12.2 When Generated` in the [spec](http://www.zigbee.org/~zigbeeor/wp-content/uploads/2014/10/07-5123-06-zigbee-cluster-library-specification.pdf)

I'm wondering if by default commands should only wait for the `messageSentHandler` frame as opposed to waiting for an explicit reply?

rcloran, 2018-02-05:
Sounds reasonable.

rcloran, 2018-02-05:
Please re-open under the new zigpy project (https://github.com/zigpy/zigpy).

closed 2018-02-05 by rcloran