zigpy/bellows #94: Ability to rename end points

ryanwinter, 2018-02-06:
Current endpoints can be difficult to differentiate, especially when you have multiple of the same item (I have 10 ikea light bulbs).

The current name is constructed as a combination of the different device components:
```
self.entity_id = '%s.%s_%s_%s_%s' % (
            self._domain,
            slugify(manufacturer),
            slugify(model),
            ieeetail,
            endpoint.endpoint_id,
```

Something similar to Z-wave's rename would allow the user to rename their devices which would make config implementation (automation, configuration, etc) easier and less prone to error.

A suggest would be to add an additional table to zigbee.db that contains the mapping of the devices (configurable) name to the endpoint. This could, be default, be filled out in the same way as the names are currently defined, but also allow a user to override with their own custom naming.

rcloran, 2018-02-06:
This looks like it's entirely in Home Assistant code. What am I missing? Zigpy/bellows have no notion of a name of a device whatsoever, and I'm not convinced they should.

ryanwinter, 2018-02-06:
I am looking at how the z-wave components are renamed. I believe this is a backend service openzwave.

rcloran, 2018-02-06:
Yes, OpenZWave maintains device names. Zigpy has no notion of a device name. I'm not sure it should -- its purpose is to speak ZigBee, and there's nothing fundamental about ZigBee that requires this. Why would your design be better than extending the `device_config` in Home Assistant to allow a custom entity there? You can also already set a friendly name for your device, for example:

```
homeassistant:
  customize:
    binary_sensor.centralite_3320l_0bbc8eb6_1:
      friendly_name: Door sensor
```

ryanwinter, 2018-02-06:
Ignoring the customize/friendly name as this is only for UI rendering where I am talking about name for groupings/automations.

Neither design is necessarily better, but a feature like this needs to go somewhere. From what I have seen using other external components (zwave, mqtt) the device naming has been external to HA which is why I proposed here first.

If you're adamant that this is the totally wrong place, then I will propose at the HA layer and see if I can get traction there.

Perhaps is would be possible to fit this into something like MQTT where you can define your own entities .

closed 2018-02-06 by rcloran