zigpy/bellows #99: Errors when adding devices

DarkProjects, 2018-02-10:
I have been able to add and control Sengled bulbs, but when I add other devices I am getting the bellow error everytime.
2018-02-09 23:22:44 INFO (MainThread) [bellows.zigbee.application] Device 0x406c (00:0d:6f:00:0b:12:51:f4) joined the network
2018-02-09 23:22:45 WARNING (MainThread) [bellows.zigbee.application] Unexpected message send notification

Even though it says it joined, it has not. I cannot see the status of the device, and the LED keeps blinking indicating it has not been paired yet. I see this issue on Visonic MCT-340 E contact sensors and Centralite 3305-P motion sensors.

rcloran, 2018-02-10:
What version? Please provide debug logs.

CSCG, 2018-02-10:
Also watching this, whatever version is built into HA 0.62. I'm getting this issue trying to connect to Sengled bulbs now too, even though I've got 6 other ones paired.

DarkProjects, 2018-02-12:
HA 0.62 same issue on both 0.4.0 and 0.5.0

Where would I find the logs?

CSCG, 2018-02-13:
I plan on getting another mSD card to install a dev copy of HA without HASSIO to try to debug this this week. I will report back, as I'm not finding a good way in HASSIO command line to issue any direct bellows CLI commands...

rcloran, 2018-02-13:
@DarkProjects: https://home-assistant.io/components/logger/ - turn on debug logging as documented there. 

@CSCG : some people have reported issues with adding > N devices, where N seems to vary. Can you try patch bellows with https://github.com/Yoda-x/bellows/commit/9b31e42b78cad9949930d55f585d9574cea27105 ? Please do send a PR of a cleaned up version of that if it works.

DarkProjects, 2018-02-13:
Thanks! I will do that tonight when I get home and post the logs.

rcloran, 2018-02-13:
@CSCG : I forgot to add that I was able to remove and re-add a Sengled bulb to my system (A19 soft white) without issues. I think your issue is different from this one.

walthowd, 2018-02-13:
I also have a mix of about twenty Sengeld A19 soft white and BR30s working fine with 0.40 and 0.50.

DarkProjects, 2018-02-14:
```
korey@DarkProjects:~$ cat .homeassistant/home-assistant.log |grep "bellows"
2018-02-13 23:03:25 DEBUG (MainThread) [bellows.zigbee.appdb] Loading application state from zbnew333.db
2018-02-13 23:03:25 DEBUG (MainThread) [bellows.uart] Sending: b'1ac038bc7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Send command version
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'004221a850ed2c7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Data frame: b'0142a1a8502805e67f627e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Application frame 0 (version) received
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'7d314321fb582815c3277e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Data frame: b'1243a1fb54fb737e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'224021fb592f15226f7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Data frame: b'2340a1fb54c6107e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'334121fb792b15a2d77e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Data frame: b'3441a1fb54d32a7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'8430fc7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'444621fb7d5e291514417e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Data frame: b'4546a1fb5435d07e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'8520dd7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'554721fb4d2815713f7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Data frame: b'5647a1fb54a9ec7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'8610be7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'664421fb55d515b18f7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Data frame: b'6744a1fb54948f7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'774521fb4a2b1524a97e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Data frame: b'7045a1fb5481b57e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'8070787e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'004a21fb629e15b2107e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Data frame: b'014aa1fb63a4387e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Send command networkInit
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'7d314b21bf676c7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Data frame: b'124ba5bf5463787e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Application frame 23 (networkInit) received
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Data frame: b'224bb1b1c450837e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Application frame 25 (stackStatusHandler) received
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.ezsp] Send command getNetworkParameters
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'2348218038017e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Data frame: b'3348a180542b5d82da3cd804e887e62b944127abedce677302c1bb557e'
2018-02-13 23:03:26 DEBUG (MainThread) [bellows.uart] Sending: b'8430fc7e'
[... cut ...]
2018-02-13 23:04:42 DEBUG (MainThread) [bellows.uart] Data frame: b'2358b1ed542e14ab59954b65aa559208639bb3bd12316b8af8c763c7ec793fb7b8cac179227e'
2018-02-13 23:04:42 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
2018-02-13 23:04:42 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received
2018-02-13 23:04:42 DEBUG (MainThread) [bellows.zigbee.zcl] [0x1694:1:0x0019] ZCL request 0x0101: [0, 4174, 7, 520573712, 0]
2018-02-13 23:04:42 DEBUG (MainThread) [bellows.zigbee.zcl] [0x1694:1:0x0019] No handler for cluster command 1
```

CSCG, 2018-02-14:
@rcloran I snagged another pi and mSD, and I've got a fresh install of HASS on Hassbian. I tested against my old, not working bulbs, and they all paired just fine. The theoretical device limit of 6 still is a problem however, but I plan on testing out the Yoda zha fork tonight or tomorrow. 

I can confirm this appears to be different than this issue so I'll refrain from muddying the waters further; @rcloran do you hang out in the Discord or anything? I want to help debugging in the future, also.

dmulcahey, 2018-02-18:
@rcloran I also can't get centralite devices to pair using HA dev on a clean install... Here are the logs from a clean installs of HA dev and HA v .60. dev has logging cranked way up (I added additional log entries to zippy and bellows)

HA dev log:

```
grep 'bellows\|zigpy' home-assistant.log
2018-02-18 13:09:43 DEBUG (MainThread) [zigpy.appdb] Loading application state from zigbee.db
2018-02-18 13:09:43 DEBUG (MainThread) [bellows.uart] Sending: b'1ac038bc7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Send command version
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'004221a850ed2c7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Data frame: b'0142a1a8502805e67f627e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Application frame 0 (version) received
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'7d314321fb582815c3277e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Data frame: b'1243a1fb54fb737e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'224021fb592f15226f7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Data frame: b'2340a1fb54c6107e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'334121fb792b15a2d77e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Data frame: b'3441a1fb54d32a7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'8430fc7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'444621fb7d5e291514417e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Data frame: b'4546a1fb5435d07e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'8520dd7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'554721fb4d2815713f7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Data frame: b'5647a1fb54a9ec7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'8610be7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'664421fb55d515b18f7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Data frame: b'6744a1fb54948f7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'774521fb6206146b557e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Data frame: b'7045a1fb63c7017e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'8070787e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'004a21fb462a0bc9207e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Data frame: b'014aa1fb54e28c7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'7d314b21fb47161587ad7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Data frame: b'124ba1fb547eb07e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'224821fb4f2015ced07e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Data frame: b'2348a1fb5443d37e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
2018-02-18 13:09:44 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
[... cut ...]
2018-02-18 12:47:40 DEBUG (MainThread) [bellows.uart] Data frame: b'6459b19754896ab658944f24ab1593499c042cabed555a7e'
2018-02-18 12:47:40 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
2018-02-18 12:47:40 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received
```

in v.60 everything works as expected. In dev, the device ultimately leaves the network and then will attempt to pair again. This eventually causes issues w/ the new unique_id handling. Does anything from these logs stand out?

rcloran, 2018-02-18:
Blame diff is 44f1a90ecd02fd5783fe5c315ff383faa8ea1d51. We're replying to ZDO messages with a different APS sequence number now.

rcloran, 2018-02-18:
Hrm, no ... I think the sequence numbering stays the same. May just be that we're not handling the coroutine properly in the ZDO reply code.

dmulcahey, 2018-02-18:
I have a couple clean test setups that I can use to try different things if you have some ideas...

rcloran, 2018-02-19:
@dmulcahey : Can you test with https://github.com/zigpy/zigpy/pull/23 please

dmulcahey, 2018-02-19:
@rcloran Will do, give me about 20 mins. Just putting my son to bed.

dmulcahey, 2018-02-19:
@rcloran there is no longer an endless loop of the device failing to join. The device is now happy and thinks it has joined the network... but I have no entities in HA. I am going to revert everything to really ensure the test setup is clean and then I'll post another log.

dmulcahey, 2018-02-19:
@rcloran here is the new log:

```
grep 'bellows\|zigpy' home-assistant.log
2018-02-18 19:59:06 DEBUG (MainThread) [bellows.zigbee.appdb] Loading application state from zigbee.db
2018-02-18 19:59:06 DEBUG (MainThread) [bellows.uart] Sending: b'1ac038bc7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Send command version
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'004221a850ed2c7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Data frame: b'0142a1a8502805e67f627e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Application frame 0 (version) received
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'7d314321fb582815c3277e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Data frame: b'1243a1fb54fb737e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'224021fb592f15226f7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Data frame: b'2340a1fb54c6107e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'334121fb792b15a2d77e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Data frame: b'3441a1fb54d32a7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'8430fc7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'444621fb7d5e291514417e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Data frame: b'4546a1fb5435d07e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'8520dd7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'554721fb4d2815713f7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Data frame: b'5647a1fb54a9ec7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'8610be7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'664421fb55d515b18f7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Data frame: b'6744a1fb54948f7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'774521fb4a2b1524a97e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Data frame: b'7045a1fb5481b57e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'8070787e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'004a21fb629e15b2107e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Data frame: b'014aa1fb63a4387e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Send command networkInit
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'7d314b21bf676c7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Data frame: b'124ba5bf5463787e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Application frame 23 (networkInit) received
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Data frame: b'224bb1b1c450837e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Application frame 25 (stackStatusHandler) received
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.ezsp] Send command getNetworkParameters
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'2348218038017e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Data frame: b'3348a180542b4661ea143f4d5b2cb956944127abedce677302c1730e7e'
2018-02-18 19:59:07 DEBUG (MainThread) [bellows.uart] Sending: b'8430fc7e'
[... cut ...]
2018-02-18 19:59:22 DEBUG (MainThread) [bellows.uart] Data frame: b'6754b1ed54f5d7b259964865ab55927063aa01cf12317a83fbc76789fc3c35e48ea3aa1deeb3aeafb0d7698c04259adf440b88a651ba7e'
2018-02-18 19:59:22 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
2018-02-18 19:59:22 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received
2018-02-18 19:59:22 DEBUG (MainThread) [bellows.uart] Data frame: b'7754b197540c716d9b944a27a81593499c3a21abed644b7e'
2018-02-18 19:59:22 DEBUG (MainThread) [bellows.uart] Sending: b'8070787e'
2018-02-18 19:59:22 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received

```

rcloran, 2018-02-19:
And still no entities in hass after checking your environment is clean? 

At this point, it looks like zigpy is doing the right thing. Can you open an issue in hass for this, and provide logs from around “Discovered endpoint information ... endpoint=2” and later?

dmulcahey, 2018-02-19:
Sure can. I’ll try again with a new venv off of HA dev in a bit and I’ll provide the logs in an issue in the HA repo. Thanks for the help.

closed 2018-02-19 by rcloran

dmulcahey, 2018-02-19:
@rcloran tried again with a completely new setup this morning. Everything works as expected now. Sorry for the confusion and thanks for the help!!