zigpy/bellows #110: Support for ZigBee "Green Power" devices such as the Hue Tap Switch

kiall, 2018-04-05:
Hey - So, I could be missing something, but as near as I can tell ZigBee "Green Power" devices, like the Philips Hue Tap Switch, are current unsupported? So far, I've tried a few things to pair a Hue Tap Switch with bellows, but have not had much luck.

Are green power devices supported? If so, is there anything special needed to pair?

mingqian, 2018-08-16:
Per my understanding, currently emberznet has very limited support for zgp functionalities, especially sink devices. For EZSP protocol, I don't think currently it supports.

Shulyaka, 2019-10-04:
Waiting for the EnOcean PTM 216Z support as well... Those battery-free devices are growing popularity now.

Adminiuga, 2019-10-05:
Where can you get one?

Adminiuga, 2019-10-05:
I think later emberznet versions had some support for ZGP, but most popular HUSBZB-1 stick is still on 4.0 version and unlikely to be updated. 
But I see that Conbee added ZGP endpoint so my guess they are gonna support it at some point.

Hedda, 2020-11-19:
FYI bellows and zigpy doesn't yet support ZGP (Zigbee Green Power) but @zoic21 started working on it now, see https://github.com/zigpy/zigpy/pull/519

Hedda, 2021-02-15:
FYI, zoic21 have made progress with ZGP for zigpy and is requesting regression testing of his new pull request in https://github.com/zigpy/zigpy/pull/656

> Can someone can test with this code if all it's ok ? No side effect. If it I think we can merge this first step (after I fix coveralls).

That is, he is asking people do functional and non-functional tests to ensure that his patch does not break any existing functions. If it does not break anything obvious in zigpy then it could possible be merged 'as-is' even if ZGP might be be fully production ready.

puddly, 2026-10-31:
This feature is now done and will be part of Home Assistant Core 2026.11.0. If you want to test it out now, please update to the beta.

closed 2026-10-31 by puddly