zigpy/bellows #121: error: UNKNOWN FRAME RECEIVED: b'\xff8\xbe~'

henryptung, 2018-06-01:
I'm trying to set up a new HUSBZB-1 stick with Home Assistant, but can't seem to get it to work. (Can verify that it is a HUSBZB-1 stick, not a HUSBZ-1 stick).

Looking into it some more, seems like the `bellows` library isn't able to interface with it, so I'm putting the I tried interfacing directly with it using the `bellows` CLI. I got this output:

```
$ bellows -v DEBUG -d /dev/ttyUSB1 info                                                                                 
debug: Using selector: EpollSelector
debug: Connected. Resetting.
debug: Sending: b'1ac038bc7e'
error: UNKNOWN FRAME RECEIVED: b'\xff8\xbe~'
```

at which point it hangs until I kill it with an interrupt. Subsequent executions simply hang after sending:

```
$ bellows -v DEBUG -d /dev/ttyUSB1 info                                                                                 
debug: Using selector: EpollSelector
debug: Connected. Resetting.
debug: Sending: b'1ac038bc7e'
```

until I reset the stick by unplugging and replugging it. Am I missing something basic about setup, or is this a known problem?

Am happy to provide more info/testing, but I don't know how to debug from this point.

AndreasBomholtz, 2018-06-28:
Can you please validate that you are using the correct USB port, because from what I can read, it the the dongle might support zwave and zigbee on two different ports.

henryptung, 2018-07-16:
@AndreasBomholtz Ah, sorry about the delayed response! Of note, was definitely using the correct port (checked attributes using `udevadm`).

But it actually turned out to be a defective stick - I raised the issue with the seller, got it returned, and the replacement just arrived. CLI working out-of-the-box now.

closed 2018-07-16 by henryptung