zigpy/bellows #123: ValueError: 255 is not a valid EmberNodeType

cavokz, 2018-06-24:
Hi there,

  thank you for writing this nice tool! I'm trying to use it with Home Assistant but it gets stuck for more than 10 seconds etc etc so I'm investigating via command line (BTW, is there a way to test it without installing, just from the local build dir? I could not find the way).

Apparently something unexpected happens in the communication over ttyUSB. I'm using the Elelabs ZigBee USB Adapter (https://elelabs.com/products/elelabs_usb_adapter.html)

This is the output I get, it's from a just pulled version from github (would be nice to have in the debug traces also the git hash of the version in use, for your reference).

```
$ bellows -d /dev/ttyUSB0 -v debug info                                                                                             debug: Using selector: EpollSelector
debug: Connected. Resetting.
debug: Sending: b'1ac038bc7e'
debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
debug: Send command version
debug: Sending: b'004221a850ed2c7e'
debug: Data frame: b'0142a1a8522815d2e7ae7e'
debug: Sending: b'8160597e'
debug: Application frame 0 (version) received
debug: Switching to eszp version 6
debug: Send command version
debug: Sending: b'7d31432157542a7d3315a67e'
debug: Data frame: b'1243a157542a13b059f4b6467e'
debug: Sending: b'82503a7e'
debug: Application frame 0 (version) received
debug: Configuring...
debug: Send command setConfigurationValue
debug: Sending: b'22402157547919b059c46e7e'
debug: Data frame: b'2340a157547915fbd27e'
debug: Sending: b'83401b7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'3341215754797d38b75995c77e'
debug: Data frame: b'3441a157547915904d7e'
debug: Sending: b'8430fc7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'44462157547938b35902ad7e'
debug: Data frame: b'4546a157547915640c7e'
debug: Sending: b'8520dd7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'554721575479144d593eca7e'
debug: Data frame: b'5647a157547915ce557e'
debug: Sending: b'8610be7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command networkInit
debug: Sending: b'66442157543dfba67e'
debug: Data frame: b'6744a157543d86938b7e'
debug: Sending: b'87009f7e'
debug: Application frame 23 (networkInit) received
debug: Send command getEui64
debug: Sending: b'77452157540c28a17e'
debug: Data frame: b'7045a157540c97518b9a4a4aa755b2e47e'
debug: Sending: b'8070787e'
debug: Application frame 38 (getEui64) received
[00:0d:6f:00:0e:d2:e3:82]
debug: Send command getNodeId
debug: Sending: b'004a2157540dd0a47e'
debug: Data frame: b'014aa157540deb4d269b7e'
debug: Sending: b'8160597e'
debug: Application frame 39 (getNodeId) received
[65534]
debug: Send command networkState
debug: Sending: b'7d314b21575432e26d7e'
debug: Data frame: b'124ba15754321508aa7e'
debug: Sending: b'82503a7e'
debug: Application frame 24 (networkState) received
[<EmberNetworkStatus.NO_NETWORK: 0>]
debug: Send command getNetworkParameters
debug: Sending: b'224821575402db807e'
debug: Data frame: b'2348a1575402864d59944a25aa55924963b1d8abed31988bfdc6638974c37e'
debug: Sending: b'83401b7e'
debug: Application frame 40 (getNetworkParameters) received
Exception in callback SerialTransport._read_ready()
handle: <Handle SerialTransport._read_ready()>
Traceback (most recent call last):
  File "/usr/lib/python3.6/asyncio/events.py", line 145, in _run
    self._callback(*self._args)
  File "/usr/local/lib/python3.6/dist-packages/pyserial_asyncio-0.4-py3.6.egg/serial_asyncio/__init__.py", line 106, in _read_ready
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.6/dist-packages/bellows-0.6.0-py3.6.egg/bellows/uart.py", line 64, in data_received
    self.frame_received(frame)
  File "/usr/local/lib/python3.6/dist-packages/bellows-0.6.0-py3.6.egg/bellows/uart.py", line 86, in frame_received
    self.data_frame_received(data)
  File "/usr/local/lib/python3.6/dist-packages/bellows-0.6.0-py3.6.egg/bellows/uart.py", line 107, in data_frame_received
    self._application.frame_received(self._randomize(data[1:-3]))
  File "/usr/local/lib/python3.6/dist-packages/bellows-0.6.0-py3.6.egg/bellows/ezsp.py", line 170, in frame_received
    result, data = t.deserialize(data, schema)
  File "/usr/local/lib/python3.6/dist-packages/bellows-0.6.0-py3.6.egg/bellows/types/__init__.py", line 9, in deserialize
    value, data = type_.deserialize(data)
  File "/usr/local/lib/python3.6/dist-packages/bellows-0.6.0-py3.6.egg/bellows/types/basic.py", line 10, in deserialize
    r = cls(int.from_bytes(data[:cls._size], 'little', signed=cls._signed))
  File "/usr/lib/python3.6/enum.py", line 291, in __call__
    return cls.__new__(cls, value)
  File "/usr/lib/python3.6/enum.py", line 533, in __new__
    return cls._missing_(value)
  File "/usr/lib/python3.6/enum.py", line 546, in _missing_
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 255 is not a valid EmberNodeType
^C
Aborted!
```

Is there anything I can do to help? I'm not in the details of ZHA.. :(

Thanks!

cavokz, 2018-07-28:
I tried 'python3 setup.py develop', it wanted to install stuff around. My life is to short for having this step in my development-test cycle, I need something way more handy. Is it really the way you guys are developing? Would you accept changes that allow execution in place?

AndreasBomholtz, 2018-08-11:
You just run the command once and then when you run bellows, it is using the local code. So after that step it is executing in place.

cavokz, 2018-08-11:
Great, thank you for the clarification. :)

cavokz, 2018-08-16:
Nice! I'll test it asap. Thanks.