zigpy/bellows #137: ZHA not able to pair with OSRAM Smart Plug

darootler, 2018-12-18:
Hi,

i am using Home Assistant version 0.84.3 on hass.io. Yesterday i installed Elelabs ZigBee Raspberry Pi Shield and everthing went fine. But unfortunately i am not able to join an Osram Smart Plug.

Here are the related logs:

```
2018-12-18 18:25:13 INFO (MainThread) [homeassistant.components.zha] Permitting joins for 60s
2018-12-18 18:25:13 DEBUG (MainThread) [bellows.ezsp] Send command permitJoining
2018-12-18 18:25:13 DEBUG (MainThread) [bellows.uart] Sending: b'6754215754082915ba7e'
2018-12-18 18:25:13 DEBUG (MainThread) [bellows.uart] Data frame: b'7754a157540815f7ce7e'
2018-12-18 18:25:13 DEBUG (MainThread) [bellows.uart] Sending: b'8070787e'
2018-12-18 18:25:13 DEBUG (MainThread) [bellows.ezsp] Application frame 34 (permitJoining) received
2018-12-18 18:25:39 DEBUG (MainThread) [bellows.uart] Data frame: b'0754b157540e61432561fb25006b22359d4e27ab08ef7e'
2018-12-18 18:25:39 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2018-12-18 18:25:39 DEBUG (MainThread) [bellows.ezsp] Application frame 36 (trustCenterJoinHandler) received
2018-12-18 18:25:39 INFO (MainThread) [zigpy.application] Device 0xf174 (XX:XX:XX:XX:XX:XX:XX:XX) joined the network
2018-12-18 18:25:39 INFO (MainThread) [zigpy.device] [0xf174] Discovering endpoints
2018-12-18 18:25:39 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast
2018-12-18 18:25:39 DEBUG (MainThread) [bellows.uart] Sending: b'71552157541e15c6a8944a20aa5592099d4e27aaeccd66ff0ce70e7e'
2018-12-18 18:25:39 DEBUG (MainThread) [bellows.uart] Data frame: b'1055a157541e158b72197e'
2018-12-18 18:25:39 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2018-12-18 18:25:39 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received
2018-12-18 18:25:44 DEBUG (MainThread) [bellows.uart] Data frame: b'2055b157541515c6a8944a20aa5592099d4e2792eca867bc787e'
2018-12-18 18:25:44 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
2018-12-18 18:25:44 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received
2018-12-18 18:25:46 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast
2018-12-18 18:25:46 DEBUG (MainThread) [bellows.uart] Sending: b'035a2157541e15c6a8944a20aa5592099d4e27a9efcd65ff0c7b687e'
2018-12-18 18:25:46 DEBUG (MainThread) [bellows.uart] Data frame: b'315aa157541e1588f4287e'
2018-12-18 18:25:46 DEBUG (MainThread) [bellows.uart] Sending: b'8430fc7e'
2018-12-18 18:25:46 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received
2018-12-18 18:25:51 DEBUG (MainThread) [bellows.uart] Data frame: b'415ab157541515c6a8944a20aa5592099c4e2791efa8674d207e'
2018-12-18 18:25:51 DEBUG (MainThread) [bellows.uart] Sending: b'8520dd7e'
2018-12-18 18:25:51 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received
2018-12-18 18:25:53 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast
2018-12-18 18:25:53 DEBUG (MainThread) [bellows.uart] Sending: b'155b2157541e15c6a8944a20aa5592099d4e27a8eecd64ff0c02f77e'
2018-12-18 18:25:53 DEBUG (MainThread) [bellows.uart] Data frame: b'525ba157541e158919a57e'
2018-12-18 18:25:53 DEBUG (MainThread) [bellows.uart] Sending: b'8610be7e'
2018-12-18 18:25:53 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received
2018-12-18 18:25:57 DEBUG (MainThread) [bellows.uart] Data frame: b'625bb157541515c6a8944a20aa5592099d4e2790eea86716ec7e'
2018-12-18 18:25:57 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
2018-12-18 18:25:57 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received
2018-12-18 18:25:57 ERROR (MainThread) [zigpy.device] Failed ZDO request during device initialization: Message send failure: EmberStatus.DELIVERY_FAILED
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/zigpy/device.py", line 51, in _initialize
    epr = await self.zdo.request(0x0005, self.nwk, tries=3, delay=2)
  File "/usr/local/lib/python3.6/site-packages/zigpy/util.py", line 52, in retry
    r = await func()
  File "/usr/local/lib/python3.6/site-packages/zigpy/device.py", line 89, in request
    expect_reply=expect_reply,
  File "/usr/local/lib/python3.6/site-packages/bellows/zigbee/application.py", line 241, in request
    v = await send_fut
zigpy.exceptions.DeliveryError: Message send failure: EmberStatus.DELIVERY_FAILED
```

Am i doing something wrong here? If someone needs more information please let me know.

Regards
Richard

walthowd, 2018-12-18:
I couldn't find anyone that has used the OSRAM Smart Plug successfully - The bulbs have been widely used without issue. If possible, I would try running the latest zigpy/bellows modules as they have some changes that allow devices that bind to a certain cluster to join successfully, but as you are on hass.io I don't know that you can easily upgrade to the latest code.

Adminiuga, 2018-12-18:
Are those the same as Sylvania Smart+ plugs? Model: 72922-A or 72922?
I have two of those and they work fine. Make sure the plug is very close to RPi Shield. 3ft/1m or so during the pairing. I had random sensitivity issues with RPi Shield.

darootler, 2018-12-18:
The model number is 4058075036239. The distance between the raspberry and the plug is under one meter while pairing.

Regards

Adminiuga, 2018-12-18:
do you have a link for you plug?
I have this one https://www.amazon.com/Sylvania-Smart-Home-72922-LIGHTIFY/dp/B01M6UM8QD

darootler, 2018-12-18:
Here is the link: https://www.amazon.de/Osram-schaltbare-fernbedienbar-Lichtsteuerung-kompatibel/dp/B074PZLX2P/ref=sr_1_3?ie=UTF8&qid=1545163438&sr=8-3&keywords=osram+smart+plug

darootler, 2018-12-18:
I just did a little research about versions being used within hass.io:

[image]

Seems pretty actual to me.

Regards

Adminiuga, 2018-12-18:
I don't have any experience with that plug. Try reseting it and joining again. Seems like it is not responding at all and `bellows` can't discover any endpoints on it. 
while joining, keep pressing the button on it every 1s to keep it awake
Check Smartthings forum and Device Handles if they do anything special for it

darootler, 2018-12-18:
I already tried resetting and keep pressing the button, without success.

Is there a way to define device handlers within Home Assistant?

Regards
Richard

darootler, 2018-12-27:
I am having the same errors while joining Xiaomi Aqara Temperature Humidity Sensor. Is there anyone out who is able to help?

Regards
Richard

Hepokatti, 2018-12-30:
I have same plugs (x3) and they are working nicely. Pairing was extremely easy, plugs were in different rooms and I triggered zha.permit from phone.... Just remember reset plug first before pairing.

My problem is Xiaomi sensors, which are dropping offline after short time (few hours) and they need to be re-paired to get them back online.

Setup:
Hass.io 0.84.6 on Pi 3+ with Elelabs ZigBee Raspberry Pi Shield

darootler, 2018-12-30:
> I have same plugs (x3) and they are working nicely. Pairing was extremely easy, plugs were in different rooms and I triggered zha.permit from phone.... Just remember reset plug first before pairing. [...]

Hi,

nice to have someone who is using the same setup.

With resetting the plug you mean holding down the button for about 10 seconds, right?

I don't know what i am doing wrong, i am not able to pair anything, but i followed the instructions from elelabs. Have you upgraded the firmware of the Pi Shield? Can you post your configuration or the steps you did for configuring the Pi Shield?

Regards
Richard

darootler, 2018-12-30:
I just followed your steps and i am not able to pair anything, always the same error:

`2018-12-30 22:44:33 ERROR (MainThread) [zigpy.device] Failed ZDO request during device initialization: Message send failure: EmberStatus.DELIVERY_FAILED
Traceback (most recent call last):
  File "/config/deps/lib/python3.6/site-packages/zigpy/device.py", line 51, in _initialize
    epr = await self.zdo.request(0x0005, self.nwk, tries=3, delay=2)
  File "/config/deps/lib/python3.6/site-packages/zigpy/util.py", line 52, in retry
    r = await func()
  File "/config/deps/lib/python3.6/site-packages/zigpy/device.py", line 89, in request
    expect_reply=expect_reply,
  File "/config/deps/lib/python3.6/site-packages/bellows/zigbee/application.py", line 241, in request
    v = await send_fut
zigpy.exceptions.DeliveryError: Message send failure: EmberStatus.DELIVERY_FAILED`

Regards
Richard

darootler, 2019-01-03:
Seems that my Elelabs ZigBee Raspberry Pi Shield is defect.

Sorry and thank you guys.

Regards
Richard

closed 2019-01-03 by darootler