zigpy/bellows #141: Is there a process to get new devices added

w1ll1am23, 2019-01-14:
I have a device (Lutron Connected Bulb remote) that pairs to ZHA and a zha entity shows up in Home Assistant. Once paired, the pressing any button on the device results in nothing happening (as far as I can tell) It would be nice to have this device show up as a sensor in Home Assistant for a zha entity with 4 attributes (one for each button) 

So the question is, is there anything that I can do to help get this added/supported in ZHA/Home Assistant? Thanks.

dmulcahey, 2019-01-14:
If you try dev with homeassistant.core set to debug you should see events. Can you try that and report back?

w1ll1am23, 2019-01-15:
Are you saying the logging isn't in production yet so I would need to run Home Assistant dev? If so I can try to do that tomorrow. I am running 0.85.1 now with debug logging and the pressing the buttons on the remote results in no log entry at all.

dmulcahey, 2019-01-15:
Would need to be dev. We switched the way the events got hooked up. What’s the signature for the device? I can tell you if it will work if you post it.

w1ll1am23, 2019-01-15:
Can you tell me how to get the signature of the device?

Adminiuga, 2019-02-07:
@w1ll1am23 if the device is joined and is in the `zigbee.db` then post the output of `bellows devices` or `bellows -d <path_to_zigbee_database>/zigbee.db devices`

Or better, enable debug logging in home assistant, factory reset the device and rejoin it. In `home-assistant.log` you should see something like 
```
2019-02-07 09:34:56 INFO (MainThread) [zigpy.application] Device 0xc203 (00:15:8d:00:02:49:d0:85) joined the network
2019-02-07 09:34:56 DEBUG (MainThread) [zigpy.application] Skip initialization for existing device 00:15:8d:00:02:49:d0:85
2019-02-07 09:34:56 INFO (MainThread) [zigpy.device] [0xc203] Discovering endpoints
```
After `discovering endpoints` line, you are gonna get a bunch of other messages, but look for:
```
2019-02-07 09:35:08 INFO (MainThread) [zigpy.device] [0xc203] Discovered endpoints: [1, 2, 3]
2019-02-07 09:35:08 INFO (MainThread) [zigpy.endpoint] [0xc203:1] Discovering endpoint information
...
2019-02-07 09:35:11 INFO (MainThread) [zigpy.endpoint] [0xc203:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=24321 device_version=1 input_clusters=[0, 3, 25, 65535, 18] output_clusters=[0, 4, 3, 5, 25, 65535, 18]>
...
2019-02-07 09:35:41 INFO (MainThread) [zigpy.endpoint] [0xc203:3] Discovered endpoint information: <SimpleDescriptor endpoint=3 profile=260 device_type=24323 device_version=1 input_clusters=[3, 12] output_clusters=[4, 3, 5, 12]
```

You are going to need the `Discovered endpoints` line, indicating the number of enpoints and then the `Discovered Endpoint Information`. In the example above there are **3** endpoints total, but information was discovered only for two (endpoint 1 & 3), but such are Xiaomi devices, with those you may have to mske a few tries. Being really close to Zigbee stick really helps.

w1ll1am23, 2019-02-07:
@Adminiuga Thanks! This is what my log looked light after pairing the remote with debug logs enabled.

https://pastebin.com/raw/yRNvxJ54

Adminiuga, 2019-02-08:
```
For the records
```
2019-02-07 18:37:44 INFO (MainThread) [zigpy.device] [0x6e04] Discovered endpoints: [1]
2019-02-07 18:37:45 INFO (MainThread) [zigpy.endpoint] [0x6e04:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=49246 device_type=2080 device_version=2 input_clusters=[0, 4096, 65280, 64580] output_clusters=[4096, 3, 6, 8, 4, 5, 0, 65280]>
2019-02-07 18:37:45 DEBUG (MainThread) [zigpy.endpoint] [0x6e04:1] Manufacturer: Lutron
2019-02-07 18:37:45 DEBUG (MainThread) [zigpy.endpoint] [0x6e04:1] Model: LZL4BWHL01 Remote
```

Adminiuga, 2019-02-08:
But this issue belongs either to [Zigpy](https://github.com/zigpy/zigpy) or better to [ZHA Device Handlers](https://github.com/dmulcahey/zha-device-handlers).
`Bellows` is just the low level radio lib. Do you mind moving it to ZHA Device handlers?
Also, this is second device with `profile_id != 260` May require custom handling like https://github.com/zigpy/zigpy/issues/121

w1ll1am23, 2019-02-08:
Moved issue to zha-device-handlers. https://github.com/dmulcahey/zha-device-handlers/issues/32

closed 2019-02-08 by w1ll1am23