zigpy/bellows #144: unexpected response warning messages from bellows

jr3us, 2019-02-09:
I am currently using HA v86.4 on hassbian stretch and using an HUSBZB-1 for Z-Wave and Zigbee and

NOTE that I have 2 Zigbee Bulbs, 2 Zigbee outlets, and 1 z-wave outlet in use currently with no issues  controlling them.

I am getting warning messages from bellows.zigbee.application occurring occasionally as follows:

Unexpected response TSN=216 command=1 args=[[<ReadAttributeRecord attrid=1291 status=0 value=0>]]
10:24 AM util/async_.py (WARNING)
Unexpected message send notification
10:18 AM util/async_.py (WARNING)
Unexpected response TSN=193 command=1 args=[[<ReadAttributeRecord attrid=1291 status=0 value=0>]]
10:18 AM util/async_.py (WARNING)
Unexpected message send notification
10:18 AM util/async_.py (WARNING)
Unexpected response TSN=193 command=1 args=[[<ReadAttributeRecord attrid=1291 status=0 value=0>]]
10:18 AM util/async_.py (WARNING)
Unexpected message send notification
10:18 AM util/async_.py (WARNING)
Unexpected response TSN=193 command=1 args=[[<ReadAttributeRecord attrid=1291 status=0 value=0>]]
10:18 AM util/async_.py (WARNING)
Unexpected response TSN=75 command=1 args=[[<ReadAttributeRecord attrid=1291 status=0 value=0>]]
9:49 AM util/async_.py (WARNING)
Unexpected message send notification
9:32 AM util/async_.py (WARNING)
Unexpected response TSN=7 command=1 args=[[<ReadAttributeRecord attrid=1291 status=0 value=0>]]
9:32 AM util/async_.py (WARNING)
Unexpected message send notification
9:32 AM util/async_.py (WARNING)
Unexpected response TSN=7 command=1 args=[[<ReadAttributeRecord attrid=1291 status=0 value=0>]]
9:32 AM util/async_.py (WARNING)
Unexpected message send notification
9:16 AM util/async_.py (WARNING)
Unexpected response TSN=197 command=1 args=[[<ReadAttributeRecord attrid=1291 status=0 value=0>]]
9:16 AM util/async_.py (WARNING)
Unexpected message send notification
9:16 AM util/async_.py (WARNING)
Unexpected response TSN=197 command=1 args=[[<ReadAttributeRecord attrid=1291 status=0 value=0>]]
9:16 AM util/async_.py (WARNING)
Unexpected response TSN=147 command=1 args=[[<ReadAttributeRecord attrid=1291 status=0 value=0>]]

Let me know if I can be of assistance finding the solution.

Regards.

Yoda-x, 2019-02-09:
These come in my experience from retransmits of the ncp to the host.. The host already processed the frame and removed the  entry from the table. The host not responds quick enough to send an ack to the ncp, so the ncp resend it.

If you use a raspi with sdcard reducing the log verbosity may help to reduce Ioload having less writes to the SD. 

But information about your system would be helpful.

jr3us, 2019-02-10:
I am using a raspberry pi 3b and wired network connection. I have not enabled any logging specifically.
I am running hassbian stretch updated in the past week. I have a hue hub with 6 bulbs associated.
I also have a handful of internet sensors like weather.
I am running on a sandisk 128gb class 10 sd card using MySQL.

Adminiuga, 2019-03-04:
You can ignore those. Usually you would get those when a reply fails to reach NCP within 10s of original request.

dmulcahey, 2019-03-04:
@Adminiuga we should change these to debug

Adminiuga, 2019-05-23:
Fixed in #151

closed 2019-05-23 by Adminiuga