zigpy/bellows #158: Zigbee Sensor Error

Richdem, 2019-04-28:
I have just finishing moving all my Zigbee light bulbs from Wink Hub 2 over to HA using a GoControl HUSBZB dongle. Everything paired without any issues but I noticed for each light 2 other components were generated (see below)
[image] 

I am getting lots of errors in the log now to do with the sensor portion of the lights. These only seem to be coming from 2 types of light, an RGB & Tuneable (soft to daylight) My standard soft lights do not have these sensors.

[image] 

Can these sensors be disabled at all?  I am not interested in using them and would rather they not show up in the unused entity section and the log.

dmulcahey, 2019-04-28:
We don’t poll sensors... are you using custom components?

dmulcahey, 2019-04-28:
Also, this should be in the HomeAssistant repo... not bellows.

Richdem, 2019-04-28:
> Also, this should be in the HomeAssistant repo... not bellows.

I am sorry,

How do I move this? 

I just paired all of my Ecosmart Zigbee bulbs with HA. The only custom component I have is a light slider. Is this what you mean

Richdem, 2019-04-28:
How do I move this

Richdem, 2019-04-28:
I do have a custom light slider loaded, is that what you mean?

Sorry for posting in the wrong section

closed 2019-05-23 by Adminiuga