zigpy/bellows #162: Command response race condition

presslab-us, 2019-05-23:
It seems like there is a race conditioning happening when handing a response to a command.

The reply wait here:
https://github.com/zigpy/bellows/blob/0b211f949e09d43be8bb1070d702ef1140a96b37/bellows/zigbee/application.py#L357-L358

will be satisfied by either `incomingMessageHandler` or `messageSentHandler`, whichever one gets there first.  If `messageSentHandler` gets there first, then the `incomingMessageHandler` will be ignored with something like
` [bellows.zigbee.application] [0xd8eb:1:0x0101]: Unexpected response TSN=8 command=256 args=[<Status.SUCCESS: 0>]` in the log and the response will be lost.

The calling method will not see the actual response from `incomingMessageHandler` it will only see that `messageSentHandler` was successful.

This problem explains this workaround here:

https://github.com/home-assistant/home-assistant/blob/7f7435f00302add4053b8894059262c9bdd405ca/homeassistant/components/zha/core/channels/__init__.py#L61-L63

That workaround also incorrectly assumes all responses have the `Status` in `result[1]` but that's just the case for `Default response`.

Adminiuga, 2019-05-23:
There are two awaits:
- send await -- indicating whether the far end device ACKed our message or if there was no ACK and EZSP called `messageSentHandler` with a failure status
- reply await -- when we actually await for a reply. If we never get, then the reply future gets cancelled and in this case there won't be `Status` in `result[1]` which is the `Default response`

presslab-us, 2019-05-23:
Ah yes, I see that.  So any idea why a Bool is returned sometimes?  Where can that even come from?

Adminiuga, 2019-05-23:
it is possible to receive `[bellows.zigbee.application] [0xd8eb:1:0x0101]: Unexpected response TSN=8 command=256 args=[<Status.SUCCESS: 0>]` if you received `messageSent` frame with a failure response, but then still get a reply. 
Do you have a trace?

presslab-us, 2019-05-23:
2019-05-23 09:00:59 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 00:0d:6f:00:02:fb:87:8c/0xd8eb
    2019-05-23 09:00:59 DEBUG (MainThread) [bellows.ezsp] Send command setExtendedTimeout
    2019-05-23 09:00:59 DEBUG (MainThread) [bellows.uart] Sending: b'7465215754549935a2964a4aa7559370227e'
    2019-05-23 09:00:59 DEBUG (MainThread) [bellows.uart] Data frame: b'4065a1575454c11d7e'
    2019-05-23 09:00:59 DEBUG (MainThread) [bellows.uart] Sending: b'8520dd7e'
    2019-05-23 09:00:59 DEBUG (MainThread) [bellows.ezsp] Application frame 126 (setExtendedTimeout) received
    2019-05-23 09:00:59 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast
    2019-05-23 09:00:59 DEBUG (MainThread) [bellows.uart] Sending: b'056a2157541e155981904b24ab5493099d4e27a0e6cd6680fd9aed7e'
    2019-05-23 09:00:59 DEBUG (MainThread) [bellows.uart] Data frame: b'516aa157541e15a7f5907e'
    2019-05-23 09:00:59 DEBUG (MainThread) [bellows.uart] Sending: b'8610be7e'
    2019-05-23 09:00:59 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received
    2019-05-23 09:01:00 DEBUG (MainThread) [bellows.ezsp] Send command nop
    2019-05-23 09:01:00 DEBUG (MainThread) [bellows.uart] Sending: b'166b2157542fe1047e'
    2019-05-23 09:01:00 DEBUG (MainThread) [bellows.uart] Data frame: b'626ba157542f7f017e'
    2019-05-23 09:01:00 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
    2019-05-23 09:01:00 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received
    2019-05-23 09:01:01 DEBUG (MainThread) [bellows.uart] Data frame: b'726bb1575415155981904b24ab5493099d4e27bfe7a86723b17e'
    2019-05-23 09:01:01 DEBUG (MainThread) [bellows.uart] Sending: b'8070787e'
    2019-05-23 09:01:01 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received
    2019-05-23 09:01:01 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.140096877516280] 'bool' object is not subscriptable

That last error is because my code is expecting a `foundation.Status` and getting a `bool`.  It's only two seconds from when the packet is sent to when the `bool` is received so it shouldn't be because of timeout.

Adminiuga, 2019-05-23:
My guess that depends on  cluster command, whether it uses a `Default response` or if it has its own response with a bool param? https://github.com/home-assistant/home-assistant/blob/7f7435f00302add4053b8894059262c9bdd405ca/homeassistant/components/zha/core/channels/__init__.py#L61-L63 is a generic wrapper and IMO maybe it is better to log and return the actual response? As not all cluster commands use the `default response`

Adminiuga, 2019-05-23:
do you know what ZCL command originated this sendUnicast?

Adminiuga, 2019-05-23:
i do need a more friendly logging in bellows...

presslab-us, 2019-05-23:
Yes, I agree with that wholeheartedly, I have changed that code you mentioned to return the whole list instead of parsing it.

The issue I am seeing is very occasional.  Nearly all the time I get the response I am expecting, in `result[0]` which is a `foundation.Status` and it works fine.

I am sending the 'lock_door' command like so: `    success = (await self._doorlock_channel.lock_door())[0]`

`success` should contain `foundation.Status`.  Sometimes it contains `bool`.  You can see the log above it does not even wait for the `incomingMessageHandler`.

presslab-us, 2019-05-23:
Here is the schema I am using for the response:

    0x0000: ('lock_door_response', (foundation.Status, ), True),
    0x0001: ('unlock_door_response', (foundation.Status, ), True),

Adminiuga, 2019-05-23:
The messageSent handler decodes to `<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 55531, <EmberApsFrame profileId=260 clusterId=257 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=20>, 10, <EmberStatus.DELIVERY_FAILED: 102>`

Uhm, that should rise an exception in `await req.send`. I don't understand why you get the bool though

presslab-us, 2019-05-23:
Ah, that makes sense.  I have seen `EmberStatus.DELIVERY_FAILED` in the past, but the command was still sent as the lock would actuate as expected.  It seems weird that it would get `EmberStatus.DELIVERY_FAILED` after only two seconds.

presslab-us, 2019-05-23:
I would it expect it to retry sending if it received `EmberStatus.DELIVERY_FAILED`, is this something it can do?

presslab-us, 2019-05-23:
2019-05-23 10:06:15 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 00:0d:6f:00:02:fb:87:8c/0xd8eb
    2019-05-23 10:06:15 DEBUG (MainThread) [bellows.ezsp] Send command setExtendedTimeout
    2019-05-23 10:06:15 DEBUG (MainThread) [bellows.uart] Sending: b'4106215754549935a2964a4aa75593ad217e'
    2019-05-23 10:06:15 DEBUG (MainThread) [bellows.uart] Data frame: b'1506a157545405867e'
    2019-05-23 10:06:15 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
    2019-05-23 10:06:15 DEBUG (MainThread) [bellows.ezsp] Application frame 126 (setExtendedTimeout) received
    2019-05-23 10:06:15 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast
    2019-05-23 10:06:15 DEBUG (MainThread) [bellows.uart] Sending: b'52072157541e155981904b24ab5493099d4e27b3f5ca6693fcc68a5a7e'
    2019-05-23 10:06:15 DEBUG (MainThread) [bellows.uart] Data frame: b'2607a157541e159b2c4b7e'
    2019-05-23 10:06:15 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
    2019-05-23 10:06:15 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received
    2019-05-23 10:06:15 DEBUG (MainThread) [bellows.zigbee.application] res: [<EmberStatus.SUCCESS: 0>, 41]
    2019-05-23 10:06:19 DEBUG (MainThread) [bellows.ezsp] Send command nop
    2019-05-23 10:06:19 DEBUG (MainThread) [bellows.uart] Sending: b'63042157542f9bbc7e'
    2019-05-23 10:06:19 DEBUG (MainThread) [bellows.uart] Data frame: b'3704a157542f30b17e'
    2019-05-23 10:06:19 DEBUG (MainThread) [bellows.uart] Sending: b'8430fc7e'
    2019-05-23 10:06:19 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received
    2019-05-23 10:06:20 DEBUG (MainThread) [bellows.uart] Data frame: b'4704b1575415155981904b24ab5493099d4e2782f5a8675a3c7e'
    2019-05-23 10:06:20 DEBUG (MainThread) [bellows.uart] Sending: b'8520dd7e'
    2019-05-23 10:06:20 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received
    2019-05-23 10:06:20 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.140628868705752] 'bool' object is not subscriptable

I added a debug print of the result from `res = await self._ezsp.sendUnicast`.  It looks like it sent fine.

I see in both these cases there is a rogue `nop` in there.  Maybe the `nop` is messing stuff up?  Why is the `nop` being sent anyway?

Adminiuga, 2019-05-23:
nah, the `nop`s are from watchdog and do not interfere with communication. 
In the trace you send earlier, the `sendUnicast` command is using seq # 11  but the `messageSentHangler` has sequence of 20, so don't assume the "messageSent" handler was for the preceding `sendUnicast` message.

Adminiuga, 2019-05-23:
if you don't mind more noisy logs, then try https://github.com/Adminiuga/bellows/commit/57e4d1171ce11308a6be36002083e5992cce6b98 which should make easier correlate requests and responses correctly

Adminiuga, 2019-05-23:
and for `zigpy`  you can get some addiotional debugging for outgoing requests with https://github.com/Adminiuga/zigpy/commits/enhancement/zcl-debug-log

presslab-us, 2019-05-23:
I put the debugging code in and of course it's not doing it now.  I will update when I can capture the failure.

presslab-us, 2019-05-23:
Ok, I have made the antenna reception worse and I can reproduce the failure.

    2019-05-23 12:32:50 INFO (MainThread) [zigpy.zcl] [0xd8eb:1:0x0101] tsn #008 'lock_door' command: ()
    2019-05-23 12:32:50 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 00:0d:6f:00:02:fb:87:8c/0xd8eb
    2019-05-23 12:32:50 DEBUG (MainThread) [bellows.ezsp] Send command setExtendedTimeout: (00:0d:6f:00:02:fb:87:8c, True)
    2019-05-23 12:32:50 DEBUG (MainThread) [bellows.uart] Sending: b'2260215754549935a2964a4aa75593440e7e'
    2019-05-23 12:32:50 DEBUG (MainThread) [bellows.uart] Data frame: b'2360a157545473b27e'
    2019-05-23 12:32:50 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
    2019-05-23 12:32:50 DEBUG (MainThread) [bellows.ezsp] Application frame 126 (setExtendedTimeout) received: b''
    2019-05-23 12:32:50 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 55531, <EmberApsFrame profileId=260 clusterId=257 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=8>, 8, b'\x01\x08\x00')
    2019-05-23 12:32:50 DEBUG (MainThread) [bellows.uart] Sending: b'33612157541e155981904b24ab5493099d4e27a3e5cd6683fdbf657e'
    2019-05-23 12:32:50 DEBUG (MainThread) [bellows.uart] Data frame: b'3461a157541e155c25cd7e'
    2019-05-23 12:32:50 DEBUG (MainThread) [bellows.uart] Sending: b'8430fc7e'
    2019-05-23 12:32:50 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received: b'00ee'
    2019-05-23 12:32:55 DEBUG (MainThread) [bellows.uart] Data frame: b'4461b1575415155981904b24ab5493099c4e2745e5a86799c47e'
    2019-05-23 12:32:55 DEBUG (MainThread) [bellows.uart] Sending: b'8520dd7e'
    2019-05-23 12:32:55 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received: b'00ebd804010101010140000000ee086600'
    2019-05-23 12:32:55 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 55531, <EmberApsFrame profileId=260 clusterId=257 sourceEndpoint=1 destinationEndpoint=1 options=64 groupId=0 sequence=238>, 8, <EmberStatus.DELIVERY_FAILED: 102>, b'']
    2019-05-23 12:32:55 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.139832190942288] 'bool' object is not subscriptable

Adminiuga, 2019-05-23:
Well, I think this `messageSentHandler` is for the `sendUnicast` in the trace. EZSP thinks it was not able to send the message, hence the `Delivery_FAilure` status.  I see it does use an extended timeout, so it think it is an end device, which I assume it true, but I can't explain why it comes only 5 seconds in.

What does the sniffer say? do you see the ACK for data sent from EZSP to the lock?

presslab-us, 2019-05-23:
Yes, it is an end device, battery powered.  My sniffer is not so useful because its antenna is not as good as the antenna in my coordinator, so it's only seeing some things with this reduced signal.

I did some more investigation and it looks like the exception is being raised here:

https://github.com/home-assistant/home-assistant/blob/1de0a0bbb95e177f7f109099bb63d785b070f922/homeassistant/components/zha/core/channels/__init__.py#L64-L72

My logging was not set properly to see this particular debug message.  I guess it should probably be an error or at least warn message.  This explains the `bool` return also.

I think that can be easily fixed, but I guess the two outstanding issues I see with bellows is the timeout duration, and also having some kind of packet re-transmission on failure.  What should the extended timeout be?  I know the routers will hold the packet for 7 seconds on sleeping devices, so it should be at least that.  I'd guess 10 seconds?

Adminiuga, 2019-05-23:
How are you invoking `await door_lock_cluster.lock()` ? Config panel or ZHA entity?

presslab-us, 2019-05-23:
I am using this:

https://github.com/presslab-us/home-assistant/tree/zha-implement-closure

I found this about the Ember timeout duration: https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2012/06/29/how_does_the_emberzn-po1M.  So it should retry 3 times with a each try timing out after 9.28 seconds for a total of 27.84 seconds.

Adminiuga, 2019-05-23:
> with a each try timing out after 9.28 seconds for a total of 27.84 seconds.
Correct. for end devices bellows is waiting for 28s to receive an answer.

presslab-us, 2019-05-23:
As an aside, is it necessary to `Send command setExtendedTimeout` for every transmission?  Seems like this could be done once on startup for each end device.  I'm not sure but I would guess since it is using the EUI64 address it is persistent.

presslab-us, 2019-05-27:
I did some more tests.  It seems that sometimes it does seem to have the usual 3 tries and 9 second timeout, and sometimes it does not.  Maybe if it is gone for too long it does not retry as long/often.  Anyway this functionality seems to be part of the EmberZNet firmware and not bellows so I am going to close this.  Thanks again for all you help, the additional debug logging was very useful.

closed 2019-05-27 by presslab-us