zigpy/bellows #166: PAIRING WSDCGQ01LM Temperature/Humidity/Pressure sensor error

ChumKiu, 2019-05-29:
Hi, I'm try to use bellow to make my didactical home automation system.
By using bellow I'm able to drive directly IKEA and OSRAM light bulb with no problem.
Issue are when I try to use the temp/Humidity/Pressor sensor on the subject.

Here below the sequence after launch the permit command:
```
(bellows) root@raspberrypi:~/zigbee# bellows permit
Joins are permitted for the next 30s...
Device 0x985e (00:15:8d:00:02:b5:40:20) left the network
Device 0xf4af (00:15:8d:00:02:b5:40:20) joined the network
[0xf4af] Discovering endpoints
Tries remaining: 3
Exception running handler
Traceback (most recent call last):
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/bellows/ezsp.py", line 194, in handle_callback
    handler(*args)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/bellows/zigbee/application.py", line 139, in ezsp_callback_handler
    self._handle_frame(*args)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/bellows/zigbee/application.py", line 160, in _handle_frame
    tsn, command_id, is_reply, args = self.deserialize(device, aps_frame.sourceEndpoint, aps_frame.clusterId, message)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/zigpy/application.py", line 73, in deserialize
    return sender.deserialize(endpoint_id, cluster_id, data)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/zigpy/device.py", line 97, in deserialize
    return self.endpoints[endpoint_id].deserialize(cluster_id, data)
KeyError: 1
Exception running handler
Traceback (most recent call last):
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/bellows/ezsp.py", line 194, in handle_callback
    handler(*args)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/bellows/zigbee/application.py", line 139, in ezsp_callback_handler
    self._handle_frame(*args)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/bellows/zigbee/application.py", line 160, in _handle_frame
    tsn, command_id, is_reply, args = self.deserialize(device, aps_frame.sourceEndpoint, aps_frame.clusterId, message)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/zigpy/application.py", line 73, in deserialize
    return sender.deserialize(endpoint_id, cluster_id, data)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/zigpy/device.py", line 97, in deserialize
    return self.endpoints[endpoint_id].deserialize(cluster_id, data)
KeyError: 1
[0xf4af] Discovered endpoints: [1, 2, 3]
[0xf4af:1] Discovering endpoint information
Tries remaining: 3
warning: Unexpected response TSN=3 command=266 args=b'\x00\x00!\x05#'
warning: [0xf4af:1] Message on unknown cluster 0x0405
[0xf4af:1] Discovered endpoint information: 41
warning: Unexpected response TSN=2 command=32772 args=[<Status.SUCCESS: 0>, 62639, <SimpleDescriptor endpoint=1 profile=260 device_type=24321 device_version=1 input_clusters=[0, 3, 25, 65535, 18] output_clusters=[0, 4, 3, 5, 25, 65535, 18]>]
warning: Unexpected response TSN=4 command=266 args=b'\x00\x00)\x80\x0b'
warning: [0xf4af:1] Message on unknown cluster 0x0402
warning: Unexpected response TSN=5 command=266 args=b'\x00\x00!\x17#'
warning: [0xf4af:1] Message on unknown cluster 0x0405
warning: Unexpected response TSN=6 command=266 args=b'\x00\x00)\x83\x0b'
warning: [0xf4af:1] Message on unknown cluster 0x0402
warning: Unexpected response TSN=7 command=266 args=b'\x00\x00!,#'
warning: [0xf4af:1] Message on unknown cluster 0x0405
Done
(bellows) root@raspberrypi:~/zigbee#
```
after that the sensor is not added to the devices db and then I'm unable to drive it.
It's seems to me that the application is unable to correctly recognize the answer from the sensor, is possible to update the code in order to get it working also with this type of sensors?
Could you please let me know how I can fix it?
Thanks!!

Adminiuga, 2019-05-29:
Xiaomi devices are tricky to pair, as they stop responding to request pretty quickly. You need to try and pair it again and while pairing, keep the device alive by pressing the button on the device every 1.5s

ChumKiu, 2019-05-30:
Thank @Adminiuga I've found this link:

https://www.reddit.com/r/homeautomation/comments/7awxz2/xiaomi_aqara_temperature_humidity_and_pressure/

and now the sensor was added to my list:

```
	(bellows) root@raspberrypi:~/zigbee# bellows devices
	Device:
	  NWK: 0x5ced
	  IEEE: 00:15:8d:00:02:b5:40:20
	  Endpoints:
		1: profile=0x104, device_type=24321
		  Input Clusters:
			Basic (0)
			Identify (3)
			Temperature Measurement (1026)
			Relative Humidity Measurement (1029)
```

from the log, during the pairing process, I've got the following messagges:

```
(bellows) root@raspberrypi:~/zigbee# bellows permit
Joins are permitted for the next 30s...
Device 0x5ced (00:15:8d:00:02:b5:40:20) joined the network
[0x5ced] Discovering endpoints
Tries remaining: 3
Exception running handler
Traceback (most recent call last):
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/bellows/ezsp.py", line 194, in handle_callback
    handler(*args)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/bellows/zigbee/application.py", line 139, in ezsp_callback_handler
    self._handle_frame(*args)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/bellows/zigbee/application.py", line 160, in _handle_frame
    tsn, command_id, is_reply, args = self.deserialize(device, aps_frame.sourceEndpoint, aps_frame.clusterId, message)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/zigpy/application.py", line 73, in deserialize
    return sender.deserialize(endpoint_id, cluster_id, data)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/zigpy/device.py", line 97, in deserialize
    return self.endpoints[endpoint_id].deserialize(cluster_id, data)
KeyError: 1
Exception running handler
Traceback (most recent call last):
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/bellows/ezsp.py", line 194, in handle_callback
    handler(*args)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/bellows/zigbee/application.py", line 139, in ezsp_callback_handler
    self._handle_frame(*args)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/bellows/zigbee/application.py", line 160, in _handle_frame
    tsn, command_id, is_reply, args = self.deserialize(device, aps_frame.sourceEndpoint, aps_frame.clusterId, message)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/zigpy/application.py", line 73, in deserialize
    return sender.deserialize(endpoint_id, cluster_id, data)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/zigpy/device.py", line 97, in deserialize
    return self.endpoints[endpoint_id].deserialize(cluster_id, data)
KeyError: 1
Exception running handler
Traceback (most recent call last):
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/bellows/ezsp.py", line 194, in handle_callback
    handler(*args)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/bellows/zigbee/application.py", line 139, in ezsp_callback_handler
    self._handle_frame(*args)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/bellows/zigbee/application.py", line 160, in _handle_frame
    tsn, command_id, is_reply, args = self.deserialize(device, aps_frame.sourceEndpoint, aps_frame.clusterId, message)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/zigpy/application.py", line 73, in deserialize
    return sender.deserialize(endpoint_id, cluster_id, data)
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/zigpy/device.py", line 97, in deserialize
    return self.endpoints[endpoint_id].deserialize(cluster_id, data)
KeyError: 1
Exception running handler
Traceback (most recent call last):
  File "/root/.virtualenvs/bellows/lib/python3.5/site-packages/bellows/ezsp.py", line 194, in handle_callback
[... 15 lines cut ...]
[0x5ced:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=24321 device_version=1 input_clusters=[0, 3, 25, 65535, 18] output_clusters=[0, 4, 3, 5, 25, 65535, 18]>
[0x5ced:2] Discovering endpoint information
Tries remaining: 3
[0x5ced:2] Discovered endpoint information: <SimpleDescriptor endpoint=2 profile=260 device_type=24322 device_version=1 input_clusters=[3, 18] output_clusters=[4, 3, 5, 18]>
[0x5ced:3] Discovering endpoint information
Tries remaining: 3
[0x5ced:3] Discovered endpoint information: <SimpleDescriptor endpoint=3 profile=260 device_type=24323 device_version=1 input_clusters=[3, 12] output_clusters=[4, 3, 5, 12]>
Done
```

From the ZCL Specification (ZigBee Cluster Library Specification - revision 6 Draft Version 1.0 - 14 Jan 2016) the profile 0x0104 correspond to a Standard ZigBee application profile range interval.
Concerning the cluster 1026 this correspond to 0x402 in hex notation, from chapter 4.4.1.3 this is the cluster identifier for the temperature measurement.
Next cluster is the 1029 that translated in hex notation give 0x405, from chapter 4.7.1.3 this one is related to the Relative Humidity Measurement.
From the device description (through the `bellows devices` command) there are no other clusters, but due to the sensor claim also to measure the pressure, the profile provide a cluster identification equal to 0x0403 (1027 in decimal format), but I can't see this cluster so may be somethings were not correctly addressed during the pairing procedure, I'll try next time to see if this is a problem into the pairing or effectively the sensor don't have this ability, so this point is left for future investigatio [... cut ...]
Now the task is use bellows to read the temperature value and the humidity but through the:

`(bellows) root@raspberrypi:~/zigbee# bellows zcl 00:15:8d:00:02:b5:40:20 1 1029 commands
`

I'm unable to see the supported command and from the specification I can see anything of useful at this moment or I'm not able now to found it because I'm study it now.

Could you please point me into the right direction and also give me some suggest if I've to change the code to implement support for such devices?

Thanks for your time!
ChumKiu

ChumKiu, 2019-05-30:
Hi @Adminiuga,
in trying to better understand how the code work I've found into the zigpy source code, at the module\quirks\xiaomi\__init__.py file the `class TemperatureHumiditySensor(CustomDevice):` some code that should be related to the analyzed sensor (I've found it through a scan over the source file with the word lumi.sensor_ht that I've found into the log output during the pairing process).
At the end of the code I see the replacement statements that in my thinking should model the sensor and effectively I can see all the info that is reported by the `bellow devices` command and effectively the pressure (0x403) cluster isn't listed.
Then at this point I kindly ask you some info/hint about:
- how to write a custom class for sensor?
- why the 0x403 cluster was omitted from this class?

and, with the current 0x402 (temperature) and 0x405 (humidity) cluster how I can retrieve the temperature and humidity value from the sensore by means of the bellow CLI due to missing command list from the commands directive?

I ask to be patient in this regard as I am now beginning to study how the code is organized, I know Pyton and other programming languages, but this is a field of study that I have recently begun to address.

I know that I've to do other testing and also read more documents to keep a better knowledge on this subject I kindly ask you to give me some help in this sense. I don't like to use ready to made solution without have a deep understanding on it so this is the way I've choose, study the ZigBee and then deep understanding of the bellow and the zigpy framework in order to be able to build my little Home Automation System from scratch. I think also that this approach may be useful also for others that desire gain more knowledge on this amazing field of application.

Thanks!!!

ChumKiu

ChumKiu, 2019-05-30:
Hi,
I've did a little step, to know where the command list are listed I've performed a search for a know command to an endpoint and cluster that is supposed to be always supported in a standard way so I've choosed to use the cluster 0, here attached an image with the bellows output and where I'm able to found the listed command inside the ZIGPY source code.

```
(bellows) root@raspberrypi:~/zigbee# bellows zcl 00:15:8d:00:02:b5:40:20 1 0 commands
reset_fact_default
```

So, if I perform a search for the cluster_id = 0x0405 for example I'm unable to found anything of similar hance this should be my explanation as starting point because nothing command list is emitted from bellows when I try to perfom this request:

```
(bellows) root@raspberrypi:~/zigbee# bellows zcl 00:15:8d:00:02:b5:40:20 1 1026 commands
```
where, 1026 in decimal means 0x402 in hex, and is related to the Temperature Measurements cluster as per ZigBee ZCL specification, see my previous post for the doc reference.

Performing a `cluster_id =` simple text string search through the file `general.py` source file I can see that the maximum cluster reference is up to 0x0021...

Any idea?

[image]

ChumKiu, 2019-05-30:
Ok, I'm going back to the specification, under the chapter 4.4 Temperature Measurement and more specifically at the paragraph 4.4.2.3 it is written that "No cluster specific commands are received by the server cluster" so is all correct, no command to use in this place because there are only attributes!
With this in mind I've also discovered that the help for the zcl section of bellows states:

```
Usage: bellows zcl [OPTIONS] NODE ENDPOINT CLUSTER COMMAND [ARGS]...

  Peform ZCL operations against a device

Options:
  -D, --database FILE  [required]
  --help               Show this message and exit.

Commands:
  command
  commands
  configure-reporting
  read-attribute
  write-attribute
```
yep now  I can see that read-attribute may be my command to use... my fault in reading/understanding the bellows help have drive me into the specification and then I've also understand that for some cluster there are only attribute and no command so "not all evils come to harm" :)

Next step then is to issue this attribute reading command to read the sensor temperature hence I've written:

```
(bellows) root@raspberrypi:~/zigbee# bellows zcl 00:15:8d:00:02:b5:40:20 1 1026 read-attribute 0
```
but I got an EmberStatus.DELIVERY_FAILED exception.

So I've did another test in reading an attribute for one of the light bulb that I've at now:

```
(bellows) root@raspberrypi:~/zigbee# bellows zcl 90:fd:9f:ff:fe:da:f0:ce 1 6 read-attribute 0
0=Bool.true
```
yep! It worked and effectively the lamp is switched on so the error encountered with the temperature sensor will drive me to think that this sensor, for some reason, coud be detached from my network, don't know the reason I've to investigate on this point, make sense?

Thank!!!!

ChumKiu

ChumKiu, 2019-05-30:
After some test in pairing the sensor I've got it in my devices database but the attribute interrogation was not succesfully:

```
(bellows) root@raspberrypi:~# bellows devices
Device:
  NWK: 0x8aea
  IEEE: 7c:b0:3e:aa:00:ad:90:6b
  Endpoints:
    3: profile=0xc05e, device_type=DeviceType.EXTENDED_COLOR_LIGHT
      Input Clusters:
        Basic (0)
        Identify (3)
        Groups (4)
        Scenes (5)
        On/Off (6)
        Level control (8)
        Color Control (768)
        LightLink (4096)
        Manufacturer Specific (64527)
      Output Clusters:
        Ota (25)
Device:
  NWK: 0xbf75
  IEEE: 00:15:8d:00:02:b5:40:20
  Endpoints:
    1: profile=0x104, device_type=24321
      Input Clusters:
        Basic (0)
        Identify (3)
        Temperature Measurement (1026)
        Relative Humidity Measurement (1029)
Device:
  NWK: 0x1348
  IEEE: 90:fd:9f:ff:fe:da:f0:ce
  Endpoints:
    1: profile=0xc05e, device_type=DeviceType.COLOR_TEMPERATURE_LIGHT
      Input Clusters:
        Basic (0)
        Identify (3)
        Groups (4)
        Scenes (5)
        On/Off (6)
        Level control (8)
        Color Control (768)
        Diagnostic (2821)
        LightLink (4096)
      Output Clusters:
        Scenes (5)
        Ota (25)
        Poll Control (32)
        LightLink (4096)
(bellows) root@raspberrypi:~# bellows zcl 00:15:8d:00:02:b5:40:20 1 1026 read-attribute 0
Attribute 0 not successful. Status=134
```
and after some interrogation I've got again the exception error DeliveryError...

Adminiuga, 2019-05-30:
Eh, so Xiaomi are not really good test subjects, because they loosely adhere to Zigbee specs.
Check section 2.2 of ZCL specs, as it talks briefly about "client/server" clusters. In `bellows/zigpy`, `output_cluster` means `client cluster` and `input_cluster` means `server cluster`. 
As you already found out, each cluster has attributes and commands. Sensor type devices usually just have the attributes and you can get those with `bellows zcl <IEEE> >endpoint_id> <cluster_id> read-attribute <attribute_id>`

Now, the reason you get `EmberStatus.DELIVERY_FAILED` to `bellows zcl 00:15:8d:00:02:b5:40:20 1 1026 read-attribute 0` (assuming your sensor is not too far from its parent device) is because Xiaomi is not polling its parent device often enough, I think xiaomi were designed to push data to the gateway and not to be actively polled. I don't recall exactly if Xiaomi replies at all to `read_attributes` ZCL command, but you can try pressing the button on the sensor right after you send the command to awake the device and force it to poll its parent. 

Regarding the missing "pressure" cluster in output of `bellows devices`.  Two things at play here:
1. In your logs, you can see actually what device reports for its endpoints and clusters on each endpoints
```
[0x5ced] Discovered endpoints: [1, 2, 3]
warning: Unexpected response TSN=4 command=266 args=b'\x05\x00B\x0elumi.sensor_ht'
warning: [0x5ced:1] Message on unknown cluster 0x0000
[0x5ced:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=24321 device_version=1 input_clusters=[0, 3, 25, 65535, 18] output_clusters=[0, 4, 3, 5, 25, 65535, 18]>
[0x5ced:2] Discovered endpoint information: <SimpleDescriptor endpoint=2 profile=260 device_type=24322 device_version=1 input_clusters=[3, 18] output_clusters=[4, 3, 5, 18]>
[0x5ced:3] Discovered endpoint information: <SimpleDescriptor endpoint=3 profile=260 device_type=24323 device_version=1 input_clusters=[3, 12] output_clusters=[4, 3, 5, 12]>
```
as you can see device itself does not report correct cluster ids for temperature/humidity.  Which brings us to the second part: `zigpy quirks`. Because this device does not behave accordingly to zigbee specs, zigpy quirks allows alteration of device aspects and behavior (keep in mind there's also `zha-device-handler` python module used with HomeAssistant which has more zigpy quirks defined, which are not applied when in use with `bellows`) So in `zigpy.quirks` there's a `CustomDevice` class defined specifically for this sensor which matches the endpoint info (profile, device_type, input_cluster/output_cluster lists etc) and if there's a match it replaces vanilla implementation for device/cluster classes with a custom ones. eg https://github.com/zigpy/zigpy/blob/287f0630b34d9a456e00b36010f38955f9c2b694/zigpy/quirks/xiaomi/__init__.py#L6-L39 replaces the above discovered information with just one endpoint and Temperature + Humidity clusters. 
There's no `pressure` cluster id in the replacement and you would need to confirm if the device really supports pressure measurements. based on the `model` attribute `warning: Unexpected response TSN=4 command=266 args=b'\x05\x00B\x0elumi.sensor_ht'` to me it looks the `_ht` part designating only humidity and temperature. Now the seller description may say otherwise, but you really would need to use a zigbee sniffer and see if it is sending any data on pressure cluster id. I doubt it does, although I do see that some sellers list this model as pressure sensor. If you need pressure, than get "aqara" sensors like https://www.aliexpress.com/item/2018-Xiaomi-Aqara-Smart-Air-Pressure-Temperature-Humidity-Environment-Sensor-Smart-control-via-Mihome-APP-Zigbee/32867769187.html

Adminiuga, 2019-05-30:
> `Attribute 0 not successful. Status=134`

`Status=134` means "unsupported attribute". I guess you can't really poll Xiaomi Mija sensors. There are xiaomi mija and xiaomi aqara sensors, both are somewhat zigbee based, but aqara ones behave more or less logically compared to mija ones.

Adminiuga, 2019-05-30:
If you want to experiment with Zigbee to learn it, my advise is to get a none xiaomi sensor, as it is much harder to tell whether behavior you see is because you are doing something wrong is it because xiaomi is not following the standards. Centralite or SmartThings sensors are much better in this regard.

ChumKiu, 2019-05-30:
I've changed the sensor with another XIAOMI model, now is a water detection sensor model SJCGQ11LM, after some try I'm succesfully in pairing:

```
Device:
  NWK: 0x2552
  IEEE: 00:15:8d:00:02:4b:cb:c4
  Endpoints:
    1: profile=0x104, device_type=DeviceType.IAS_ZONE
      Input Clusters:
        Basic (0)
        Power Configuration (1)
        Identify (3)
        IAS Zone (1280)
```

but now I've found a strange behavior, if I try to read the attribute 1 (ZoneType from ZCL specification for the IAS Zone devices, cluster 0x0500 - 1280 in decimal) I got the  EmberStatus.DELIVERY_FAILED error, but if during the request I push one time the central button (the blu led on the cover will blink) I can obtain the right answer from the sensor:

```
(bellows) root@raspberrypi:~# bellows zcl 00:15:8d:00:02:4b:cb:c4 1 1280 read-attribute 1
1=42
```
42 is in decimal format, translating it in hex give 0x002A that from the ZoneType attribute table means that this is a Water sensor.
Later I tried to read the ZoneStatus attribute so:

```
(bellows) root@raspberrypi:~# bellows zcl 00:15:8d:00:02:4b:cb:c4 1 1280 read-attribute 2
2=0
```
putting the two contact short-circuited to simulate the water presence give:

```
(bellows) root@raspberrypi:~# bellows zcl 00:15:8d:00:02:4b:cb:c4 1 1280 read-attribute 2
2=1
```
but to obtain these answers is necessary to press one time the button on the sensor just after the bellows command is issued, this mode of operation obviously is not feasible into a real application, so there should be another way to keep the sensor alive because it seems that the sensor go in a sleep mode (may be to save battery) hence there is a need to find a weak up way.

Something have some idea?

Thank!
ChumKiu

[image]
[image]

Adminiuga, 2019-05-30:
Check the whole 8.2 section of ZCL specification revision 6.

Adminiuga, 2019-05-30:
And take a look at Homeassistant ZHA component implementation.

ChumKiu, 2019-05-30:
Hi @Adminiuga,
thanks for your answers, I've see these one just now when I've do a browser refresh, may be when you wrote I'm also writing my experience.

I will read carefully your answer and then will go back with my testing.

Thanks for now!
ChumKiu

ChumKiu, 2019-05-31:
Hi @Adminiuga,
thank for your useful info.

> 
> Eh, so Xiaomi are not really good test subjects, because they loosely adhere to Zigbee specs.
> Check section 2.2 of ZCL specs, as it talks briefly about "client/server" clusters. In `bellows/zigpy`, `output_cluster` means `client cluster` and `input_cluster` means `server cluster`.
>

Following your suggestion I have read more carefully the paragraph 2.2.2 Client/Server Model of the ZCL and I've discovered that a Server Device, like the XIAOMI temperature sensor, can be interrogated from the Client (in this way my Raspberry Pi), but there is also another (very interesting) possibility that the information is transmitted from the server side (for example the temperature sensor) to the client (my RPi) through the so called `report attribute command` as mentioned into the paragraph 2.2.2.

[image]

going to the paragraph 2.5.11 we have:

[image]

then going to the paragraph 2.5.7 give:

[image]

so my thinking, basing on these information, to overcome the trouble of the XIAOMI sensor when is queried by the client (my RPi) could be to force the sensor itself to send back a report of the temperature attribute to my RPi, in this way I should avoid to get the `EmberStatus.DELIVERY_FAILED` message when I try to query directly the sensor from my RPi. This behaviour make sense to me from a battery point of view, the sensor will be in sleep state most of the time and awake briefly only to send back the temperature or other phisical quantity to the client (that have to process this information) otherwise the sensor must keep  continuously powered the radio section (that is responsible for the major part of the current consumption) shortening the useful life of the battery.

> Now, the reason you get `EmberStatus.DELIVERY_FAILED` to `bellows zcl 00:15:8d:00:02:b5:40:20 1 1026 read-attribute 0` (assuming your sensor is not too far from its parent device) is because Xiaomi [...]

So if my reasoning is correct, after pairing the sensor, I'll have also to send the `configure reporting command`, hence now I've to discover how this can be do practically by means of the `bellows` CLI.

Make sense for you?

Thank!
ChumKiu

ChumKiu, 2019-05-31:
> 
> 
> > `Attribute 0 not successful. Status=134`
> 
> `Status=134` means "unsupported attribute". I guess you can't really poll Xiaomi Mija sensors. There are xiaomi mija and xiaomi aqara sensors, both are somewhat zigbee based, but aqara ones behave more or less logically compared to mija ones.

Thank @Adminiuga ,
I've found the table reference on the ZCL specification paragraph 2.6.3 Enumerations, for reference I will report here the whole table taken directly from the document (134 in hex is 0x86):

[image]
[image]
[image]

may be useful integrate into the answer also the meaning as specified into the ZCL document.

ChumKiu, 2019-05-31:
I'm try to use bellows CLI to send a configure-reporting command to the AQARA door switch MCCGQ11LM, here the device description when succesfully paired (after some attempts):

```
Device:
  NWK: 0x941d
  IEEE: 00:15:8d:00:02:bf:d6:d8
  Endpoints:
    1: profile=0x104, device_type=24321
      Input Clusters: < - - - SERVER CLUSTER
        Basic (0)
        Identify (3)
      Output Clusters:  <- - - CLIENT CLUSTER
        Basic (0)
        Groups (4)
        On/Off (6)
```
as @Adminiuga has pointed me the Input Cluster in bellows are the Server Cluster into the ZCL specification and the Output Cluster in bellows are the so called Client Cluster into the ZCL Specification then to perform a reporting, If I've correctly understand, I've to work with the Output Cluster and in this case with the On/Off.
So I've set this command:

```
bellows zcl 00:15:8d:00:02:bf:d6:d8 1 6 configure-reporting 0 1800 2700 1
```
the attribute is equal to 0 (OnOff) other values was taken from the xiaomi.py code from 
[puddly](https://gist.github.com/puddly/7db5ecc9b4f618e37583922dce2f4b83) but I'm unable to get it working, here below the message from `bellows`:

```
Device 00:15:8d:00:02:bf:d6:d8 has no cluster 6 on endpoint 1
```

of course there is no a Server Cluster 6 but how I can specify in `bellows` the Output Cluster that i would use?

Thank for your help!
ChumKiu

Adminiuga, 2019-05-31:
You can `configure-reporting` only for attributes you can read, as in attribute of a `server/input` cluster. 

The `OnOff` cluster of this door switch is an `output/client` attribute, means it does not receive OnOff cluster commands, BUT sends OnOff cluster commands like `On`, `Off`, `Toggle` etc. I have to find a device yet which allows me to read attributes from "client clusters".

ChumKiu, 2019-06-03:
> You can `configure-reporting` only for attributes you can read, as in attribute of a `server/input` cluster.
>
> The `OnOff` cluster of this door switch is an `output/client` attribute, means it does [...]

Hi @Adminiuga,
thank for your support, but I'm confused on this point, I'm not trying to send a output command but a request for send back to me a report every fixed time and from I can understand this task should be possibile by a Output Cluster or a Client Cluster, isn't it?

-----------------------------------------------------
UPGRADING BELLOWS TO THE LATEST RELEASE
-----------------------------------------------------
As second question I kindly request how I can update my bellow/zigpy installation.
Actually I've installed (by issuing the `pip install bellows`):

```
(bellows) root@raspberrypi:~# pip list | grep zigpy
zigpy            0.2.0
(bellows) root@raspberrypi:~# pip list | grep bellows
bellows          0.7.0
```

but I can see that there are update on zigpy and bellows so I kindly ask you how I've to do to manually update my current version to the latest?
I've to dowload manually the `bellows-dev.zip` archivie from `https://github.com/zigpy/bellows` and then after unzipping inside my virtualenv install it with the command `python setup.py install --user` ?
I'm not so practic with manual installation and related stuff...

Thank!!
ChumKiu

nerdosity, 2020-06-11:
Hi, was this issue ever resolved? 
I am trying to migrate from zigate to Silicon Labs EZSP in latest home assistant, on a Synology NAS with latest home assistant in docker.
Everything worked wonderfully, until I tried to join my two Temperature/humidity sensors. Both were just ignored.
Is there any way to get them paired?

nerdosity, 2020-06-12:
`2020-06-12 16:29:53 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x1c6f, <EmberApsFrame profileId=260 clusterId=6 sourceEndpoint=1 destinationEndpoint=1 options=64 groupId=0 sequence=255>, 0, b'\x01\xff\x00')
2020-06-12 16:29:53 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received: b'00cb'
2020-06-12 16:29:53 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'000401060001014001000043ffca6f1cffff0508ff0b0000'
2020-06-12 16:29:53 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=260 clusterId=6 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=67>, 255, -54, 0x1c6f, 255, 255, b'\x08\xff\x0b\x00\x00']
2020-06-12 16:29:53 DEBUG (MainThread) [zigpy.zcl] [0x1c6f:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=255 command_id=Command.Default_Response>
2020-06-12 16:29:53 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received: b'006f1c04010600010140000000cb000000'
2020-06-12 16:29:53 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 7279, <EmberApsFrame profileId=260 clusterId=6 sourceEndpoint=1 destinationEndpoint=1 options=64 groupId=0 sequence=203>, 0, <EmberStatus.SUCCESS: 0>, b'']
2020-06-12 16:29:53 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1c6f:1:0x0006]: executed 'off' command with args: '()' kwargs: '{}' result: [0, <Status.SUCCESS: 0>]
2020-06-12 16:29:54 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2020-06-12 16:29:54 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2020-06-12 16:30:04 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2020-06-12 16:30:04 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2020-06-12 16:30:12 DEBUG (MainThread) [bellows.ezsp] Send command sendBroadcast: (<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>, <EmberApsFrame profileId=0 clusterId=54 sourceEndpoint=0 destinationEndpoint=0 options=0 groupId=0 sequence=1>, 0, 2, b'\x01<\x00')
2020-06-12 16:30:12 DEBUG (MainThread) [bellows.ezsp] Application frame 54 (sendBroadcast) received: b'00cc'
2020-06-12 16:30:12 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'0500003600000000000000ccff000000ffff03013c00'
2020-06-12 16:30:12 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST_LOOPBACK: 5>, <EmberApsFrame profileId=0 clusterId=54 sourceEndpoint=0 destinationEndpoint=0 options=0 groupId=0 sequence=204>, 255, 0, 0x0000, 255, 255, b'\x01<\x00']
2020-06-12 16:30:12 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
2020-06-12 16:30:13 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received: b'06fcff00003600000000000000cc020000'
2020-06-12 16:30:13 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_BROADCAST: 6>, 65532, <EmberApsFrame profileId=0 clusterId=54 sourceEndpoint=0 destinationEndpoint=0 options=0 groupId=0 sequence=204>, 2, <EmberStatus.SUCCESS: 0>, b'']
2020-06-12 16:30:13 DEBUG (MainThread) [bellows.ezsp] Send command permitJoining: (60,)
2020-06-12 16:30:13 DEBUG (MainThread) [bellows.ezsp] Application frame 34 (permitJoining) received: b'00'
2020-06-12 16:30:14 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'000401000001010001000075ffc430edffff1d1c34121c0a02ff4c0600100121c70b21a81324010000000021af01205e'
2020-06-12 16:30:14 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=260 clusterId=0 sourceEndpoint=1 destinationEndpoint=1 options=256 groupId=0 sequence=117>, 255, -60, 0xed30, 255, 255, b'\x1c4\x12\x1c\n\x02\xffL\x06\x00\x10\x01!\xc7\x0b!\xa8\x13$\x01\x00\x00\x00\x00!\xaf\x01 ^']
2020-06-12 16:30:14 DEBUG (MainThread) [zigpy.zcl] [0xed30:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=False disable_default_response=True> manufacturer=4660 tsn=28 command_id=Command.Report_Attributes>
2020-06-12 16:30:14 DEBUG (MainThread) [zigpy.zcl] [0xed30:1:0x0000] ZCL request 0x000a: [[<Attribute attrid=65282 value=<TypeValue type=LVList, value=[<TypeValue type=Bool, value=Bool.true>, <TypeValue type=uint16_t, value=3015>, <TypeValue type=uint16_t, value=5032>, <TypeValue type=uint40_t, value=1>, <TypeValue type=uint16_t, value=431>, <TypeValue type=uint8_t, value=94>]>>]]
2020-06-12 16:30:14 DEBUG (MainThread) [zigpy.zcl] [0xed30:1:0x0000] Attribute report received: 65282=[<TypeValue type=Bool, value=Bool.true>, <TypeValue type=uint16_t, value=3015>, <TypeValue type=uint16_t, value=5032>, <TypeValue type=uint40_t, value=1>, <TypeValue type=uint16_t, value=431>, <TypeValue type=uint8_t, value=94>]
2020-06-12 16:30:14 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
[... cut ...]
2020-06-12 16:30:50 DEBUG (MainThread) [zigpy.zcl] [0xc7dc:1:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=98 command_id=Command.Report_Attributes>
2020-06-12 16:30:50 DEBUG (MainThread) [zigpy.zcl] [0xc7dc:1:0x0406] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=bitmap8, value=bitmap8.1>>]]
2020-06-12 16:30:50 DEBUG (MainThread) [zigpy.zcl] [0xc7dc:1:0x0406] Attribute report received: occupancy=1
2020-06-12 16:30:50 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc7dc:1:0x0500]: Updated alarm state: 1
`

nerdosity, 2020-07-12:
Hi, I updated to latest HA version, but problem is still present.
How can I debug?

closed 2020-12-17 by Adminiuga