zigpy/bellows #168: permit-with-key not working

rayzorben, 2019-06-04:
Trying to use permit-with-key, encountered an error with zigpy that I worked around, but now bellows is giving the error `'EmberKeyData' object has no attribute 'contents'`. This seems to be failing in 
```python
ezsp.py _ezsp_frame(self, name, *args): data = t.serialize(args, c[1]))
```

The command I am running is 
```bash
sudo /srv/homeassistant/bin/bellows -d /dev/zigbee permit-with-key -D zigbee.db 00:0d:6f:00:0b:5d:xx:xx 313233343536E672
```

Apologies for not having more information, I am fairly new to python and not sure how I can get more details.

Adminiuga, 2019-06-04:
yep, I think it is a bug. can you try the following patch
```diff --git a/bellows/zigbee/application.py b/bellows/zigbee/application.py
index 1f5d4f4..cef0632 100644
--- a/bellows/zigbee/application.py
+++ b/bellows/zigbee/application.py
@@ -380,7 +380,9 @@ class ControllerApplication(zigpy.application.ControllerApplication):
         if key is None:
             raise Exception("Invalid install code")
 
-        v = await self._ezsp.addTransientLinkKey(node, key)
+        transient_key = t.EmberKeyData()
+        transient_key.contents = key
+        v = await self._ezsp.addTransientLinkKey(node, transient_key)
         if v[0] != t.EmberStatus.SUCCESS:
             raise Exception("Failed to set link key")
 
```

rayzorben, 2019-06-04:
So that you can verify, here is the code:

```python
  key = zigpy.util.convert_install_code(code)
        if key is None:
            raise Exception("Invalid install code")

        transient_key = t.EmberKeyData()
        transient_key.contents = key
        v = await self._ezsp.addTransientLinkKey(node, transient_key)
        #v = await self._ezsp.addTransientLinkKey(node, key)
        if v[0] != t.EmberStatus.SUCCESS:
            raise Exception("Failed to set link key")
```

It seems to make it past that, but now I get an assertion aerror in `ezsp.py`

```python 
Exception in callback SerialTransport._read_ready()
handle: <Handle SerialTransport._read_ready()>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/srv/homeassistant/lib/python3.7/site-packages/serial_asyncio/__init__.py", line 106, in _read_ready
    self._protocol.data_received(data)
  File "/srv/homeassistant/lib/python3.7/site-packages/bellows/uart.py", line 65, in data_received
    self.frame_received(frame)
  File "/srv/homeassistant/lib/python3.7/site-packages/bellows/uart.py", line 87, in frame_received
    self.data_frame_received(data)
  File "/srv/homeassistant/lib/python3.7/site-packages/bellows/uart.py", line 108, in data_frame_received
    self._application.frame_received(self._randomize(data[1:-3]))
  File "/srv/homeassistant/lib/python3.7/site-packages/bellows/ezsp.py", line 229, in frame_received
    assert expected_id == frame_id
AssertionError
```

Adminiuga, 2019-06-04:
88 == 0x58 == `invalid command` 

Which bellows version are you running? If 0.8.0, I'd like to see the log for ezsp just prior the exception, to see what was sent and what we received. `invalid_command` frame should contain code explaining why the command is invalid, may give some hints.
If you are not on 0.8.0, cherry pick https://github.com/zigpy/bellows/commit/d8c1358a2b4ab10d24ea9a8db88bfc87b6f2596a and post the logs with ezsp data please.

rayzorben, 2019-06-04:
Thanks @Adminiuga stupid question, how do I see the logs?

rayzorben, 2019-06-04:
Maybe this is what you want?

```
debug: Using selector: EpollSelector
debug: Loading application state from zigbee.db
debug: Considering <class 'zigpy.quirks.xiaomi.TemperatureHumiditySensor'>
debug: Fail because endpoint list mismatch: dict_keys([1, 2, 3]) {3}
debug: Considering <class 'zigpy.quirks.xiaomi.AqaraTemperatureHumiditySensor'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.xiaomi.AqaraOpenCloseSensor'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.xiaomi.AqaraWaterSensor'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.smartthings.SmartthingsTemperatureHumiditySensor'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.kof.CeilingFan'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.xiaomi.TemperatureHumiditySensor'>
debug: Fail because endpoint list mismatch: dict_keys([1, 2, 3]) {3}
debug: Considering <class 'zigpy.quirks.xiaomi.AqaraTemperatureHumiditySensor'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.xiaomi.AqaraOpenCloseSensor'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.xiaomi.AqaraWaterSensor'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.smartthings.SmartthingsTemperatureHumiditySensor'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.kof.CeilingFan'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Attribute id: 4 value: b'OSRAM'
debug: Attribute id: 5 value: b'LIGHTIFY BR RGBW'
debug: Attribute id: 4 value: b'OSRAM'
debug: Attribute id: 5 value: b'LIGHTIFY BR RGBW'
debug: Attribute id: 7 value: 1
debug: Attribute id: 7 value: 1
debug: Attribute id: 7 value: 370
debug: Attribute id: 7 value: 370
debug: Attribute id: 0 value: 0
debug: Attribute id: 0 value: 255
debug: Attribute id: 0 value: 0
debug: Attribute id: 0 value: 255
debug: Attribute id: 4 value: 24701
debug: Attribute id: 3 value: 24939
debug: Attribute id: 4 value: 24701
debug: Attribute id: 3 value: 24939
debug: Attribute id: 1291 value: 400
debug: Attribute id: 1291 value: 400
debug: Sending: b'1ac038bc7e'
debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
debug: Send command version
debug: Sending: b'004221a850ed2c7e'
debug: Data frame: b'0142a1a8502805e67f627e'
debug: Sending: b'8160597e'
debug: Application frame 0 (version) received
debug: Send command setConfigurationValue
debug: Sending: b'7d314321fb582815c3277e'
debug: Data frame: b'1243a1fb54fb737e'
debug: Sending: b'82503a7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'224021fb592f15226f7e'
debug: Data frame: b'2340a1fb54c6107e'
debug: Sending: b'83401b7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
debug: Sending: b'334121fb792b15a2d77e'
debug: Data frame: b'3441a1fb54d32a7e'
debug: Sending: b'8430fc7e'
debug: Application frame 83 (setConfigurationValue) received
debug: Send command setConfigurationValue
[... 84 lines cut ...]
debug: Data frame: b'5556a18ec5c948b959fb472574b87e'
debug: Sending: b'8610be7e'
debug: Application frame 38 (getEui64) received
debug: Send command addTransientLinkKey
debug: Sending: b'56572107c5c948b959fb47257d31a832c030645f2bf1399b365b955afd6cee7e'
debug: Data frame: b'6657a1f065442d7e'
debug: Sending: b'87009f7e'
debug: Application frame 88 (invalidCommand) received
Exception in callback SerialTransport._read_ready()
handle: <Handle SerialTransport._read_ready()>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/srv/homeassistant/lib/python3.7/site-packages/serial_asyncio/__init__.py", line 106, in _read_ready
    self._protocol.data_received(data)
  File "/srv/homeassistant/lib/python3.7/site-packages/bellows/uart.py", line 64, in data_received
    self.frame_received(frame)
  File "/srv/homeassistant/lib/python3.7/site-packages/bellows/uart.py", line 86, in frame_received
    self.data_frame_received(data)
  File "/srv/homeassistant/lib/python3.7/site-packages/bellows/uart.py", line 107, in data_frame_received
    self._application.frame_received(self._randomize(data[1:-3]))
  File "/srv/homeassistant/lib/python3.7/site-packages/bellows/ezsp.py", line 169, in frame_received
    assert expected_id == frame_id
AssertionError
```

rayzorben, 2019-06-04:
```bash
pi@hassbian:/home/homeassistant/.homeassistant $ sudo /srv/homeassistant/bin/bellows -v DEBUG -d /dev/zigbee permit-with-key -D zigbee.db 00:0d:6f:00:0b:5d:E3:91 31323334353672E6
debug: Using selector: EpollSelector
debug: Loading application state from zigbee.db
debug: Considering <class 'zigpy.quirks.xiaomi.TemperatureHumiditySensor'>
debug: Fail because endpoint list mismatch: dict_keys([1, 2, 3]) {3}
debug: Considering <class 'zigpy.quirks.xiaomi.AqaraTemperatureHumiditySensor'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.xiaomi.AqaraOpenCloseSensor'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.xiaomi.AqaraWaterSensor'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.smartthings.SmartthingsTemperatureHumiditySensor'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.kof.CeilingFan'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.xiaomi.TemperatureHumiditySensor'>
debug: Fail because endpoint list mismatch: dict_keys([1, 2, 3]) {3}
debug: Considering <class 'zigpy.quirks.xiaomi.AqaraTemperatureHumiditySensor'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.xiaomi.AqaraOpenCloseSensor'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.xiaomi.AqaraWaterSensor'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.smartthings.SmartthingsTemperatureHumiditySensor'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Considering <class 'zigpy.quirks.kof.CeilingFan'>
debug: Fail because endpoint list mismatch: dict_keys([1]) {3}
debug: Attribute id: 4 value: b'OSRAM'
debug: Attribute id: 5 value: b'LIGHTIFY BR RGBW'
debug: Attribute id: 4 value: b'OSRAM'
debug: Attribute id: 5 value: b'LIGHTIFY BR RGBW'
debug: Attribute id: 7 value: 1
debug: Attribute id: 7 value: 1
debug: Attribute id: 7 value: 370
debug: Attribute id: 7 value: 370
debug: Attribute id: 0 value: 0
debug: Attribute id: 0 value: 255
debug: Attribute id: 0 value: 0
debug: Attribute id: 0 value: 255
debug: Attribute id: 4 value: 24701
debug: Attribute id: 1291 value: 400
debug: Attribute id: 1291 value: 400
debug: Attribute id: 3 value: 24939
debug: Attribute id: 4 value: 24701
debug: Attribute id: 3 value: 24939
debug: Sending: b'1ac038bc7e'
debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
debug: Send command version: (4,)
debug: Sending: b'004221a850ed2c7e'
debug: Data frame: b'0142a1a8502805e67f627e'
debug: Sending: b'8160597e'
debug: Application frame 0 (version) received: b'04021054'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_STACK_PROFILE: 12>, 2)
debug: Sending: b'7d314321fb582815c3277e'
debug: Data frame: b'1243a1fb54fb737e'
debug: Sending: b'82503a7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_SECURITY_LEVEL: 13>, 5)
debug: Sending: b'224021fb592f15226f7e'
debug: Data frame: b'2340a1fb54c6107e'
debug: Sending: b'83401b7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_SUPPORTED_NETWORKS: 45>, 1)
debug: Sending: b'334121fb792b15a2d77e'
debug: Data frame: b'3441a1fb54d32a7e'
debug: Sending: b'8430fc7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_APPLICATION_ZDO_FLAGS: 42>, 3)
debug: Sending: b'444621fb7d5e291514417e'
[... 85 lines cut ...]
debug: Application frame 38 (getEui64) received: b'91e35d0b006f0d00'
debug: Send command addTransientLinkKey: (00:0d:6f:00:0b:5d:e3:91, <EmberKeyData contents=[187, 253, 160, 137, 172, 42, 120, 128, 28, 247, 252, 189, 166, 83, 57, 116]>)
debug: Sending: b'56572107c5c948b959fb47257d31a832c030645f2bf1399b365b955afd6cee7e'
debug: Data frame: b'6657a1f065442d7e'
debug: Sending: b'87009f7e'
debug: Application frame 88 (invalidCommand) received: b'31'
Exception in callback SerialTransport._read_ready()
handle: <Handle SerialTransport._read_ready()>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/srv/homeassistant/lib/python3.7/site-packages/serial_asyncio/__init__.py", line 106, in _read_ready
    self._protocol.data_received(data)
  File "/srv/homeassistant/lib/python3.7/site-packages/bellows/uart.py", line 64, in data_received
    self.frame_received(frame)
  File "/srv/homeassistant/lib/python3.7/site-packages/bellows/uart.py", line 86, in frame_received
    self.data_frame_received(data)
  File "/srv/homeassistant/lib/python3.7/site-packages/bellows/uart.py", line 107, in data_frame_received
    self._application.frame_received(self._randomize(data[1:-3]))
  File "/srv/homeassistant/lib/python3.7/site-packages/bellows/ezsp.py", line 170, in frame_received
    assert expected_id == frame_id
AssertionError
^C
Aborted!
```

Adminiuga, 2019-06-04:
Have some bad news (
`b'31'` -- Ezsp Status `The NCP received a command containing an unsupported frame ID.`

I'm getting the same with my HUSBZB-1, which is running Ezsp version 4 ???
But the same works with Elelabs Shield EZSP radio which is running version 6:

```
debug: MulticastTableEntry[15] = <EmberMulticastTableEntry multicastId=0x0000 endpoint=0 networkIndex=0>
debug: Send command addTransientLinkKey: (00:0d:6f:00:0b:5d:11:22, <EmberKeyData contents=[59, 62, 0, 196, 212, 139, 223, 45, 1, 51, 163, 101, 211, 196, 69, 147]>)
debug: Sending: b'23682157548537a3049f4a4aa755a9779c8af32032e366b85ea3b04db9ed436a7e'
debug: Data frame: b'3368a157548515a24f7e'
debug: Sending: b'8430fc7e'
debug: Application frame 175 (addTransientLinkKey) received: b'00'
debug: Send command setPolicy: (<EzspPolicyId.TC_KEY_REQUEST_POLICY: 5>, <EzspDecisionId.GENERATE_NEW_TC_LINK_KEY: 82>)
debug: Sending: b'34692157547f10e057b97e'
debug: Data frame: b'4469a157547f1586c07e'
debug: Sending: b'8520dd7e'
debug: Application frame 85 (setPolicy) received: b'00'
debug: Send command permitJoining: (30, True)
Joins are permitted for the next 30s...
debug: Sending: b'456e215754080bb6017e'
```

Adminiuga, 2019-06-04:
I wonder if it is possible to upgrade firmware in husbzb-1. should be standard silabs chipset.

rayzorben, 2019-06-04:
How do I tell what version of ezsp I am running? I was already trying to research how to upgrade the firmware :)

Adminiuga, 2019-06-04:
Three HUSBZB-1 sticks I had were running version 4. After Zigpy loads `zigbee.db` and `bellows` resets the controller, the very 1st command is going to be `version` and you should receive back `debug: Application frame 0 (version) received: b'06020060'`. The 1st "hexlified" byte is going to be "Protocol Version". 06 in case of Elelabs.

```
debug: Resetting EZSP
debug: Resetting ASH
debug: Sending: b'1ac038bc7e'
debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
debug: Send command version: (4,)
debug: Sending: b'004221a850ed2c7e'
debug: Data frame: b'0142a1a8522815d2e7ae7e'
debug: Sending: b'8160597e'
debug: Application frame 0 (version) received: b'06020060'
debug: Switching to eszp version 6
debug: Send command version: (6,)
```

rayzorben, 2019-06-04:
Thanks, I have gone down the route of trying to upgrade the firmware, downloaded Simplicity Studio, and got the package for the EM3581, but I am blocked because I don't own a 'kit' for EmberZNet. So I am guessing I cannot upgrade the firmware. I can not find the firmware available elsewhere.

The only reason I got the HUSBZB-1 is because it also has ZWave, but come to find out I don't think I actually own any ZWave devices. What I am most interested in getting working is integration with my Smart Meter from my electric company, and they require Zigbee 3.0 with the installation code.

I will give the Elelabs Shield a try. Any reason it has to be USB? Can I use the GPIO version: https://elelabs.com/products/elelabs_zigbee_shield.html ?

Adminiuga, 2019-06-04:
That suxx. I don't think there's a chance GoControl would release an update. Too bad, cause that stick was quite popular.
For Elelabs, it doesn't have to be USB, i've got a simple one and just connected it to serial usb adapter.

Adminiuga, 2019-06-11:
@rayzorben what was the device you tried to pair?

rayzorben, 2019-06-12:
A smart meter from my electric company, Southern California Edison.

rayzorben, 2019-07-10:
I acquired an Elelabs Zigbee Shield and was super excited to try this out. However when I try anything on it, I am getting an error
```bash
pi@hassbian:/home/homeassistant/.homeassistant $ sudo /srv/homeassistant/bin/bellows -d /dev/ttyAMA0 -v DEBUG scan
debug: Using selector: EpollSelector
debug: Connected. Resetting.
debug: Sending: b'1ac038bc7e'
error: CRC error in frame b'527e' (b'52' != b'ffff')
debug: Sending: b'a0547d3a7e'
```

And then it just hangs there. Any ideas?

Adminiuga, 2019-07-10:
Hrm, I was unsuccessful getting Elelabs Zigbee Shield to run off GPIO pins, even after troubleshooting with elelabs. It was blamed on the custom debian image I was running on PI, however the module itself works flawlessly on my USB-2-Serial converter. 

Not familiar with hassbian config, but make sure that kernel is not using the serial port for its console nor `agetty`  is configured on that port.

Elelabs also told me to disable built-in bluetooth module, which required changing some boot configuration.  Keywords to google: `dtoverlay=pi3-disable-bt`

It would be interesting to see what are you getting on the serial port. Can you modify `bellows` and insert `LOGGER.info("extracting frame: %s", binascii.hexlify(frame))` prior https://github.com/zigpy/bellows/blob/7df8062116debe997bb4bb5e731e1d84f32cb9ff/bellows/uart.py#L72

Adminiuga, 2019-07-10:
https://openenergymonitor.org/forum-archive/node/12311.html

impala454, 2019-10-19:
Hi @rayzorben did you ever get further in your attempts to connect to your smart meter?