zigpy/bellows #175: error from deserialize not catched

martin3000, 2019-07-14:
In bellows/zigbee/application.py  _handle_frame calls deserialize() which sometimes returns "None". This results in a TypeError which is not catched.
(TypeError: cannot unpack non-iterable NoneType object)

dmulcahey, 2019-07-14:
Linking original thread from HA: https://github.com/home-assistant/home-assistant/issues/25136#issuecomment-511207100

Adminiuga, 2019-07-14:
enable debug logs for `bellows` and `zigpy` and post log here. Without the logs it is hard to say what is going on. Especially with xiaomi which is known to send wrongly encoded data.

martin3000, 2019-07-14:
I just did this:
```
  def _handle_frame(self, message_type, aps_frame, lqi, rssi, sender, binding_index, address_index, message):
        try:
            device = self.get_device(nwk=sender)
        except KeyError:
            LOGGER.debug("No such device %s", sender)
            return

        device.radio_details(lqi, rssi)
        try:
            tsn, command_id, is_reply, args = self.deserialize(device, aps_frame.sourceEndpoint, aps_frame.clusterId, message)
        except TypeError as e:
==>*            LOGGER.error("TypeError parsing message (%s) on cluster %d, because %s", binascii.hexlify(message), aps_frame.clusterId, e) *
            return
        except ValueError as e:
            LOGGER.error("Failed to parse message (%s) on cluster %d, because %s", binascii.hexlify(message), aps_frame.clusterId, e)
            return

        if is_reply:
            self._handle_reply(device, aps_frame, tsn, command_id, args)
        else:
            self.handle_message(device, False, aps_frame.profileId, aps_frame.clusterId, aps_frame.sourceEndpoint, aps_frame.destinationEndpoint, tsn, command_id, args)

```

Adminiuga, 2019-07-14:
Can you provide the message data logged by `LOGGER.error("TypeError parsing message (%s) on cluster %d, because %s", binascii.hexlify(message), aps_frame.clusterId, e)` in your environment?

martin3000, 2019-07-14:
TypeError parsing message (b'18000a050042126c756d692e766962726174696f6e2e61713101002008') on cluster 0

TypeError parsing message (b'1c5f11010a01ff422e01213b0b0328210421a8010521380006240100000000082108030a21dc6498212800992100009a250b00f9ff9c04') on cluster 0

TypeError parsing message (b'18030a55002102000305213f00') on cluster 257

I think this is from the aqara sensor.

Adminiuga, 2019-07-14:
the 1st one is surprising, on the 1st glance it should deserialize correctly
`1800` -- frame control and seq
`0a` -- attr report
`0500` -- attrid, Basic Cluster, attr id 5 - Model
`42` -- string
`12` -- string length - 18 chars
`6c756d692e766962726174696f6e2e617131` -- string payload `lumi.vibration.aq1`
`0100` -- attrid, Basic Cluster, attr id 1 -   app version
`20` -- uint8, analog
`08` -- 08 uint8 value

Adminiuga, 2019-07-14:
2nd one should also deserialize correctly, although some xiaomi devices do send that data with incorrectly encoded length:
`1c5f` -- frame control + tsn
`1101` -- manufacturer id
`0a` -- attr report
`01ff` -- 0xff01 -- xiaomi manufacturer specific cluster id AKA lifeline
`42` -- String
`2e` -- String length 46
`01213b0b0328210421a8010521380006` -- 16 bytes of string payload
`240100000000082108030a21dc649821` -- 16 bytes of string payload
`2800992100009a250b00f9ff9c04` -- 14 bytes of string payload

16 + 16 + 14 == 0x2e should deserialize correctly.

Are you positive you haven't introduced a regression somewhere else?

Adminiuga, 2019-07-14:
this does not throws any exceptions in my environment (haven't tested the manufacturer specific cluster, but looks fine too)

```
import binascii

from unittest import mock
import pytest

from bellows.zigbee.application import ControllerApplication
import zigpy.device

@pytest.fixture
def app():
    ezsp = mock.MagicMock()
    return ControllerApplication(ezsp)

@pytest.fixture
def device(app):
    dev = zigpy.device.Device(app, mock.sentinel.ieee, 0xaa55)
    ep = dev.add_endpoint(1)
    ep.add_input_cluster(0)
    ep.add_input_cluster(257)
    return dev

def test_cluster_0(device):
   data = binascii.unhexlify(b'18000a050042126c756d692e766962726174696f6e2e61713101002008')
   cluster = 0x0000

   tsn, cmd_id, is_reply, args = device.deserialize(1,  # endpoint id hardcoded in dev fixture
                                                    cluster,  # cluster id
                                                    data)

def test_cluster_257(device):
    data = binascii.unhexlify(b'18030a55002102000305213f00')
    cluster = 257

    tsn, cmd_id, is_reply, args = device.deserialize(1,  # endpoint id hardcoded in dev fixture
                                                     cluster,  # cluster id
                                                     data)
```

[image]

Looks like the issue is somewhere in your local environment

martin3000, 2019-07-14:
I have ember chip and deserialization is different then.

Adminiuga, 2019-07-15:
Take a look on the code: 'self.deserialize()' is implemented in zigpy, doesn't matter if it's ember chip or not. In other words, received Zigbee data is chip implementation independent.

martin3000, 2019-07-15:
The question is: how do we get the error 
>cannot unpack non-iterable NoneType object

This means "None" is returned from deserialize(). How can that be?

closed 2019-07-15 by martin3000