zigpy/bellows #178: ZHA EmberStatus.DELIVERY_FAILED every 2 minutes while Hue bulbs are attached

skylyn, 2019-08-02:
Description of problem:
When multiple Philips Hue lights are connected to the Zigbee radio in Hass.io, logs are flooded with errors (see traceback below) every 2 minutes at most, frequently more than once per minute, and all zigbee devices are generally slow or completely unresponsive.

Traceback (if applicable):
```
2019-08-01 17:42:47 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/bellows/zigbee/application.py", line 365, in request
    res = await asyncio.wait_for(req.send, timeout=APS_ACK_TIMEOUT)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 442, in wait_for
    return fut.result()
zigpy.exceptions.DeliveryError: [0x405b:0:0x8006]: message send failure: EmberStatus.DELIVERY_FAILED
```
Additional information:
Turning all of the light bulbs attached directly to HA Zigbee mesh off at the power stops the log errors completely (has now been more than one hour without any log errors).
Zigbee lights and sensors did work for a while with several lights attached it started being intermittent around the time I upgraded from ~92 and added a few lights. Unfortunately, I don't have a specific time or event that caused the issue exactly. I'll try to do more investigating when I have time to see if specific bulb(s) are causing the issues or a specific number is.

Additional log excerpt
```
zigpy.exceptions.DeliveryError: [0x405b:0:0x8006]: message send failure: EmberStatus.DELIVERY_FAILED
zigpy.exceptions.DeliveryError: [0xe57a:0:0x8006]: message send failure: EmberStatus.DELIVERY_FAILED
zigpy.exceptions.DeliveryError: [0x405b:0:0x8000]: message send failure: EmberStatus.DELIVERY_FAILED
zigpy.exceptions.DeliveryError: [0x405b:0:0x8000]: message send failure: EmberStatus.DELIVERY_FAILED
zigpy.exceptions.DeliveryError: [0xe57a:0:0x8000]: message send failure: EmberStatus.DELIVERY_FAILED
zigpy.exceptions.DeliveryError: [0xe57a:0:0x8000]: message send failure: EmberStatus.DELIVERY_FAILED
zigpy.exceptions.DeliveryError: [0x5517:0:0x8006]: message send failure: EmberStatus.DELIVERY_FAILED
```

More logs:
[zigpyErrors.log](https://github.com/zigpy/bellows/files/3459476/zigpyErrors.log)
[zigpyErrors-withTime.log](https://github.com/zigpy/bellows/files/3459477/zigpyErrors-withTime.log)

Adminiuga, 2019-08-02:
are these all Philips HUE lights?
- 0x0ffd
- 0x405b
- 0x5517
- 0xe57a

I'm getting those errors only for some of my Zigbee End-Devices

Adminiuga, 2019-08-02:
for a single HUE light i have I only get this error twice a day.

skylyn, 2019-08-02:
I believe those are the 4 Hue bulbs I had connected to HA directly.

After I had these problems I moved them back to the Hue hub, and only left one bulb connect to HA directly as a repeater, so I'm not able to verify exactly. However, when I turned the bulbs off at the power source, all of these messages stopped, and when I moved them back to the Hue hub and turned them on, the messages did not come back.

Adminiuga, 2019-08-02:
Can you control the lights, even though you get this error? These errors are resulting from the bulbs sending ZDO `Match_Desc_Req` looking for the over the air update server. Zigpy replies to those, but for whatever reason some devices do not acknowledge the reception. When you turn off (physically) the bulb, zigpy does not receives those requests anymore, so there are no replies to requests and therefore no failures.

skylyn, 2019-08-02:
Controlling the bulbs (as well as receiving any updates from other Zigbee devices) was very buggy while in this state. 
Bulbs would show off in HA but in fact be on. In this state trying to toggle on then off could result in any one of:
- successfully turned off
- was ignored (with HA ending in off state, but bulb ending in on state)
- bulb flashed multiple times on and off (HA also toggling back and forth, or just static)

In short, there was clearly some communication happening, but actually attempting to control the devices in any meaningful way was an exercise in frustration.

Adminiuga, 2019-08-02:
ATM HUE bulbs are not sending attribute update, so there could be state discrepancy if the bulb was turned off/on. 
When having only one bulb joined to HUSBZB, there's essentially no mesh, so that bulb needs to be relatively close to HUSBZ for reliable control. 

Although, in your log with timestamps, you do get more than normal DeliveryErrors. Is HUSBZB-1 close to any other 2.4GHz sources? Xbox, wifi routers, microwave ?

skylyn, 2019-08-02:
Yes, it's very close to a wifi router.

Note that although there is only one bulb, there are 7 other sensors (motion and contact sensors) that are connected and are working now with only one bulb, but were having problems with multiple bulbs connected.

skylyn, 2019-08-02:
I originally added the first Hue bulb to act as a repeater to reach some Iris contact sensors on the other side of the house, as they were very intermittent in updating the controller. This worked wonderfully - the contact sensor updates consistently reach the controller without issue.

When I had these problems, I initially removed all Hue lights from HA and paired them back to the Hue hub, but I quickly had the same problem - contact sensors wouldn't update - prompting me to add the one hue bulb back.

Adminiuga, 2020-08-20:
Closing on no activity.

closed 2020-08-20 by Adminiuga