zigpy/bellows #179: NCP entered failed state.

causefx, 2019-08-09:
For some odd reason my bellows "crashes" every few days.  I will have this message in my logs:
```
2019-08-09 02:20:01 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
``` 
Then absolutely nothing after that as HA is no longer responsive.

I'm not sure where to troubleshoot as there is nothing beside that in the logs.

Throughout the day I will have a few of these:
```
2019-08-09 07:32:26 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/bellows/zigbee/application.py", line 365, in request
    res = await asyncio.wait_for(req.send, timeout=APS_ACK_TIMEOUT)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 442, in wait_for
    return fut.result()
zigpy.exceptions.DeliveryError: [0xbde4:0:0x8006]: message send failure: EmberStatus.DELIVERY_FAILED
```
Thanks!

Adminiuga, 2019-08-09:
What version?

causefx, 2019-08-10:
```
bellows-homeassistant==0.9.0
```
looks like 0.9.0

perosb, 2019-12-07:
In my case I see this daily.
Not sure if the cause is the same as yours.
The devices in my case are mapped into the docker container. There is nothing else than hass in docker that _should_ be using the device.

```2019-12-07 09:16:48 ERROR (bellows.thread_0) [bellows.uart] Lost serial connection: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2019-12-07 09:16:48 DEBUG (MainThread) [bellows.ezsp] /dev/ttyUSB-zha connection lost unexpectedly: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2019-12-07 09:16:48 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2019-12-07 09:16:48 DEBUG (MainThread) [bellows.ezsp] Reconnecting /dev/ttyUSB-zha serial port on 57600 bauds
2019-12-07 09:16:54 DEBUG (MainThread) [bellows.ezsp] Reconnecting /dev/ttyUSB-zha serial port on 57600 bauds
2019-12-07 09:16:54 DEBUG (MainThread) [bellows.ezsp] Resetting EZSP
2019-12-07 09:16:54 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved```

perosb, 2019-12-07:
I believe I found the root cause for my issue.
When issuing too many commands, it need more Voltage.
```
Dec 07 12:37:15 arrakis kernel: Under-voltage detected! (0x00050005)
Dec 07 12:37:18 arrakis kernel: cp210x ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
Dec 07 12:37:18 arrakis kernel: cp210x ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
Dec 07 12:37:18 arrakis kernel: cp210x ttyUSB0: usb_serial_generic_write_bulk_callback - urb stopped: -32
Dec 07 12:37:18 arrakis kernel: cp210x ttyUSB0: usb_serial_generic_write_bulk_callback - urb stopped: -32
Dec 07 12:37:18 arrakis kernel: usb 1-1.4: USB disconnect, device number 120
Dec 07 12:37:18 arrakis kernel: cp210x ttyUSB0: failed set request 0x7 status: -19
Dec 07 12:37:18 arrakis kernel: cp210x ttyUSB0: failed set request 0x12 status: -19
Dec 07 12:37:18 arrakis kernel: cp210x ttyUSB0: failed set request 0x0 status: -19
Dec 07 12:37:18 arrakis kernel: cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
Dec 07 12:37:18 arrakis kernel: cp210x 1-1.4:1.0: device disconnected
Dec 07 12:37:19 arrakis kernel: usb 1-1.4: new full-speed USB device number 121 using dwc_otg
Dec 07 12:37:19 arrakis kernel: usb 1-1.4: New USB device found, idVendor=10c4, idProduct=8293, bcdDevice= 1.00
Dec 07 12:37:19 arrakis kernel: usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Dec 07 12:37:19 arrakis kernel: usb 1-1.4: Product: Telegesis USB Device
Dec 07 12:37:19 arrakis kernel: usb 1-1.4: Manufacturer: Silicon Labs
Dec 07 12:37:19 arrakis kernel: usb 1-1.4: SerialNumber: 04000B22
Dec 07 12:37:19 arrakis kernel: cp210x 1-1.4:1.0: cp210x converter detected
Dec 07 12:37:19 arrakis kernel: usb 1-1.4: cp210x converter now attached to ttyUSB0```

Probably need a new power adapter.

Adminiuga, 2019-12-07:
Try a different power adapter. Sometimes they go bad. Try a different USB port as well

hayatae, 2020-04-12:
I just ran into a similar problem today. Seriously doubt it's a power supply issue in my case since I have home assistant running on a full desktop PC. It looks like connection to the physical device was lost and never recovered?

```
2020-04-12 08:51:04 ERROR (bellows.thread_0) [bellows.uart] Lost serial connection: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2020-04-12 08:51:04 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
```

Even though a restart was requested, the Application Controller remains not running even hours later. I'm going to try switching to a different USB port and see if there is any improvement.

meremortals70, 2020-06-28:
I am also receiving the same result when I try and change the zigbee channel

Adminiuga, 2020-06-28:
If it lost serial connection, then restart won't succeed until connection is re-established. It may never happen if the connection is not established. You'll have to enable debug logging to get more insight

closed 2020-07-28 by Adminiuga

CanDoAnything, 2020-10-30:
I've started getting these odd bellows errors recently as described above and I haven't had any luck in figuring it out. The only change I've made recently is adding a very low power case fan. I've since removed it and the problem persists. 

My setup: I'm running HA in docker on an RPi4 on raspbian. I'm using the  [GoControl CECOMINOD016164 HUSBZB-1 USB Hub](https://www.amazon.com/GoControl-CECOMINOD016164-HUSBZB-1-USB-Hub/dp/B01GJ826F8) I'm running from a USB hard drive [(this usb3 128gb)](https://www.amazon.com/Monster-Digital-Overdrive-External-SSDOM-0128-T/dp/B00PRJ36I2)

Specific errors and odd log entries:
- [bellows.zigbee.application] Watchdog heartbeat timeout:
- [bellows.uart] Reset future is None
- [bellows.uart] Lost serial connection: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
- [bellows.ezsp] NCP entered failed state. Requesting APP controller restart

Things I've tried:
1. Restarted the rpi
1.  Changed the zigbee stick to a different USB port
1. Tried a different power adapter for the rpi
1. Deleted zigbee.db and let HA recreate a fresh db
1. Changed the volume / working directory of my HA entry in my docker-compose.yml from `/home/pi/homeassistant:/config` to `/home/pi/homeassistant2:/config` (basically HA was treated as a fresh install)

Because of 4 and 5 above and the "lost serial connection" error, it makes me believe it's a hardware or power issue. Also, I've come across [this article](https://community.octoprint.org/t/offline-error-serialexception-device-reports-readiness-to-read-but-returned-no-data-device-disconnected-or-multiple-access-on-port-comm-py2793/13667/16) , which leads me to the same: a hardware or power issue. I'm thinking my next step is to try a powered USB hub? 

suggestions appreciated!! 😁

Adminiuga, 2020-10-30:
check that `/config/home-assistant_v2.db` is under 300MB

CanDoAnything, 2020-10-31:
@Adminiuga  - I'm using a MySQL instance on another machine for the main recorder/db

Vlad-Star, 2020-11-13:
Same problem here. Nortek stick, was getting NCP errors. Upgraded NCP to version 6.6.3, still getting the same errors. Situation became much worse with HA upgrade from 0.114 to 0.117.4 (heavy delays, constantly missed automations etc.). 
home-assistant_v2.db file is 392Mb.
```
2020-11-13 07:56:36 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:36 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:36 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:37 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:37 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:37 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:37 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:38 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:38 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:38 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:38 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:38 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:38 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:38 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:38 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:38 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:38 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:38 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-13 07:56:59 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.2nd_floor_salt_lamp_power is taking over 10 seconds
2020-11-13 07:56:59 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.master_bedroom_salt_lamp_power is taking over 10 seconds
2020-11-13 07:56:59 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.living_room_salt_lamp_power is taking over 10 seconds
2020-11-13 07:56:59 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.front_door_entrance_salt_lamp_power is taking over 10 seconds
2020-11-13 07:57:20 WARNING (MainThread) [homeassistant.components.sensor] Updating zha sensor took longer than the scheduled update interval 0:00:30
2020-11-13 07:57:50 WARNING (MainThread) [homeassistant.components.sensor] Updating zha sensor took longer than the scheduled update interval 0:00:30
```

CanDoAnything, 2020-11-14:
I believe the issue is a failing SD card.Run the raspberry pi health check. It's under accessories> raspberry pi diagnostics. My write speeds were absolutely terrible. 

I bought a new SD card and did a fresh install of everything and I'm working like normal. 

After some basic researching I ended up on a kingston card. https://www.amazon.com/dp/B0858B1HBT

Vlad-Star, 2020-11-14:
@CanDoAnything Thank you for the suggestion, though I doubt this is the case (see below, SD access speed is 44MB/s). 

I am running Hassio (not a docker container under Raspbian) on I have Samsung EVO 128Mb. I didn't have such issues with the initial installation HassOS 4.12 image (latest as of 08/12/2020, I believe with HA 0.111) but upgrade to HassOS 4.15/HA 0.114 made the problem "visible" (once in a while) and upgrade to HassOS 4.15/HA 0.117 made it even worse (50% of the time motion sensor doesn't trigger the basement stairs light). Surely I added few integrations, but they didn't affect functionality at the time I added them.
```
# df -h /mnt/data
Filesystem                Size      Used Available Use% Mounted on
/dev/mmcblk0p8           58.0G      8.0G     47.0G  15% /mnt/data
# hdparm -tT /dev/mmcblk0p8

/dev/mmcblk0p8:
Timing buffer-cache reads:  1020 MB in 0.50 seconds = 2049941 kB/s
Timing buffered disk reads:  130 MB in 3.01 seconds = 44188 kB/s
#

```

Vlad-Star, 2020-11-14:
I'll be happy to provide whatever logs are needed to get the problem solved - just let me know what exactly to do.

Adminiuga, 2020-11-14:
Do you have any add-ons which are disk IO heavy?

CanDoAnything, 2020-11-15:
> @CanDoAnything Thank you for the suggestion, though I doubt this is the case (see below, SD access speed is 44MB/s).

Isn't that speed sequential though? My understanding is that SD cards need good _random_ read and write speeds to perform adequately on a PI. The card I replaced was a 32gb Samsung EVO Select -- when I ran the test on that card test its sequential speeds were fine, but the random write/read IOPS were failing. 
https://www.raspberrypi.org/blog/sd-card-speed-test/ 

Depending on how much is being written to that card ~6 months might be enough to start to wear it down and cause speed degradation. Especially if you have the sqlite db saving on the card. @Adminiuga , do you agree?

In addition to that Kingson sd card; I also ordered a Sandisk extreme (the gold /red one) as i wanted to compare them since they both were rated highly. My quick bs test has the Kingston performs much better: 

https://www.amazon.com/dp/B0858B1HBT
```
Raspberry Pi Diagnostics - version 0.5
Sat Nov 14 10:47:00 2020

Test : SD Card Speed Test
Run 1
prepare-file;0;0;32768;64
seq-write;0;0;32866;64
rand-4k-write;0;0;5707;1426
rand-4k-read;12013;3003;0;0
Sequential write speed 32866 KB/sec (target 10000) - PASS
Random write speed 1426 IOPS (target 500) - PASS
Random read speed 3003 IOPS (target 1500) - PASS
Test PASS
```

https://www.amazon.com/gp/product/B07FCMBLV6
```
Raspberry Pi Diagnostics - version 0.5
Sun Nov 15 03:57:21 2020

Test : SD Card Speed Test
Run 1
prepare-file;0;0;19974;39
seq-write;0;0;19168;37
rand-4k-write;0;0;3481;870
rand-4k-read;8379;2094;0;0
Sequential write speed 19168 KB/sec (target 10000) - PASS
Random write speed 870 IOPS (target 500) - PASS
Random read speed 2094 IOPS (target 1500) - PASS
Test PASS
```

Vlad-Star, 2020-11-15:
@Adminiuga I don't think so. 

Here is a list of add-ons that I have:
- DuckDNS
- Mosquito Broker
- Samba Share
- Terminal & SSH
- Check Home Assistant config (disabled)

Integrations:
- Z-Wave
- Zigbee (ZHA)
- Ecobee
- Logitech
- MQTT Broker
- Mobile App
- Meteorologisk
- Google Cast

Plus few non-standard integrations:
- litter_robot
- greeneye_monitor
- wyzesense

GreenEye monitor may be periodically writing sensor's data (load, temperatures etc.), back to the SD card but shouldn't cause any problems, it should be few kilobytes per minute. I can disable all 3 of them if needed.

I am also attaching my full home-assistant log - according to it, something bad is happening to bellows.zigbee.application and EZSP process.
[20201115-home-assistant-log.zip](https://github.com/zigpy/bellows/files/5542607/20201115-home-assistant-log.zip)

Vlad-Star, 2020-11-15:
@CanDoAnything thank you for the link! I've ordered Kingston, it should arrive tomorrow. I will clone my existing card to it and let you know.

Adminiuga, 2020-11-15:
Yeah, random writes are more important, than sequential.  Although I had evo 32gb cards running with full debug for quite a while before it gave up.
In one specific case, it was the bad power supply causing the issue.

Vlad-Star, 2020-11-20:
@CanDoAnything @Adminiuga Thank you folks for helping me out, I really appreciate it! 

It turned out that Kingston 64G SD card had fewer blocks compared to Samsung EVO 64G, so I couldn't clone it directly. Instead, I've created a snapshot, downloaded and installed fresh HA 0.117 to Kingston SD card and then restored configuration from the snapshot. It worked like a charm and no longer reports those NCP issues. Not sure what exactly was wrong, but the issue is now resolved. 

In addition, to put a relief on SD card, I moved [recorder](https://www.home-assistant.io/integrations/recorder/) from sqlite (homeassistant_v2.db) to mysql (which I run on the adjacent server anyway).

salvq, 2021-10-02:
I suspect this is related to Wi-Fi <-> Zigbee controllers, zigbee requires stable connection, see below

**Warning about Zigbee to WiFi bridges**
zigpy requires a robust connection between the zigpy radio library and the serial port of the Zigbee radio. With solutions such as ITEAD Sonoff ZBBridge or a Ser2Net serial proxy connection over a WiFi network it is expected to see  `NCP entered failed state. Requesting APP controller restart` in the logs. This is a normal part of the operation and indicates there was a drop in communication which caused packet loss to occurred between the zigpy radio library and the Zigbee radio. The use of serial network proxies/bridges/servers over WiFi is therefore not recommended when wanting a stable Zigbee environment with zigpy.

[Github bellows](https://github.com/zigpy/bellows)

FARKIr, 2022-11-17:
> skontrolujte, či `/config/home-assistant_v2.db`je menej ako 300 MB

https://prnt.sc/501qtpxt2pqk