zigpy/bellows #182: Which Channel for bellows dump?

JimmyTheNerd, 2019-08-31:
Hello!

`bellows dump` requires the `--channel` argument, but I don't know where I could get the information from.

I'd like to log all the communication of my ZigBee network and tried all channels from 11 to 26, without any success.

Could someone please give me a hint?

Thank you!

Adminiuga, 2019-08-31:
By default bellows forms network on channel 15

Adminiuga, 2019-08-31:
Keep in mind you cannot run coordinator and dump network at the same time.
You would need a second EZSP radio or you could get a CC2531 and flash it
with zboss sniffer firmware.

JimmyTheNerd, 2019-08-31:
Thanks @Adminiuga, _something_ (binary) is now written into the dump file.

Is there a way of logging it readable for humans? :-)

Adminiuga, 2019-08-31:
Change extension to .pcap and open in Wireshark. Google on how to obtain
your network key so you could decrypt aps payload.

JimmyTheNerd, 2019-09-01:
@Adminiuga thank you. I loaded the `*.pcap` file in Wireshark and it's detecting the different frames, but I'm not able to decrypt it yet after following a few tutorials, including [this Video](https://www.youtube.com/watch?v=4vG4CVNAm_A):

The _protocol_ of all frames is detected as **IEEE 802.15.4** and not as ZigBee. It's _info value_ is _Beacon Request_. **Is that normal?**

I added the _Trust Center Link Key_ `5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39`, which I understood is equally for everyone. 

Afterwards I wanted to get the _Transport Key_ (Video at 1:24), but I can't get that information from any frame.

Thanks again.

Adminiuga, 2019-09-01:
802.15.4 is the radio layer and Zigbee is an upper layer on too of it.
After you added TC Lin key, did you successfuly join a device to get the transport key? The network key is only sent when a device joins the network, that's why you need two EZSP radios, one to act normally as a trust center and second one to do capture. Afaik it is not possible to read network key from EZSP, but you can provide your own key.

JimmyTheNerd, 2019-09-01:
> 802.15.4 is the radio layer and Zigbee is an upper layer on too of it.
Thanks @Adminiuga 

> After you added TC Lin key, did you successfuly join a device to get the transport key?
Not yet. It looks like I can't run `bellows dump -c 15 -w output.pcap` and `bellows permit` at the same time, which means I can't log the traffic during the joining.
Both commands run just fine when called separately.

Adminiuga, 2019-09-01:
Correct, and hence my previous remark that you would need two EZSP radios.
One to run coordinator and second one to dump traffic.

JimmyTheNerd, 2019-09-02:
I finally got what you meant.

Thank you very much @Adminiuga.

closed 2019-09-02 by JimmyTheNerd