zigpy/bellows #183: Feature Request: Add network support for usb devices.

rbflurry, 2019-09-07:
I am currently trying to utilize a HUSBZB-1 connected remotely on another device in my HASSIO deployment.

It involves using ser2net to make the serial device available over the network.

```
##Zwave
3333:raw:0:/dev/ttyUSB0:115200 8DATABITS NONE 1STOPBIT
##Zigbee
3334:raw:0:/dev/ttyUSB1:57600 8DATABITS NONE 1STOPBIT
```

Typically using socat you can make a "virtual" serial device to utilize this connection 

`pty,link=/dev/ttyUSB1,raw,user=0,group=0,mode=777 tcp:IP_address:3334`

However currently there is no way to get socat running in HASSIO.

I saw that network support was added to RFXtrx (https://github.com/Danielhiversen/pyRFXtrx/pull/85)

Could the same thing be added to this library and the zwave libraries?

For reference I am using the HUSBZB-1 with https://www.gl-inet.com/products/gl-ar150/

Which has the potential to be a wifi/ethernet usb or POE powered zwave/zigbee hub. ( a wink replacement for me)

Hedda, 2020-02-17:
@rbflurry FYI, copied your text plus idea and posted same request to zigpy @ https://github.com/zigpy/zigpy/issues/297

rbflurry, 2020-02-18:
Thank you. I did end up getting this running with SOCAT on the Ha docker image but it is ugly.

Adminiuga, 2020-02-18:
I wouldn't hold my breath over this happening in near future. There are two and half developers working on ZHA integration and radio libraries and frankly it is already taking more than I anticipated of the personal time. 
Also, none of the current libraries (but bellows) handle serial port disconnects/reconnects, which are more likely to happen if using serial over IP. And there could be security implications properly securing access to serial port. In other words, openly exposing your serial port to network access without any access control might not be the best idea.

I personally think if we go through the effort, then better implement something like `zigbee2mqtt` or currently being developed zwave2mqtt. I see far more potential here.

Hedda, 2020-02-19:
@foxy82 also suggested in https://github.com/zigpy/bellows/issues/183 that one other solution could be to get socat added to HASSIO as he requested to them in https://github.com/home-assistant/docker/issues/92

As I understand it, integrated socat is what Zigbee2mqtt provides in its add-on for HASSIO, it is described from an end-users point of view here https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html

rbflurry, 2020-02-19:
I later found out that socat is included with home assistant now. 

You can see how I’m currently using it here. 
https://community.home-assistant.io/t/add-network-protocol-support-to-zigbee-and-zwave-integrations/136056/10

foxy82, 2020-02-19:
Did you try the above using Hassio on docker? Sure I can run socat on the native OS however the python library that the Hassio supervisor uses to pass through serial ports to the docker filters them out. I couldn't even trick it by putting my socat port in /dev/serial-by-id

rbflurry, 2020-02-19:
@foxy82 Yes that is what I am using now. Socat inside the hassio docker container

Hedda, 2020-03-06:
Might be better if ZHA integration for Home Assistant Core interfaces with zigpy over MQTT or similar?

zigpy with the Zigbee database could then run on a separate Raspberry Pi, similar to Zigbee2MQTT?

If that was possible when your Zigbee network would not go down if reboot Home Assistant Core?

Hedda, 2020-04-21:
Have you guys heard of the new Sonoff ZBBridge (Sonoff Zigbee Bridge) that was just announced?

See related request for zigpy-> https://github.com/zigpy/zigpy/issues/385

Sonoff ZBBridge is basically a prepackaged networked EmberZNet / EZSP device with an ESP8266..

Adminiuga, 2020-08-10:
It works with latest pyserial

closed 2020-08-10 by Adminiuga