zigpy/bellows #191: Bellows Error changing Zigbee Channel

Richdem, 2019-10-03:
I have been trying to change the Zigbee channel of my HUSBZB-1 as I have been getting interference from my neighbours new mesh WiFi. I am running the latest version of HA (99.3) on Ubunter Server & Docker. When running the following command I get an error

bellows -d /dev/ttyUSB1 form -c 19

[image]

I was able to run the following to check the current Zigbee channel

bellows -d /dev/ttyUSB1 info

It was suggested by someone over at the HA community that I post here.

Adminiuga, 2019-10-03:
So, when running from cmdline like that, you need to specify the database with -D parameter, something like
`bellows -d /dev/ttyUSB1 form -D /path/to/your/zigbee.db -c 19` or `bellows -d /dev/ttyUSB1 -D  /path/to/your/zigbee.db form -c 19`

What is I'm interested in, the network is already formed, so I'm interested to see the behavior. Would it accept it and form a new network or would it just change the channel?

Adminiuga, 2019-10-03:
there's this EZSP command
[image]

But ATM there's no interface to access it, other then modifying `bellows` code. I know I did change the channel on my test environment, but I forgot were I put it. I remember I had to issue this command before the network was started.

so insert before https://github.com/zigpy/bellows/blob/7d2dbc98a2976a632d65d881d27cdf0fd022ff94/bellows/zigbee/application.py#L113 the following
```
v = await e.setRadioChannel(19)
LOGGER.debug("Setting channel result: %s", v)
```

Adminiuga, 2019-10-03:
and restart HA. To verify if channel was changed. Stop HA and run `bellows -d /dev/serial info`

Richdem, 2019-10-03:
> So, when running from cmdline like that, you need to specify the database with -D parameter, something like
> `bellows -d /dev/ttyUSB1 form -D /path/to/your/zigbee.db -c 19` or `bellows -d /dev/ttyUSB1 -D /path/to/your/zigbee.db form -c 19`
> 
> Alternatively, create directory `~/.config/bellows` and inside that folder soft link your `zigbee.db` to `app.db`. In other words, CLI bellows is looking by default `~/.config/bellows/app.db` file for its database.
> 
> What is I'm interested in, the network is already formed, so I'm interested to see the behavior. Would it accept it and form a new network or would it just change the channel?

Thank you for the reply,

I have some time over the weekend so I will try again. It is my understanding that once the channel has been changed all existing Zigbee devices would need to be re added.

Adminiuga, 2019-10-03:
To my surprize, when I've changed the channel using EZSP setRadioChannel, most devices reconnected without any issues, but xiaomi.

Richdem, 2019-10-03:
That's good to know,

The only thing is working in cmd line in Linux is not my strong point. I can see my zigbee.db across the network as I have that folder shared. How do I find the correct path when using putty to SSH  in?

Richdem, 2019-10-06:
How did you disable the ZHA component to allow you to run the commands?

closed 2019-10-08 by Richdem

MattWestb, 2020-07-20:
I think that if you changing channel to one default lighting link channel (11.15. 20 and 25 if i remember right) the lights its searching and finding the coordinator if they its LL or Z30 lights. For HA i'm not sure but Xiaomi 99.9% not being any standard and not moving with. But interesting that it was working for Adminiuga