zigpy/bellows #193: Unable to form zigbee network with EFR32 running EmberZNet NCP application

zabubu, 2019-10-29:
OS:	Ubuntu 18.04.3 LTS
Bellows Version: 0.10.0

I have a EFR32MG12 running the EmberZNet application provided by Silicon Labs.
Following are bellows command outputs:

`$ bellows -v DEBUG -d /dev/ttyACM1 info`:
```
debug: Using selector: EpollSelector
debug: Using selector: EpollSelector
debug: Connected. Resetting.
debug: Resetting EZSP
debug: Resetting ASH
debug: Sending: b'1ac038bc7e'
debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
debug: Send command version: (4,)
debug: Sending: b'004221a850ed2c7e'
debug: Data frame: b'0142a1a8532825d4f4497e'
debug: Sending: b'8160597e'
debug: Application frame 0 (version) received: b'07023066'
debug: Switching to eszp version 7
debug: Send command version: (7,)
debug: Sending: b'7d31432157542a1205877e'
debug: Data frame: b'1243a157542a12b069f2a5a17e'
debug: Sending: b'82503a7e'
debug: Application frame 0 (version) received: b'07023066'
debug: Configuring...
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_STACK_PROFILE: 12>, 2)
debug: Sending: b'22402157547919b059c46e7e'
debug: Data frame: b'2340a157547915fbd27e'
debug: Sending: b'83401b7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_SECURITY_LEVEL: 13>, 5)
debug: Sending: b'3341215754797d38b75995c77e'
debug: Data frame: b'3441a157547915904d7e'
debug: Sending: b'8430fc7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_SUPPORTED_NETWORKS: 45>, 1)
debug: Sending: b'44462157547938b35902ad7e'
debug: Data frame: b'4546a157547915640c7e'
debug: Sending: b'8520dd7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_PACKET_BUFFER_COUNT: 1>, 255)
debug: Sending: b'554721575479144d593eca7e'
debug: Data frame: b'5647a157547915ce557e'
debug: Sending: b'8610be7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command networkInit: ()
debug: Sending: b'66442157543dfba67e'
debug: Data frame: b'6744a157543d86938b7e'
debug: Sending: b'87009f7e'
debug: Application frame 23 (networkInit) received: b'93'
debug: Send command getEui64: ()
debug: Sending: b'77452157540c28a17e'
debug: Data frame: b'7045a157540ce6ef226ab5ba57c52bd27e'
debug: Sending: b'8070787e'
debug: Application frame 38 (getEui64) received: b'f35d7bfeff9ffd90'
[90:fd:9f:ff:fe:7b:5d:f3]
debug: Send command getNodeId: ()
debug: Sending: b'004a2157540dd0a47e'
debug: Data frame: b'014aa157540deb4d269b7e'
debug: Sending: b'8160597e'
debug: Application frame 39 (getNodeId) received: b'feff'
[0xfffe]
debug: Send command networkState: ()
debug: Sending: b'7d314b21575432e26d7e'
debug: Data frame: b'124ba15754321508aa7e'
debug: Sending: b'82503a7e'
debug: Application frame 24 (networkState) received: b'00'
[<EmberNetworkStatus.NO_NETWORK: 0>]
[... 117 lines cut ...]
  File "/home/abbas/.virtualenvs/bellows/lib/python3.6/site-packages/bellows/cli/application.py", line 32, in form
    return util.app(inner, app_startup=False)(ctx)
  File "/home/abbas/.virtualenvs/bellows/lib/python3.6/site-packages/bellows/cli/util.py", line 71, in inner
    loop.run_until_complete(async_inner(*args, **kwargs))
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "/home/abbas/.virtualenvs/bellows/lib/python3.6/site-packages/bellows/cli/util.py", line 58, in async_inner
    await f(ctx, *args, **kwargs)
  File "/home/abbas/.virtualenvs/bellows/lib/python3.6/site-packages/bellows/cli/application.py", line 29, in inner
    await app.initialize()
  File "/home/abbas/.virtualenvs/bellows/lib/python3.6/site-packages/bellows/zigbee/application.py", line 79, in initialize
    await self._cfg(c.CONFIG_END_DEVICE_POLL_TIMEOUT_SHIFT, 8)
  File "/home/abbas/.virtualenvs/bellows/lib/python3.6/site-packages/bellows/zigbee/application.py", line 185, in _cfg
    assert v[0] == t.EmberStatus.SUCCESS  # TODO: Better check
AssertionError
^CError in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3.6/concurrent/futures/thread.py", line 40, in _python_exit
    t.join()
  File "/usr/lib/python3.6/threading.py", line 1056, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.6/threading.py", line 1072, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
```

Adminiuga, 2019-10-29:
Apparently protocol version 7 does not have `CONFIG_END_DEVICE_POLL_TIMEOUT_SHIFT` parameter.
Any chance you could get protocol version 6?

Adminiuga, 2019-10-30:
Do you know in which version they got rid of `CONFIG_END_DEVICE_POLL_TIMEOUT_SHIFT` configuration id?

Adminiuga, 2019-10-30:
I think EmberZnet version != protocol version. From what I could find, EmberZnet 6.4.0 is the protocol version 7 and there's https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf which I assume is protocol 7.
I suspect in protocol version 6 they've changed the way the end device poll timeout is configured and essentially you just need to write 8 into `CONFIG_END_DEVICE_POLL_TIMEOUT` which is `2^8` minutes. Elelab rpi shield seems to be running version 6, but it wasn't complaining about timeout shift parameter. But I don't have Silab support access, so can't confirm.

Adminiuga, 2019-10-30:
I guess my real question is: From which protocol version we should be writing 8 into 'CONFIG_END_DEVICE_POLL_TIMEOUT`

zabubu, 2019-10-30:
I think it's safe to say that 8 should be written into `CONFIG_END_DEVICE_POLL_TIMEOUT` from v7 onward since the latest version of v7 does not have `CONFIG_END_DEVICE_POLL_TIMEOUT_SHIFT`

zabubu, 2019-10-30:
I made a very hackish change to prevent bellows from sending `CONFIG_END_DEVICE_POLL_TIMEOUT_SHIFT` but now I'm getting snagged with another error:

```
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_END_DEVICE_POLL_TIMEOUT: 19>, 8)
debug: Sending: b'664c2157547906ba5996767e'
debug: Data frame: b'674ca157547915ec1b7e'
debug: Sending: b'87009f7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_MULTICAST_TABLE_SIZE: 6>, 16)
debug: Sending: b'774d215754797d33a2594b317e'
debug: Data frame: b'704da15754791587847e'
debug: Sending: b'8070787e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_PACKET_BUFFER_COUNT: 1>, 255)
debug: Sending: b'005221575479144d59302b7e'
debug: Data frame: b'0152a15754792002f57e'
debug: Sending: b'8160597e'
debug: Application frame 83 (setConfigurationValue) received: b'35'
Traceback (most recent call last):
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/bin/bellows", line 11, in <module>
    load_entry_point('bellows-homeassistant==0.11.0.dev0', 'console_scripts', 'bellows')()
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/Click-7.0-py3.6.egg/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/Click-7.0-py3.6.egg/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/Click-7.0-py3.6.egg/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/Click-7.0-py3.6.egg/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/Click-7.0-py3.6.egg/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/Click-7.0-py3.6.egg/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/bellows_homeassistant-0.11.0.dev0-py3.6.egg/bellows/cli/application.py", line 32, in form
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/bellows_homeassistant-0.11.0.dev0-py3.6.egg/bellows/cli/util.py", line 71, in inner
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/bellows_homeassistant-0.11.0.dev0-py3.6.egg/bellows/cli/util.py", line 58, in async_inner
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/bellows_homeassistant-0.11.0.dev0-py3.6.egg/bellows/cli/application.py", line 29, in inner
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/bellows_homeassistant-0.11.0.dev0-py3.6.egg/bellows/zigbee/application.py", line 83, in initialize
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/bellows_homeassistant-0.11.0.dev0-py3.6.egg/bellows/zigbee/application.py", line 186, in _cfg
AssertionError
```

Adminiuga, 2019-10-30:
`debug: Application frame 83 (setConfigurationValue) received: b'35'` means status to `setConfigurationValue` is `OUT_OF_MEMORY`.  Which is strange, cause it is being told to allocate remaining memory. per docs

> The number of packet buffers available to the stack. When set to the special value 0xFF, the NCP will allocate all remaining configuration RAM towards packet buffers, such that the resulting count will be the largest whole number of packet buffers that can fit into the available memory

Adminiuga, 2019-10-31:
can you add these line, prior setting the buffer count to `0xff` I'd like to see what is your current setting
```
        r = await e.getConfigurationValue(c.CONFIG_PACKET_BUFFER_COUNT)
        LOGGER.info("Current buffers: %s", r)
```

zabubu, 2019-10-31:
> Do you have access to SDK? any chance you could build an image for EM3581 so I could try updating my stick to be able to test it?

I've attached the images for EM381 (both hardware and software flow) that are available in EmberZNet 6.6.3

> can you add these line, prior setting the buffer count to `0xff` I'd like to see what is your current setting

Find the output bellow:

```
debug: Send command getConfigurationValue: (<EzspConfigId.CONFIG_PACKET_BUFFER_COUNT: 1>,)
debug: Sending: b'00522157547814dda27e'
debug: Data frame: b'0152a157547815f2593f467e'
debug: Sending: b'8160597e'
debug: Application frame 82 (getConfigurationValue) received: b'004000'
Current buffers: [<EzspStatus.SUCCESS: 0>, 64]
```

[em3581-ncp-uart.zip](https://github.com/zigpy/bellows/files/3795701/em3581-ncp-uart.zip)

Adminiuga, 2019-10-31:
`64` is what I get on both my HUSBZB-1 and elelabs shield. 
Speacial value `0xff` was supposed to allocate the remaining memory. After setting it to use all remaining memory, the buffer count reads `117` on elelabs stick. Modify the code not to set the buffer count or play with a value between 64 - 117 and see if you can find the one which work. Would be interesting to see if you actually need to decrease the number of buffers.

zabubu, 2019-10-31:
Yeah, it's weird that `0xFF` returns an error code. I'll also file a ticket with silabs to get more input

Adminiuga, 2019-10-31:
Can you try [this branch](https://github.com/Adminiuga/bellows/commits/ac/buffer-allocation) and post the output?
It essentially reduces the number of buffers in the beginning, configures the system and then expands the buffers.

PS: thanks you for attachment. Gonna delete the files

zabubu, 2019-11-01:
It also responded with the `OUT_OF_MEMORY` error code:

```
debug: Using selector: EpollSelector
debug: Using selector: EpollSelector
debug: Loading application state from /home/abbas/.config/bellows/app.db
debug: Attribute id: 4 value: Samjin
debug: Attribute id: 5 value: water
debug: Attribute id: 4 value: ORVIBO
debug: Attribute id: 5 value: 895a2d80097f4ae2b2d40500d5e03dcc
debug: Checking quirks for None None (90:fd:9f:ff:fe:7b:5d:f3)
debug: Considering <class 'zigpy.quirks.smartthings.SmartthingsTemperatureHumiditySensor'>
debug: Fail because device_type mismatch on at least one endpoint
debug: Considering <class 'zigpy.quirks.smartthings.SmartthingsMultiPurposeSensor'>
debug: Fail because device_type mismatch on at least one endpoint
debug: Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
debug: Found custom device replacement for 90:fd:9f:ff:fe:7b:5d:f3: <class 'bellows.zigbee.application.EZSPCoordinator'>
debug: Checking quirks for Samjin water (28:6d:97:00:01:05:4a:ce)
debug: Considering <class 'zigpy.quirks.smartthings.SmartthingsTemperatureHumiditySensor'>
debug: Fail because device_type mismatch on at least one endpoint
debug: Considering <class 'zigpy.quirks.smartthings.SmartthingsMultiPurposeSensor'>
debug: Fail because input cluster mismatch on at least one endpoint
debug: Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
debug: Fail because device_type mismatch on at least one endpoint
debug: Checking quirks for ORVIBO 895a2d80097f4ae2b2d40500d5e03dcc (00:15:8d:00:01:bf:a5:5a)
debug: Considering <class 'zigpy.quirks.smartthings.SmartthingsTemperatureHumiditySensor'>
debug: Fail because device_type mismatch on at least one endpoint
debug: Considering <class 'zigpy.quirks.smartthings.SmartthingsMultiPurposeSensor'>
debug: Fail because input cluster mismatch on at least one endpoint
debug: Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
debug: Fail because device_type mismatch on at least one endpoint
debug: Attribute id: 4 value: Samjin
debug: Attribute id: 5 value: water
debug: Attribute id: 4 value: ORVIBO
debug: Attribute id: 5 value: 895a2d80097f4ae2b2d40500d5e03dcc
debug: Attribute id: 33 value: 200
debug: Resetting EZSP
debug: Initialize OTA providers
debug: Resetting ASH
debug: Sending: b'1ac038bc7e'
debug: OTA image directory '/home/abbas/.config/bellows/zigpy_ota/' does not exist
debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
debug: Send command version: (4,)
debug: Sending: b'004221a850ed2c7e'
debug: Data frame: b'0142a1a8532825d4f4497e'
debug: Sending: b'8160597e'
debug: Application frame 0 (version) received: b'07023066'
debug: Send command version: (7,)
debug: Sending: b'7d31432157542a1205877e'
debug: Data frame: b'1243a157542a12b069f2a5a17e'
debug: Sending: b'82503a7e'
debug: Application frame 0 (version) received: b'07023066'
debug: Switched to EZSP protocol version 7
EZSP Stack Type: 2, Stack Version: 26160
debug: Send command getStandaloneBootloaderVersionPlatMicroPhy: ()
debug: Sending: b'2240215754bbef5f7e'
debug: Data frame: b'2340a15754bb15ab5d8c4511b47e'
debug: Sending: b'83401b7e'
debug: Application frame 145 (getStandaloneBootloaderVersionPlatMicroPhy) received: b'001904180f'
bootloader: [6400, 4, 24, 15]
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_PACKET_BUFFER_COUNT: 1>, 16)
debug: Sending: b'33412157547914a2591c207e'
debug: Data frame: b'3441a157547915904d7e'
debug: Sending: b'8430fc7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_STACK_PROFILE: 12>, 2)
debug: Sending: b'44462157547919b059e6087e'
debug: Data frame: b'4546a157547915640c7e'
debug: Sending: b'8520dd7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
[... 84 lines cut ...]
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/Click-7.0-py3.6.egg/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/Click-7.0-py3.6.egg/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/Click-7.0-py3.6.egg/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/bellows_homeassistant-0.11.0.dev0-py3.6.egg/bellows/cli/application.py", line 32, in form
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/bellows_homeassistant-0.11.0.dev0-py3.6.egg/bellows/cli/util.py", line 71, in inner
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/bellows_homeassistant-0.11.0.dev0-py3.6.egg/bellows/cli/util.py", line 58, in async_inner
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/bellows_homeassistant-0.11.0.dev0-py3.6.egg/bellows/cli/application.py", line 29, in inner
[... cut ...]
  File "/home/abbas/Documents/netex/peripherals/zigbee/bellows/venv/lib/python3.6/site-packages/bellows_homeassistant-0.11.0.dev0-py3.6.egg/bellows/zigbee/application.py", line 199, in _cfg
AssertionError
^CError in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3.6/concurrent/futures/thread.py", line 40, in _python_exit
    t.join()
  File "/usr/lib/python3.6/threading.py", line 1056, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.6/threading.py", line 1072, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
```

Adminiuga, 2019-11-01:
Hrm, interesting. I think it might be hardware platform dependent. I've updated my EM3581 (many thanks) and it still works fine
```
debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
debug: Send command version: (4,)
debug: Sending: b'004221a850ed2c7e'
debug: Data frame: b'0142a1a8532825d4f4497e'
debug: Sending: b'8160597e'
debug: Application frame 0 (version) received: b'07023066'
debug: Send command version: (7,)
debug: Sending: b'7d31432157542a1205877e'
debug: Data frame: b'1243a157542a12b069f2a5a17e'
debug: Sending: b'82503a7e'
debug: Application frame 0 (version) received: b'07023066'
debug: Switched to EZSP protocol version 7
EZSP Stack Type: 2, Stack Version: 26160
...
debug: Application frame 82 (getConfigurationValue) received: b'001000'
Current buffers: [<EzspStatus.SUCCESS: 0>, 16]
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_PACKET_BUFFER_COUNT: 1>, 255)
debug: Sending: b'335121575479144d59217d387e'
debug: Data frame: b'3451a1575479158ac97e'
debug: Sending: b'8430fc7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command getConfigurationValue: (<EzspConfigId.CONFIG_PACKET_BUFFER_COUNT: 1>,)
debug: Sending: b'44562157547814c7297e'
debug: Data frame: b'4556a1575478154d59fd2e7e'
debug: Sending: b'8520dd7e'
debug: Application frame 82 (getConfigurationValue) received: b'00ff00'
New buffers: [<EzspStatus.SUCCESS: 0>, 255]
```

But what i'm noticing, this one shows `255` for the new buffer count, same as it was on the same hardware with protocol version 4. Elelabs shows `117` buffers, not `255`

To me it sounds like a BUG in EmberZnet for that platform, so feel free to open a ticket with Silabs.
Alternatively, just try to find a new value which it does not complain about it. I see the default number to be 64, see if you can increase it.

zabubu, 2019-11-01:
Yeah I filed a ticket with Silabs.

Strangely, this is only happening when I compile their example program for NCP UART. If I flash on the the pre-compiled NCP UART binary then all seems to work fine which indicates even more that it's something wonky on their end. 

Thanks for all the help!

Adminiuga, 2019-11-03:
Depending on what the example program is doing, the memory could be allocated different. However I still would expect the 0xff value to work

closed 2020-01-22 by zabubu