zigpy/bellows #207: Failed ZDO request during device initialization - on each startup

shortbloke, 2020-01-14:
Whenever I start/re-start my Home Assistant (0.103.6) instance (not sure how to determine what versions of zigpy is included), I'm seeing an error logged:
`ERROR (MainThread) [zigpy.device] Failed ZDO request during device initialization`

This is using a Nortek HUSBZB-1 USB Zigbee transceiver.

With debug logging enabled, the device which appears to be causing the error is `Device 0x0000 (00:0d:6f:00:0b:5d:d1:25) joined the network` 
Though I don't see that device in the zigbee database. Any suggestions on what might be going on here and how to fix/avoid the error?

**Debug Logs**
```
2020-01-14 14:49:39 DEBUG (MainThread) [bellows.zigbee.application] APS_UNICAST_MESSAGE_COUNT is set to 10
2020-01-14 14:49:39 DEBUG (MainThread) [bellows.zigbee.application] Ezsp adding endpoint: [<EzspStatus.SUCCESS: 0>]
2020-01-14 14:49:40 INFO (MainThread) [zigpy.application] Device 0x0000 (00:0d:6f:00:0b:5d:d1:25) joined the network
2020-01-14 14:49:40 DEBUG (MainThread) [bellows.zigbee.application] EZSP nwk=0x0000, IEEE=00:0d:6f:00:0b:5d:d1:25
2020-01-14 14:49:40 DEBUG (MainThread) [bellows.zigbee.application] Starting EZSP watchdog
2020-01-14 14:49:40 INFO (MainThread) [zigpy.device] [0x0000] Requesting 'Node Descriptor'
2020-01-14 14:49:40 DEBUG (MainThread) [zigpy.device] [0x0000] Extending timeout for 0x01 request
2020-01-14 14:49:40 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 00:0d:6f:00:0b:5d:d1:25/0x0000
2020-01-14 14:49:40 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=0 clusterId=2 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=211>, 255, 0, 0x0000, 255, 255, b'\x01\x00\x00']
2020-01-14 14:49:40 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] ZDO request 0x0002: [0x0000]
2020-01-14 14:49:40 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] Unsupported ZDO request:0x0002
2020-01-14 14:49:40 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=0 clusterId=32770 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=212>, 255, 0, 0x0000, 255, 255, b'\x01\x00\x00\x00\x00@\x8f\xcd\xabR\x80\x00\x00\x00\x80\x00\x00']
2020-01-14 14:49:40 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] ZDO request 0x8002: [<Status.SUCCESS: 0>, 0x0000, <Optional byte1=0 byte2=64 mac_capability_flags=143 manufacturer_code=43981 maximum_buffer_size=82 maximum_incoming_transfer_size=128 server_mask=0 maximum_outgoing_transfer_size=128 descriptor_capability_field=0>]
2020-01-14 14:49:40 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] Unsupported ZDO request:0x8002
2020-01-14 14:49:40 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0, <EmberApsFrame profileId=0 clusterId=2 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=211>, 2, <EmberStatus.SUCCESS: 0>, b'']
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xbd85](lumi.sensor_magnet.aq2): started initialization
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xbd85:ZDO](lumi.sensor_magnet.aq2): entry loaded from storage: ZhaDeviceEntry(name='LUMI lumi.sensor_magnet.aq2', ieee='00:15:8d:00:03:67:3f:64', last_seen=1579008703.6866663)
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xbd85:ZDO](lumi.sensor_magnet.aq2): channel: 'async_initialize' stage succeeded
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x1049](lumi.sensor_magnet.aq2): started initialization
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x1049:ZDO](lumi.sensor_magnet.aq2): entry loaded from storage: ZhaDeviceEntry(name='LUMI lumi.sensor_magnet.aq2', ieee='00:15:8d:00:02:75:33:5b', last_seen=1579012988.7722054)
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x1049:ZDO](lumi.sensor_magnet.aq2): channel: 'async_initialize' stage succeeded
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xbd85:1:0x0000]: initializing channel: from_cache: True
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xbd85:1:0x0000]: channel: 'async_initialize' stage succeeded
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xbd85:1:0x0001]: initializing channel: from_cache: True
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xbd85:1:0x0001]: channel: 'async_initialize' stage succeeded
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xbd85:1:0x0006]: initializing channel: from_cache: True
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xbd85:1:0x0006]: channel: 'async_initialize' stage succeeded
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xbd85:1:0x0006]: initializing channel: from_cache: True
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xbd85:1:0x0006]: channel: 'async_initialize' stage succeeded
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x1049:1:0x0000]: initializing channel: from_cache: True
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x1049:1:0x0000]: channel: 'async_initialize' stage succeeded
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x1049:1:0x0001]: initializing channel: from_cache: True
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x1049:1:0x0001]: channel: 'async_initialize' stage succeeded
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x1049:1:0x0006]: initializing channel: from_cache: True
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x1049:1:0x0006]: channel: 'async_initialize' stage succeeded
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x1049:1:0x0006]: initializing channel: from_cache: True
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x1049:1:0x0006]: channel: 'async_initialize' stage succeeded
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xbd85](lumi.sensor_magnet.aq2): power source: Battery or Unknown
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xbd85](lumi.sensor_magnet.aq2): completed initialization
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x1049](lumi.sensor_magnet.aq2): power source: Battery or Unknown
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x1049](lumi.sensor_magnet.aq2): completed initialization
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xa42f](lumi.sensor_magnet.aq2): started initialization
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xa42f:ZDO](lumi.sensor_magnet.aq2): entry loaded from storage: ZhaDeviceEntry(name='LUMI lumi.sensor_magnet.aq2', ieee='00:15:8d:00:02:53:70:bc', last_seen=1579012668.572292)
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xa42f:ZDO](lumi.sensor_magnet.aq2): channel: 'async_initialize' stage succeeded
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x5e3c](lumi.sensor_magnet.aq2): started initialization
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x5e3c:ZDO](lumi.sensor_magnet.aq2): entry loaded from storage: ZhaDeviceEntry(name='LUMI lumi.sensor_magnet.aq2', ieee='00:15:8d:00:02:75:32:10', last_seen=1579012401.3020155)
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x5e3c:ZDO](lumi.sensor_magnet.aq2): channel: 'async_initialize' stage succeeded
2020-01-14 14:49:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0xa42f:1:0x0000]: initializing channel: from_cache: True
[... cut ...]
ec:1b:bd:ff:fe:78:3a:89	62600	2
```

**Database (SELECT * FROM "endpoints")**
```
00:0d:6f:00:12:35:22:91	1	260	48879	1
00:15:8d:00:03:67:3f:64	1	260	24321	1
00:15:8d:00:02:75:33:5b	1	260	24321	1
00:15:8d:00:02:53:70:bc	1	260	24321	1
00:15:8d:00:02:75:32:10	1	260	24321	1
00:15:8d:00:03:03:3d:99	1	260	260	1
00:15:8d:00:02:37:8a:ca	1	260	24321	1
00:15:8d:00:02:37:8a:ca	2	260	24322	1
00:15:8d:00:02:37:8a:ca	3	260	24323	1
ec:1b:bd:ff:fe:78:31:0a	1	260	1026	1
ec:1b:bd:ff:fe:78:3a:89	1	260	1026	1
```

Adminiuga, 2020-01-14:
Is this the second zigbee stick you are trying?
Seems like the new one is `00:0d:6f:00:0b:5d:d1:25` and old one used to be `00:0d:6f:00:12:35:22:91` ?

If this is correct, try deleting all references to `00:0d:6f:00:12:35:22:91` from the zigbee.db
```
DELETE FROM attributes WHERE ieee='00:0d:6f:00:12:35:22:91';
DELETE FROM output_clusters WHERE ieee='00:0d:6f:00:12:35:22:91';
DELETE FROM clusters WHERE ieee='00:0d:6f:00:12:35:22:91';
DELETE FROM endpoints WHERE ieee='00:0d:6f:00:12:35:22:91';
DELETE FROM node_descriptors WHERE ieee='00:0d:6f:00:12:35:22:91';
```
and restart HA

shortbloke, 2020-01-14:
Thanks @Adminiuga 
I previously had a different ZigBee adapter in my Pi3. This install on the Pi4 has only ever had the Nortek stick, but I did copy the config and files from the Pi3, so I guess the database had the old device though it wasn't showing anywhere and I don't see any mention of the "00:0d:6f:00:12:35:22:91" in the logs either.

Deleting the entry from all the database fields appears to have fixed it.

closed 2020-01-14 by shortbloke