zigpy/bellows #215: [SUGGESTION] Rename the bellows library to zigpy-ezsp for clarity and consistency

Hedda, 2020-02-17:
Please consider renaming the bellows library and repo to "zigpy-ezsp" for clarity and consistency.

zigpy is an organization which can have many members and is upstream for multiple projects:

- https://github.com/zigpy

bellows came first before broke out zigpy and newer dependencies follow same naming-standard.

There are now already these libraries for zigpy where only bellows do not follow the standard:

- https://github.com/zigpy/zigpy/
- https://github.com/zigpy/bellows
- https://github.com/zigpy/zigpy-deconz
- https://github.com/zigpy/zigpy-xbee
- https://github.com/doudz/zigpy-zigate/
- https://github.com/sanyatuning/zigpy-cc

Yes I have OCD but others might also find this could be a logical choice that makes sense.

ZHA in Home Assistant is gaining traction and zigpy will get more and more radio libraries.

If the "bellows" library was renamed "zigpy-ezsp" all would follow the same naming-standard.

IMHO best for clarity and consistency was if bellows also follow the same naming-standard.

mickeprag, 2020-02-17:
Also, in the same track, why is the pypi library called "bellows-homeassistant" instead of just "bellows"? It does not have any homeassistant dependencies and can be reused by other projects as well (as I am).

I vote for a namechange from "bellows-homeassistant" to "zigpy-ezsp" (ar the author suggests).

Hedda, 2020-02-17:
@mickeprag As I understand from https://github.com/zigpy/zigpy/issues/190 and more I read Home Assistant community forums, the reason was that @rcloran who initially created bellows (and zigpy) went missing for a long while (I'm only guessing that it was for personal reasons) before he could give more developers write access permissions to the existing projects that he has created. Anyway, the same goes for the use of "zigpy-homeassistant" (https://pypi.org/project/zigpy-homeassistant/) instead of just "zigpy" (https://pypi.org/project/zigpy/) on pypi.

You can also reas a somewhat related discussion in https://github.com/zigpy/zigpy/issues/113

Hedda, 2020-03-30:
@rcloran any options for or against renaming bellows since you are the original creator of it & zigpy?

Hedda, 2020-03-30:
@rcloran Would you be OK if someone else in the zigpy org renamed the bellows library & its repo?

Hedda, 2020-04-08:
For reference, this suggestion is related and very similar to https://github.com/zigpy/zha-device-handlers/issues/327

Hedda, 2020-04-15:
For reference, see related https://github.com/zigpy/zigpy/pull/363 in which @Adminiuga changed  zigpy module name back to just "zigpy"

Hedda, 2020-04-28:
@Adminiuga sorry for bikeshedding but wondering how you feel about this renaming suggestion?

Apologies if my OCD ranting is not really constructive here but doesn't Parkinson's law of triviality clearly states that all bikesheds should have a corrugated galvanised iron roof and be black in colour ;)

Hedda, 2020-05-26:
@rcloran Please let us know if you are totally against renaming bellows library, this repository on GitHub, and the PyPI project. Hope that is OK that if we do not hear from you then hope that it is OK if we will assume that you are OK with renaming them to something else with "zigpy" as the prefix.

Hedda, 2020-08-10:
@Adminiuga OK to bump for comment/feedback on this suggestion now ;)

roger-, 2020-08-19:
+1 for renaming.

Always wondered what this was about (until I needed it) since the name is inconsistent with everything else.

Hedda, 2020-08-20:
@roger- Out if curiosity, what did you need it for? Troubleshooting as an end-user or developing something with it as a developer?

Mostly just thinking that it is very interesting to know of more third-party projects that use zigpy and/or bellows without HA's ZHA :)

roger-, 2020-08-20:
@Hedda 

Actually just via ZHA (troubleshooting), but I could imagine using this for a standalone project.

Hedda, 2020-09-07:
@roger- Maybe you would be willing to volunteer for this renaming task and submit a pull request if you would take up the task?

Adminiuga, 2020-09-08:
At this moment we're not considering renaming `bellows`. Yes, it's name does not match zigpy and all other lib, but that's for historical reason. Zigpy came out for `bellows`. `Bellows`  was the name picked by Russel and I can't nor really want to change it. 
Renaming it won't fix anything and proper renaming, would require too many changes -- it doesn't worth it. Sorry.

closed 2020-09-08 by Adminiuga