zigpy/bellows #219: Invalid syntax when trying to use CLI

Nixellion, 2020-03-11:
As part of my troubleshooting of not working Elelabs USB stick I downloaded bellows-homeassistant from pip3, and here's what I get when I try to run `bellows devices` or just `bellows`:

```
Traceback (most recent call last):
  File "/usr/local/bin/bellows", line 7, in <module>
    from bellows.cli.main import main
  File "/usr/local/lib/python3.5/dist-packages/bellows/cli/__init__.py", line 2, in <module>
    from . import application
  File "/usr/local/lib/python3.5/dist-packages/bellows/cli/application.py", line 10, in <module>
    import bellows.zigbee.application
  File "/usr/local/lib/python3.5/dist-packages/bellows/zigbee/application.py", line 11, in <module>
    import zigpy.application
  File "/usr/local/lib/python3.5/dist-packages/zigpy/application.py", line 7, in <module>
    import zigpy.appdb
  File "/usr/local/lib/python3.5/dist-packages/zigpy/appdb.py", line 4, in <module>
    import zigpy.device
  File "/usr/local/lib/python3.5/dist-packages/zigpy/device.py", line 8, in <module>
    import zigpy.endpoint
  File "/usr/local/lib/python3.5/dist-packages/zigpy/endpoint.py", line 8, in <module>
    import zigpy.zcl
  File "/usr/local/lib/python3.5/dist-packages/zigpy/zcl/__init__.py", line 123
    ):
```

What's the issue? Debian 9

mickeprag, 2020-03-11:
Perhaps blir because you are running python 3.5? Try 3.6+ first.

Nixellion, 2020-03-11:
Okay, so it requires python 3.7 I suppose, dist-upgrade to debian 10 solved the issue

closed 2020-03-11 by Nixellion