zigpy/bellows #242: [FYI] DEFARO Zigbee SprutStick project based on Silicon Labs EFR32

Hedda, 2020-04-17:
FYI, might be cool to follow this "DEFARO Zigbee Sprut Stick" project from @sprut ( sprut.ai ) by @sprut666666 as it would be nice to also have support for newer Zigbee adapters based on Silicon Labs EFR32 which uses newer SDK

Just read about thier prototypes for new EmberZNet / EZSP based "SprutStick 2" and "SprutStick Pro" USB-sticks they are working on which will be based on a EFR32MG13P and MGM12P chips from the Silicon Labs EFR32 series which feature a Zigbee 3.0 stack 

https://www.silabs.com/wireless/zigbee/efr32mg12-series-1-modules/device.mgm12p32f1024ge

Made in Russia and posts are in all Russian and therefore had to use Google translate to read so don't unstand everything but could be interesting for users of bellows

https://sprut.ai/client/article/2583

https://www.instagram.com/sprut666666/

Anyway, apparently thet are testing it with Home Assistant, presuably via bellows using ZHA / zigpy

Hedda, 2020-04-21:
Also see #243 as that hardware is based on the same family and series of chip from Silicon Labs.

Hedda, 2020-06-23:
Not sure how missed it but apparently Elelabs Zigbee USB adapter and Pi shield is based on EFR32MG1

Elelabs Zigbee Raspberry Pi Shield
- https://elelabs.com/products/elelabs-zigbee-shield.html

Elelabs Zigbee USB Adapter
- https://elelabs.com/products/elelabs-usb-adapter.html

@MattWestb stumbled on that information and posted in https://github.com/Koenkk/zigbee-herdsman/issues/168

I thought that Elelabs had not publicly published exactly which chip type they were using other than it was from Silicon Labs, so I had no idea that they are based on EFR32 MG1 (EFR32 Series 1).

I was sure that Elelabs only marketed those as alternatives Nortek GoControl QuickStick Combo Model HUSBZB-1 if you only want a Zigbee adapter compatible the bellows library as used in the ZHA integration component for Home Assistant. I know that the fact that HUSBZB-1 is not sold outside the United States was the main reason that Elelabs started making those, at least they at first have been targeting Home Assistant's ZHA users, and later also openHAB users.

Regardless Elelabs must only ship with old firmware version(s) with their products since it is compatible with the bellows library out-of-the-box and the fact is that the bellows library is currently only compatible with the older versions of EZSP protocols.

sprut666666, 2020-07-08:
Hi, and thanks for your interest in the product.
The stick was developed for [Sprut.hub](https://spruthub.ru) and works with the latest version of SDK (the latest firmware with the latest version of SDK will always be provided)
The first batch of sticks was sold in the first 5 minutes after the start [https://spruthub.ru/stick](https://spruthub.ru/stick)
The next batch will be already large and based on the chip https://www.silabs.com/wireless/zigbee/efr32mg21-series-2-socs/device.efr32mg21a010f1024im32

Hedda, 2020-07-26:
@sprut666666 Any chance you would consider donating series 1 and/or series 2 adapters to @Adminiuga for bellows and zigpy development?

Adminiuga, 2020-07-26:
I appreaciate, but shipping is going to be a killer and ATM it's lack of time is the limiting factor. But once bellows supports v8, whether it is mg1 or mg2 series, to `bellows` it would be transparent.

I'd rather have someone explain me, why with the current `bellows` settings, Aqara no-neutral US wall switch, joins fine out of the box for the 1st time, but if you reset it and try to join again, it doesn't want to join. (joins and leaves). 
Still have to try it with the gateway app from Silabs, should offer some clues.

sprut666666, 2020-07-26:
> @sprut666666 Any chance you would consider donating series 1 and/or series 2 adapters to @Adminiuga for bellows and zigpy development?
The second batch of sticks is still in production. 
And for development it is better to use the dev kit https://www.instagram.com/p/B7qswCSCrfY/ 
Moreover, now it seems you can buy for $ 99 https://www.silabs.com/development-tools/wireless/efr32xg22-wireless-starter-kit

sprut666666, 2020-07-26:
> I appreaciate, but shipping is going to be a killer and ATM it's lack of time is the limiting factor. But once bellows supports v8, whether it is mg1 or mg2 series, to `bellows` it would be [...]

Quite right, firmware 6.7.6 globally does not differ from series to series.

I'm sorry, but I'm not using zigpy and ha. On the problem, you need to look at silabs policy and traffic zigbee sniff.
We did not notice such a problem, the Aqara no-neutral work perfectly.  List of tested equipment here https://v2.sprut.ai/catalog?tags=SprutHub

Hedda, 2020-11-19:
@sprut666666 Which firmware are you using on SprutStick adapters? If standard EmberZNet with EZSP API firmware then can just close this as SprutStick with such firmware should be compatible with bellows and zigpy out-of-the-box without modifcations.

sprut666666, 2020-11-20:
> @sprut666666 Which firmware are you using on SprutStick adapters? If standard EmberZNet with EZSP API firmware then can just close this as SprutStick with such firmware should be compatible with bellows and zigpy out-of-the-box without modifcations.

It's standard EmberZNet + minor edits.

Hedda, 2020-11-25:
@sprut666666 OK so they work out-of-the-box with Home Assistant's ZHA via bellows EZSP support without firmware upgrades?

https://www.home-assistant.io/integrations/zha/

sprut666666, 2020-11-25:
> @sprut666666 OK so they work out-of-the-box with Home Assistant's ZHA via bellows EZSP support without firmware upgrades?
> 
> https://www.home-assistant.io/integrations/zha/

If ZHA can 6.8.2.0 then yes

Hedda, 2020-11-25:
> If ZHA can 6.8.2.0 then yes

It does support but so far 6.8.2.x has proven to be unstable and therefore 6.7.x.x is still recommended for stability in production.

sprut666666, 2020-11-25:
> > If ZHA can 6.8.2.0 then yes
> 
> It does support but so far 6.8.2.x has proven to be unstable and therefore 6.7.x.x is still recommended for stability in production.

I didn't notice any problems other than ZLL in 6.8.2.0, on the contrary, the stability increased.

Hedda, 2020-11-25:
As I have seen, most reports of instability with EmberZNet 6.8.x.x firmware comes from those who have hacked Sonoff ZBBridge with Tasmota and using its serial TCP server as a remote adapter with Home Assistant ZHA -> https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

I believe that @Adminiuga is still beta testing EmberZNet 6.8.x.x firmware with bellows using Elelabs ELU013 and/or ELR023. Those adapters are both based on EFR32MG13P -> https://github.com/zha-ng/EZSP-Firmware/tree/master/Elelabs-ELU013

Elelabs themselves have not yet officially released any EmberZNet 6.8.x.x firmware updates but said back in September that they had plans to release EmberZNet 6.8.x.x firmware upgrades in October -> https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/issues/3

sprut666666, 2020-11-25:
We have tested 6.8 since the silabs beta, we have observed that the stability has increased significantly, but ZLL 🤬🤬🤬

MattWestb, 2020-11-25:
Thats sound bad that ZLL is going down then most users is using lights in there HA systems :-((
But the good is the overall stability is improved and hope silabs is fixing the ZLL problems i future of EZSP release.

Great feedback sprut666666 !!!

Hedda, 2020-11-25:
@sprut666666 have you tested both upgraded and downgrading firmware on DEFARO SprutStick Pro with Elelabs EZSP Utility:

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility

Great thing about that Elelabs EZSP Firmware Update Utility is should work with all adapters using standard Silabs bootloader.

Hedda, 2020-12-10:
@sprut / @sprut666666 FYI, "DEFARO SprutStick Pro" is now listed as zigpy/bellows compatible adapters -> https://github.com/zigpy/bellows/pull/374

closed 2020-12-10 by Hedda

MattWestb, 2021-01-21:
@sprut666666 ZHA is starting "recommending" EZSP 6.7.8.0 after some critical bug fixes that is not implanted in 6.8 and 6.9 branch.
Is it anny plan doing one update for the first and second gen of your EFR32 1/2 ?
I can only finding 6.7.6.0 on the [download page](https://defaro.ru/index.php/firmware/258-firmware-sprutstick-pro).

And one more question on [ZHA wiki supported list](https://github.com/zigpy/zigpy/wiki/Coordinator-Firmware-Updates#recommended-firmware-version-for-ezsp-coordinators) its only on devices listed but the sprutstick-pro do have 2 hardware version. Should it being listed as 2 separate or only the new one ?