zigpy/bellows #243: [HARDWARE] Ebyte E180-ZG120B-TB Zigbee 3.0 development board with E180-ZG120B module based on EFR32MG1B Series 1 (Silicon Labs EFR32 family)

Hedda, 2020-04-21:
Can anyone confirm if the EFR32MG1B based E180-ZG120B module from Ebyte works with bellows?

Ebyte is making an inexpensive development board called "E180-ZG120B-TB" made for testing this. 

- http://www.ebyte.com/en/product-view-news.aspx?id=896
  - http://www.ebyte.com/en/downpdf.aspx?id=896
- http://www.cdebyte.net/test-board/zigbee-3-0-module-usb-test-board-kits.html

This development is sold for less than $9 US-dollar on Aliexpress or about twice that on eBay UK:

- https://www.aliexpress.com/item/4000538003718.html
- https://www.ebay.co.uk/itm/EFR32-Zigbee-3-0-Test-Board-USB-Port-2-4GHZ-E180-ZG120B-TB-Test-Kit/233517910970

Looks like you can remove some jumpers to disable the USB converter if want to use serial directly.

EFR32MG1B Series 1 is a 20dbm powerful Zigbee 3.0 radio capable SoC / chip in Silicon Labs EFR32 family which I understand also used the Ember / EZSP interface so could be made compatible with bellows if flashed with the right firmware? So I guess main question is which exact firmware to use?

- https://www.silabs.com/wireless/zigbee/efr32mg1-series-1-socs
- https://www.silabs.com/documents/public/data-sheets/efr32mg1-datasheet.pdf

Ebyte is now making two EFR32MG1B based modules called E180-ZG120A and E180-ZG120B

- http://www.ebyte.com/en/product-view-news.aspx?id=842
- http://www.ebyte.com/en/downpdf.aspx?id=842

Both E180-ZG120A and E180-ZG120B modules are now sold on eBay and Aliexpress at low prices. 

- https://www.aliexpress.com/wholesale?catId=0&SearchText=%22E180-ZG120B%22

Example: 

- https://www.aliexpress.com/item/4000473346234.html

Hedda, 2020-04-21:
Also, see #242  as that hardware is based on the same family and series of chip from Silicon Labs.

Hedda, 2020-04-21:
Silicon Labs also has an official devkit called "EFR32™ Mighty Gecko Wireless Starter Kit", it, however, cost almost $500 US-dollars ($499 to be precise), though it contains three complete dev boards.

- https://www.silabs.com/products/development-tools/wireless/mesh-networking/mighty-gecko-starter-kit
- https://www.silabs.com/wireless/zigbee/efr32mg1-series-1-socs/device.efr32mg1b132f256gm32#software

Hedda, 2020-04-21:
Z-Smart Systems now has driver code supporting latest Silicon Labs Ember dongles at @zsmartsystems

- https://github.com/zsmartsystems
  - https://github.com/zsmartsystems/com.zsmartsystems.zigbee
    - https://github.com/zsmartsystems/com.zsmartsystems.zigbee.sniffer

Adminiuga, 2020-04-21:
Does ebyte provide any firmware? 
I've ordered one we'll see what firmware it comes with. But in later version of EZSP they changed commands id to 16 bit.

Hedda, 2020-04-21:
Believe that the Ebyte module only comes with standalone bootloader firmware preloaded. 

I don't own one myself yet but would be nice if could just load up their SDK (Simplicity Studio) and it would detect the hardware once it is plugged in. Guessing that you might need a serial number for SDK access or kit part number for the official "EFR32 Mighty Gecko Wireless Starter Kit" (that kit part number is "SLWSTK6000B") as it is kind of a similar development board with the same type of chip(?).

https://www.silabs.com/products/development-tools/software/simplicity-studio

I would think you have to flash application bootloader firmware with Zigbee stack yourself, ...but where to get that Zigbee stack and application bootloader firmware I don't know, maybe from a Silicon Labs SDK? As I understand they call protocol such as the Zigbee stack for "apps".

- https://www.silabs.com/documents/public/user-guides/ug103-06-fundamentals-bootloading.pdf

Silicon Labs has different user forums here:

- https://www.silabs.com/community/wireless/zigbee-and-thread/forum

- https://www.silabs.com/community/software/simplicity-studio

@cdjackson from @zsmartsystems also mention two main NCP images in their README.md

- https://github.com/zsmartsystems/com.zsmartsystems.zigbee

  "Currently there are two main NCP images - one that supports hardware flow control with a baud rate of 115200, and one that supports software flow control with a rate of 57600."

Perhaps he would be willing to hint where to get the correct Zigbee coordinator firmware for EFR32?

@cdjackson also keep adding new EZSP support from UG100 to com.zsmartsystems.zigbee

- https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pulls?q=ezsp

- https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf

Hedda, 2020-04-21:
@sprut / @sprut666666 as mentioned in https://github.com/zigpy/bellows/issues/242 is also working on similar hardware.

Adminiuga, 2020-04-21:
So it boils down to getting Silicon Labs SDK which you only get with their dev-kit. But once you get SDK you can also get new image for the same chip as on HUSBZB-1 stick.

Adminiuga, 2020-04-21:
There're three dev boards in the mighty Gecko Silicon labs devkit. Just need to find two more volunteers willing to shell out $133 + taxes for the dev board and access to SDK which would allow to upgrade HUSBZB-1 to the latest emberznet.

Hedda, 2020-04-21:
Yes, same SDK. Some posts on their forum hint that you don't need a real serial number for the SDK

- https://www.silabs.com/community/wireless/zigbee-and-thread/forum.topic.html/who_can_share_emberznetkitregistrationserialnu-oa2A
- https://www.silabs.com/community/wireless/zigbee-and-thread/forum.topic.html/what_is_lowest_costkittogainaccesstoemberzne-B0dB.html
- https://www.silabs.com/community/wireless/zigbee-and-thread/forum.topic.html/how_to_get_the_ember-vpHN.html

You might only need to enter the kit part number for the official "EFR32 Mighty Gecko Wireless Starter Kit", and that kit part number is "SLWSTK6000B" (as it is kind of a similar development board with the same type of chip) and then just click a checkbox to accept the license agreement.

Then once someone has created hex/bin files with EmberZNet firmware then anyone can flash them themselves (I understand that is already done for getting Telegesis adapters to work with bellows).

Adminiuga, 2020-04-21:
not sure if we're reading the same posts, but those do tell you have to have access to dev-kit serial number
https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2017/11/22/access_to_siliconla-jk1S

Hedda, 2020-04-21:
> not sure if we're reading the same posts, but those do tell you have to have access to dev-kit serial number
> https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2017/11/22/access_to_siliconla-jk1S

Note that the original post there was posted more than two years ago so might have changed.

sslin (Employee) teplied Feb 21 2019, 2:40 AM in that thread with this:

Simplicity Studio is free to download and use even if you don't buy any kits.

JTAG driver will be installed automatically when you install the Simplicity Studio.

Some sdks/stack are free to download within the Simplicity Studio like Bluetooth, Flex, MCU, etc ...

With the serial number you got from the Mesh starter kit, you will be granted the access right to Zigbee and Thread for life long.

cdjackson, 2020-04-21:
The HUSBZB does not contain a bootloader so cannot be updated.

Adminiuga, 2020-04-21:
> Some sdks/stack are free to download within the Simplicity Studio like Bluetooth, Flex, MCU, etc .

Well, that might be true, but last time I've looked Emberznet stack was not accessible without a kit. 

> The HUSBZB does not contain a bootloader so cannot be updated

You send it a bootloader frame and it enters into bootload mode and you can upload .ebl image using xmodem.

Adminiuga, 2020-04-21:
from HUSBZB in bootloader
```
EM3581 Serial Bootloader v5.4.1.0 b962                                       
1. upload ebl                                                                
2. run                                                                       
3. ebl info                                                                  
BL >
```

cdjackson, 2020-04-21:
> You send it a bootloader frame and it enters into bootload mode and you can upload .ebl image using xmodem.

Really? Have you tried this? For the HUSBs that I've looked they have not shown a bootloader being installed.

Adminiuga, 2020-04-21:
send it `launchStandaloneBootloader()` EZSP command and afterwards connect to it with a serial terminal (might need different baudrate, i don't remember the details) It did enter the boot loader mode

[image]

cdjackson, 2020-04-21:
Sure - I’m very familiar with the protocols and have supported many companies with implementing Silabs systems. I don’t personally have one of these devices as I’m in Europe so there’s little point here, but I’ve had someone else test this and it didn’t work - maybe there was a problem with that device or maybe not all support the boot loader (or maybe the previous person to test this didn’t test it well).

Adminiuga, 2020-04-21:
I think it is might be up to the vendor to include boot loader into image, but both of Nortek GoControl HUSBZB-1 sticks respond `launchStandaloneBootloader` command and were ready to accept `.ebl` image.

cdjackson, 2020-04-21:
Does it respond to getStandaloneBootloaderVersionPlatMicroPhy?

Adminiuga, 2020-04-21:
```
debug: Send command getStandaloneBootloaderVersionPlatMicroPhy: ()
debug: Sending: b'2240215754bbef5f7e'
debug: Data frame: b'2340a15754bb05e65d9e49e6697e'
debug: Sending: b'83401b7e'
debug: Application frame 145 (getStandaloneBootloaderVersionPlatMicroPhy) received: b'1054040a03'
[21520, 4, 10, 3]
debug: Closed serial connection
```

cdjackson, 2020-04-21:
Thanks. I might need to try and get hold of one to create a new firmware and try this.

Adminiuga, 2020-04-21:
https://github.com/zigpy/bellows/pull/249

Hedda, 2020-04-22:
There are otherwise plenty of instructions out there on how to flash Telegesis ETRX357 USB dongles, as you can't use them with bellows until you have first flashed them with other EmberZNet firmware

https://github.com/zigpy/bellows/blob/dev/README.md

- Telegesis ETRX357USB (Note! This first have to be flashed with other EmberZNet firmware)
- Telegesis ETRX357USB-LRS (Note! This first have to be flashed with other EmberZNet firmware)
- Telegesis ETRX357USB-LRS+8M (Note! This first have to be flashed with other EmberZNet firmware)

https://community.home-assistant.io/t/eu-usb-sticks-for-the-new-zigbee-component/16718/10

Quote bellow originally posted by Frank Aug '17 in linked thread:

I can confirm that the ETRX357USB-LRS+8M can be flashed with an appropriate EZSP firmware without the use of any programming hardware.

So, without any guarantees (you might -but probably won’t- brick your device), here’s how:

Download the firmware. I used some random blob someone put on github that at least seemed to have the right name. What could possibly go wrong, right? :smile: https://github.com/yqyunjie/Zigbee-Project/blob/master/firmware/EmberZNet/EM35x-EZSP/build/em35x-ezsp-images/EM357/em357-ncp-uart-xon-xoff-use-with-serial-uart-bl-500.ebl?raw=true 282

(Install USB-to-serial drivers for the device. Linux will load the driver automatically when you plug in; not sure about other OSes.)

Install a serial port communication app that supports X-MODEM. (debian/ubuntu: sudo apt-get install minicom) Run it and configure it (sudo minicom -s) to use the fake serial port (/dev/ttyUSB0) at 19200 baud 8N1. Disable both hardware (RTS/CTS) and software (XON/XOFF) flow control.

Type ‘AT’ , you should get OK in response. Now type ‘AT+BLOAD’ . The device will reboot into the bootloader.

Change the baud rate in the serial port communication app setting to 115200 baud. (exit minicom using CTRL-A Q, run ‘sudo minicom -s’ again)

Pressing enter in the terminal should now show you a three-option boot loader menu. Choose option 1.

‘C’ characters will start showing. Don’t wait for this to finish, but start an X-MODEM upload of the firmware you downloaded earlier (use CTRL-A S in minicom). You have 60 seconds to start the upload.

After the upload finished, you should return to the menu. Now select option 2, to reboot into the new firmware. You’re done.

Enjoy!

Hedda, 2020-04-22:
Think these might finally a couple of easier how-to guide for getting EmberZNet onto EFR32 or EM35x.

# QSG106: Getting Started with EmberZNet PRO

- https://www.silabs.com/documents/public/quick-start-guides/qsg106-efr32-zigbee-pro.pdf

- https://www.silabs.com/support/mesh-networking-training/zigbee/zigbee-3-0-dyi-building-a-zigbee-3-0-switch-and-light-from-scratch

It does however sound as if you need to register EFR32MG development kit with a serial number to get a logon, but once you have you are allowed to reuse it within the same organization or company.

- https://siliconlabs.force.com/KitRegistration

Could be worth it for you to try to contact Silicon Labs Salesforce and explain that you would like a logon for an open-source organization/project without owning a dev kit, as an "organization" is a loose term, zigpy is technically an organization. 

After all, projects like zigpy does help Silicon Labs sell chips. Guess that the answer might differ depending on how Silicon Labs feeling towards helping open-source projects which are not-for-profit.

@Adminiuga Another suggestion could be for you and @dmulcahey to ask if maybe Home Assistant core team member would be willing to donate an official EFR32MG development kit to you ZHA devs.

Again it sounds as if you would only need one login for "team zigpy" and then several people on "team zigpy" would be allowed to use that same login with Simplicity Studio and other Silicon Labs software.

Hedda, 2020-04-22:
> I think it is might be up to the vendor to include boot loader into image

Guess they might sometimes recommend disabling bootloader to prevent any firmware upgrades.

Here are otherwise some Silicon Labs recommendations for building firmware images for production:

- https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2017/12/28/building_firmwareim-1OPr

kpszupin (Employee) originally posted 12/29/2017 but it has probably updated since then:

Some recommendations when building firmware images for production:

1. Build a single manufacturing hex file for each device with bootloader, application image, flash protection patches, and any tokens.
2. Enable write protection on the bootloader flash area (0-0x4000) by clearing the lower eight bits of the first byte in the lock bits page (0x0fe04000). Each bit represents 2KB of flash, so 8 bits represents 16KB total for the standard bootloader: --patch 0x0fe04000:0x00
3. Disable the debug port on the EFR32 by clearing the four LSBs of the DLW register in the lock bits page. This prevents debugger access from reading/writing to the device. Debug access can only be gained again by erasing the device. --patch 0x0fe041FC:0xF0
4. Program the AES/encryption key (if applicable) and the ECDSA public key tokens. These can be in the same token file as any other tokens you are programming. --tokenfile keyfile.txt
5. Lock the lock bits page on EFR32 by clearing bit 1 of the ULW register in the lock bits page. This is a requirement to secure the AES and ECDSA tokens which reside in the lock bits page. --patch 0x0fe041F8:0xFD

It is possible to combine these patches into a single hex file for manufacturing with a Simplicity Commander command, which can be put into a batch file or post-build script. For example:

  commander.exe convert gecko-bootloader.s37 app_image.s37 --patch 0x0fe04000:0x00 --patch 0x0fe041FC:0xF0 --patch 0x0fe041F8:0xFD  —tokengroup znet —tokenfile mfg_tokens.txt -o manufacturing_image.hex -d  EFR32MG1P233F256GM48

Hedda, 2020-04-22:
Another tip is this "Zigbee Boot Camp Course" wiki for Silicon Labs @MarkDing has put on GitHub :

- https://github.com/MarkDing/IoT-Developer-Boot-Camp/wiki/Zigbee-Boot-Camp

- https://github.com/MarkDing/IoT-Developer-Boot-Camp

He has also put together this PDF with Silicon Labs ZigBee Onboarding Roadmap for beginners:

- https://github.com/MarkDing/IoT-Developer-Boot-Camp/wiki/files/ZB-Zigbee-Boot-Camp/Silicon-Labs-ZigBee-Onboarding-Roadmap.pdf

Slightly off-topic question; but once have EFR32 based adapter with latest EmberZNet Zigbee stack:

Would it be possible for zigpy to form a Zigbee 3.0 network using latest EmberZNet ZigBee Stack?

- https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2018/04/24/how_to_form_zigbee3-zrzB
- https://www.silabs.com/support/mesh-networking-training/zigbee/zigbee-3-0-dyi-building-a-zigbee-3-0-switch-and-light-from-scratch

currently, as I understand, zigpy & bellows form a Zigbee HA (Zigbee Home Automation) network?

Hedda, 2020-04-22:
Slightly off-topic question; but once have EFR32 based adapter with latest EmberZNet Zigbee stack:

Would it be possible for zigpy to form a Zigbee 3.0 network using latest EmberZNet ZigBee Stack?

- https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2018/04/24/how_to_form_zigbee3-zrzB
- https://www.silabs.com/support/mesh-networking-training/zigbee/zigbee-3-0-dyi-building-a-zigbee-3-0-switch-and-light-from-scratch

currently, as I understand, zigpy & bellows form a Zigbee HA (Zigbee Home Automation) network?

Hedda, 2020-04-28:
FYI @SillyDay don't own a E180-ZG120B but has tried to build a firmware for it as per discussion here:

https://github.com/Koenkk/zigbee-herdsman/issues/168

SillyDay GitHub repo for EFR32 firmware:

https://github.com/SillyDay/EFR32

tofurky, 2020-05-31:
not quite the latest, but i've obtained the following:
```
   148224  2018-11-02 13:39   developer/sdks/gecko_sdk_suite/v2.4/protocol/zigbee/ncp-images/em3581/ncp-uart-rts-cts-use-with-serial-uart-btl-6.4.1.ebl
   146560  2018-11-02 13:39   developer/sdks/gecko_sdk_suite/v2.4/protocol/zigbee/ncp-images/em3581/ncp-uart-xon-xoff-use-with-serial-uart-btl-6.4.1.ebl
   147904  2018-11-02 13:39   developer/sdks/gecko_sdk_suite/v2.4/protocol/zigbee/ncp-images/em3581/ncp-uart-sw.ebl
   149632  2018-11-02 13:39   developer/sdks/gecko_sdk_suite/v2.4/protocol/zigbee/ncp-images/em3581/ncp-uart-hw.ebl
```
any idea which of these, if sent with xmodem after 'bellows bootloader', would be *least* likely to brick a perfectly working HUSBZB-1? :-) (EZSP Stack Type: 2, Stack Version: 21520 - "EM3581 Serial Bootloader v5.4.1.0 b962")

btw, i'm fully willing to provide these to anyone who needs them, but am not certain on the best way to go about doing so. i could throw them in a repo? 😬

AFAICT 6.4.1.0 is the last release to include prebuilt NCP firmwares (per https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.4.pdf).

btw, bringing things back to the original topic, also seeing these:
```
   198592  2018-11-02 13:39   developer/sdks/gecko_sdk_suite/v2.4/protocol/zigbee/ncp-images/efr32mg12p433f1024gm68-brd4170a-gb868-native/ncp-spi-use-with-ezsp-spi-btl-6.4.1.ebl
   189504  2018-11-02 13:39   developer/sdks/gecko_sdk_suite/v2.4/protocol/zigbee/ncp-images/efr32mg12p433f1024gm68-brd4170a/ncp-spi-use-with-ezsp-spi-btl-6.4.1.ebl
   182336  2018-11-02 13:39   developer/sdks/gecko_sdk_suite/v2.4/protocol/zigbee/ncp-images/efr32mg12p433f1024gm68-brd4170a/ncp-uart-rts-cts-use-with-serial-uart-btl-6.4.1.ebl
   191360  2018-11-02 13:39   developer/sdks/gecko_sdk_suite/v2.4/protocol/zigbee/ncp-images/efr32mg12p433f1024gm68-brd4170a-gb868-native/ncp-uart-rts-cts-use-with-serial-uart-btl-6.4.1.ebl
```

Adminiuga, 2020-05-31:
I'd think either `developer/sdks/gecko_sdk_suite/v2.4/protocol/zigbee/ncp-images/em3581/ncp-uart-sw.ebl` or `developer/sdks/gecko_sdk_suite/v2.4/protocol/zigbee/ncp-images/em3581/ncp-uart-xon-xoff-use-with-serial-uart-btl-6.4.1.ebl`

the HW ones AFAIK use baudrate 115200.

Hedda, 2020-06-01:
@tofurky Cool but maybe start a new issue for "HUSBZB-1 firmware upgrade" as that is off-topic here?

Both use EZSP APIs but HUSBZB-1 is based on the older EM358x family, not the newer EFR32 family.

Hedda, 2020-06-01:
FYI, @s-hadinger has now got a Sonoff ZBBridge and started reviewing the board in it (for Tasmota)

- https://github.com/arendst/Tasmota/issues/8583

First signs it is the EFR32 Zigbee module it has inside it has EmberZNet / Ember based firmware.

Hedda, 2020-06-02:
There is now also a deep dive follow-up discussion here in parallel about hacking or flashing its EFR32:

- https://github.com/Koenkk/zigbee-herdsman/issues/168

s-hadinger, 2020-06-18:
Hi, I have the EFR32 on the Sonoff Zigbee bridge flashed with EmberZNet v6.7.6.0 and supporting protocol EZSP v8. I'm struggling to send EZSP frames beyond the first v4 version frame.

I see that bellows send a 0xFF extended frame control when switching to protocol v5, but I can't find any mention of this 0xFF in the v8 documentation (UG100).

Can someone give me a link or a piece of documentation about this value?

Adminiuga, 2020-06-18:
v8 changed frame format and expanded command_id to 2 bytes and `bellows` currently does not support that. 
In EZSP Reference guide check the protocol format, around page 10

s-hadinger, 2020-06-18:
Thanks. I solved my problem as soon, but it needed to specify frame version to 1 in the high control byte.

s-hadinger, 2020-06-18:
Btw adding support in bellows may not be a big change. You need to set control bytes to `0001` and extend command to 2 bytes. Still there is no command number greater than 0xFF so adding `00` after the command should do the work.

Adminiuga, 2020-06-18:
Haven't looked yet, still waiting for my ebyte modules to show up. Just need to figure a way to switch between the versions, since most will be running on V7, V6 or even V5.
And need a way to implement different command sets between the versions, as there were some changes. e.g. I don't see RCE4 frames in the later protocols. Probably would be easier with SDK access to track changes between protocol versions.

Hedda, 2020-06-22:
@Adminiuga have you read about @s-hadinger development progress with Sonoff ZBBridge (Sonoff Zigbee Bridge)? He has added a TCP serial bridge to Tasmota firmware for remote access:

- https://github.com/arendst/Tasmota/issues/8583

FYI; @mtx512 also builds custom EZSP NCP firmware for Sonoff ZBBridge and other EFR32 devices:

- https://github.com/mtx512/efr32

Hedda, 2020-06-22:
@mtx512 also hosts new EZSP NCP firmware for "icc-a-1" EFR32 MG1 based module which can be desoldered from an IKEA Trådfri bulb and hacked for the same reason, as @basilfx documented here:

- https://github.com/basilfx/TRADFRI-Hacking

MattWestb, 2020-06-22:
@s-hadinger @Adminiuga In  EmberZNet Changed release 6.7.0.0 to V8 and not longer supporting older versions(" no longer use "Legacy Frame ID""). 6.6.4.0 its the latest that having possibility to going down to older protocols.
https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Modul#firmware. Both its in the latest SDK so perhaps @mtx512 can doing one 6.6.4.0 for you s-hadinger.

Hedda, 2020-06-23:
Not sure how I missed this but apparently Elelabs Zigbee adapter and shield is based on EFR32MG1:

Elelabs Zigbee Raspberry Pi Shield
- https://elelabs.com/products/elelabs-zigbee-shield.html

Elelabs Zigbee USB Adapter
- https://elelabs.com/products/elelabs-usb-adapter.html

@MattWestb stumbled on that information and posted in https://github.com/Koenkk/zigbee-herdsman/issues/168

I thought that Elelabs had not publicly published exactly which chip type they were using other than it was from Silicon Labs, so I had no idea that they are based on EFR32 MG1 series.

I was sure that Elelabs only marketed those as alternatives Nortek GoControl QuickStick Combo Model HUSBZB-1 if you only want a Zigbee adapter compatible the bellows library as used in the ZHA integration component for Home Assistant. I know that the fact that HUSBZB-1 is not sold outside the United States was the main reason that Elelabs started making those, at least they at first have been targeting Home Assistant's ZHA users, and later also openHAB users.

Regardless Elelabs must only ship with old firmware version(s) with their products since it is compatible with the bellows library out-of-the-box and the fact is that the bellows library is currently only compatible with the older versions of EZSP protocols.

Hedda, 2020-06-23:
> Probably would be easier with SDK access to track changes between protocol versions.

@Adminiuga Do you think, if you ask, that Home Assistant founders (or Nabu Casa) would sponsor you with $479 to buy the latest "EFR32xG21 Wireless Gecko Starter Kit" (article: SLWSTK6006A)?

- https://www.silabs.com/products/development-tools/wireless/efr32xg21-wireless-starter-kit

It contains three starter kits and includes full access Silicon Labs Zigbee SDK after register on Silabs.

Silicon Labs recently also released a less expensive Wireless Gecko Starter Kit which only contain one mainboard and low-power radios but I am not sure really if that includes Zigbee SDK access as well.

https://www.silabs.com/products/development-tools/wireless/efr32xg22-wireless-starter-kit

Anyway, Silabs "EFR32xG22 Wireless Gecko Starter Kit" (SLWSTK6021A) just released only cost $99

Hedda, 2020-06-23:
It has been confirmed by @jnicolson that buying just that $99 kit does give you full access to Silabs Zigbee stack (he contacted SiLabs support specifically asking this question and got confirmation it does, and he has since already purchased one and now have full access to the Silabs Zigbee stack).

Hedda, 2020-06-23:
> I'm currently running 6.7.6.0 of EFR32MG21, using protocol v8. The stack is correctly initialized, based on the work from bellows, the network is up and EFR32 is acting as coordinator. But I'm struggling to have devices pairing to the coordinator.
> 
> My first attempts showed a tentative of devices (Xiaomi and OSRAM) to pair, and unpairing because of a TC key issue. Now I'm not receiving any message when trying to pair. I'm stuck...

@s-hadinger sorry for cross-posting confusion but did you mean to write quoted to @Adminiuga 

You posted that in https://github.com/Koenkk/zigbee-herdsman/issues/168 but think you meant to post it here in https://github.com/zigpy/bellows/issues/243

Adminiuga, 2020-06-23:
@s-hadinger yeah, I suspect something has changed between the versions. Don't they use EZSP Values now, instead of EZSP Policy IDs ?
Need to make sure the commands ID are still correct, as there were changes (commands added/deleted/replaced) between the versions, but bellows most likely still has the V5 command set. 

One thing to try: change https://github.com/zigpy/bellows/blob/1dfe9c79ef6daea8aa06c4ddde4cf33f6c863033/bellows/zigbee/application.py#L237 to `t.EzspDecisionId.ALLOW_TC_KEY_REQUESTS`

s-hadinger, 2020-06-23:
Yes, thanks. I'm taking help from where I can.

I'm currently running 6.7.6.0 of EFR32MG21, using protocol v8. The stack is correctly initialized, based on the work from bellows, the network is up and EFR32 is acting as coordinator. But I'm struggling to have devices pairing to the coordinator.

My first attempts showed a tentative of devices (Xiaomi and OSRAM) to pair, and unpairing because of a TC key issue. Now I'm not receiving any message when trying to pair. I'm stuck...

The example I found from bellows called `networkInit`, I'm calling `formNetwork` instead after each boot, and I don't believe it makes any difference. 

I have tried many different security settings, and tried to mimick the one from bellows:
- `setConcentrator(false, 0xFFF9, 0x0258, 0x0708, 2, 5, 0)`
- `setInitialSecurityState` with the following flags: EMBER_GLOBAL_LINK_KEY + EMBER_HAVE_NETWORK_KEY + EMBER_HAVE_PRECONFIGURED_KEY + EMBER_REQUIRE_ENCRYPTED_KEY
- I have entered two different keys for preConfiguredKey and networkKey (basically random stuff)
- `setPolicy(EZSP_TRUST_CENTER_POLICY, EZSP_DECISION_ALLOW_JOINS | EZSP_DECISION_ALLOW_UNSECURED_REJOINS)` (0x03)
- `setPolicy(EZSP_TC_KEY_REQUEST_POLICY, EZSP_ALLOW_TC_KEY_REQUEST_AND_GENERATE_NEW_KEY)` (0x52)
- `setPolicy(EZSP_APP_KEY_REQUEST_POLICY, EZSP_ALLOW_APP_KEY_REQUESTS)` (0x61)
- when calling `formNetwork()` I use the join flag `EMBER_USE_MAC_ASSOCIATION`

- I do receive `stackStatusHandler(EMBER_NETWORK_UP)` callback.
- then send `permitJoining(0x3C)` (status = OK)
- then try to pair a Xiamo device, but nothing happens (no callback)

s-hadinger, 2020-06-23:
@Adminiuga The value for `EZSP_TC_KEY_REQUEST_POLICY` is now `EZSP_ALLOW_TC_KEY_REQUEST_AND_GENERATE_NEW_KEY` which matches the value for 0x52 `GENERATE_NEW_TC_LINK_KEY` 

The alternative is 0x51 `EZSP_ALLOW_TC_KEY_REQUESTS_AND_SEND_CURRENT_KEY`

Adminiuga, 2020-06-23:
Hrm, without logs hard to say. But if you are not seeing anything, I'd use a zigbee sniffer to see what is going on, at least to see if the device is requesting beacons and beacons are being sent out. 

For TC_KEY policy, yes, change it to `EZSP_ALLOW_TC_KEY_REQUESTS_AND_SEND_CURRENT_KEY`

s-hadinger, 2020-06-23:
Argh. I found the problem. I put the same key for preConfiguredKey and networkKey, it does not complain but it makes pairing fail. Now I can try again.

Thanks for your help, I may need it to go further.

Adminiuga, 2020-06-23:
Yeah, without a matching TC Link key, devices won't be able to get the network key.

s-hadinger, 2020-06-23:
I also realized that I set `rxRadioPower` to 1 because I had no clue about the value. I found in the doc that for EFR32 it should be 20, this also explains the erratic behavior.

Now if I can benefit from you experience:

I paired an OSRAM switch mini, but didn't receive any `childJoinHandler` message. I recevied the following and pairing seemed to work:
- `trustCenterJoinHandler` `2400E4E9DD92020AAA3EB07C01010000` (key sent in clear)
- `incomingMessageHandler` `450004000013000000000100000194C1E4E9FFFF0C00E4E9DD92020AAA3EB07C8E` (incoming ZDO message)
- `messageSentHandler` `3F0006FDFF0000130000000001000001000000` which looks to be EMBER_INCOMING_MANY_TO_ONE_ROUTE_REQUEST

But, with a Xiaomi device, I get the following:
- `childJoinHandler` `23000001B8D4AE506B03008D150004` child joining
- `trustCenterJoinHandler` `2400B8D4AE506B03008D150001010000` same as before
and 2 minutes later
- `childJoinHandler` `23000000B8D4AE506B03008D150004` child leaved

Do you have any idea why I don't receive `childJoinHandler` with OSRAM, and why Xiaomi device leaves? In parallel I will try to set up a Zigbee sniffer.

Adminiuga, 2020-06-23:
`childJoinHandler` is when an EndDevice joins directly to NCP. `trustCenterJoinHandler` is the handler notifying about a device being joined.

s-hadinger, 2020-06-24:
Ah got it. The CC2530 Z-Stack I'm used to is of higher level interface. Since the OSRAM wall switch is also a router, it is considered as a neighbour, not a child. This does not explain why the Xiaomi leaves after 2 minutes, but thanks. I'm progressing.

Hedda, 2020-06-24:
@NilsBohr you think Elelabs might be willing to donate a few EFR32MG13P adapters to bellows and zigbee-herdsman devs?

Thinking that such sponsoring will benefit Elelabs as a company since more developers working on this will probably lead to more users buying your Silicon Labs based adapters as long as you also provide firmware compatible with those open source projects.

bellows is primarily used by Home Assistant ZHA integration and zigbee-herdsman is used by Zigbee2mqtt and IoBroker.

> Hello guys,
> 
> That's Kit from Elelabs.
> 
> > I thought that Elelabs had not published exactly which chip type they were using other than they were from Silicon Labs, so I had no idea that they are based on EFR32 MG1 (EFR32 Mighty Gecko Series 1).
> 
> Jumping here in the discussion, if you don't mind.
> 
> We had sold:
> 
> * EZBUSBA model using version EM357
> * ELU012 model using EFR32MG1
>   **Now we are selling ELU013, which is using EFR32MG13P.**

> The Raspberry shield also had different models:
> 
> * EZBPIS based on EM357
> * ELR022 based on EFR32MG1B232 (32Kb RAM)
>   **Now we are selling ELR023, which is also based on EFR32MG13P732 (64Kb RAM)**

NilsBohr, 2020-06-24:
@Hedda 

Yes, we would be glad to.
Please reach out to me at sne@elelabs.com

Hedda, 2020-06-24:
> > @NilsBohr you think Elelabs might be willing to donate a few EFR32MG13P adapters to bellows and [zigbee-herdsman](https://github.com/Koenkk/zigbee-herdsman/issues/168) devs?
> > 
> > Thinking that such sponsoring will benefit Elelabs as a company since more developers working on this will probably lead to more users buying your Silicon Labs based adapters as long as you also provide firmware compatible with those open source projects.
> > 
> > bellows is primarily used by [Home Assistant's ZHA integration](https://www.home-assistant.io/integrations/zha/), and zigbee-herdsman is used by [Zigbee2mqtt](https://www.zigbee2mqtt.io/) and [IoBroker](https://www.iobroker.net/).
> > 
> 
> Yes, we would be glad to.
> Please reach out to me at [sne@elelabs.com](mailto:sne@elelabs.com)

I am not a skilled enough programmer myself, but maybe @Adminiuga @puddly @dmulcahey & @s-hadinger are interested(?) 👍 

PS: @NilsBohr  Suggest that you also post the same offer in the zigbee-herdsman discussion here -> https://github.com/Koenkk/zigbee-herdsman/issues/168

Hedda, 2020-06-24:
> Silabs just released a new $99 "EFR32xG22 Wireless Gecko Starter Kit" (SLWSTK6021A) which only contain one mainboard and two low-power radios (Zigbee Green Power compatible):
> 
> * https://www.silabs.com/products/development-tools/wireless/efr32xg22-wireless-starter-kit
> 
> (Google search shows that relatively inexpensive kit can already be bought from resellers globally).
> 
> While such low-power radios are not really usable as Zigbee coordinators, for $99 that kit can be useful to developers if buying such an official kit also gives full access to the Zigbee Stack/SDK and libraries.

FYI, those who are interested in the $99 SLWSTK6021A started kit might also be interested in buying article SLWRB4180A for around $49 at the same time as that is the matching EFR32MG21 based high-power (+20 dBm) radio board for the official dev board kit.

https://www.silabs.com/products/development-tools/wireless/slwrb4180a-efr32-wireless-gecko-radio-board

(Google search shows that SLWRB4180A can be bought from many of resellers globally who also stock the SLWSTK6021A kit).

Adminiuga, 2020-06-24:
> This does not explain why the Xiaomi leaves after 2 minutes, but thanks. I'm progressing

👍 if you figure this one out, let me know, as i'm troubleshooting the same issue. Have two EZSP running 6.3.3 version, the dev works fine, but it is only a few devices, but in prod, ZB3 joins, stays a bit (works fine during that period) and then leaves. Both prod and dev running the same version, just number of devices is different. I have strong suspicion it might be related to some ZB3 settings/policy settings, as downgrading to protocol version 6 or 5 fixes this issue.

Is the Xiaomi a ZB3 device?

s-hadinger, 2020-06-24:
@Adminiuga I doubt the Xiaomi runs ZB3. When I was using CC2530, it was a ZB1.2 network. I need to re-read the doc carefully to make it ZB1.2 devices compatible. This means sending the network key in clear, and pre-registering the "well-known" Link Key `ZigBeeAlliance09`. It looks from the doc that you need to add it explicitly.

I will also try a configuration without Trust Center. It's not ideal but for domestic Zigbee networks as we use in Tasmota, that's enough to start with.

[164 more replies]

closed 2020-12-17 by Adminiuga