zigpy/bellows #252: problem on run bellows zcl

MohsenZamany, 2020-04-22:
I installed bellows on rasbian stretch and python 3.8.2
after running "bellows zcl" face with following error

missing 1 required positional argument: 'device_config'

the full stack is:

Traceback (most recent call last):
  File "/home/pi/bel38/bin/bellows", line 8, in <module>
    sys.exit(main())
  File "/home/pi/bel38/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/bel38/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/pi/bel38/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/bel38/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/bel38/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/bel38/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/bel38/lib/python3.8/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/bel38/lib/python3.8/site-packages/bellows/cli/application.py", line 311, in commands
    ezsp = bellows.ezsp.EZSP()
TypeError: __init__() missing 1 required positional argument: 'device_config'

plz hellp me

Adminiuga, 2020-04-22:
Thanks for reporting this. `bellows` is in flux now with syncing changes to zigpy, so some bugs are possible. 
You could try new installing the previous version of the package which is named`bellows-homeassistant`. Run 
pip3 uninstall bellows
pip3 install bellows-homeassistant==0.15.2

MohsenZamany, 2020-04-22:
Thanks @Adminiuga 

I installed bellows-homeassistant and run the command again but face with  this error:

AttributeError: module 'bellows.zigbee' has no attribute 'exceptions'

the full stack is: 

EZSP Stack Type: 2, Stack Version: 22304
Device 0x0000 (00:0d:6f:00:0e:3f:97:c3) joined the network
Traceback (most recent call last):
  File "/home/pi/belhass/lib/python3.8/site-packages/bellows/cli/application.py", line 329, in command
    v = await getattr(cluster, command)(*parameters, manufacturer=manufacturer)
  File "/home/pi/belhass/lib/python3.8/site-packages/zigpy/device.py", line 201, in request
    raise zigpy.exceptions.DeliveryError(
zigpy.exceptions.DeliveryError: [0x2424:2:0x0006]: Message send failure

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/belhass/bin/bellows", line 8, in <module>
    sys.exit(main())
  File "/home/pi/belhass/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/belhass/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/pi/belhass/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/belhass/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/belhass/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/belhass/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/belhass/lib/python3.8/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/belhass/lib/python3.8/site-packages/bellows/cli/util.py", line 66, in inner
    loop.run_until_complete(async_inner(*args, **kwargs))
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/pi/belhass/lib/python3.8/site-packages/bellows/cli/util.py", line 53, in async_inner
    await f(ctx, *args, **kwargs)
  File "/home/pi/belhass/lib/python3.8/site-packages/bellows/cli/application.py", line 333, in command
    except bellows.zigbee.exceptions.DeliveryError as e:
AttributeError: module 'bellows.zigbee' has no attribute 'exceptions'

and my command is :

bellows zcl 03:a9:dc:0a:00:4b:12:00 2 6 command on

Adminiuga, 2020-04-22:
that was fixed in #233 but the root cause in your case the device did not receive the message. 
Could be it's changed it's address or if it is a battery operated device it went to sleep. 
Bellows is not really meant to be run from command line, because when it is not running it might miss messages from the network about devices leaving/rejoining.

Any reason you don't want to run ZHA integration of home assisitant?

PR #254 (2020-04-22): Fix zcl commands CLI command
Fix zcl commands CLI command.
Set default logging level to warning.

Fixes #252
closed 2020-04-22 by Adminiuga