zigpy/bellows #255: 0x37 (EZSP_ERROR_INVALID_ID) on Nortek HUSBZB-1 while attempting to set EzspConfigId.CONFIG_TRANSIENT_KEY_TIMEOUT_S

Hueburtsonly, 2020-04-25:
I'm attempting to set up a Zigbee network centered around a Nortek HUSBZB-1. `bellows info` works fine, but `bellows form` and `bellows permit` both fail; debug output is included below. I'm seeing that my device is running protocol version 4 ( 04021054 ), is it possible that my firmware needs updating?

As noted in the issue title, the problem comes when attempting to `setConfigurationValue` on `EzspConfigId.CONFIG_TRANSIENT_KEY_TIMEOUT_S: 54`. `0x37` is returned, indicating "The configuration id was not recognized." What I'm not sure about, is whether this is a case of out-of-date firmware on the HUSBZB-1, or perhaps an oversight in zigpy/bellows? Any advice would be most greatly appreciated.

**bellows info**
```
$ pip show zigpy bellows
Name: zigpy
Version: 0.20.0
Summary: Library implementing a ZigBee stack
Home-page: http://github.com/zigpy/zigpy
Author: Russell Cloran
Author-email: rcloran@gmail.com
License: GPL-3.0
Location: /usr/local/lib/python3.7/dist-packages
Requires: voluptuous, pycryptodome, aiohttp, crccheck
Required-by: bellows
---
Name: bellows
Version: 0.16.0
Summary: Library implementing EZSP
Home-page: http://github.com/zigpy/bellows
Author: Russell Cloran
Author-email: rcloran@gmail.com
License: GPL-3.0
Location: /usr/local/lib/python3.7/dist-packages
Requires: click-log, voluptuous, Click, pyserial-asyncio, pure-pcapy3, zigpy
Required-by:

$ bellows -v DEBUG -d /dev/ttyUSB1 info
debug: Using selector: EpollSelector
debug: Using selector: EpollSelector
debug: Connected. Resetting.
debug: Resetting EZSP
debug: Resetting ASH
debug: Sending: b'1ac038bc7e'
debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
debug: Send command version: (4,)
debug: Sending: b'004221a850ed2c7e'
debug: Data frame: b'0142a1a8502805e67f627e'
debug: Sending: b'8160597e'
debug: Application frame 0 (version) received: b'04021054'
EZSP Stack Type: 2, Stack Version: 21520
debug: Configuring...
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_STACK_PROFILE: 12>, 2)
debug: Sending: b'7d314321fb582815c3277e'
debug: Data frame: b'1243a1fb54fb737e'
debug: Sending: b'82503a7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_SECURITY_LEVEL: 13>, 5)
debug: Sending: b'224021fb592f15226f7e'
debug: Data frame: b'2340a1fb54c6107e'
debug: Sending: b'83401b7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_SUPPORTED_NETWORKS: 45>, 1)
debug: Sending: b'334121fb792b15a2d77e'
debug: Data frame: b'3441a1fb54d32a7e'
debug: Sending: b'8430fc7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_PACKET_BUFFER_COUNT: 1>, 255)
debug: Sending: b'444621fb55d51534da7e'
debug: Data frame: b'4546a1fb5435d07e'
debug: Sending: b'8520dd7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command networkInit: ()
debug: Sending: b'554721bfb6607e'
debug: Data frame: b'5647a1bfc7db3e7e'
debug: Sending: b'8610be7e'
debug: Application frame 23 (networkInit) received: b'93'
debug: Send command getEui64: ()
[... 185 lines cut ...]
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/bellows/cli/application.py", line 53, in permit
    return util.app(inner)(ctx)
  File "/usr/local/lib/python3.7/dist-packages/bellows/cli/util.py", line 68, in inner
    loop.run_until_complete(async_inner(*args, **kwargs))
  File "/usr/lib/python3.7/asyncio/base_events.py", line 579, in run_until_complete
    return future.result()
  File "/usr/local/lib/python3.7/dist-packages/bellows/cli/util.py", line 52, in async_inner
    ctx.obj["device"], ctx.obj["baudrate"], database_file, startup=app_startup
  File "/usr/local/lib/python3.7/dist-packages/bellows/cli/util.py", line 137, in setup_application
    app_config, startup
  File "/usr/local/lib/python3.7/dist-packages/zigpy/application.py", line 62, in new
    await app.startup(auto_form)
  File "/usr/local/lib/python3.7/dist-packages/bellows/zigbee/application.py", line 138, in startup
    await self.initialize()
  File "/usr/local/lib/python3.7/dist-packages/bellows/zigbee/application.py", line 82, in initialize
    await self._cfg(t.EzspConfigId[config], value)
  File "/usr/local/lib/python3.7/dist-packages/bellows/zigbee/application.py", line 230, in _cfg
    assert v[0] == t.EmberStatus.SUCCESS  # TODO: Better check
AssertionError
```
FWIW, my broader goal is to `form` a network, `permit` a bunch of Xiaomi WSDCGQ11LM Temp/Humidity Sensors to join, and then regularly pull temp/humidity figures via the CLI (and maybe one day a python script).

Adminiuga, 2020-04-25:
Can you uninstall bellows and install bellows-homeassistant with zigpy-homeassistant==0.19.0 ?

puddly, 2020-04-26:
I also ran into this problem when testing out the new config changes. I don't know much about EZSP but I "fixed" it by just ignoring the key if the status is `t.EzspStatus.ERROR_INVALID_ID`: https://github.com/puddly/bellows/commit/7236e94fa31290aef0608fd3ff2722327904f05a

Adminiuga, 2020-04-26:
@puddly  what is the stack version reported after the reset. 

I go through bthe changes, to see if we were setting this before.

Adminiuga, 2020-04-26:
I'm confused. We were setting it before https://github.com/zigpy/bellows/blob/d26b5bbb077c1f93cab4dd4d3db4f9e0ecb28719/bellows/zigbee/application.py#L93 and we're setting it in the new version , but why it fails now?

Adminiuga, 2020-04-26:
Order of setting config IDs should not matter but for the buffer counts

puddly, 2020-04-26:
I'll make some debug logs comparing the two releases.

puddly, 2020-04-26:
Here's a startup log from the latest version:

```
[2020-04-26 15:43:38] INFO [zigpy.ota.provider.initialize_provider:146] Trådfri: OTA provider enabled
[2020-04-26 15:43:55] DEBUG [zigpy.ota.provider.refresh_firmware_list:160] Trådfri: Finished downloading firmware update list
[2020-04-26 15:43:55] DEBUG [bellows.ezsp.reset:75] Resetting EZSP
[2020-04-26 15:43:55] DEBUG [bellows.uart.reset:202] Resetting ASH
[2020-04-26 15:43:55] DEBUG [bellows.uart.write:178] Sending: b'1ac038bc7e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.rstack_frame_received:137] RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp._command:116] Send command version: (4,)
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'004221a850ed2c7e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.data_frame_received:110] Data frame: b'0142a1a8502805e67f627e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'8160597e'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp.frame_received:230] Application frame 0 (version) received: b'04021054'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp.version:95] EZSP Stack Type: 2, Stack Version: 21520
[2020-04-26 15:43:56] DEBUG [bellows.ezsp._command:116] Send command setConfigurationValue: (<EzspConfigId.CONFIG_INDIRECT_TRANSMISSION_TIMEOUT: 18>, 7680)
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'7d314321fb462a0b0ed87e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.data_frame_received:110] Data frame: b'1243a1fb54fb737e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'82503a7e'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp.frame_received:230] Application frame 83 (setConfigurationValue) received: b'00'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp._command:116] Send command setConfigurationValue: (<EzspConfigId.CONFIG_PAN_ID_CONFLICT_REPORT_THRESHOLD: 34>, 2)
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'224021fb7628157d310f7e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.data_frame_received:110] Data frame: b'2340a1fb54c6107e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'83401b7e'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp.frame_received:230] Application frame 83 (setConfigurationValue) received: b'00'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp._command:116] Send command setConfigurationValue: (<EzspConfigId.CONFIG_SOURCE_ROUTE_TABLE_SIZE: 26>, 16)
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'334121fb4e3a15d2a07e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.data_frame_received:110] Data frame: b'3441a1fb54d32a7e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'8430fc7e'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp.frame_received:230] Application frame 83 (setConfigurationValue) received: b'00'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp._command:116] Send command setConfigurationValue: (<EzspConfigId.CONFIG_SECURITY_LEVEL: 13>, 5)
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'444621fb592f15bdb17e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.data_frame_received:110] Data frame: b'4546a1fb5435d07e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'8520dd7e'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp.frame_received:230] Application frame 83 (setConfigurationValue) received: b'00'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp._command:116] Send command setConfigurationValue: (<EzspConfigId.CONFIG_APPLICATION_ZDO_FLAGS: 42>, 3)
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'554721fb7d5e2915defb7e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.data_frame_received:110] Data frame: b'5647a1fb54a9ec7e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'8610be7e'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp.frame_received:230] Application frame 83 (setConfigurationValue) received: b'00'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp._command:116] Send command setConfigurationValue: (<EzspConfigId.CONFIG_KEY_TABLE_SIZE: 30>, 4)
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'664421fb4a2e157d31e67e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.data_frame_received:110] Data frame: b'6744a1fb54948f7e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'87009f7e'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp.frame_received:230] Application frame 83 (setConfigurationValue) received: b'00'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp._command:116] Send command setConfigurationValue: (<EzspConfigId.CONFIG_SUPPORTED_NETWORKS: 45>, 1)
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'774521fb792b15b85c7e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.data_frame_received:110] Data frame: b'7045a1fb5481b57e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'8070787e'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp.frame_received:230] Application frame 83 (setConfigurationValue) received: b'00'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp._command:116] Send command setConfigurationValue: (<EzspConfigId.CONFIG_TRANSIENT_KEY_TIMEOUT_S: 54>, 180)
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'004a21fb629e15b2107e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.data_frame_received:110] Data frame: b'014aa1fb63a4387e'
[2020-04-26 15:43:56] DEBUG [bellows.uart.write:178] Sending: b'8160597e'
[2020-04-26 15:43:56] DEBUG [bellows.ezsp.frame_received:230] Application frame 83 (setConfigurationValue) received: b'37'
[2020-04-26 15:43:56] ERROR [zigpy.application.new:67] Couldn't start application
[... cut ...]
[2020-04-26 16:07:16] DEBUG [bellows.uart.data_frame_received:109] Data frame: b'4469a1cb542a15b2ae487e'
[2020-04-26 16:07:16] DEBUG [bellows.uart.write:177] Sending: b'8520dd7e'
[2020-04-26 16:07:16] DEBUG [bellows.ezsp.frame_received:221] Application frame 99 (getMulticastTableEntry) received: b'00000000'
[2020-04-26 16:07:16] DEBUG [bellows.multicast._initialize:27] MulticastTableEntry[15] = <EmberMulticastTableEntry multicastId=0x0000 endpoint=0 networkIndex=0>
```

`EzspConfigId.CONFIG_TRANSIENT_KEY_TIMEOUT_S` is the only config to fail with the old version as well, it just ignored it.

puddly, 2020-04-26:
I believe the migration to the new config schema dropped the `optional` argument that was originally there:

```python
await self._cfg(c.CONFIG_TRANSIENT_KEY_TIMEOUT_S, 180, True)
```

It looks like the only config key for which it was ever used.

Adminiuga, 2020-04-26:
I think we just need to drop this parameter since even HUSBZB-1 with quite an old firmware does not support it.

Adminiuga, 2020-04-26:
It looks like it was added in the later versions of Emberznet and is deprecated in the latest. And yep, the only one marked optional.  I'm open to ideas how to mark a key as optional in Schema without overloading it too much

puddly, 2020-04-26:
It looks to be the only deprecated config key as well. The simplest approach would be just handling it in `_cfg` as a special case, since it really is the only one?

puddly, 2020-04-26:
I can't any documentation about when it was introduced but looking around GitHub, the newest EmberZNet codebase I can find without that config key is `0x5601`. It is present in `0x5730`.

On the other hand, `EZSP_ERROR_INVALID_ID` does tell you that the config ID is invalid, so you could also just log a warning and ignore it.

Adminiuga, 2020-04-26:
It seems to be deprecated in Emberznet Pro Release 6.7 rev 3.8. So I'm gonna just remove this one and rely on the default value of 360s

PR #256 (2020-04-26): Remove default value for CONFIG_TRANSIENT_KEY_TIMEOUT_S
Remove default value for CONFIG_TRANSIENT_KEY_TIMEOUT_S. 
Fixes #255
closed 2020-04-26 by Adminiuga