zigpy/bellows #259: Integration without Home Assistant

rossnelson, 2020-04-27:
We are currently evaluating zigpy and bellows for use in our custom IoT solution. Our gateway runs containerized, so we are attempting to build a docker image that runs inside our k8s distribution alongside our zwave integration. So far, I haven't been able to get bellows to work with my Nortek HUSBZB-1. I get the same error whether installed at the system level or inside a docker image. So I'll include info on both and hopefully, someone will be able to point me in the right direction. 

### Docker

`Dockerfile`
```
FROM python:buster

RUN apt update \
  && apt install -y vim git build-essential \
  && pip install bellows \
  && mkdir -p /root/.confg/bellows \
  && touch /root/.confg/bellows/app.db

CMD ['bellows', 'form']
```

```
$ docker run \
  -it \
  --privileged=true \
  -v $(pwd)/data:/root/.config/bellows \
  -v /dev/ttyUSB1:/dev/ttyUSB1 \ # could vary per system/device
  -e EZSP_DEVICE=/dev/ttyUSB1 \
  {image id}
```

### System 

`$ cat /etc/os-release`
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
```

`$ pip3 show zigpy bellows`
```
Name: zigpy
Version: 0.20.1a1
Summary: Library implementing a ZigBee stack
Home-page: http://github.com/zigpy/zigpy
Author: Russell Cloran
Author-email: rcloran@gmail.com
License: GPL-3.0
Location: /home/pi/.local/lib/python3.7/site-packages
Requires: pycryptodome, voluptuous, aiohttp, crccheck
Required-by: bellows
---
Name: bellows
Version: 0.16.0
Summary: Library implementing EZSP
Home-page: http://github.com/zigpy/bellows
Author: Russell Cloran
Author-email: rcloran@gmail.com
License: GPL-3.0
Location: /home/pi/.local/lib/python3.7/site-packages
Requires: click-log, Click, pure-pcapy3, zigpy, pyserial-asyncio, voluptuous
Required-by:
```

The error I see when running `bellows form` in either the system or from docker is this:

```
Traceback (most recent call last):
  File "/usr/local/bin/bellows", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/bellows/cli/application.py", line 34, in form
    return util.app(inner, app_startup=False)(ctx)
  File "/usr/local/lib/python3.8/site-packages/bellows/cli/util.py", line 68, in inner
    loop.run_until_complete(async_inner(*args, **kwargs))
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/usr/local/lib/python3.8/site-packages/bellows/cli/util.py", line 51, in async_inner
    application = await setup_application(
  File "/usr/local/lib/python3.8/site-packages/bellows/cli/util.py", line 136, in setup_application
    app = await bellows.zigbee.application.ControllerApplication.new(
  File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 60, in new
    app = cls(config)
TypeError: Can't instantiate abstract class ControllerApplication with abstract methods probe
```

Adminiuga, 2020-05-02:
There are some changes being made.  Use zigpy==0.20.1a3 and bellows 0.16.1a1
I think those should be published by now

Adminiuga, 2020-05-04:
Fixed in `bellows==0.16.1`

`pip3 show zigpy bellows`
```
Name: zigpy
Version: 0.20.1
Summary: Library implementing a ZigBee stack
Home-page: http://github.com/zigpy/zigpy
Author: Russell Cloran
Author-email: rcloran@gmail.com
License: GPL-3.0
Location: /tmp/venv/lib/python3.7/site-packages
Requires: voluptuous, aiohttp, crccheck, pycryptodome
Required-by: bellows
---
Name: bellows
Version: 0.16.1
Summary: Library implementing EZSP
Home-page: http://github.com/zigpy/bellows
Author: Russell Cloran
Author-email: rcloran@gmail.com
License: GPL-3.0
Location: /tmp/venv/lib/python3.7/site-packages
Requires: zigpy, click-log, Click, pyserial-asyncio, pure-pcapy3, voluptuous
Required-by: 
```

`bellows -v DEBUG form`
```
debug: Using selector: EpollSelector
debug: Loading application state from /home/lex/.config/bellows/app.db
debug: Attribute id: 4 value: Jasco Products
debug: Attribute id: 5 value: 45857
debug: Checking quirks for None None (00:0d:6f:00:0b:5d:e3:b6)
debug: Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
debug: Found custom device replacement for 00:0d:6f:00:0b:5d:e3:b6: <class 'bellows.zigbee.application.EZSPCoordinator'>
debug: Checking quirks for Jasco Products 45857 (00:22:a3:00:00:1b:94:bb)
debug: Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
debug: Fail because endpoint list mismatch: {1} {1, 2}
debug: Attribute id: 4 value: Jasco Products
debug: Attribute id: 5 value: 45857
debug: Attribute id: 770 value: 10000
debug: Attribute id: 769 value: 1
debug: Attribute id: 768 value: 0
debug: Attribute id: 772 value: 11
debug: Attribute id: 0 value: 0
debug: Attribute id: 7 value: 1
debug: Attribute id: 0 value: 0
debug: Attribute id: 1024 value: 0
debug: Attribute id: 7 value: 1
debug: Adding group: 546, Group 0x0222
debug: Adding group: 16387, No name group 0x4003
debug: Using selector: EpollSelector
debug: Resetting EZSP
debug: Resetting ASH
debug: Sending: b'1ac038bc7e'
debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
debug: Send command version: (4,)
debug: Sending: b'004221a850ed2c7e'
debug: Data frame: b'0142a1a8522815d2e7ae7e'
debug: Sending: b'8160597e'
debug: Application frame 0 (version) received: b'06020060'
debug: Send command version: (6,)
debug: Sending: b'7d31432157542a7d3315a67e'
debug: Data frame: b'1243a157542a13b059f4b6467e'
debug: Sending: b'82503a7e'
debug: Application frame 0 (version) received: b'06020060'
debug: Switched to EZSP protocol version 6
debug: EZSP Stack Type: 2, Stack Version: 24576
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_ADDRESS_TABLE_SIZE: 5>, 16)
[... 157 lines cut ...]
debug: Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x0000, <EmberApsFrame profileId=0 clusterId=2 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=1>, 2, b'\x01\x00\x00')
debug: Sending: b'705d2157541e15b259944a27aa5592099d4e27aaefcd668bfd19017e'
debug: Data frame: b'005da157541e1512ce907e'
debug: Sending: b'8160597e'
debug: Application frame 52 (sendUnicast) received: b'00a0'
debug: Data frame: b'105db157546f15b259964a25aa1593499ceed8abedce9874fec763894b587e'
debug: Sending: b'82503a7e'
debug: Data frame: b'205db157546f15b25996ca25aa1593499cefd8abedce9874ecc76389fc7e7f2826668cef8fbeeb5bd5d2f6567e'
debug: Sending: b'83401b7e'
debug: Data frame: b'305db157541515b259944a27aa5592099d4e270befce677d4f7e'
debug: Sending: b'8430fc7e'
debug: Application frame 69 (incomingMessageHandler) received: b'0000000200000040010000a0ff000000ffff03010000'
debug: Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=0 clusterId=2 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=160>, 255, 0, 0x0000, 255, 255, b'\x01\x00\x00']
debug: [0x0000:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
debug: [0x0000:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_req
debug: Application frame 69 (incomingMessageHandler) received: b'0000000280000040010000a1ff000000ffff110100000000408fcdab528000412c800000'
debug: Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=0 clusterId=32770 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=161>, 255, 0, 0x0000, 255, 255, b'\x01\x00\x00\x00\x00@\x8f\xcd\xabR\x80\x00A,\x80\x00\x00']
debug: [0x0000:zdo] ZDO request ZDOCmd.Node_Desc_rsp: [<Status.SUCCESS: 0>, 0x0000, <Optional byte1=0 byte2=64 mac_capability_flags=143 manufacturer_code=43981 maximum_buffer_size=82 maximum_incoming_transfer_size=128 server_mask=11329 maximum_outgoing_transfer_size=128 descriptor_capability_field=0>]
debug: [0x0000:zdo] Unsupported ZDO request:ZDOCmd.Node_Desc_rsp
debug: Application frame 63 (messageSentHandler) received: b'00000000000200000040010000a0020000'
debug: Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0, <EmberApsFrame profileId=0 clusterId=2 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=160>, 2, <EmberStatus.SUCCESS: 0>, b'']
Shutting down ControllerApplication
debug: Cancelling watchdog
debug: Closed serial connection
```

closed 2020-05-04 by Adminiuga

rossnelson, 2020-05-06:
Great thanks for the quick reply. I’ll check this out on my end.