zigpy/bellows #263: AttributeError: module 'zigpy.application' has no attribute 'CONFIG_SCHEMA'

jonwaland, 2020-05-06:
Any suggestions on whats broken here:

( Home Assistant 0.109.4, HUSBZB-1 stick - just upgraded and might have been a bit over zealous checking python libraries)

issue with bellows-homeassistant-0.14.0 and 0.15.2

```
2020-05-06 18:22:32 ERROR (MainThread) [homeassistant.setup] Setup failed for zha: unknown error
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/setup.py", line 137, in _async_setup_component
    component = integration.get_component()
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/loader.py", line 270, in get_component
    cache[self.domain] = importlib.import_module(self.pkg_path)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/zha/__init__.py", line 14, in <module>
    from . import api
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/zha/api.py", line 18, in <module>
    from .core.const import (
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/zha/core/__init__.py", line 4, in <module>
    from .device import ZHADevice
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/zha/core/device.py", line 25, in <module>
    from . import channels, typing as zha_typing
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/zha/core/channels/__init__.py", line 9, in <module>
    from . import (  # noqa: F401 # pylint: disable=unused-import
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/zha/core/channels/base.py", line 25, in <module>
    from ..helpers import LogMixin, safe_read
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/zha/core/helpers.py", line 11, in <module>
    from .registries import BINDABLE_CLUSTERS
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/zha/core/registries.py", line 7, in <module>
    import bellows.zigbee.application
  File "/srv/homeassistant/lib/python3.8/site-packages/bellows/zigbee/application.py", line 21, in <module>
    CONFIG_SCHEMA = zigpy.application.CONFIG_SCHEMA.extend(
AttributeError: module 'zigpy.application' has no attribute 'CONFIG_SCHEMA'
```

jonwaland, 2020-05-06:
and at cli

```
(homeassistant) homeassistant@JonLabNUC:~/.homeassistant$ bellows
Traceback (most recent call last):
  File "/srv/homeassistant/bin/bellows", line 5, in <module>
    from bellows.cli.main import main
  File "/srv/homeassistant/lib/python3.8/site-packages/bellows/cli/__init__.py", line 2, in <module>
    from . import application
  File "/srv/homeassistant/lib/python3.8/site-packages/bellows/cli/application.py", line 10, in <module>
    import bellows.zigbee.application
  File "/srv/homeassistant/lib/python3.8/site-packages/bellows/zigbee/application.py", line 21, in <module>
    CONFIG_SCHEMA = zigpy.application.CONFIG_SCHEMA.extend(
AttributeError: module 'zigpy.application' has no attribute 'CONFIG_SCHEMA'
```

Adminiuga, 2020-05-06:
Uninstall zigpy and install zigpy-homeassistant

jonwaland, 2020-05-07:
ok...

```
(homeassistant) homeassistant@JonLabNUC:~/.homeassistant$ pip3 uninstall zigpy
Found existing installation: zigpy 0.20.1
Uninstalling zigpy-0.20.1:
  Would remove:
    /srv/homeassistant/lib/python3.8/site-packages/zigpy-0.20.1.dist-info/*
    /srv/homeassistant/lib/python3.8/site-packages/zigpy/*
Proceed (y/n)? y
  Successfully uninstalled zigpy-0.20.1
(homeassistant) homeassistant@JonLabNUC:~/.homeassistant$ pip3 install --upgrade zigpy-homeassistant
Requirement already up-to-date: zigpy-homeassistant in /srv/homeassistant/lib/python3.8/site-packages (0.19.0)
Requirement already satisfied, skipping upgrade: pycryptodome in /srv/homeassistant/lib/python3.8/site-packages (from zigpy-homeassistant) (3.9.7)
Requirement already satisfied, skipping upgrade: aiohttp in /srv/homeassistant/lib/python3.8/site-packages (from zigpy-homeassistant) (3.6.1)
Requirement already satisfied, skipping upgrade: crccheck in /srv/homeassistant/lib/python3.8/site-packages (from zigpy-homeassistant) (0.6)
Requirement already satisfied, skipping upgrade: voluptuous in /srv/homeassistant/lib/python3.8/site-packages (from zigpy-homeassistant) (0.11.7)
Requirement already satisfied, skipping upgrade: attrs>=17.3.0 in /srv/homeassistant/lib/python3.8/site-packages (from aiohttp->zigpy-homeassistant) (19.3.0)
Requirement already satisfied, skipping upgrade: multidict<5.0,>=4.5 in /srv/homeassistant/lib/python3.8/site-packages (from aiohttp->zigpy-homeassistant) (4.5.2)
Requirement already satisfied, skipping upgrade: async-timeout<4.0,>=3.0 in /srv/homeassistant/lib/python3.8/site-packages (from aiohttp->zigpy-homeassistant) (3.0.1)
Requirement already satisfied, skipping upgrade: chardet<4.0,>=2.0 in /srv/homeassistant/lib/python3.8/site-packages (from aiohttp->zigpy-homeassistant) (3.0.4)
Requirement already satisfied, skipping upgrade: yarl<2.0,>=1.0 in /srv/homeassistant/lib/python3.8/site-packages (from aiohttp->zigpy-homeassistant) (1.3.0)
Requirement already satisfied, skipping upgrade: idna>=2.0 in /srv/homeassistant/lib/python3.8/site-packages (from yarl<2.0,>=1.0->aiohttp->zigpy-homeassistant) (2.8)
(homeassistant) homeassistant@JonLabNUC:~/.homeassistant$ bellows
Traceback (most recent call last):
  File "/srv/homeassistant/bin/bellows", line 5, in <module>
    from bellows.cli.main import main
  File "/srv/homeassistant/lib/python3.8/site-packages/bellows/cli/__init__.py", line 2, in <module>
    from . import application, dump, ncp, network
  File "/srv/homeassistant/lib/python3.8/site-packages/bellows/cli/application.py", line 6, in <module>
    import bellows.ezsp
  File "/srv/homeassistant/lib/python3.8/site-packages/bellows/ezsp.py", line 6, in <module>
    from bellows.commands import COMMANDS
  File "/srv/homeassistant/lib/python3.8/site-packages/bellows/commands.py", line 1, in <module>
    import bellows.types as t
  File "/srv/homeassistant/lib/python3.8/site-packages/bellows/types/__init__.py", line 2, in <module>
    from .named import *  # noqa: F401,F403
  File "/srv/homeassistant/lib/python3.8/site-packages/bellows/types/named.py", line 3, in <module>
    import zigpy.types as ztypes
ModuleNotFoundError: No module named 'zigp
```y'

jonwaland, 2020-05-07:
uninstalled zigpy and bellows-homeassistant

then re-installed. 

working - thanks for the tips

closed 2020-05-07 by jonwaland