zigpy/bellows #266: Problem initiating an Elelabs zigbee dongle; Application frame 83 (setConfigurationValue) received: b'35'

unitware, 2020-05-26:
Hi, I have just purchased the Elelabs USB-dongle and also their RPi shield. both have the same issue and I would appreciate any help in what is going on. 

There seems to be a problem setting the channel?
Im just beginning with bellows and home assistant (same problem there) so any hints are appreciated

```
$ git clone https://github.com/zigpy/bellows.git
$ cd bellows
$ git describe --tags
0.16.a2-20-g191cafe
$ python3 setup.py install
...
$
$ touch db
$
$ bellows -d /dev/ttyUSB0 -b 115200 -v debug form -D db -c 16 
debug: Using selector: EpollSelector
debug: Loading application state from db
debug: Using selector: EpollSelector
debug: Resetting EZSP
debug: Resetting ASH
debug: Sending: b'1ac038bc7e'
debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
debug: Send command version: (4,)
debug: Sending: b'004221a850ed2c7e'
debug: Data frame: b'0142a1a8522825d2e23b7e'
debug: Sending: b'8160597e'
debug: Application frame 0 (version) received: b'06023060'
debug: Send command version: (6,)
debug: Sending: b'7d31432157542a7d3315a67e'
debug: Data frame: b'1243a157542a13b069f4b3d37e'
debug: Sending: b'82503a7e'
debug: Application frame 0 (version) received: b'06023060'
debug: Switched to EZSP protocol version 6
debug: EZSP Stack Type: 2, Stack Version: 24624
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_KEY_TABLE_SIZE: 30>, 4)
debug: Sending: b'2240215754790bb65943cb7e'
debug: Data frame: b'2340a157547915fbd27e'
debug: Sending: b'83401b7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_INDIRECT_TRANSMISSION_TIMEOUT: 18>, 7680)
debug: Sending: b'33412157547907b247f69f7e'
debug: Data frame: b'3441a157547915904d7e'
debug: Sending: b'8430fc7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_APPLICATION_ZDO_FLAGS: 42>, 3)
debug: Sending: b'4446215754793fb159e15f7e'
debug: Data frame: b'4546a157547915640c7e'
debug: Sending: b'8520dd7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_ADDRESS_TABLE_SIZE: 5>, 16)
debug: Sending: b'55472157547910a259e2867e'
debug: Data frame: b'5647a157547915ce557e'
debug: Sending: b'8610be7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_MAX_END_DEVICE_CHILDREN: 17>, 32)
debug: Sending: b'66442157547904925969837e'
debug: Data frame: b'6744a157547915e1597e'
debug: Sending: b'87009f7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_TRUST_CENTER_ADDRESS_CACHE_SIZE: 25>, 2)
debug: Sending: b'7745215754790cb0595fa87e'
debug: Data frame: b'7045a1575479158ac67e'
debug: Sending: b'8070787e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_SECURITY_LEVEL: 13>, 5)
debug: Sending: b'004a215754797d38b7599a2e7e'
debug: Data frame: b'014aa15754791573c57e'
debug: Sending: b'8160597e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_SUPPORTED_NETWORKS: 45>, 1)
debug: Sending: b'7d314b2157547938b3592f227e'
debug: Data frame: b'124ba157547915d99c7e'
debug: Sending: b'82503a7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_MULTICAST_TABLE_SIZE: 6>, 16)
debug: Sending: b'2248215754797d33a25978647e'
debug: Data frame: b'2348a157547915f6907e'
debug: Sending: b'83401b7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_STACK_PROFILE: 12>, 2)
debug: Sending: b'33492157547919b05925ba7e'
debug: Data frame: b'3449a1575479159d0f7e'
debug: Sending: b'8430fc7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_SOURCE_ROUTE_TABLE_SIZE: 26>, 16)
debug: Sending: b'444e215754790fa2596c007e'
debug: Data frame: b'454ea1575479200fb87e'
debug: Sending: b'8520dd7e'
debug: Application frame 83 (setConfigurationValue) received: b'35'
Traceback (most recent call last):
  File "/home/magnus/.espressif/python_env/idf4.1_py3.6_env/bin/bellows", line 11, in <module>
debug: Closed serial connection
    load_entry_point('bellows==0.17.0.dev0', 'console_scripts', 'bellows')()
  File "/home/magnus/.espressif/python_env/idf4.1_py3.6_env/lib/python3.6/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/magnus/.espressif/python_env/idf4.1_py3.6_env/lib/python3.6/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/magnus/.espressif/python_env/idf4.1_py3.6_env/lib/python3.6/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/magnus/.espressif/python_env/idf4.1_py3.6_env/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/magnus/.espressif/python_env/idf4.1_py3.6_env/lib/python3.6/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/magnus/.espressif/python_env/idf4.1_py3.6_env/lib/python3.6/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/magnus/.espressif/python_env/idf4.1_py3.6_env/lib/python3.6/site-packages/bellows-0.17.0.dev0-py3.6.egg/bellows/cli/application.py", line 44, in form
  File "/home/magnus/.espressif/python_env/idf4.1_py3.6_env/lib/python3.6/site-packages/bellows-0.17.0.dev0-py3.6.egg/bellows/cli/util.py", line 75, in inner
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "/home/magnus/.espressif/python_env/idf4.1_py3.6_env/lib/python3.6/site-packages/bellows-0.17.0.dev0-py3.6.egg/bellows/cli/util.py", line 61, in async_inner
  File "/home/magnus/.espressif/python_env/idf4.1_py3.6_env/lib/python3.6/site-packages/bellows-0.17.0.dev0-py3.6.egg/bellows/cli/application.py", line 42, in inner
  File "/home/magnus/.espressif/python_env/idf4.1_py3.6_env/lib/python3.6/site-packages/bellows-0.17.0.dev0-py3.6.egg/bellows/zigbee/application.py", line 138, in startup
  File "/home/magnus/.espressif/python_env/idf4.1_py3.6_env/lib/python3.6/site-packages/bellows-0.17.0.dev0-py3.6.egg/bellows/zigbee/application.py", line 88, in initialize
  File "/home/magnus/.espressif/python_env/idf4.1_py3.6_env/lib/python3.6/site-packages/bellows-0.17.0.dev0-py3.6.egg/bellows/zigbee/application.py", line 230, in _cfg
AssertionError
```

Adminiuga, 2020-05-26:
1.that is not the most recent version of bellows, this log was collected before Apr 15th?
2. `debug: Application frame 83 (setConfigurationValue) received: b'35'` -- means the stick is out of memory

The out of memory is strange, cause I'm running on my HUSBZB with 32 source routes without any issues and the same initializes  fine on the older elelabs stick I have. I'd escalate with Elelabs, must be something related to this specific firmware as a lot of people are running these parameters without issues.

Alternatively, try a separate venv with `"bellows-homeassistant==0.15.2"` and `"zigpy-homeassistant==0.19.0"` before big changes

Adminiuga, 2020-05-26:
or try changing the source route table size to `8` or decrease the number `EZSP_CONFIG_MAX_END_DEVICE_CHILDREN` is https://github.com/zigpy/bellows/blob/191cafe5ffad9bcb3dfe72dedeb78c25c2760b9c/bellows/config/ezsp.py#L77

unitware, 2020-05-26:
Thanks, i did clone the code today, git describe might be misleading if no new tag is set, this is 20 commits after the tag

I’ll try the older version tomorrow

//M

Skickat från min iPhone

unitware, 2020-05-28:
```
Alternatively, try a separate venv with "bellows-homeassistant==0.15.2" and "zigpy-homeassistant==0.19.0" before big changes
```
Running on old versions work, can I do this in a recent homeassistant install in any way?

```
or try changing the source route table size to 8 or decrease the number EZSP_CONFIG_MAX_END_DEVICE_CHILDREN 
```
This did not work (not sure I did it correctly though, I changed the constant in bellows/types/named.py

Adminiuga, 2020-05-28:
Nope, that's wrong. You should change default values for that parameter in https://github.com/zigpy/bellows/blob/191cafe5ffad9bcb3dfe72dedeb78c25c2760b9c/bellows/config/ezsp.py

Alternatively, if using with home assistant 0.110 or newer, add to configuration.yaml
```
zha:
  zigpy_config:
    ezsp_config:
      - CONFIG_SOURCE_ROUTE_TABLE_SIZE: 8
```

Although personally, I'd reduce the number of max end devices, 32 is way too high and increase source routes number to 16

unitware, 2020-05-28:
Thanks I'm getting somewhere now, but without the dash in the config:
````
zha:
  zigpy_config:
    ezsp_config:
      CONFIG_SOURCE_ROUTE_TABLE_SIZE: 8
````

Is there a way to set the channel and network key this way to? I'm running hassio and I cant get the bellows to install without gcc (pycryptodome setup).

Adminiuga, 2020-05-28:
you would need to form the network again, meaning you would have to re-pair all the devices. ATM not possible to update the network key or channel.

1. stop HA
2. run `bellows -d /dev/ttyUSB-your-zigbee-device leave` -- this will kill the current network
3. run `bellows -d /dev/ttyUSB-your-zigbee-device form -c 15 --network-key 1,2,3,4,5,6,,16`

or you could put under `zigpy_config:`
```
zigpy_config:
  network:
    key: [1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16]   ## 16 bytes of network key
    channel: 15
```
and restart HA

Hedda, 2020-06-02:
@Adminiuga should network key & channel configuration be in current or next ZHA integration docs?

https://www.home-assistant.io/integrations/zha/

https://github.com/home-assistant/home-assistant.io/blob/current/source/_integrations/zha.markdown

https://github.com/home-assistant/home-assistant.io/blob/next/source/_integrations/zha.markdown

Adminiuga, 2020-06-02:
Don't think so, it's not supported by all radios yet

unitware, 2020-06-04:
Hi, I'm still struggling a bit. now I can start bellows most of the times but it is sometimes failing on startup/initiation

````
magnus@uw-build-ubuntu:~/work/bellows/bellows$ bellows -d /dev/ttyUSB0 -b 115200 -vDEBUG permit -D db 
debug: Using selector: EpollSelector
debug: Loading application state from db
debug: Checking quirks for None None (cc:cc:cc:ff:fe:a5:f3:56)
debug: Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
debug: Found custom device replacement for cc:cc:cc:ff:fe:a5:f3:56: <class 'bellows.zigbee.application.EZSPCoordinator'>
debug: Using selector: EpollSelector
debug: Resetting EZSP
debug: Resetting ASH
debug: Sending: b'1ac038bc7e'
debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
debug: Send command version: (4,)
debug: Sending: b'004221a850ed2c7e'
debug: Data frame: b'0142a1a8522825d2e23b7e'
debug: Sending: b'8160597e'
debug: Application frame 0 (version) received: b'06023060'
debug: Send command version: (6,)
debug: Sending: b'7d31432157542a7d3315a67e'
debug: Data frame: b'1243a157542a13b069f4b3d37e'
debug: Sending: b'82503a7e'
debug: Application frame 0 (version) received: b'06023060'
debug: Switched to EZSP protocol version 6
debug: EZSP Stack Type: 2, Stack Version: 24624
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_SECURITY_LEVEL: 13>, 5)
debug: Sending: b'2240215754797d38b7596ac97e'
debug: Data frame: b'2340a157547915fbd27e'
debug: Sending: b'83401b7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_SUPPORTED_NETWORKS: 45>, 1)
debug: Sending: b'33412157547938b359dfc57e'
debug: Data frame: b'3441a157547915904d7e'
debug: Sending: b'8430fc7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_INDIRECT_TRANSMISSION_TIMEOUT: 18>, 7680)
debug: Sending: b'44462157547907b2472bf77e'
debug: Data frame: b'4546a157547915640c7e'
debug: Sending: b'8520dd7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_ADDRESS_TABLE_SIZE: 5>, 16)
debug: Sending: b'55472157547910a259e2867e'
debug: Data frame: b'5647a157547915ce557e'
debug: Sending: b'8610be7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_STACK_PROFILE: 12>, 2)
debug: Sending: b'66442157547919b05908357e'
debug: Data frame: b'6744a157547915e1597e'
debug: Sending: b'87009f7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_MULTICAST_TABLE_SIZE: 6>, 16)
debug: Sending: b'7745215754797d33a25955eb7e'
debug: Data frame: b'7045a1575479158ac67e'
debug: Sending: b'8070787e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_SOURCE_ROUTE_TABLE_SIZE: 26>, 8)
debug: Sending: b'004a215754790fba592a817e'
debug: Data frame: b'014aa15754791573c57e'
debug: Sending: b'8160597e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_APPLICATION_ZDO_FLAGS: 42>, 3)
debug: Sending: b'7d314b215754793fb159ccd07e'
debug: Data frame: b'124ba157547915d99c7e'
debug: Sending: b'82503a7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_TRUST_CENTER_ADDRESS_CACHE_SIZE: 25>, 2)
debug: Sending: b'2248215754790cb05972277e'
debug: Data frame: b'2348a15754792090667e'
debug: Sending: b'83401b7e'
[... 484 lines cut ...]
debug: Send command nop: ()
debug: Sending: b'31792157542f5c947e'
debug: Data frame: b'1479a157542ff7a57e'
debug: Sending: b'82503a7e'
debug: Application frame 5 (nop) received: b''
debug: Data frame: b'2479b157541515f7c3904b25aa5d9a099c4e27ece3a867b4d17e'
debug: Sending: b'83401b7e'
debug: Application frame 63 (messageSentHandler) received: b'00459a04010000080840000000470e6600'
debug: Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 39493, <EmberApsFrame profileId=260 clusterId=0 sourceEndpoint=8 destinationEndpoint=8 options=64 groupId=0 sequence=71>, 14, <EmberStatus.DELIVERY_FAILED: 102>, b'']
debug: [0x9a45] Delivery error for seq # 0x0d, on endpoint id 8 cluster 0x0000: message send failure
debug: [0x9a45:8] Manufacturer: None
debug: [0x9a45:8] Model: None
debug: Checking quirks for None None (f4:ce:36:74:f3:6f:5a:ed)
debug: Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
debug: Fail because endpoint list mismatch: {1} {8}
debug: Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
debug: Fail because endpoint list mismatch: {1} {8}
debug: Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
debug: Fail because endpoint list mismatch: {1} {8}
Done
Shutting down ControllerApplication
debug: Cancelling watchdog
debug: Closed serial connection
magnus@uw-build-ubuntu:~/work/bellows/bellows$ 
````

Adminiuga, 2020-06-04:
Hard to say why it is leaving. Take zigbee packet capture when joining both to bellows and to Nordic controller and compare. Unfortunately with zigbee, when device is leaving is does not provide a reason to the other party. If you can enable debug on the lock itself, then maybe it could provide more insight in what exactly it did not like.

Adminiuga, 2020-06-04:
Elelabs stick seems to  be running protocol version 6,  EZSP 6.0.3 stack.  Could be related to ZB3.0 interop?

unitware, 2020-06-05:
Thanks, I'm new to this and hav litttle knowledge on the EZSP stack versions. Is 6.0.3 an old or new one? Can I get hold of a correct version somehow? (Elelabs have yet to answer my first email, but might respond if I have a more specific question, but my hopes are not to high at the moment)

What usb-adapter / rpi-shield is best today? I mainly selected the elelabs since there were hints of support for the closure/door_lock cluster which is my main interest at this time.

Yoda-x, 2020-06-05:
EZSP stack version 6 had a significant change for the end-device-poll-timeout registry compared to 5.
I used the elelab usb stick without problems after below change in the in init procedure: to check the stack version

     self._stackversion = await e.version()
     ....
     if self._stackversion > 0x64ff:
         await self._cfg(c.CONFIG_END_DEVICE_POLL_TIMEOUT, 9)
         # with ezsp_callback_handlerstack 6.5 timeout changes
         # timeout_shift is gone
         # The timeout corresponding to a nonzero value N is 2^N minutes
     else:
         await self._cfg(c.CONFIG_END_DEVICE_POLL_TIMEOUT, 120)
         await self._cfg(c.CONFIG_END_DEVICE_POLL_TIMEOUT_SHIFT, 8)`

unitware, 2020-06-05:
thanks, the version seems to be 24 624 (0x6030)
```
debug: Application frame 0 (version) received: b'06023060'
debug: Switched to EZSP protocol version 6
debug: EZSP Stack Type: 2, Stack Version: 24624
```

Should the old way be used? or was the check >= 0x64ff in error?
I did this diff to my code, still not working (the device leaves after the transport key):

```
diff --git a/bellows/zigbee/application.py b/bellows/zigbee/application.py
index f4a1426..09c5f97 100644
--- a/bellows/zigbee/application.py
+++ b/bellows/zigbee/application.py
@@ -88,11 +88,20 @@ class ControllerApplication(zigpy.application.ControllerApplication):
             await self._cfg(t.EzspConfigId[config], value)
 
         c = t.EzspConfigId
+
         if self._ezsp.ezsp_version >= 7:
             await self._cfg(c.CONFIG_END_DEVICE_POLL_TIMEOUT, 8)
+
+        elif self._ezsp.ezsp_version > 6:
+            await self._cfg(c.CONFIG_END_DEVICE_POLL_TIMEOUT, 9)
+             # with ezsp_callback_handlerstack 6.5 timeout changes
+             # timeout_shift is gone
+             # The timeout corresponding to a nonzero value N is 2^N minutes
+
         else:
             await self._cfg(c.CONFIG_END_DEVICE_POLL_TIMEOUT, 60)
             await self._cfg(c.CONFIG_END_DEVICE_POLL_TIMEOUT_SHIFT, 8)
+
         await self._cfg(
             t.EzspConfigId.CONFIG_PACKET_BUFFER_COUNT,
             ezsp_config[c.CONFIG_PACKET_BUFFER_COUNT.name],
```

Adminiuga, 2020-07-23:
This should be fixed in 0.18.0 (ha 0.113.x)

closed 2020-07-23 by Adminiuga

Yuubari, 2020-09-08:
I get the exact same issue on bellows 0.18.1 and homeassistant 0.114.4 with Elelabs USB adaptor:

```
2020-09-08 16:13:57 DEBUG (MainThread) [bellows.ezsp] Resetting EZSP
2020-09-08 16:13:58 DEBUG (bellows.thread_0) [bellows.uart] Resetting ASH
2020-09-08 16:13:58 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'1ac038bc7e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Send command version: (4,)
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'004221a850ed2c7e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'0142a1a8522825d2e23b7e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Application frame 0 (version) received: b'06023060'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Send command version: (6,)
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'7d31432157542a7d3315a67e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'1243a157542a13b069f4b3d37e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Application frame 0 (version) received: b'06023060'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Switched to EZSP protocol version 6
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] EZSP Stack Type: 2, Stack Version: 24624
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue: (<EzspConfigId.CONFIG_STACK_PROFILE: 12>, 2)
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'22402157547919b059c46e7e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'2340a157547915fbd27e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received: b'00'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue: (<EzspConfigId.CONFIG_MULTICAST_TABLE_SIZE: 6>, 16)
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'3341215754797d33a25999b07e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'3441a157547915904d7e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received: b'00'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue: (<EzspConfigId.CONFIG_ADDRESS_TABLE_SIZE: 5>, 16)
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'44462157547910a2591d887e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'4546a157547915640c7e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received: b'00'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue: (<EzspConfigId.CONFIG_SUPPORTED_NETWORKS: 45>, 1)
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'55472157547938b359fda37e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'5647a157547915ce557e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received: b'00'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue: (<EzspConfigId.CONFIG_APPLICATION_ZDO_FLAGS: 42>, 3)
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'6644215754793fb1590f627e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'6744a157547915e1597e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received: b'00'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue: (<EzspConfigId.CONFIG_TRUST_CENTER_ADDRESS_CACHE_SIZE: 25>, 2)
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'7745215754790cb0595fa87e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'7045a1575479158ac67e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received: b'00'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue: (<EzspConfigId.CONFIG_KEY_TABLE_SIZE: 30>, 4)
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'004a215754790bb659b32c7e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'014aa15754791573c57e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received: b'00'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Send command setConfigurationValue: (<EzspConfigId.CONFIG_SECURITY_LEVEL: 13>, 5)
[... cut ...]
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'3349215754790fa259b1687e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'3449a157547920fbf97e'
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2020-09-08 16:13:59 DEBUG (MainThread) [bellows.ezsp] Application frame 83 (setConfigurationValue) received: b'35'
2020-09-08 16:13:59 ERROR (MainThread) [zigpy.application] Couldn't start application
2020-09-08 16:13:59 DEBUG (bellows.thread_0) [bellows.uart] Closed serial connection
```

Adminiuga, 2020-09-08:
Should be fixed HA beta version.
or add to `configuraiton.yaml`
```yaml
zha:
  zigpy_config:
    ezsp_config:
      CONFIG_SOURCE_ROUTE_TABLE_SIZE: 8
```